evcc-io/evcc #18544: Variable vehicleTitle wird bei Versendung von Nachrichten nicht gefüllt

scharrin2, 2025-02-02:
### Describe the bug

Hallo zusammen,

ich habe zu diesem Problem bereits Bugs gefunden, welche geschlossen wurden. Trotzdem existiert dieses Problem bei mir. Ich benutze die Version v0.133.0

In der Nachricht verwende ich unter anderem folgenden title:

title: "{{.vehicleTitle}} connected on wallbox {{.title}}"

unter vehicles ist diese wie folgt gefüllt:
title: Audi Q4 e-tron

In der Weboberfläche wird der Name des Autos auch korrekt angezeigt.

Trotzdem steht in den Pushover-Nachrichten folgender Text:
<no value> disconnected of wallbox Garage

### Steps to reproduce

1. Stecker der Wallbox aus Auto entfernen
2. Pushover-Nachricht empfangen
3.
...

### Configuration details

```yaml
Konfiguration (/etc/evcc.yaml):

network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: ts-453be.fritz.box
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token: *****
sponsortoken: *****

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
telemetry: true

plant: *****

# log settings
log: debug
levels:
[... 292 lines cut ...]

Version: `0.133.0`
```

### Log details

```text
Welchen Bereich brauche ich hier von den Log-Files?
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.133.0 (b7a9abb6)

StefanSchoof, 2025-02-02:
Welches war den altes Issue?

Klingt nach einen Duplikat von https://github.com/evcc-io/evcc/issues/7193 und wurde wahrscheinlich daher geschlossen

andig, 2025-02-02:
So isses. Leider ein doofes Timingproblem.

closed 2025-02-02 by andig