evcc-io/evcc #18550: Add ETEK EKEPC2 charge controller

joe-0310, 2025-02-02:
Hello, I would like to suggest adding a template for a pretty cheap charge controller EKEPC2 from aliexpress I recently integrated with some Homeassistant workarounds.

Works pretty reliably for me. I hope this Info can be usefull for people using this charger or someone can help me to turn this behavior into an evcc template.

Full documentation of the charge controller here:
[https://www.etek-electric.com/epc-ev-charge-controller/ekepc2-cs-ev-charging-station-controller](url)

[image]

Here is my temporary solution (evcc config yaml) I used to get the charger working:
```yaml
chargers:
  - type: custom
    name: my_charger
    status:
      # Read the status of the charger
      # Either A,B,C or F
      source: http
      uri: http://10.10.69.110:8123/api/states/sensor.evcc_car_charger_state
      method: GET
      headers:
        - Authorization: Bearer "your_token"
      insecure: true
      jq: .state
      timeout: 2s
    enabled:
      # Is the charger enabled (1) or not (0)
      source: http
      uri: http://10.10.69.110:8123/api/states/switch.car_charger_enable
      method: GET
      headers:
        - Authorization: Bearer "your_token"
      insecure: true
      jq: if .state == "on" then 1 else 0 end
      timeout: 2s
    enable:
      # Enable the charger
      source: modbus
      id: 2
      uri: 10.10.69.100:502
      register:
        address: 134
        type: writeholding
        decode: int16
    maxcurrent:
      # Set the maximum current
      source: modbus
      id: 2
      uri: 10.10.69.100:502
      register:
        address: 97
        type: writeholding
        decode: int16
```
I am reading status and enabled from HA. Enable end maxcurrent are being written in unused modbus registers of the EKEPC2, then read by HA, mapped and the correct values are being written in the correct registers of the EKEPC2.

I had to to implement the following mapping and automations in HA:
```yaml
- id: 2d78b0db02b3489c96796e3a1c438752
  alias: Write Mapped Charging Current to evcc
  triggers:
  - entity_id: sensor.car_charger_set_pwm
    trigger: state
  - trigger: time_pattern
    seconds: '0'
  actions:
  - action: modbus.write_register
    metadata: {}
    data:
      hub: EW11
      value: '{{ states(''sensor.car_charger_set_pwm'') | int * 167 }}'
      address: 109
      slave: 2
- id: '1737739324768'
  alias: Write mapped set enable to evcc
  description: ''
  triggers:
  - trigger: state
    entity_id:
    - switch.car_charger_enable
  - trigger: time_pattern
    seconds: '0'
  conditions: []
  actions:
  - choose:
    - conditions:
      - condition: state
        entity_id: switch.car_charger_enable
        state: 'on'
      sequence:
      - action: automation.turn_on
        metadata: {}
        data: {}
        target:
          entity_id: automation.write_mapped_charging_current_to_evcc
      - action: automation.trigger
        metadata: {}
        data:
          skip_condition: true
        target:
          entity_id: automation.write_mapped_charging_current_to_evcc
      - action: modbus.write_register
        metadata: {}
        data:
          hub: EW11
          address: 89
          slave: 2
          value: 0
    - conditions:
      - condition: state
        entity_id: switch.car_charger_enable
        state: 'off'
      sequence:
      - action: automation.turn_off
        metadata: {}
        data:
          stop_actions: true
        target:
          entity_id: automation.write_mapped_charging_current_to_evcc
      - action: modbus.write_register
        metadata: {}
        data:
          hub: EW11
          address: 89
          slave: 2
          value: 2
```
The first automation writes the mapped charging current to the EKEPC2. This automation is disabled if the charger is disabled.
The second automation writes the vaue 2 to the modbus register 89 of the EKEPC2 to disable it and value 0 to enable it.
Important notice: when disabled, the EKEPC2 enters an "emergency stop" (status code 19, not referenced in the documentation) and doesn't report the status correctly anymore. Thus in this case I manually had to map this status code as either "Ready" or "Connected". To know if a car is connected, I read the CP_pos voltage from register 153 and the status code from register 141 and do the following mapping in my HA config:
```yaml
  - sensor:
      name: "Car Charger State Text"
      unique_id: c93e1446-4b41-4781-a1bb-2a261066a526
      state: >-
        {% set d = {
        '1': 'Ready', 
        '3': 'Connected',
        '5': 'Charging',
        '19': 'Stopped' } %}
        {% if states('sensor.car_charger_state') in d.keys() %}
          {{ d.get(states('sensor.car_charger_state')) }}
        {% else %}
          Unknown
        {% endif %}
        
  - sensor:
      - name: "EVCC Car Charger State"
        unique_id: 765c4f06-5bdb-44ac-80df-6277637cd8cd
        state: >
          {% set state = states('sensor.car_charger_state') | int %}
          {% set voltage = states('sensor.car_charger_cp_pos_voltage') | int %}
          
          {% if state == 0 %}
            F
          {% elif state == 1 %}
            A
          {% elif state == 2 %}
            F
          {% elif state in [3, 4] %}
            B
          {% elif state == 5 %}
            C
          {% elif state == 19 and voltage < 300 %}
            A
          {% elif state == 19 %}
            B
          {% else %}
            F
          {% endif %}
        availability: >
          {{ states('sensor.car_charger_state') not in ['unknown', 'unavailable'] }}
        icon: >
          {% set state = states('sensor.car_charger_state') | int %}
          {% if state == 5 %}
            "mdi:ev-station"
          {% elif state == 1 %}
            "mdi:power-plug-off-outline"
          {% elif state in [3, 4] %}
            "mdi:power-plug-outline"
          {% else %}
            "mdi:alert"
          {% endif %}
```
Here is the needed modbus config:
```yaml
  sensors:
    - name: Car charger State #state of the car charger I used to do the previous mapping
      unique_id: 65ee4d14-c0be-4ecb-98cf-0d490e2f77dc
      scan_interval: 5
      slave: 2
      address: 141
      data_type: int16
      
    - name: Car charger set PWM #evcc writes amp setting to temporary spare register 97, HA reads the value, scales it by factor 167 and writes it to register 109
      unique_id: 71298412-3502-47b0-b7ac-ba40a773f1e0
      scan_interval: 1
      slave: 2
      address: 97 # spare register written to 109
      unit_of_measurement: A
      data_type: int16

    - name: Car charger CP pos voltage #when the charger is disabled (status code 19), this is needed to determine if a car is connected
      unique_id: 48f27a17-1f73-41f5-bb87-4ae34ad94579
      scan_interval: 1
      slave: 2
      address: 153
      data_type: int16

  switches:
    - name: "Car Charger enable" #evcc writes to temporary spare register 134, HA reads the value, maps it and writes to register 89
      unique_id: d5110855-050e-478c-9ef4-f7710f039abc
      slave: 2
      address: 134 # spare register written to 89
      write_type: holding
      command_on: 1
      command_off: 0
      verify:
        input_type: holding
        address: 134
        state_on: 1
        state_off: 0
```

andig, 2025-02-02:
@joe-0310 do you want to bundle your custom config as tempalte and open a PR?

joe-0310, 2025-02-02:
@andig yes I want to create a template with help of one of the devs and test it. didn't have the time/knowledge to fork and test on my own

andig, 2025-02-02:
Just create a custom charger and use modbus directly. No reason calling a home automation for that.

joe-0310, 2025-02-03:
I had some difficulties writing scaled values to modbus (only reading and than scaling worked) furthermore I don't know how to setup the mapping of the status A B C F directly in evcc.

andig, 2025-02-03:
> I had some difficulties writing scaled values to modbus

What exactly did you try/observe

> furthermore I don't know how to setup the mapping of the status A B C F directly in evcc.

The go plugin might help.

joe-0310, 2025-02-03:
Thank you for your help. Tbh I am totally new to evcc.

> > I had some difficulties writing scaled values to modbus
> 
> What exactly did you try/observe
> 

When I added the "scale: 167" attribute to the "maxcurrent" section, the yaml config did not load.

> > furthermore I don't know how to setup the mapping of the status A B C F directly in evcc.
> 
> The go plugin might help.

Regarding the mapping of the status, I had no clue how to attack this problem only using evcc, so I used HA. Next week I will have the time to check out your suggestion. Thank you.

joe-0310, 2025-10-24:
Hi Joao, yes better to take a break and get you head free. I can recommend you to get a EW11 or EE11 (fairly inexpensive) after that I can help you with the setup and next steps in Home Assistant developer tools to change the address.
Sth I noticed as well is, that they swapped A and B terminals over the years. I found pictures and documentation on the EKEPC2 which showed terminals A B and others B A printed on the housing, so you could try swapping the lines in you haven't tried that yet.

ct3fu, 2025-10-24:
I'll try and then I'll leave the result here

ct3fu, 2025-10-25:
Using the code available from Joe-0310 it was possible to unlock Address 255 to 1 and here is what I used.

web_server:
  port: 80

uart:
  id: uart_etek
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 9600
  stop_bits: 1
  parity: NONE

modbus:
  id: modbus_bus
  uart_id: uart_etek
  flow_control_pin: GPIO4

modbus_controller:
  - id: etek_controller
    address: 255
    modbus_id: modbus_bus
    update_interval: 5s

sensor:
  - platform: modbus_controller
    modbus_controller_id: etek_controller
    name: "CP Voltage"
    id: cp_voltage
    address: 153
    register_type: holding
    value_type: S_WORD
    unit_of_measurement: "V"
    accuracy_decimals: 1
    filters:
      - multiply: 0.01

number:
  - platform: modbus_controller
    modbus_controller_id: etek_controller
    name: "Corrente Máxima ETEK"
    id: corrente_maxima_etek
    address: 0x66
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "A"
    min_value: 6
    max_value: 32
    step: 1

patrick-foubert-56770, 2025-12-02:
Hi,

I'm working on a esp32 module to make interface with EKEPC2 to provide an solution for EVCC, my approch is #, okay i'm working also with home assistant, but my goal is to made coding on esp32 to have an independant module from home assistant.

patrick-foubert-56770, 2025-12-03:
[debug.yaml](https://github.com/user-attachments/files/23912075/debug.yaml)

Hi, my code is dirty (coming from more complex), perhaps need cleaning under esphome, but it's set on boot the remote control to stop

On web interface if you change remote control to start (and your electrical wiring is correct) for an Etek module, charge must start @ 6A

closed 2026-01-13 by andig

joe-0310, 2026-10-15:
Better late than never...

## Native evcc integration (no Home Assistant needed)

Following up on @andig's suggestion: here is a `type: custom` charger that talks to the EKEPC2 **directly via Modbus**. All the mapping I previously did in HA (status codes, enable/stop, current scaling) now runs inside evcc using the `go` plugin with `in`/`out` transformations. The spare-register workaround (97/134) and both HA automations are no longer needed.

### Setup

- EKEPC2 on RS485, **9600 8N1**. The manual mentions 38400 in places, but the controller only answers at 9600. This was also confirmed independently [here](https://github.com/francois-baptiste/esphome-acrel-smartevse-gateway/commit/21fa348e6b89a5a63effb9ff1c4cc4fe8be37a0e).
- Modbus address changed from the factory broadcast `255` by writing register `100` (see my earlier comment).
- Gateway: EW11/EE11 in *TCP server* mode with protocol *Modbus* (Modbus TCP ⇄ RTU translation), so no `rtu: true` is needed.

### Registers used

| Register | Access | Meaning | Used for |
|---|---|---|---|
| 89 | R/W | Remote start/stop (`0` none, `1` start, `2` stop) | `enable` / `enabled` |
| 109 | R/W | Max output PWM duty cycle × 100 | `maxcurrent` |
| 141 | R | Working state | `status` |
| 153 | R | CP positive voltage | `status` (plug detection in state 19) |

### Status mapping

| Reg 141 | evcc |
|---|---|
| 1 | A |
| 3, 4 | B |
| 5 | C |
| 19 (remote stop) and CP < 300 | A |
| 19 (remote stop) otherwise | B |
| anything else | F |

State `19` isn't in the documentation. While stopped via reg 89 = 2, the controller stops reporting the plug state, so CP voltage (reg 153) is used to tell A from B.

### Current scaling

Reg 109 is the duty cycle in 0.01 %. Per IEC 61851, duty % = A / 0.6, so the value is `A × 1000/6`. This matches the controller's own factory gear table (regs 128–133, e.g. 32 A → `5333`). The `× 167` factor I used before is a close approximation but overshoots slightly (32 A → `5344`). `maxcurrentmillis` is included because the 0.01 % duty resolution allows ~6 mA steps.

### Config

Adjust `uri` and `id` to your setup. `power`/`energy` come from my separate energy meter (Modbus ID 1); drop or replace them as needed.

```yaml
chargers:
  - name: etek
    type: custom

    status:
      source: go
      script: |
        etekStatus := "F"
        switch {
        case etekState == 1:
          etekStatus = "A"
        case etekState == 3 || etekState == 4:
          etekStatus = "B"
        case etekState == 5:
          etekStatus = "C"
        case etekState == 19 && etekCp < 300:
          etekStatus = "A"
        case etekState == 19:
          etekStatus = "B"
        }
        etekStatus
      in:
        - name: etekState
          type: int
          config:
            source: modbus
            uri: 10.10.69.100:502
            id: 2
            timeout: 2s
            register:
              address: 141
              type: holding
              decode: int16
        - name: etekCp
          type: int
          config:
            source: modbus
            uri: 10.10.69.100:502
            id: 2
            timeout: 2s
            register:
              address: 153
              type: holding
              decode: int16

    enabled:
      source: go
      script: |
        etekRemote != 2
      in:
        - name: etekRemote
          type: int
          config:
            source: modbus
            uri: 10.10.69.100:502
            id: 2
            timeout: 2s
            register:
              address: 89
              type: holding
              decode: int16

    enable:
      source: go
      script: |
        etekCmd := 2
        if enable {
          etekCmd = 0
        }
        etekCmd
      out:
        - name: etekCmd
          type: int
          config:
            source: modbus
            uri: 10.10.69.100:502
            id: 2
            timeout: 2s
            register:
              address: 89
              type: writeholding
              decode: uint16

    maxcurrent:
      source: go
      script: |
        int(math.Round(float64(maxcurrent) * 1000 / 6))
      out:
        - name: etekDuty
          type: int
          config:
            source: modbus
            uri: 10.10.69.100:502
            id: 2
            timeout: 2s
            register:
              address: 109
              type: writeholding
              decode: uint16

    maxcurrentmillis:
      source: go
      script: |
        int(math.Round(maxcurrentmillis * 1000 / 6))
      out:
        - name: etekDutyMillis
          type: int
          config:
            source: modbus
            uri: 10.10.69.100:502
            id: 2
            timeout: 2s
            register:
              address: 109
              type: writeholding
              decode: uint16

    # optional: external energy meter
    power:
      source: modbus
      uri: 10.10.69.100:502
      id: 1
      register:
        address: 260
        type: holding
        decode: int32
    energy:
      source: modbus
      uri: 10.10.69.100:502
      id: 1
      scale: 0.01
      register:
        address: 270
        type: holding
        decode: int32
```

### Notes

- **No periodic rewrites.** Regs 89 and 109 are retained through power loss (persistent storage). evcc only writes on change, which avoids flash/EEPROM wear. My old HA automation rewrote every minute; that isn't needed.
- **Enable value.** I use `0` (no remote control) to release and `2` to stop, because that's what works on my unit. If `0` doesn't resume a stopped session on yours, try `1` (start).
- **One Modbus master.** Don't keep HA (or an ESPHome DLB loop) writing regs 89/109 in parallel, or they will fight evcc.

andig, 2026-10-15:
@joe-0310 seems you may have missed that the modbus implementation has already been added in https://github.com/evcc-io/evcc/pull/26669?

joe-0310, 2026-10-15:
> 29\. [@joe-0310](https://github.com/joe-0310) seems you may have missed that the modbus implementation has already been added in [#26669](https://github.com/evcc-io/evcc/pull/26669)?

@andig I wanted to keep my custom config because of my separate energy meter and wanted to share it just in case...

andig, 2026-10-15:
You can always configure a separate meter on the loadpoint ;)