evcc-io/evcc #18551: PV Charging erratic

danst0, 2025-02-02:
### Describe the bug

Hi,
I have a RCT PV and an HardyBart Wallbox.
Today and yesterday we had a very sunny day with about 4kW coming from the roof at noon.
Now evcc decided to charge at 11kW (see pictures).

[image]
[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: https
  host: evcc.me # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 5b

interval: 30s # control cycle interval

sponsortoken: eyJU

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: rct-power 
  usage: grid  
  host: 10..209  
  name: grid1
- type: template
  template: rct-power 
  usage: pv  
  host: 10..09  
  name: pv2
- type: template
  template: rct-power 
  usage: battery  
  host: 10..9  
  name: battery3

chargers:
- type: template
  template: hardybarth-salia 
  host: 10..10  
  name: wallbox4

loadpoints:
- title: Carport
  charger: wallbox4
  mode: pv

site:
[... 218 lines cut ...]
[site ] DEBUG 2025/02/02 14:37:27 grid power: -2365W
[site ] DEBUG 2025/02/02 14:37:27 pv 1 power: 3256W
[site ] DEBUG 2025/02/02 14:37:27 site power: -2385W
[lp-1 ] DEBUG 2025/02/02 14:37:27 charger status: C
[lp-1 ] DEBUG 2025/02/02 14:37:27 soc estimated: 81.13% (vehicle: 80.00%)
[lp-1 ] DEBUG 2025/02/02 14:37:27 vehicle soc: 81%
[lp-1 ] DEBUG 2025/02/02 14:37:27 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/02/02 14:37:27 vehicle range: 262km
[lp-1 ] DEBUG 2025/02/02 14:37:27 !! pvScalePhases DOWN activePhases: 3, available power: 13425W, scalable: false
[lp-1 ] DEBUG 2025/02/02 14:37:27 !! pvScalePhases UP maxPhases: 3, available power: 13425W, scalable: false
[lp-1 ] DEBUG 2025/02/02 14:37:27 !! pvScalePhases EXIT
[lp-1 ] DEBUG 2025/02/02 14:37:27 pv charge current: 19.5A = 16A + 3.46A (-2385W @ 3p)
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

danst0, 2025-02-02:
small addition: it seems evcc adds the charging power to the total power used to charge. This may be due to my electric installation. The Wallbox is _not_ fed through the rct pv inverter... How can I reflect that in my meters. There is a mqtt based metering of the total power used.

VolkerK62, 2025-02-02:
you must create a custom grid meter with the `calc` plugin, that is calculating the grid power by adding the charge power

danst0, 2025-02-02:
Thanks for the hint! Now it works
I did not find a good description of a Custom MQTT sensor.
I now used:
```
meters:
- type: custom
  name: mqtt_grid
  power:
    source: mqtt
    topic: vzlogger/data/chn2/agg
    timeout: 40s
```

closed 2025-02-02 by andig