evcc-io/evcc #18555: Kein Schnelladen möglich, EVCC & Wallbox zeigen aber schnelladen an

netguru76, 2025-02-02:
### Describe the bug

Nutze EVCC schon lange und in 132.10 hat alles funktioniert.

Cupra Born geht an der Cupra (ELLI)-Wallbox nicht mehr ins "schnelladen".
Wähle ich Schnellladen, oder Mini-PV mit Schnelladeboost  zeigt EVCC 10,4kw an, auch die Ellii-Wallbox zeigt per LED Schnelladen an, am Auto kommen aber nur 4,1kw an.
In den Bestandteilen des Ladestroms zeigt EVCC im Gui anstatt die Batterie ein "?" an. Also die angeblich 10,4k teilen sich in 5,6kw Sonnen und 4,8k "?" auf, wo normalerweise die Batterie angezeigt wird, auch wird bei der Batterie selbst kein wird kein  Entladen angezeigt (siehe Bild).

[image]
[image]

### Steps to reproduce

1. EVVC im Lademodus schnell
2. Auto lädet nur mit 4,3kw
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: 573a08db636491298c96b423afd20888ddf4455baa815addda579ada2522108b

interval: 15s # control cycle interval

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.39  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.39  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.39  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3
- type: template
  template: sungrow-hybrid 
  id: 2  
  host: 192.168.178.38  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv4

[... 750 lines cut ...]
[eebus ] DEBUG 2025/02/02 11:13:16 Recv: d:_i:47859_Elli-Wallbox-2145A1RSRN:NodeManagement to NodeManagement read 17 NodeManagementDestinationListData
[eebus ] DEBUG 2025/02/02 11:13:16 Send: d:_i:47859_Elli-Wallbox-2145A1RSRN reply 9 17 NodeManagementDestinationListData
[eebus ] DEBUG 2025/02/02 11:13:16 Recv: d:_i:47859_Elli-Wallbox-2145A1RSRN:NodeManagement to NodeManagement read 18 NodeManagementUseCaseData
[eebus ] DEBUG 2025/02/02 11:13:16 Send: d:_i:47859_Elli-Wallbox-2145A1RSRN reply 10 18 NodeManagementUseCaseData
[eebus ] DEBUG 2025/02/02 11:13:17 Recv: d:_i:47859_Elli-Wallbox-2145A1RSRN:NodeManagement to NodeManagement notify 22 NodeManagementUseCaseData
[eebus ] DEBUG 2025/02/02 11:13:18 Recv: d:_i:47859_Elli-Wallbox-2145A1RSRN:NodeManagement to NodeManagement notify 23 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2025/02/02 11:13:18 ski 3fe117f1b5e4a0528f9c1e8be9cca352d3264cf2 event cem-evcc-UseCaseSupportUpdate
[eebus ] DEBUG 2025/02/02 11:13:18 Send: d:_i:47859_Elli-Wallbox-2145A1RSRN call 11 NodeManagementSubscriptionRequestCall
```

### What type of operating system or environment does evcc run on?

other

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.133.0

andig, 2025-02-02:
@DerAndereAndi mal wieder Elli- lohnt es hier rein zu schauen?

DerAndereAndi, 2025-02-02:
Hallo, die Software der Elli spinnt. Es gab einen Verbindungsabbruch und die EEBUS Software auf der Wallbox ist gecrasht. Dann gibt es einen neuen Verbindungsaufbau während das Fahrzeug angesteckt ist. Was danach passiert ist in dem Log nicht zu sehen, aber ich vermute die Wallbox hat nicht mehr gemeldet dass ein Auto angesteckt ist und lädt. Während die Verbindung zwischen evcc und der Wallbox weg ist, darf das Auto mit 6A laden, das ist ein Fallback Modus in der Wallbox.

Das ganze lässt sich nicht lösen außer Elli fixt die Bugs. Was kann ein Anwender tun: Auto abstecken, ein paar Sekunden warten, wieder anstecken.

closed 2025-02-02 by andig

andig, 2025-02-02:
@DerAndereAndi ich überlege ob sich das- die crashes- irgendwie ohne Logfile lesen identifizieren liesse. Idee: Fehler melden und darauf hinweisen dass es an Elli liegt, nicht an Stack oder evcc. Auf welches Event müsste man als gute Indikation für den Crash schauen- remove SKI?

netguru76, 2025-02-02:
An und abstecken hatte leider nichts bewirkt .
Wenn die Verbindung zwischen Wallbox und evcc weg ist , lädt man nur mit 4,1 kw , korrekt .
Allerdings wird das dann auch immer durch die LED (eines gelb statt grün ) angezeigt .
Hier ist alles grün .
Seit der alten Version muss sich da in evcc was geändert haben .

andig, 2025-02-02:
@netguru76 das mag alles sein. Dennoch können wir dir leider nicht helfen. Die Software der Box ist so fehlerhaft dass ein standard-konformer EEbus Stack zusammen mit evcc damit leider nicht stabil läuft. Der Ball liegt bei der Firma die dir dieses Schmuckstück verkauft hat. Es günstige Alternativen die jahrelang stabil laufen…

netguru76, 2025-02-02:
Welche?
Kaufe ungern alle zwei Jahre ein wallbox inklusive kosten für den Anschluß ...
Ist die elli 2 fehlerfrei(er)

DerAndereAndi, 2025-02-02:
> Seit der alten Version muss sich da in evcc was geändert haben .

Nein, es hat sich auf EEBUS Seite in evcc nichts. gar nichts geändert.

> ich überlege ob sich das- die crashes- irgendwie ohne Logfile lesen identifizieren liesse. Idee: Fehler melden und darauf hinweisen dass es an Elli liegt, nicht an Stack oder evcc

In dem konkreten Fall war es nicht mal ein Crash, aber die Box hat einen Fehler gemeldet und die Verbindung selbst geschlossen. 

Die Software in der Wallbox ist leider Schrott. Siehe https://github.com/evcc-io/evcc/discussions/15367 Ich renne hier seit mehreren Jahren (!!) hinterher, irgendwann mal muss gut sein. Es wird für diese Wallbox keine Bugfixes geben und von außerhalb kann man hier rein gar nichts machen. Mit Verlaub: Es gibt niemanden der EEBUS besser kennt als ich, hier ist nichts zu machen. Ich habe echt keine Lust mehr mich immer und immer wieder zu wiederholen. Diese Box ist Schrott und das kann nur Elli und ihr Lieferant beheben.

Die Elli 2 ist von einem anderen Zulieferer, die läuft deutlich besser und bekommt Updates.

andig, 2025-02-02:
Ich würde kein Elli Gerät kaufen angesichts der Unfähigkeit auch nur die einfachsten Fehler zu beheben…

DerAndereAndi, 2025-02-02:
> angesichts der Unfähigkeit auch nur die einfachsten Fehler zu beheben…

Ich habe und verwende die Elli 2. Elli schreibt selbst keine Software, die Geräte sind von unterschiedlichen Zulieferern. Das ist ein Vergleich von Äpfeln mit Birnen.

andig, 2025-02-02:
> Elli schreibt selbst keine Software, die Geräte sind von unterschiedlichen Zulieferern. Das ist ein Vergleich von Äpfeln mit Birnen.

Das ist ein schönes Argument- hilft aber dem Kunden aber leider gar nicht. Daher würde ich persönlich keine Produkte von Elli kaufen. Aber das ist hier OT, da hast Du recht...

netguru76, 2025-02-02:
Mir ging es eher um eine Alternative bzw von elli ein günstiges Angebot für die Version 2 zu bekommen .

netguru76, 2025-02-13:
@DerAndereAndi @andig 
Ich bin zurück auf 131.8 und werde schauen, ob der Sprung auf 131.9 der break ist, mit 131.8 funktoniert wieder alles. Es lag also nicht auf dem move zu 133.
Ich nutze zwei Sungrow Hybrid, da gab es ein Update in der 131.9, vielleicht liegt es daran.

Melde mich, ob es an den Änderungen zur 131.9 liegt.
Vielleicht ist das ja leicht zu fixen.

TSNr1, 2025-03-04:
@netguru76 hast du schon was rausgefunden?

Alternativ würde ich einfach mal einen Neustart der Elli probieren. Die macht solche Spierenzchen nämlich auch ohne das irgendein Lademanagement dran hängt.

Getreu dem altbewehrten Motto "Reboot tut gut" einfach mal neustarten.

netguru76, 2025-03-04:
Klappt mal und mal nicht .
Prinzipiell komm ich zurecht 👍

TSNr1, 2025-03-04:
Ich bin mittlerweile auf 200.5 da läuft bei mir alles reibungslos.
zwei Elli Charger Connect V1 WLAN
Ein Tesla MY
Ein Skoda Enyaq