evcc-io/evcc #18565: Solar Edge

Tubbi22, 2025-02-02:
### Describe the bug

Hallo, ich habe mein evcc so weit am laufen. Offenbar passen die Werte jedoch nicht. 

Batterie wird richtig angezeigt, Netz wohl auch.
Einzig bei PV scheint nur ein Wechselrichter angezeigt zu werden. Es sind 2 SE 10KRWB angeschlossen. Einer ist "Solar edge leader" einer WR ist "RS486 follower"

Danke

[image]

### Steps to reproduce

1. keine Fehlermeldungen
2. Daten werden als Erzeugung / in angezeigt, sind aber zu gering (ca 50%) 
3.
...

### Configuration details

```yaml
meters:

name: my_pv
type: template
template: solaredge-hybrid
usage: pv
modbus: tcpip
id: 1
host: 192.168.1.xxx
port: 502
timeout: 10s
name: my_grid
type: template
template: tibber-pulse
usage: grid
token: xxxx
timeout: 10s
name: my_battery
type: template
template: solaredge-hybrid
usage: battery
modbus: tcpip
id: 1
host: 192.168.1.xxxx
port: 502
timeout: 10s
```

### Log details

```text
[lp-1 ] DEBUG 2025/01/28 14:39:46 charge power: 1905W
[lp-1 ] DEBUG 2025/01/28 14:39:46 charge currents: [9 0 0]A
[site ] DEBUG 2025/01/28 14:39:46 grid power: -1585W
[site ] DEBUG 2025/01/28 14:39:46 grid currents: [0 0 0]A
[site ] DEBUG 2025/01/28 14:39:47 battery 1 power: -0W
[site ] DEBUG 2025/01/28 14:39:47 battery 1 soc: 99%
[site ] DEBUG 2025/01/28 14:39:47 pv 1 power: 2922W
[site ] DEBUG 2025/01/28 14:39:47 site power: -1585W
[lp-1 ] DEBUG 2025/01/28 14:39:47 charge voltages: [229 233 233]V
[lp-1 ] DEBUG 2025/01/28 14:39:47 detected connected phases: 3p
[lp-1 ] DEBUG 2025/01/28 14:39:47 detected active phases: 1p
[lp-1 ] DEBUG 2025/01/28 14:39:47 charge total import: 1901.959kWh
[lp-1 ] DEBUG 2025/01/28 14:39:47 charger status: C
[lp-1 ] DEBUG 2025/01/28 14:39:47 vehicle status: C
[lp-1 ] DEBUG 2025/01/28 14:39:47 vehicle soc: 25%
[lp-1 ] DEBUG 2025/01/28 14:39:47 vehicle range: 0km
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

andig, 2025-02-02:
> Einzig bei PV scheint nur ein Wechselrichter angezeigt zu werden. Es sind 2 SE 10KRWB angeschlossen. Einer ist "Solar edge leader" einer WR ist "RS486 follower"

Sagt mir nichts. Da muss jemand ran der sich mit SE auskennt.

StephanKleinEppendorf, 2025-02-03:
Moin
Beide Wechselrichter haben in deiner Configuration die id 1 (Modbus ID), das kann eigentlich nicht richtig sein.
Sind die beiden WR per RS485 mit einander verbunden?

CaptainRoot, 2025-02-03:
Moin,
laut deiner configuration haben beide Wechselrichter die ID 1 (Modbus ID) das kann eigentlich nicht richtig sein.
Sind deine beiden Wechselrichter per RS485 miteinander verbunden?

CaptainRoot, 2025-02-03:
Moin,
Prüf bitte einmal ob der Haupt Wechselrichter auch die Modbus id 1 hat, nicht das dein Solareur dem Follower die 1 gegeben hat und dem HauptWechselrichter eine andere id.

Tubbi22, 2025-02-03:
Das sieht richtig aus, denke ich

[image]

Tubbi22, 2025-02-03:
> Wenn ich die Dokumentation und den Quellcode richtig verstanden habe musst du die Wechselrichter einzeln anlegen. Und dann beide in side referenzieren
> 
> https://docs.evcc.io/docs/reference/configuration/site
> 
> ` meters:
> 
> name: my_pv type: template template: solaredge-hybrid usage: pv modbus: tcpip id: 1 host: 192.168.1.xxx port: 502 timeout: 10s
> 
> name: my_pv_2 type: template template: solaredge-hybrid usage: pv modbus: tcpip id: ? host: 192.168.1.xxx port: 502 timeout: 10s `

>Hi, hatte ich auch schon versucht und erst so verstanden: je WR ein Eintrag - das hat nicht geklappt. Im SE Menü ist ein WR als Lead devinniert mit Device ID 1, der andere als "Follower", ohne eigene ID