evcc-io/evcc #18569: Enphase get too much requests

jaytonic, 2025-02-02:
### Describe the bug

I've an enphase system, so I get both PV requests and grid request going to it.
The issue is that both the grid and the PV part requests the same file. Also, only the PV template requests the 5 times(according to what I read here: https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/enphase.yaml )

So with the solar request, it makes 6 requests. When I manually request the file, it takes ~2-3s to request it.
Currently when I run my EVCC instance, I do get a lot of timeout. It varies a lot but it goes up to 4/5 of the requests being timed out.
I've other stuff requesting data(READ ONLY, nothing active) from my enphase envoy(home assistant and potentially the app).

I see two issues:

1.  There is no way to set a timeout with the integration
2. It's a shame that the same exact file get requested 6 times. 

Do you see any chance to reduce the number of requests made, especially for the PV implementation?

### Steps to reproduce

1. Use both the grid and the PV implementation of enphase
2. Run it in the day with some other system running
3. Check the timeouts
...

### Configuration details

```yaml
meters:
  - name: my_grid
    type: template
    template: enphase
    usage: grid
    host: 192.168.0.196 # IP address or hostname
    token: AAAAA

  - name: my_pv
    type: template
    template: enphase
    usage: pv
    host: 192.168.0.196 # IP address or hostname
    token: AAAAA

tariffs:
  currency: CHF
  grid:
    type: fixed
    price: 0.2917 # EUR/kWh (default)
    zones:
      - days: Mo-So
        hours: 21-0
        price: 0.1881 # EUR/kWh (weekdays 2-6 hours)
      - days: Mo-So
        hours: 0-7
        price: 0.1881 # EUR/kWh (weekdays 2-6 hours)
chargers:
  - name: easee_home
    type: template
    template: easee
    user: ABC
    password: DEF
    charger: IJK
    timeout: 30s # optional
    authorize: 
vehicles:
  - name: vehicle_1
    title: Audi Q6
    type: custom
    soc:
      source: js
      vm: shared
      script: |
        var lp = state.loadpoints[0];
        if (lp.chargepower > 0) lp.vehicleSoc+=0.1; else lp.vehicleSoc-=0.1;
        if (lp.vehicleSoc < 15) lp.vehicleSoc = 80;
        if (lp.vehicleSoc > 80) lp.vehicleSoc = 15;
        lp.vehicleSoc;
    range:
      source: js
      vm: shared
      script: |
        var lp = state.loadpoints[0]
        var range = (44 * lp.vehicleSoc) / 15;
        range
    capacity: 44
site:
  title: Maison
  meters:
    grid: my_grid
    pv: my_pv

loadpoints:
  - title: Couvert
    charger: easee_home
    mode: pv
    vehicle: vehicle_1

sponsortoken: my-token
interval: 60s
```

### Log details

```text
[site ] DEBUG 2025/02/02 16:24:53 ----
[lp-1 ] DEBUG 2025/02/02 16:24:53 charge power: 0W
[lp-1 ] DEBUG 2025/02/02 16:24:53 charge currents: [0 0 0]A
[site ] DEBUG 2025/02/02 16:24:54 grid power: -340W
[site ] DEBUG 2025/02/02 16:24:57 pv 1 power: 852W
[site ] DEBUG 2025/02/02 16:25:08 grid currents: [-0.907 -1.02 0.712]A
[site ] DEBUG 2025/02/02 16:25:08 site power: -340W
[lp-1 ] DEBUG 2025/02/02 16:25:08 charge total import: 1.744kWh
[lp-1 ] DEBUG 2025/02/02 16:25:08 charger status: A
[lp-1 ] DEBUG 2025/02/02 16:25:08 plan: charge 2h1m28s between 2025-02-03 04:13:32 +0100 CET until 2025-02-03 06:15:00 +0100 CET (power: 14490W, avg cost: 0.188)
[site ] DEBUG 2025/02/02 16:25:53 ----
[lp-1 ] DEBUG 2025/02/02 16:25:53 charge power: 0W
[lp-1 ] DEBUG 2025/02/02 16:25:53 charge currents: [0 0 0]A
[site ] ERROR 2025/02/02 16:26:03 pv 1 power: Get "https://192.168.0.196/production.json": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[site ] ERROR 2025/02/02 16:26:03 grid power: Get "https://192.168.0.196/production.json": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[site ] DEBUG 2025/02/02 16:26:53 ----
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

dgaust, 2025-02-02:
Yeah, I had a similar issue, but I really think it's on the enphase firmware - responses from the old APIs are ridiculously slow. Not sure what version you're on but I have version 8ish.

This is not a resolution for your issue, but I use this https://github.com/vk2him/Enphase-Envoy-mqtt-json as part of my Home Assistant monitoring. For some reason, the 1s resolution of this monitoring is no problem for the envoy-s. Since it publishes to MQTT, I just point EVCC to the mqtt values for my production and usage

```
meters:
  - type: custom
    name: grid1
    power:
      source: mqtt
      topic: /envoy/json
      jq: .[1].activePower | tonumber
      timeout: 30s
      scale: 1

  - type: custom
    name: mqtt_production
    power:
      source: mqtt
      topic: /envoy/json
      jq: .[0].activePower | tonumber
      timeout: 30s # don't accept values older than timeout
      scale: 1
```

jaytonic, 2025-02-03:
> Yeah, I had a similar issue, but I really think it's on the enphase firmware - responses from the old APIs are ridiculously slow. Not sure what version you're on but I have version 8ish.
>
> This is [...]

Two questions though:
1. My home assistant already receives the data(at 1s update rate) from   my enphase. Is there a way to publish data from this directly to MQTT from my home assistant?
2. I see you use only two values from enphase, from what I see from the enphase template, there are other(like currents), don't you need them?

Anyway, since Your python script, home assistant and the enphase apps are able to get some live data without any issue, I guess there is some kind of issue with the way evcc request data. From what I see, the fact that it requests 6x the data on each refresh could be it, or other app might use another endpoint?

andig, 2025-02-03:
If enphase did send cache headers, evcc should already do caching. Seems it doesn't though. Right now there is no additional logic in the templates that could help here.

VolkerK62, 2025-02-03:
what about a `timeout` parameter within the template?

dgaust, 2025-02-03:
> Two questions though:
> 
> 1. My home assistant already receives the data(at 1s update rate) from   my enphase. Is there a way to publish data from this directly to MQTT from my home assistant?

It probably doesn't get it every second, but every 60s. The standard enphase HA integration only updates in 60s intervals.

> 3. I see you use only two values from enphase, from what I see from the enphase template, there are other(like currents), don't you need them?> 

For EVCC it only needs these to calculate the production and consumption graphs. I display them in HA in my energy dashboard, but it's just information.

> Anyway, since Your python script, home assistant and the enphase apps are able to get some live data without any issue, I guess there is some kind of issue with the way evcc request data. From what [...]

The live data is a different endpoint to the production.json endpoint that EVCC uses. The overrequest might be an issue, but I've been having the same issues with the stock HA enphase integration that hits those same API endpoints.

andig, 2025-02-04:
@Maschga du hattest vor einiger Zeit einen Caching PR gemacht. Siehst Du eine Möglichkeit, den hier nochmal einzubringen? Konkret: ich stelle mir vor, eine Middleware in die HTTP Response mit einzubauen die Cache header injiziert. Das müsste passieren bevor der HTTP Cache die Resonpse zu sehen bekommt. Wäre das eine Option?

Maschga, 2025-02-04:
Hi, ich weiß nicht genau auf welchen PR du dich beziehst. Hast du vielleicht die falsche Person gepingt oder übersehe ich den PR? 😅

Lenart12, 2025-02-04:
Untested... something like this?

[0001-feat-http-Add-force-cache-headers-option.patch](https://github.com/user-attachments/files/18660035/0001-feat-http-Add-force-cache-headers-option.patch)

andig, 2025-02-04:
Oh yeah- looks straight forward. Would you potentially be able to test and open a PR?

Lenart12, 2025-02-04:
Another option which I would be more inclined against, but has more change of breaking things. LMK for which one I create a PR.

[0001-feat-http-Replace-second-layer-of-http-cache-with-se.patch](https://github.com/user-attachments/files/18660411/0001-feat-http-Replace-second-layer-of-http-cache-with-se.patch)

andig, 2025-02-04:
I need to think a bit about this. Once you have a single value caching that is nice. The problem here is that we have no notion of "single value".

andig, 2025-02-04:
Did think (a bit). Not sure there's an analytical solution. I like the second approach because it's one config options less. If it doesn't work we can still retract it :)

closed 2025-02-05 by andig

ButtiBBQ, 2025-08-17:
Just to let you know. this one was improved, by possibility to switch to https, better Caching, Mutex for same url (only one request to the gateway at a time), and same url within the envoy template. So if you have a own config and not using the template, always use https://envoy.local/production.json?details=1

PR #18601 (2025-02-05): feat(http): Replace second layer of http cache with setting cache headers
Fixes #18569