evcc-io/evcc #18578: Ladevorgang über EVCC wird nicht beendet

steinpilz71, 2025-02-03:
### Describe the bug

Ladevorgang über EVCC wird nicht beendet, wenn zuwenig PV-Überschuss vorhanden ist.
Und Ladevorgang über EVCC wird nicht beendet , wenn über Strompreisanstieg nicht mehr geladen werden soll. Wallbox läd bis das Auto voll ist bzw. am Auto eingestellter Soc erreicht ist.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
log: info
network:
  host: homeassistant.local
levels:
  ocpp: trace
tariffs:
  grid:
    type: template
    template: tibber
    token: "xxx" # access token
    homeId: "f70b1bb4-eb1a-42b3-8b51-4f240b78fa39"
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # EUR/kWh
mqtt:
    broker: homeassistant.local:1883
    user: evcc
    password: evcc
modbusproxy:
  - port: 502
    device: /dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0
    baudrate: 9600
    comset: "8N1"
meters:
  - name: my_grid
    type: template
    template: growatt-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: homeassistant.local
    port: 502
  - name: my_pv
    type: template
    template: growatt-hybrid
    usage: pv
    modbus: tcpip
    id: 1
    host: homeassistant.local
    port: 502
  - name: my_battery
    type: template
    template: growatt-hybrid
    usage: battery
    capacity: 4.8
    modbus: tcpip
    id: 1
    host: homeassistant.local
    port: 502
#  - name: my_grid
#    type: custom
#    power:
#      source: mqtt
#      topic: pvdataevcc/sensor/power_grid_1
#      scale: 1
#      timeout: 30s
#    energy:
#      source: mqtt
#      topic: pvdataevcc/sensor/energy_grid_1
#      scale: 1
#      timeout: 30s
#  - name: my_pv
#    type: custom
#    power:
#      source: mqtt
#      topic: pvdataevcc/sensor/power_pv_1
#      scale: 1
#      timeout: 30s
#    energy:
#      source: mqtt
#      topic: pvdataevcc/sensor/energy_pv_1
#      scale: 1
#      timeout: 30s
#  - name: my_battery
#    type: custom
#    power:
#      source: mqtt
#      topic: pvdataevcc/sensor/power_battery_1
#      timeout: 30s
#      scale: -1
#    energy:
#      source: mqtt
#      topic: pvdataevcc/sensor/energy_battery_1
#      scale: 1
#      timeout: 30s
#    soc:
#      source: mqtt
#      topic: pvdataevcc/sensor/soc_battery_1
#      timeout: 30s
chargers:
  - name: my_charger
    type: template
    template: phoenix-ev-eth
    
    # Modbus TCP
    modbus: tcpip
    id: 255
    host: 192.168.2.8 # Hostname
    port: 502 # Port # IP-Adresse oder Hostname
site:
  - title: Zuhause # display name for UI

    meters:
      grid: my_grid # grid meter reference
      pv:
        - my_pv # pv meter reference
      battery:
        - my_battery
    residualPower: 50
loadpoints:
  - title: Carport # display name for UI
    charger: my_charger
    mode: pv # charge mode (off, now, minpv, pv)
    soc:
        estimate: true
        poll:
            mode: charging
            interval: 600m
vehicles:
 - name: Surprise
   title: Surprise
   capacity: 60 # kWh
   type: template
   template: tesla-command # change `tesla` to `tesla-command`
   accessToken: xxx # generetad by tesla.evcc.io
   refreshToken: xxx # generetad by tesla.evcc.io
   
sponsortoken: xxx
```

### Log details

```text
[lp-1 ] DEBUG 2025/02/03 12:56:44 charge power: 0W
[site ] DEBUG 2025/02/03 12:56:45 grid power: 15W
[site ] DEBUG 2025/02/03 12:56:45 pv 1 power: 2150W
[site ] DEBUG 2025/02/03 12:56:45 battery 1 power: -1590W
[site ] DEBUG 2025/02/03 12:56:45 battery 1 soc: 51%
[site ] DEBUG 2025/02/03 12:56:45 site power: -1525W
[lp-1 ] DEBUG 2025/02/03 12:56:45 charger status: B
[site ] DEBUG 2025/02/03 12:56:54 ----
[lp-1 ] DEBUG 2025/02/03 12:56:54 charge power: 0W
[site ] DEBUG 2025/02/03 12:56:55 grid power: 6W
[site ] DEBUG 2025/02/03 12:56:55 pv 1 power: 2160W
[site ] DEBUG 2025/02/03 12:56:55 battery 1 power: -1600W
[site ] DEBUG 2025/02/03 12:56:55 battery 1 soc: 51%
[site ] DEBUG 2025/02/03 12:56:55 site power: -1544W
[lp-1 ] DEBUG 2025/02/03 12:56:55 charger status: B
[site ] DEBUG 2025/02/03 12:57:04 ----
[lp-1 ] DEBUG 2025/02/03 12:57:04 charge power: 0W
[site ] DEBUG 2025/02/03 12:57:05 grid power: -60W
[site ] DEBUG 2025/02/03 12:57:05 pv 1 power: 2160W
[site ] DEBUG 2025/02/03 12:57:05 battery 1 power: -1670W
[site ] DEBUG 2025/02/03 12:57:05 battery 1 soc: 51%
[site ] DEBUG 2025/02/03 12:57:05 site power: -1680W
[lp-1 ] DEBUG 2025/02/03 12:57:05 charger status: B
[site ] DEBUG 2025/02/03 12:57:14 ----
[lp-1 ] DEBUG 2025/02/03 12:57:14 charge power: 0W
[site ] DEBUG 2025/02/03 12:57:15 grid power: 0W
[site ] DEBUG 2025/02/03 12:57:15 pv 1 power: 2170W
[site ] DEBUG 2025/02/03 12:57:15 battery 1 power: -1650W
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

StefanSchoof, 2025-02-03:
`charge power: 0W`

Wenn da gerade ein Ladevorgang ist, misst dein Charger meter den nicht.

andig, 2025-02-03:
> charger status: B

Da ist keiner.

closed 2025-02-03 by andig