evcc-io/evcc #18582: Only Priority 0 and 1 can be selected for loadpoints in new evcc configuration GUI

walburgf, 2025-02-03:
### Describe the bug

When using the new GUI for creating a loadpoint, only Piority 0 and 1 can be choosen.
In evcc.yaml, priority up to 9 could be  defined. It should be possible to define the same number of priorities as in evcc.yaml

[image]

### Steps to reproduce

1. create a new loadpoint an try to set priority of the loadpoint
2.
3.
...

### Configuration details

```yaml
[main  ] INFO 2025/02/03 16:33:03 evcc 0.133.0 (b0c7750b)
[main  ] INFO 2025/02/03 16:33:03 using config file: /home/evcc/evcc.yaml
[db    ] INFO 2025/02/03 16:33:03 using sqlite database: /var/lib/evcc/evcc.db
[mqtt  ] INFO 2025/02/03 16:33:03 connecting evcc-1225127123 at tcp://192.168.1.227:1883
[mqtt  ] DEBUG 2025/02/03 16:33:03 tcp://192.168.1.227:1883 connected
[lp-1  ] WARN 2025/02/03 16:33:07 deprecated: title setting is ignored, please remove
[lp-1  ] WARN 2025/02/03 16:33:07 deprecated: soc setting is ignored, please remove
[lp-2  ] WARN 2025/02/03 16:33:07 deprecated: mode setting is ignored, please remove
[lp-2  ] WARN 2025/02/03 16:33:07 deprecated: soc setting is ignored, please remove
[lp-2  ] DEBUG 2025/02/03 16:33:07 set title: Garage rechts
[lp-2  ] DEBUG 2025/02/03 16:33:07 set priority: 1
[lp-2  ] DEBUG 2025/02/03 16:33:07 set smart cost limit: <nil>
[lp-2  ] DEBUG 2025/02/03 16:33:07 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
[lp-2  ] DEBUG 2025/02/03 16:33:07 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-2  ] DEBUG 2025/02/03 16:33:07 set session energy limit: 0
[lp-2  ] DEBUG 2025/02/03 16:33:07 set session soc limit: 0
[lp-2  ] DEBUG 2025/02/03 16:33:07 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0xc000d61148}
[lp-2  ] DEBUG 2025/02/03 16:33:07 set default mode: pv
[lp-2  ] DEBUG 2025/02/03 16:33:07 set phases: 0
[lp-2  ] DEBUG 2025/02/03 16:33:07 charge total import: 2700.202kWh

Konfiguration (/home/evcc/evcc.yaml):

# open evcc at http://evcc-pve.walburg.net:7070
network:
  schema: http
  host: evcc-pve.walburg.net # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  site: debug
  lp-2: trace
  script: trace
  cache: error

database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db

# unique installation id
plant: *****
telemetry: true
sponsortoken: *****
interval: 15s # control cycle interval, default is 30s

# ############################################################################################################
chargers:

- name: AC_Thor_9s_charger
  type: template
  template: ac-thor
  modbus: tcpip
  id: 1
  tempsource: "2"
  host: 192.168.1.231
  port: 5021

# ############################################################################################################
loadpoints:

- charger: AC_Thor_9s_charger
  title: AC Thor 9s
  priority: 0
  enable:
    threshold: -100
    delay: 30s
  disable:
    threshold: -50
    delay: 60s

Version: `0.133.0 (b0c7750b)`
```

### Log details

```text
[main ] INFO 2025/02/03 16:32:26 evcc 0.133.0 (b0c7750b)
[main ] INFO 2025/02/03 16:32:26 using config file: /etc/evcc.yaml
[db ] INFO 2025/02/03 16:32:26 using sqlite database: /var/lib/evcc/evcc.db
[mqtt ] INFO 2025/02/03 16:32:27 connecting evcc-1562902702 at tcp://192.168.1.227:1883
[mqtt ] DEBUG 2025/02/03 16:32:27 tcp://192.168.1.227:1883 connected
[main ] INFO 2025/02/03 16:32:27 listening at :7070
[proxy-5021] DEBUG 2025/02/03 16:32:27 modbus proxy for 192.168.1.248:502 listening at :5021
[proxy-5022] DEBUG 2025/02/03 16:32:27 modbus proxy for 192.168.1.247:502 listening at :5022
[lp-1 ] WARN 2025/02/03 16:32:30 deprecated: title setting is ignored, please remove
[lp-1 ] WARN 2025/02/03 16:32:30 deprecated: soc setting is ignored, please remove
[lp-2 ] WARN 2025/02/03 16:32:30 deprecated: mode setting is ignored, please remove
[lp-2 ] WARN 2025/02/03 16:32:30 deprecated: soc setting is ignored, please remove
[lp-2 ] DEBUG 2025/02/03 16:32:30 set title: Garage rechts
[lp-2 ] DEBUG 2025/02/03 16:32:30 set priority: 1
[lp-2 ] DEBUG 2025/02/03 16:32:30 set smart cost limit: <nil>
[lp-2 ] DEBUG 2025/02/03 16:32:30 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
[lp-2 ] DEBUG 2025/02/03 16:32:30 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-2 ] DEBUG 2025/02/03 16:32:30 set session energy limit: 0
[lp-2 ] DEBUG 2025/02/03 16:32:30 set session soc limit: 0
[lp-2 ] DEBUG 2025/02/03 16:32:30 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0xc000f3b858}
[lp-2 ] DEBUG 2025/02/03 16:32:30 set default mode: pv
[lp-2 ] DEBUG 2025/02/03 16:32:30 set phases: 0
[lp-2 ] DEBUG 2025/02/03 16:32:30 charge total import: 2700.202kWh
[site ] INFO 2025/02/03 16:32:31 site config:
[site ] INFO 2025/02/03 16:32:31 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2025/02/03 16:32:31 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2025/02/03 16:32:31 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2025/02/03 16:32:31 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site ] INFO 2025/02/03 16:32:31 vehicles:
[site ] INFO 2025/02/03 16:32:31 vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1 ] INFO 2025/02/03 16:32:31 loadpoint 1:
[lp-1 ] INFO 2025/02/03 16:32:31 mode: off
[lp-1 ] INFO 2025/02/03 16:32:31 charger: power ✓ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1 ] INFO 2025/02/03 16:32:31 meters: charge ✓
[lp-1 ] INFO 2025/02/03 16:32:31 charge: power ✓ energy ✗ currents ✗
[lp-2 ] INFO 2025/02/03 16:32:31 loadpoint 2:
[lp-2 ] INFO 2025/02/03 16:32:31 mode: pv
[lp-2 ] INFO 2025/02/03 16:32:31 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-2 ] INFO 2025/02/03 16:32:31 meters: charge ✓
[lp-2 ] INFO 2025/02/03 16:32:31 charge: power ✓ energy ✓ currents ✓
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.133.0 (b0c7750b)

VolkerK62, 2025-02-03:
Ich vermute, das ist abhängig von der Zahl der Laodpoints.
Bei 2 Loadpoints reicht 0 und 1.
Wenn du weitere Loadpoints anlegst, vergrößert sich die Auswahl.

TobiasHuber1980, 2025-02-03:
> Ich vermute, das ist abhängig von der Zahl der Laodpoints. Bei 2 Loadpoints reicht 0 und 1. Wenn du weitere Loadpoints anlegst, vergrößert sich die Auswahl.

das kann ich bestätigen!

[image]

steps56, 2025-02-03:
Habe ich Etwas verpasst ?
Wo hat man denn diese Gui ?
Nutze HA und Evcc Addon, Alles in evcc.yaml konfiguriert.
Addon hat da nur paar Sachen in Menüs...

andig, 2025-02-04:
@naltatis ich würde hier auch weniger Magie vorschlagen und immer Prio 0-9 erlauben.

walburgf, 2025-02-04:
> Danke für Nichts bzw. das Nichterklären 🤔🙄👎.

de GUI ist in der aktuellen Beta Version enthalten

steps56, 2025-02-04:
> > Danke für Nichts bzw. das Nichterklären 🤔🙄👎.
> 
> de GUI ist in der aktuellen Beta Version enthalten

Danke.
Bedeutet da Beta/nightly (?) ja nicht in HA nutzbar/installierbar, dass das nur auf Windows oder Linux PC/Nas oder (?) sichtbar ist?
(Habe gerade vorher Andig dazu angeschrieben)

Oder gibt es ein Beta Addon für HA... und falls ja, wo/wie ?
Sorry, bin da nur etwas verwirrt, da bisher sinnvoll Evcc nur unter HA für micht nutzbar ist.
Grüße und danke nochmals auch für weitere Infos.

closed 2025-02-07 by andig