evcc-io/evcc #18583: Battery Control: Incorrect/Switched description for minsoc and maxsoc

Maschga, 2025-02-03:
### Describe the bug

Die Batterie wird nach derImplementierung  bis zum maxsoc geladen, also bis zu 97%. Die Beschreibung im Bild müsste also zu maxsoc und nicht zu minsoc:

[image]

### Steps to reproduce

1. Nightly installieren
2. Die RCT Batterie im Config UI hinzufügen
3. Advanced Einstellungen ansehen

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

Maschga, 2025-02-03:
Im Zuge dessen: Es kam auch noch der Wunsch auf, (zumindest für das RCT Battery Control) die Ladeleistung festsetzen zu können, beispielsweise auf 6kW anstatt von 10kW, um die Batterie etwas zu schonen.
Ist das Offtopic hier oder sollte das Issue hier genutzt werden, um generell BatteryControl spezifische Parameter zu überdenken?

VolkerK62, 2025-02-03:
Die Beschreibung ist falsch.
Die gehört zum `minsoc` beim vehicle.

naltatis, 2025-02-07:
Duplikat zu https://github.com/evcc-io/evcc/issues/17662

closed 2025-02-07 by andig