evcc-io/evcc #18584: SOC für Polestar lässt sich nicht auslesen

Fuellhase, 2025-02-03:
### Describe the bug

Requesting SOC for Polestar 2 under openWB Version: 1.9.303.0 doesn't work anymore.

### Steps to reproduce

1. Request SOC
2. See that SOC is 0
3. View Status messages
2.
3.
...

### Configuration details

```yaml
The configuration was not changed. Until noon it still worked.
```

### Log details

```text
2025-02-03 17:16:40: PID: 2923: Lp1: Error from EVCC: 2025/02/03 17:16:40 rpc error: code = Unknown desc = Message: Validation error of type FieldUndefined: Field 'getBatteryData' in type 'Query' is undefined @ 'getBatteryData', Locations: [{Line:1 Column:36}], Extensions: map[], Path: [] (LV0)

2025-02-03 17:16:35: PID: 2923: Lp1: Requesting SoC (LV0)
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

ottomitschke, 2025-02-03:
`lp-1 ] INFO 2025/02/03 19:29:58 car connected
[lp-1 ] DEBUG 2025/02/03 19:29:58 vehicle api refresh
[lp-1 ] DEBUG 2025/02/03 19:29:58 pv timer elapse
[lp-1 ] DEBUG 2025/02/03 19:29:58 pv timer inactive
[lp-1 ] DEBUG 2025/02/03 19:29:58 charge total import: 18173.915kWh
[main ] ERROR 2025/02/03 19:29:58 vehicle status: Message: Validation error of type FieldUndefined: Field 'getBatteryData' in type 'Query' is undefined @ 'getBatteryData', Locations: [{Line:1 Column:36}], Extensions: map[], Path: []
[lp-1 ] DEBUG 2025/02/03 19:29:58 pv charge current: 0A = 0A + -0.813A (561W @ 3p)
[lp-1 ] DEBUG 2025/02/03 19:29:58 pv enable timer reset
[lp-1 ] DEBUG 2025/02/03 19:29:58 pv timer inactive
[site ] DEBUG 2025/02/03 19:29:58 ----
[lp-1 ] DEBUG 2025/02/03 19:29:59 charge power: 0W
[lp-1 ] DEBUG 2025/02/03 19:30:02 charge currents: [0 0 0]A
[site ] DEBUG 2025/02/03 19:30:02 pv 1 power: 4W
[site ] DEBUG 2025/02/03 19:30:02 battery 1 power: 427W
[site ] DEBUG 2025/02/03 19:30:02 grid power: 17W
[site ] DEBUG 2025/02/03 19:30:02 battery 1 soc: 14%
[site ] DEBUG 2025/02/03 19:30:02 site power: 544W
[lp-1 ] DEBUG 2025/02/03 19:30:03 charge total import: 18173.915kWh
[lp-1 ] DEBUG 2025/02/03 19:30:03 charger status: B
[main ] ERROR 2025/02/03 19:30:03 vehicle status: Message: Validation error of type FieldUndefined: Field 'getBatteryData' in type 'Query' is undefined @ 'getBatteryData', Locations: [{Line:1 Column:36}], Extensions: map[], Path: []
[lp-1 ] DEBUG 2025/02/03 19:30:03 pv charge current: 0A = 0A + -0.788A (544W @ 3p) `

hjwsg, 2025-02-03:
Same problem here since today.

loebse, 2025-02-04:
Polestar has changed the API again, I am working on a fix already, expect a PR soon

loebse, 2025-02-04:
Create PR https://github.com/evcc-io/evcc/pull/18594 that fixes the issue @andig

closed 2025-02-04 by andig

Fuellhase, 2025-03-02:
Hallo EVCC!Könnte mir bitte jemand mit wenigen Worten erklären, wie man den Polestar SOC Fix anwendet?Ich bin leider nur Elektronik-Ingenieur und kein Programmier-Profi. Go kenne ich bisher nur als Brettspiel!Danke!Peter FüllhaseZur Quelle 1448341 AltenbergeMail: ***@***.***

Closed #18584 as completed via #18594.

andig, 2025-03-02:
Der Fehler ist lange behoben…