evcc-io/evcc #18602: maximale Leistung zusätzlich zum maximalen Ladestrom

Captainbrille, 2025-02-04:
Hallo,

Mein Cupra Born auf MEB Platform kann einphasig maximal 32A laden und dreiphasig dann 16A, was 11kW entsprechen.
Um dies maximal auszunutzen und bis 7,2kW einphasig laden zu können und unnötig phasenwechsel zu machen, habe ich also automatischen Wechsel und 6 - 32A eingestellt.
Dies ist jedoch für die Berechnung bei geplanter Ladung falsch. Es wird dann von 22kw ausgegangen, obwohl das Fahrzeug maximal 11kw laden kann.

Ich wünsche mir daher eine Einstellung, entweder in der yaml oder gui in der eine maximale Leistung eingestellt werden kann.

[image]

StefanSchoof, 2025-02-04:
Siehe https://github.com/evcc-io/evcc/issues/14661

Captainbrille, 2025-02-04:
> Siehe [#14661](https://github.com/evcc-io/evcc/issues/14661)

Sehr gut. So wie da beschrieben wäre es auch eine gute Lösung.

closed 2025-02-04 by andig