evcc-io/evcc #18603: possible mqtt deadlock

jeffborg, 2025-02-05:
### Describe the bug

evcc stops updating meters after some time, no mqtt activity from meters solar etc..
Also disconnects from mqtt broker as indicated by the broker updating the evcc/status topic to offline

### Steps to reproduce

1. run evcc 0.133.0
2. wait some time
3. evcc will stop updating

### Configuration details

```yaml
interval: 30s

site:
  - title: Home
    circuit: main
    meters:
      grid: grid
      pv:
        - pv

circuits:
- name: main
  meter: grid
  maxpower: 2000
  GetMaxPower:
    source: mqtt
    topic: amber/max_demand
log: trace
levels:
  ocpp: trace

meters:
  - name: grid
    type: custom
    power:
      source: mqtt
      topic: shellies/house-energy-monitor/emeter/0/power
      timeout: 30s
      scale: 1
    energy:
      source: mqtt
      topic: shellies/house-energy-monitor/emeter/0/total
      timeout: 30s
      scale: 0.001
  - name: pv
    type: custom
    power:
      source: mqtt
      topic: energy/solar
      timeout: 30s
      scale: 1
      jq: .OutputPower
    energy:
      source: mqtt
      topic: energy/solar
      timeout: 30s
      jq: .TotalGenerateEnergy
chargers:
  - name: solax
    type: template
    template: ocpp
    remotestart: true
    metervalues: -Current.Offered
loadpoints:
  - title: Garage
    charger: solax
[... 611 lines cut ...]
[cache ] TRACE 2025/02/05 10:59:57 lp-1/chargerStatusReason: unknown
[cache ] TRACE 2025/02/05 10:59:57 lp-1/enabled: true
[cache ] TRACE 2025/02/05 10:59:57 lp-1/mode: pv
[mqtt  ] TRACE 2025/02/05 10:59:57 recv evcc/loadpoints/1/disableThreshold/set: '1312'
[site  ] TRACE 2025/02/05 10:59:57 POST https://api.evcc.io/v1/charge
[site  ] TRACE 2025/02/05 10:59:57 {"instanceId":"XXXXXXXXX","chargePower":1517,"greenPower":1481.0700000000002,"chargeEnergy":0,"greenEnergy":0}
--
{"status":"ok"}
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

jeffborg, 2025-02-05:
The log looks cut off but here is the end of it. there is no activity apart from ocpp

```
[cache ] TRACE 2025/02/05 10:59:57 lp-1/smartCostNextStart: 0001-01-01 00:00:00 +0000 UTC
[cache ] TRACE 2025/02/05 10:59:57 lp-1/chargeVoltages: [244.4 0 0]
[cache ] TRACE 2025/02/05 10:59:57 lp-1/chargeCurrent: 6
[lp-1  ] DEBUG 2025/02/05 10:59:57 charge total import: 324.435kWh
[cache ] TRACE 2025/02/05 10:59:57 lp-1/sessionEnergy: 0
[cache ] TRACE 2025/02/05 10:59:57 lp-1/sessionSolarPercentage: 0
[cache ] TRACE 2025/02/05 10:59:57 lp-1/sessionPricePerKWh: <nil>
[cache ] TRACE 2025/02/05 10:59:57 lp-1/sessionPrice: <nil>
[cache ] TRACE 2025/02/05 10:59:57 lp-1/sessionCo2PerKWh: <nil>
[cache ] TRACE 2025/02/05 10:59:57 lp-1/chargedEnergy: 0
[cache ] TRACE 2025/02/05 10:59:57 lp-1/chargeDuration: 0s
[cache ] TRACE 2025/02/05 10:59:57 lp-1/chargeTotalImport: 324.435
[cache ] TRACE 2025/02/05 10:59:57 lp-1/effectivePriority: 0
[cache ] TRACE 2025/02/05 10:59:57 lp-1/effectivePlanId: 2
[lp-1  ] DEBUG 2025/02/05 10:59:57 charger status: B
[cache ] TRACE 2025/02/05 10:59:57 lp-1/effectivePlanTime: 2025-02-05 15:00:00 +1100 AEDT
[cache ] TRACE 2025/02/05 10:59:57 lp-1/effectivePlanSoc: 50
[cache ] TRACE 2025/02/05 10:59:57 lp-1/effectiveMinCurrent: 6
[cache ] TRACE 2025/02/05 10:59:57 lp-1/effectiveMaxCurrent: 32
[cache ] TRACE 2025/02/05 10:59:57 lp-1/effectiveLimitSoc: 100
[cache ] TRACE 2025/02/05 10:59:57 lp-1/vehicleWelcomeActive: false
[cache ] TRACE 2025/02/05 10:59:57 lp-1/connected: true
[cache ] TRACE 2025/02/05 10:59:57 lp-1/charging: false
[cache ] TRACE 2025/02/05 10:59:57 lp-1/chargerStatusReason: unknown
[cache ] TRACE 2025/02/05 10:59:57 lp-1/enabled: true
[cache ] TRACE 2025/02/05 10:59:57 lp-1/mode: pv
[mqtt  ] TRACE 2025/02/05 10:59:57 recv evcc/loadpoints/1/disableThreshold/set: '1312'
[site  ] TRACE 2025/02/05 10:59:57 POST https://api.evcc.io/v1/charge
[site  ] TRACE 2025/02/05 10:59:57 {"instanceId":"XXXXXXXXX","chargePower":1517,"greenPower":1481.0700000000002,"chargeEnergy":0,"greenEnergy":0}
--
{"status":"ok"}
[ocpp  ] TRACE 2025/02/05 11:00:01 recv SN10052307203612: [2,"176301","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2025-02-05T00:00:00Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"6.1"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"243.7","context":"Sample.Periodic"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"324435.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"1506.0","context":"Sample.Periodic"}]}],"transactionId":1738268539}]
[ocpp  ] TRACE 2025/02/05 11:00:01 send SN10052307203612: [3,"176301",{}]
[ocpp  ] TRACE 2025/02/05 11:00:02 recv SN10052307203612: [2,"176302","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":100,\"cpVolt\":615,\"transactionId\":1738268539,\"totalPowerFactor\":246,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":1,\"ccVolt\":246,\"offset0\":2006,\"offset1\":2045,\"leakcurr\":20,\"AMBTemp\":410,\"lockStatus\":1,\"ctCurrent\":15}"}]
[ocpp  ] TRACE 2025/02/05 11:00:02 send SN10052307203612: [3,"176302",{"status":"Accepted"}]
[ocpp  ] TRACE 2025/02/05 11:00:11 recv SN10052307203612: [2,"176303","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2025-02-05T00:00:10Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"6.1"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"244.1","context":"Sample.Periodic"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"324448.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"1508.0","context":"Sample.Periodic"}]}],"transactionId":1738268539}]
[ocpp  ] TRACE 2025/02/05 11:00:11 send SN10052307203612: [3,"176303",{}]
[ocpp  ] TRACE 2025/02/05 11:00:12 recv SN10052307203612: [2,"176304","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":100,\"cpVolt\":615,\"transactionId\":1738268539,\"totalPowerFactor\":250,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":1,\"ccVolt\":250,\"offset0\":2006,\"offset1\":2045,\"leakcurr\":20,\"AMBTemp\":410,\"lockStatus\":1,\"ctCurrent\":15}"}]
[ocpp  ] TRACE 2025/02/05 11:00:12 send SN10052307203612: [3,"176304",{"status":"Accepted"}]
[ocpp  ] TRACE 2025/02/05 11:00:18 recv SN10052307203612: [2,"176305","Heartbeat",{}]
[ocpp  ] TRACE 2025/02/05 11:00:18 send SN10052307203612: [3,"176305",{"currentTime":"2025-02-05T00:00:18Z"}]
[ocpp  ] TRACE 2025/02/05 11:00:21 recv SN10052307203612: [2,"176306","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2025-02-05T00:00:20Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"6.1"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"244.4","context":"Sample.Periodic"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"324448.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"1510.0","context":"Sample.Periodic"}]}],"transactionId":1738268539}]
[ocpp  ] TRACE 2025/02/05 11:00:21 send SN10052307203612: [3,"176306",{}]
[... cut ...]
[ocpp  ] TRACE 2025/02/05 11:07:53 recv SN10052307203612: [2,"176478","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":100,\"cpVolt\":615,\"transactionId\":1738268539,\"totalPowerFactor\":246,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":1,\"ccVolt\":246,\"offset0\":2006,\"offset1\":2048,\"leakcurr\":10,\"AMBTemp\":427,\"lockStatus\":1,\"ctCurrent\":15}"}]
[ocpp  ] TRACE 2025/02/05 11:07:53 send SN10052307203612: [3,"176478",{"status":"Accepted"}]

```

andig, 2025-02-05:
@jeffborg please omit `cache` from the logs- it's just noise. If there's a deadlock we need to know where. For that purpose please start with `--profile` and add the goroutine profile from /debug/pprof.

andig, 2025-02-05:
> Also disconnects from mqtt broker

Also not visible in the log afaikt. Anyway the log is too noisy to see much.

jeffborg, 2025-02-05:
@andig worked it out what --profile does, just using nightly as both nightly and 0.133.0 have the same issue

First time running and this happened - 2nd start was ok so will be another comment once it stops functioning I'll get the profile out.

```
[main  ] INFO 2025/02/06 08:46:45 evcc 0.133.0 (6fdda39e)
[main  ] INFO 2025/02/06 08:46:45 using config file: /config/evcc.yaml
[db    ] INFO 2025/02/06 08:46:45 using sqlite database: /config/.evcc/evcc.db
[mqtt  ] INFO 2025/02/06 08:46:47 connecting evcc-719386966 at tcp://mqtt-mosquitto:1883
[mqtt  ] DEBUG 2025/02/06 08:46:47 tcp://mqtt-mosquitto:1883 connected
[main  ] INFO 2025/02/06 08:46:47 listening at :7070
fatal error: concurrent map iteration and map write

goroutine 1 [running]:
github.com/evcc-io/evcc/util.CaptureLogs(...)
	github.com/evcc-io/evcc/util/log.go:160
github.com/evcc-io/evcc/cmd.runRoot(0x55cc6a0, {0x289aeb3?, 0x7?, 0x288e9d8?})
	github.com/evcc-io/evcc/cmd/root.go:177 +0x784
github.com/spf13/cobra.(*Command).execute(0x55cc6a0, {0x400011a0d0, 0x3, 0x3})
	github.com/spf13/cobra@v1.8.1/command.go:989 +0x81c
github.com/spf13/cobra.(*Command).ExecuteC(0x55cc6a0)
	github.com/spf13/cobra@v1.8.1/command.go:1117 +0x344
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.8.1/command.go:1041
github.com/evcc-io/evcc/cmd.Execute()
	github.com/evcc-io/evcc/cmd/root.go:118 +0x24
main.main()
	github.com/evcc-io/evcc/main.go:44 +0x38

goroutine 67 [chan receive]:
github.com/evcc-io/evcc/util.(*Tee).Run(0x4000c948e0, 0x400010aaf0)
	github.com/evcc-io/evcc/util/tee.go:39 +0x44
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:153 +0x33c

goroutine 9 [select]:
database/sql.(*DB).connectionOpener(0x400043b860, {0x3198cb8, 0x40004cea50})
	database/sql/sql.go:1253 +0x80
created by database/sql.OpenDB in goroutine 1
	database/sql/sql.go:833 +0x128

goroutine 10 [chan receive]:
github.com/evcc-io/evcc/cmd.configureDatabase.func2()
	github.com/evcc-io/evcc/cmd/setup.go:550 +0x64
created by github.com/evcc-io/evcc/cmd.configureDatabase in goroutine 1
	github.com/evcc-io/evcc/cmd/setup.go:549 +0x88

goroutine 36 [select]:
google.golang.org/grpc/internal/grpcsync.(*CallbackSerializer).run(0x400190aa90, {0x3198cb8, 0x4000614d20})
	google.golang.org/grpc@v1.70.0/internal/grpcsync/callback_serializer.go:88 +0xc4
created by google.golang.org/grpc/internal/grpcsync.NewCallbackSerializer in goroutine 1
	google.golang.org/grpc@v1.70.0/internal/grpcsync/callback_serializer.go:52 +0x118

goroutine 37 [select]:
google.golang.org/grpc/internal/grpcsync.(*CallbackSerializer).run(0x400190aac0, {0x3198cb8, 0x4000614d70})
	google.golang.org/grpc@v1.70.0/internal/grpcsync/callback_serializer.go:88 +0xc4
created by google.golang.org/grpc/internal/grpcsync.NewCallbackSerializer in goroutine 1
	google.golang.org/grpc@v1.70.0/internal/grpcsync/callback_serializer.go:52 +0x118

goroutine 38 [select]:
google.golang.org/grpc/internal/grpcsync.(*CallbackSerializer).run(0x400190aaf0, {0x3198cb8, 0x4000614dc0})
	google.golang.org/grpc@v1.70.0/internal/grpcsync/callback_serializer.go:88 +0xc4
created by google.golang.org/grpc/internal/grpcsync.NewCallbackSerializer in goroutine 1
	google.golang.org/grpc@v1.70.0/internal/grpcsync/callback_serializer.go:52 +0x118

goroutine 46 [IO wait]:
internal/poll.runtime_pollWait(0x7f577d7e20, 0x72)
	runtime/netpoll.go:351 +0xa0
internal/poll.(*pollDesc).wait(0x4000764a00?, 0x40008bc000?, 0x0)
	internal/poll/fd_poll_runtime.go:84 +0x28
[... 126 lines cut ...]

goroutine 68 [chan receive]:
github.com/evcc-io/evcc/util/pipe.(*Dropper).pipe(...)
	github.com/evcc-io/evcc/util/pipe/limiter.go:78
created by github.com/evcc-io/evcc/util/pipe.(*Dropper).Pipe in goroutine 1
	github.com/evcc-io/evcc/util/pipe/limiter.go:90 +0x94

goroutine 69 [chan receive]:
github.com/evcc-io/evcc/util.(*ParamCache).Run(0x4000c94900, 0x4001c18bd0)
	github.com/evcc-io/evcc/util/param.go:61 +0x5c
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:157 +0x464

goroutine 70 [chan receive]:
github.com/evcc-io/evcc/util/pipe.(*Dropper).pipe(...)
	github.com/evcc-io/evcc/util/pipe/limiter.go:78
created by github.com/evcc-io/evcc/util/pipe.(*Dropper).Pipe in goroutine 1
	github.com/evcc-io/evcc/util/pipe/limiter.go:90 +0x94

goroutine 71 [select]:
github.com/evcc-io/evcc/server.(*SocketHub).Run(0x4001c4a690, 0x4001c18c40, 0x4000c94900)
	github.com/evcc-io/evcc/server/socket.go:147 +0x84
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:174 +0x710
```

jeffborg, 2025-02-05:
> Also not visible in the log afaikt. Anyway the log is too noisy to see much.

Can tell because of this `options.SetWill(topic, "offline", 1, true)` even though no log in evcc the topic `evcc/status` changes to offline.

jeffborg, 2025-02-06:
@andig here is your answer about which locks are causing an issue.

Seems in loadpoint_api.go SetDisableThreshold & GetMaxCurrent are deadlocked! 

Yes I have home assistant setting the enable / disable threasholds externally. but this shouldn't cause any issues and clearly it is. This is not an issue in 0.132.1.

Also attached is the debug2 version just in case.
[goroutine_debug_2.txt](https://github.com/user-attachments/files/18680808/goroutine_debug_2.txt)

debug=1 output
```
goroutine profile: total 46
3 @ 0x85de8 0x185b4 0x18184 0x2030038 0x8e604
#	0x2030037	github.com/evcc-io/evcc/util/pipe.(*Dropper).pipe+0x57	github.com/evcc-io/evcc/util/pipe/limiter.go:78

3 @ 0x85de8 0x6157c 0x6a7864 0x8e604
#	0x6a7863	google.golang.org/grpc/internal/grpcsync.(*CallbackSerializer).run+0xc3	google.golang.org/grpc@v1.70.0/internal/grpcsync/callback_serializer.go:88

1 @ 0x1e0f4 0x882d0 0x5c486c 0x8e604
#	0x882cf		os/signal.signal_recv+0x2f	runtime/sigqueue.go:152
#	0x5c486b	os/signal.loop+0x1b		os/signal/signal_unix.go:23

1 @ 0x42d3c 0x84bf4 0x2006a44 0x2006860 0x2003df8 0x2018a04 0x2019318 0x31fd88 0x3219f4 0xf9a224 0x33e72c 0x31e93c 0x8e604
#	0x2006a43	runtime/pprof.writeRuntimeProfile+0xb3			runtime/pprof/pprof.go:793
#	0x200685f	runtime/pprof.writeGoroutine+0x4f			runtime/pprof/pprof.go:752
#	0x2003df7	runtime/pprof.(*Profile).WriteTo+0x147			runtime/pprof/pprof.go:374
#	0x2018a03	net/http/pprof.handler.ServeHTTP+0x443			net/http/pprof/pprof.go:272
#	0x2019317	net/http/pprof.Index+0xc7				net/http/pprof/pprof.go:388
#	0x31fd87	net/http.HandlerFunc.ServeHTTP+0x37			net/http/server.go:2220
#	0x3219f3	net/http.(*ServeMux).ServeHTTP+0x1b3			net/http/server.go:2747
#	0xf9a223	github.com/gorilla/mux.(*Router).ServeHTTP+0x193	github.com/gorilla/mux@v1.8.1/mux.go:212
#	0x33e72b	net/http.serverHandler.ServeHTTP+0xbb			net/http/server.go:3210
#	0x31e93b	net/http.(*conn).serve+0x4fb				net/http/server.go:2092

1 @ 0x85de8 0x176a0 0x172d8 0x1ee0638 0x8e604
#	0x1ee0637	github.com/evcc-io/evcc/core.(*Site).loopLoadpoints+0x67	github.com/evcc-io/evcc/core/site.go:1000

1 @ 0x85de8 0x176a0 0x172d8 0xd75800 0x8e604
#	0xd757ff	github.com/eclipse/paho%2emqtt%2egolang.startIncoming.func1+0xcf	github.com/eclipse/paho.mqtt.golang@v1.5.0/net.go:136

1 @ 0x85de8 0x176a0 0x172d8 0xd75f5c 0x8e604
#	0xd75f5b	github.com/eclipse/paho%2emqtt%2egolang.startIncomingComms.func1+0x42b	github.com/eclipse/paho.mqtt.golang@v1.5.0/net.go:224

1 @ 0x85de8 0x176a0 0x172d8 0xd77eb4 0x8e604
#	0xd77eb3	github.com/eclipse/paho%2emqtt%2egolang.startComms.func1+0x93	github.com/eclipse/paho.mqtt.golang@v1.5.0/net.go:417

1 @ 0x85de8 0x185b4 0x18164 0x202f354 0x8e604
#	0x202f353	github.com/evcc-io/evcc/server/updater.(*watch).watchReleases+0x53	github.com/evcc-io/evcc/server/updater/watch.go:32

1 @ 0x85de8 0x185b4 0x18164 0x20ce4cc 0x8e604
#	0x20ce4cb	github.com/evcc-io/evcc/cmd.runRoot.func2+0x2b	github.com/evcc-io/evcc/cmd/root.go:271

1 @ 0x85de8 0x185b4 0x18164 0x20ce5f0 0x8e604
#	0x20ce5ef	github.com/evcc-io/evcc/cmd.runRoot.func1+0x8f	github.com/evcc-io/evcc/cmd/root.go:265

1 @ 0x85de8 0x185b4 0x18164 0xd6cf58 0x8e604
#	0xd6cf57	github.com/eclipse/paho%2emqtt%2egolang.(*client).internalConnLost.func1+0xb7	github.com/eclipse/paho.mqtt.golang@v1.5.0/client.go:544

1 @ 0x85de8 0x185b4 0x18164 0xfce564 0x8e604
#	0xfce563	github.com/evcc-io/evcc/charger/ocpp.(*Connector).WatchDog+0x43	github.com/evcc-io/evcc/charger/ocpp/connector.go:86

1 @ 0x85de8 0x185b4 0x18184 0x1e4e30c 0x8e604
#	0x1e4e30b	github.com/evcc-io/evcc/push.(*Hub).Run+0x5b	github.com/evcc-io/evcc/push/hub.go:100

1 @ 0x85de8 0x185b4 0x18184 0x1ee0278 0x8e604
#	0x1ee0277	github.com/evcc-io/evcc/core.(*Site).Prepare.func1+0x77	github.com/evcc-io/evcc/core/site.go:965

1 @ 0x85de8 0x185b4 0x18184 0x1f9648c 0x8e604
#	0x1f9648b	github.com/evcc-io/evcc/server.(*MQTT).Run+0x32b	github.com/evcc-io/evcc/server/mqtt.go:292

[... 160 lines cut ...]
```

Logs just stop at 10:42:21 as you mentioned earlier not really much to see here.
```
[circuit-main] TRACE 2025/02/06 10:42:21 validate power: 947.56W + (0W -> 0W) <= 14900W ok
[mqtt  ] TRACE 2025/02/06 10:42:21 recv evcc/loadpoints/1/enableThreshold/set: '-384'
[lp-1  ] DEBUG 2025/02/06 10:42:21 set enable threshold: -384
[mqtt  ] TRACE 2025/02/06 10:42:21 send evcc/loadpoints/1/enableThreshold/set: ''
[site  ] DEBUG 2025/02/06 10:42:21 ----
[lp-1  ] DEBUG 2025/02/06 10:42:21 charge power: 0W
[lp-1  ] DEBUG 2025/02/06 10:42:21 charge currents: [0 0 0]A
[circuit-main] DEBUG 2025/02/06 10:42:21 power: 947.56W
[circuit-main] DEBUG 2025/02/06 10:42:21 current: 0A
[mqtt  ] TRACE 2025/02/06 10:42:21 recv evcc/loadpoints/1/enableThreshold/set: ''
[site  ] DEBUG 2025/02/06 10:42:21 grid power: 948W
[site  ] DEBUG 2025/02/06 10:42:21 pv 1 power: 1390W
[site  ] DEBUG 2025/02/06 10:42:21 site power: 823W
[lp-1  ] DEBUG 2025/02/06 10:42:21 charge voltages: [246 0 0]V
[lp-1  ] DEBUG 2025/02/06 10:42:21 detected connected phases: 1p
[lp-1  ] DEBUG 2025/02/06 10:42:21 charge total import: 332.822kWh
[mqtt  ] TRACE 2025/02/06 10:42:21 recv shellies/house-energy-monitor/emeter/0/power: '1037.15'
[lp-1  ] DEBUG 2025/02/06 10:42:21 charger status: B
[mqtt  ] TRACE 2025/02/06 10:42:21 recv shellies/house-energy-monitor/emeter/0/total: '12856409.6'
[mqtt  ] TRACE 2025/02/06 10:42:21 recv evcc/loadpoints/1/disableThreshold/set: '1318'
```

andig, 2025-02-06:
Looks like a deadlock here:

```
1 @ 0x85de8 0x62418 0x623f5 0x87868 0x1ecfc8c 0x1ecfc39 0x1ed2324 0x1ed2700 0x1ed1f5c 0x1ed3698 0x1ed3a64 0x1ec9dd0 0x1edf560 0x1ee056c 0x20ce2cc 0x8e604
#	0x87867		sync.runtime_SemacquireRWMutexR+0x27					runtime/sema.go:100
#	0x1ecfc8b	sync.(*RWMutex).RLock+0x7b						sync/rwmutex.go:72
#	0x1ecfc38	github.com/evcc-io/evcc/core.(*Loadpoint).GetMaxCurrent+0x28		github.com/evcc-io/evcc/core/loadpoint_api.go:582
#	0x1ed2323	github.com/evcc-io/evcc/core.(*Loadpoint).effectiveMaxCurrent+0x23	github.com/evcc-io/evcc/core/loadpoint_effective.go:156
#	0x1ed26ff	github.com/evcc-io/evcc/core.(*Loadpoint).EffectiveMaxPower+0x1f	github.com/evcc-io/evcc/core/loadpoint_effective.go:205
#	0x1ed1f5b	github.com/evcc-io/evcc/core.(*Loadpoint).nextVehiclePlan+0x3fb		github.com/evcc-io/evcc/core/loadpoint_effective.go:85
#	0x1ed3697	github.com/evcc-io/evcc/core.(*Loadpoint).GetPlanGoal+0xb7		github.com/evcc-io/evcc/core/loadpoint_plan.go:69
#	0x1ed3a63	github.com/evcc-io/evcc/core.(*Loadpoint).plannerActive+0x1c3		github.com/evcc-io/evcc/core/loadpoint_plan.go:118
#	0x1ec9dcf	github.com/evcc-io/evcc/core.(*Loadpoint).Update+0x70f			github.com/evcc-io/evcc/core/loadpoint.go:1838
#	0x1edf55f	github.com/evcc-io/evcc/core.(*Site).update+0x52f			github.com/evcc-io/evcc/core/site.go:901
#	0x1ee056b	github.com/evcc-io/evcc/core.(*Site).Run+0x22b				github.com/evcc-io/evcc/core/site.go:1024
#	0x20ce2cb	github.com/evcc-io/evcc/cmd.runRoot.func4+0x2b				github.com/evcc-io/evcc/cmd/root.go:312

vs

1 @ 0x85de8 0x62418 0x623f5 0x878c8 0x9ce18 0x1ece22c 0x1f955f4 0x1f955a4 0xd84adc 0xd85050 0xd7b1c4 0x8e604
#	0x878c7		sync.runtime_SemacquireRWMutex+0x27									runtime/sema.go:105
#	0x9ce17		sync.(*RWMutex).Lock+0xf7										sync/rwmutex.go:153
#	0x1ece22b	github.com/evcc-io/evcc/core.(*Loadpoint).SetDisableThreshold+0x3b					github.com/evcc-io/evcc/core/loadpoint_api.go:394
#	0x1f955f3	github.com/evcc-io/evcc/server.(*MQTT).listenLoadpointSetters.pass[...].func9+0x23			github.com/evcc-io/evcc/server/helper.go:17
#	0x1f955a3	github.com/evcc-io/evcc/server.(*MQTT).listenLoadpointSetters.floatSetter.setterFunc[...].func21+0x43	github.com/evcc-io/evcc/server/mqtt_setter.go:25
#	0xd84adb	github.com/evcc-io/evcc/plugin/mqtt.(*Client).ListenSetter.func1+0x4b					github.com/evcc-io/evcc/plugin/mqtt/client.go:193
#	0xd8504f	github.com/evcc-io/evcc/plugin/mqtt.(*Client).listen.func1+0x21f					github.com/evcc-io/evcc/plugin/mqtt/client.go:214
#	0xd7b1c3	github.com/eclipse/paho%2emqtt%2egolang.(*router).matchAndDispatch.func2+0x7f3	
```

Apart from this shouldn't be happening, you've selected "no external automation". Where would these MQTT requests be coming from?

jeffborg, 2025-02-06:
@andig Just dynamically setting the enable and disable thresholds. It wasn't an issue in 0.132.1 or hasn't been an issue beforehand. This shouldn't occur full stop. 

I hammered a 2nd copy on my computer against these 2 methods and couldn't get a result. But that was go from the container. 

My gut feeling it's an issue in go itself and if I rebuild 0.132.1 with the latest go because the Dockerfile doesn't limit the point version I may see the issue in 0.132.1 as well.

andig, 2025-02-06:
So there _is_ external automation? Just to get the picture straight. Could you try the nightly, too?

jeffborg, 2025-02-06:
@andig that was the nightly for the profile same as 0.133.0. Never bothered with 0.133.0 for profile.

GrimmiMeloni, 2025-02-06:
@andig I went through the call chain. The thread in `Loadpoint.Update()` (routine 153) already holds the RLock when requesting RLock. As I learned from you long ago that Locks are non-reentrant in Go, I think we have our smoking gun.... 💣 

```
goroutine 153 [sync.RWMutex.RLock, 18 minutes]:
sync.runtime_SemacquireRWMutexR(0x4000d7d3e8?, 0x4c?, 0x0?)
        runtime/sema.go:100 +0x28
sync.(*RWMutex).RLock(...)
        sync/rwmutex.go:72
github.com/evcc-io/evcc/core.(*Loadpoint).GetMaxCurrent(0x4001678908)
        github.com/evcc-io/evcc/core/loadpoint_api.go:582 +0x7c
github.com/evcc-io/evcc/core.(*Loadpoint).effectiveMaxCurrent(0x4001678908)
        github.com/evcc-io/evcc/core/loadpoint_effective.go:156 +0x24
github.com/evcc-io/evcc/core.(*Loadpoint).EffectiveMaxPower(0x4001678908)
        github.com/evcc-io/evcc/core/loadpoint_effective.go:205 +0x20
github.com/evcc-io/evcc/core.(*Loadpoint).nextVehiclePlan(0x4001678908)
        github.com/evcc-io/evcc/core/loadpoint_effective.go:85 +0x3fc
github.com/evcc-io/evcc/core.(*Loadpoint).GetPlanGoal(0x4001678908)
        github.com/evcc-io/evcc/core/loadpoint_plan.go:69 +0xb8          <<<<<<<<<<<<<<<<<< holds RLock()

github.com/evcc-io/evcc/core.(*Loadpoint).plannerActive(0x4001678908)
        github.com/evcc-io/evcc/core/loadpoint_plan.go:118 +0x1c4
github.com/evcc-io/evcc/core.(*Loadpoint).Update(0x4001678908, 0x4089b47ae147ae14, 0x0, {0x4000e6e808, 0x30, 0x30}, 0x0, 0x0, 0x0, 0x4000a367a0, ...)
        github.com/evcc-io/evcc/core/loadpoint.go:1838 +0x710
...
```

andig, 2025-02-06:
Nope. That's what RLocks are for- they just allow ready and you can have any number of those. I still don't get it. The RLock will only "lock" if a Lock is already being held.

andig, 2025-02-06:
Of course. But the writer doesn‘t have the lock either. So who has?

GrimmiMeloni, 2025-02-06:
Still 153.
It has the Read Lock from the initial locking I pointed out earlier.
While it is doing its thing the MQTT thread requests a write lock and blocks on that call.
Then the Loadpoint routine (153) reaches the point where it wants to once more acquire the Read Lock. And that Blocks, because the MQTT has requested a write lock.

It is exactly the behavior described in the docs I quoted.

andig, 2025-02-06:
Whaaaaat. Youre right. Never noticed that.

jeffborg, 2025-02-06:
@andig this line is the cause of it https://github.com/evcc-io/evcc/commit/285b22ee6e2021c0573fb331454ef015ebe0e543#diff-64161edf08f807019f3ade5b7d32ce0f06b9a42bab3450090d8223f3ec6c2ee0R85

Just did a git bisect and given the previous explanations makes total sense.

closed 2025-02-07 by andig

jeffborg, 2025-02-07:
@andig not fixed by 18650

```
goroutine profile: total 43
5 @ 0x85de8 0x46a48 0x85090 0x1103f8 0x11171c 0x11170d 0x1f6418 0x2086a4 0x319f54 0x2a0058 0x2a01c0 0x31ea74 0x8e604
#	0x8508f		internal/poll.runtime_pollWait+0x9f		runtime/netpoll.go:351
#	0x1103f7	internal/poll.(*pollDesc).wait+0x27		internal/poll/fd_poll_runtime.go:84
#	0x11171b	internal/poll.(*pollDesc).waitRead+0x1fb	internal/poll/fd_poll_runtime.go:89
#	0x11170c	internal/poll.(*FD).Read+0x1ec			internal/poll/fd_unix.go:165
#	0x1f6417	net.(*netFD).Read+0x27				net/fd_posix.go:55
#	0x2086a3	net.(*conn).Read+0x33				net/net.go:189
#	0x319f53	net/http.(*connReader).Read+0x223		net/http/server.go:798
#	0x2a0057	bufio.(*Reader).fill+0xf7			bufio/bufio.go:110
#	0x2a01bf	bufio.(*Reader).Peek+0x5f			bufio/bufio.go:148
#	0x31ea73	net/http.(*conn).serve+0x633			net/http/server.go:2127

3 @ 0x85de8 0x185b4 0x18184 0x2030be8 0x8e604
#	0x2030be7	github.com/evcc-io/evcc/util/pipe.(*Dropper).pipe+0x57	github.com/evcc-io/evcc/util/pipe/limiter.go:78

1 @ 0x1e0f4 0x882d0 0x5c4a2c 0x8e604
#	0x882cf		os/signal.signal_recv+0x2f	runtime/sigqueue.go:152
#	0x5c4a2b	os/signal.loop+0x1b		os/signal/signal_unix.go:23

1 @ 0x42d3c 0x84bf4 0x20075f4 0x2007410 0x20049a8 0x20195b4 0x2019ec8 0x31fd88 0x3219f4 0xf9aba4 0x33e72c 0x31e93c 0x8e604
#	0x20075f3	runtime/pprof.writeRuntimeProfile+0xb3			runtime/pprof/pprof.go:793
#	0x200740f	runtime/pprof.writeGoroutine+0x4f			runtime/pprof/pprof.go:752
#	0x20049a7	runtime/pprof.(*Profile).WriteTo+0x147			runtime/pprof/pprof.go:374
#	0x20195b3	net/http/pprof.handler.ServeHTTP+0x443			net/http/pprof/pprof.go:272
#	0x2019ec7	net/http/pprof.Index+0xc7				net/http/pprof/pprof.go:388
#	0x31fd87	net/http.HandlerFunc.ServeHTTP+0x37			net/http/server.go:2220
#	0x3219f3	net/http.(*ServeMux).ServeHTTP+0x1b3			net/http/server.go:2747
#	0xf9aba3	github.com/gorilla/mux.(*Router).ServeHTTP+0x193	github.com/gorilla/mux@v1.8.1/mux.go:212
#	0x33e72b	net/http.serverHandler.ServeHTTP+0xbb			net/http/server.go:3210
#	0x31e93b	net/http.(*conn).serve+0x4fb				net/http/server.go:2092

1 @ 0x85de8 0x176a0 0x172d8 0x1ee0f68 0x8e604
#	0x1ee0f67	github.com/evcc-io/evcc/core.(*Site).loopLoadpoints+0x67	github.com/evcc-io/evcc/core/site.go:1000

1 @ 0x85de8 0x176a0 0x172d8 0xd759c0 0x8e604
#	0xd759bf	github.com/eclipse/paho%2emqtt%2egolang.startIncoming.func1+0xcf	github.com/eclipse/paho.mqtt.golang@v1.5.0/net.go:136

1 @ 0x85de8 0x176a0 0x172d8 0xd7611c 0x8e604
#	0xd7611b	github.com/eclipse/paho%2emqtt%2egolang.startIncomingComms.func1+0x42b	github.com/eclipse/paho.mqtt.golang@v1.5.0/net.go:224

1 @ 0x85de8 0x176a0 0x172d8 0xd78074 0x8e604
#	0xd78073	github.com/eclipse/paho%2emqtt%2egolang.startComms.func1+0x93	github.com/eclipse/paho.mqtt.golang@v1.5.0/net.go:417

1 @ 0x85de8 0x185b4 0x18164 0x202ff04 0x8e604
#	0x202ff03	github.com/evcc-io/evcc/server/updater.(*watch).watchReleases+0x53	github.com/evcc-io/evcc/server/updater/watch.go:32

1 @ 0x85de8 0x185b4 0x18164 0x20cfa8c 0x8e604
#	0x20cfa8b	github.com/evcc-io/evcc/cmd.runRoot.func2+0x2b	github.com/evcc-io/evcc/cmd/root.go:271

1 @ 0x85de8 0x185b4 0x18164 0x20cfbb0 0x8e604
#	0x20cfbaf	github.com/evcc-io/evcc/cmd.runRoot.func1+0x8f	github.com/evcc-io/evcc/cmd/root.go:265

1 @ 0x85de8 0x185b4 0x18184 0x1e4e6cc 0x8e604
#	0x1e4e6cb	github.com/evcc-io/evcc/push.(*Hub).Run+0x5b	github.com/evcc-io/evcc/push/hub.go:100

1 @ 0x85de8 0x185b4 0x18184 0x1ee0ba8 0x8e604
#	0x1ee0ba7	github.com/evcc-io/evcc/core.(*Site).Prepare.func1+0x77	github.com/evcc-io/evcc/core/site.go:965

1 @ 0x85de8 0x185b4 0x18184 0x1f96dbc 0x8e604
#	0x1f96dbb	github.com/evcc-io/evcc/server.(*MQTT).Run+0x32b	github.com/evcc-io/evcc/server/mqtt.go:292

1 @ 0x85de8 0x185b4 0x18184 0x202fdf8 0x8e604
#	0x202fdf7	github.com/evcc-io/evcc/server/updater.Run+0x197	github.com/evcc-io/evcc/server/updater/run.go:22

1 @ 0x85de8 0x185b4 0x18184 0x20d3e04 0x8e604
#	0x20d3e03	github.com/evcc-io/evcc/cmd.configureDatabase.func2+0x63	github.com/evcc-io/evcc/cmd/setup.go:550
[... 127 lines cut ...]
#	0x1edfe8f	github.com/evcc-io/evcc/core.(*Site).update+0x52f			github.com/evcc-io/evcc/core/site.go:901
#	0x1ee0ebf	github.com/evcc-io/evcc/core.(*Site).Run+0x24f				github.com/evcc-io/evcc/core/site.go:1022
#	0x20cf88b	github.com/evcc-io/evcc/cmd.runRoot.func4+0x2b				github.com/evcc-io/evcc/cmd/root.go:312

1 @ 0x85de8 0x62418 0x623f5 0x878c8 0x9ce18 0x1ece66c 0x1f95fe4 0x1f95f94 0xd84c9c 0xd85210 0xd7b384 0x8e604
#	0x878c7		sync.runtime_SemacquireRWMutex+0x27									runtime/sema.go:105
#	0x9ce17		sync.(*RWMutex).Lock+0xf7										sync/rwmutex.go:153
#	0x1ece66b	github.com/evcc-io/evcc/core.(*Loadpoint).SetEnableThreshold+0x3b					github.com/evcc-io/evcc/core/loadpoint_api.go:390
#	0x1f95fe3	github.com/evcc-io/evcc/server.(*MQTT).listenLoadpointSetters.pass[...].func8+0x23			github.com/evcc-io/evcc/server/helper.go:17
#	0x1f95f93	github.com/evcc-io/evcc/server.(*MQTT).listenLoadpointSetters.floatSetter.setterFunc[...].func20+0x43	github.com/evcc-io/evcc/server/mqtt_setter.go:25
#	0xd84c9b	github.com/evcc-io/evcc/plugin/mqtt.(*Client).ListenSetter.func1+0x4b					github.com/evcc-io/evcc/plugin/mqtt/client.go:193
#	0xd8520f	github.com/evcc-io/evcc/plugin/mqtt.(*Client).listen.func1+0x21f					github.com/evcc-io/evcc/plugin/mqtt/client.go:214
#	0xd7b383	github.com/eclipse/paho%2emqtt%2egolang.(*router).matchAndDispatch.func2+0x7f3				github.com/eclipse/paho.mqtt.golang@v1.5.0/router.go:218

[... cut ...]
#	0xaf93f		syscall.Syscall+0x1f				syscall/syscall_linux.go:73
#	0xad5ef		syscall.read+0x3f				syscall/zsyscall_linux_arm64.go:736
#	0x11173f	syscall.Read+0x21f				syscall/syscall_unix.go:183
#	0x11172c	internal/poll.ignoringEINTRIO+0x20c		internal/poll/fd_unix.go:745
#	0x111614	internal/poll.(*FD).Read+0xf4			internal/poll/fd_unix.go:161
#	0x1f6417	net.(*netFD).Read+0x27				net/fd_posix.go:55
#	0x2086a3	net.(*conn).Read+0x33				net/net.go:189
#	0x31992f	net/http.(*connReader).backgroundRead+0x3f	net/http/server.go:690

```

jeffborg, 2025-02-07:
@andig  I've found only 2 places with these recursive read locks in master ce7f71e0

```
1 @ 0x85de8 0x62418 0x623f5 0x87868 0x1ed33b8 0x1ed3365 0x1ed390c 0x1ed303c 0x1ed26dc 0x1ed3fc8 0x1ed4394 0x1eca570 0x1edfe90 0x1ee0ec0 0x20cf88c 0x8e604
#	0x87867		sync.runtime_SemacquireRWMutexR+0x27					runtime/sema.go:100
#	0x1ed33b7	sync.(*RWMutex).RLock+0x77						sync/rwmutex.go:72
>>> #	0x1ed3364	github.com/evcc-io/evcc/core.(*Loadpoint).getMeasuredPhases+0x24	github.com/evcc-io/evcc/core/loadpoint_phases.go:45
#	0x1ed390b	github.com/evcc-io/evcc/core.(*Loadpoint).maxActivePhases+0x2b		github.com/evcc-io/evcc/core/loadpoint_phases.go:95
#	0x1ed303b	github.com/evcc-io/evcc/core.(*Loadpoint).effectiveMaxPower+0x2b	github.com/evcc-io/evcc/core/loadpoint_effective.go:216
#	0x1ed26db	github.com/evcc-io/evcc/core.(*Loadpoint).nextVehiclePlan+0x3fb		github.com/evcc-io/evcc/core/loadpoint_effective.go:85
>>> #	0x1ed3fc7	github.com/evcc-io/evcc/core.(*Loadpoint).GetPlanGoal+0xb7		github.com/evcc-io/evcc/core/loadpoint_plan.go:69
#	0x1ed4393	github.com/evcc-io/evcc/core.(*Loadpoint).plannerActive+0x1c3		github.com/evcc-io/evcc/core/loadpoint_plan.go:118
#	0x1eca56f	github.com/evcc-io/evcc/core.(*Loadpoint).Update+0x70f			github.com/evcc-io/evcc/core/loadpoint.go:1845
#	0x1edfe8f	github.com/evcc-io/evcc/core.(*Site).update+0x52f			github.com/evcc-io/evcc/core/site.go:901
#	0x1ee0ebf	github.com/evcc-io/evcc/core.(*Site).Run+0x24f				github.com/evcc-io/evcc/core/site.go:1022
#	0x20cf88b	github.com/evcc-io/evcc/cmd.runRoot.func4+0x2b				github.com/evcc-io/evcc/cmd/root.go:312

```

```
1 @ 0x85de8 0x62418 0x623f5 0x87868 0x1ed3ba0 0x1ed3b45 0x1ed2208 0x1ed26f0 0x1ed3e38 0x1ed4204 0x1eca570 0x1edfd00 0x1ee0d0c 0x20cf6fc 0x8e604
#	0x87867		sync.runtime_SemacquireRWMutexR+0x27					runtime/sema.go:100
#	0x1ed3b9f	sync.(*RWMutex).RLock+0x7f						sync/rwmutex.go:72
>>> #	0x1ed3b44	github.com/evcc-io/evcc/core.(*Loadpoint).GetPlanRequiredDuration+0x24	github.com/evcc-io/evcc/core/loadpoint_plan.go:49
#	0x1ed2207	github.com/evcc-io/evcc/core.(*Loadpoint).nextActivePlan+0xa7		github.com/evcc-io/evcc/core/loadpoint_effective.go:43
#	0x1ed26ef	github.com/evcc-io/evcc/core.(*Loadpoint).nextVehiclePlan+0x40f		github.com/evcc-io/evcc/core/loadpoint_effective.go:85
>>> #	0x1ed3e37	github.com/evcc-io/evcc/core.(*Loadpoint).GetPlanGoal+0xb7		github.com/evcc-io/evcc/core/loadpoint_plan.go:69
#	0x1ed4203	github.com/evcc-io/evcc/core.(*Loadpoint).plannerActive+0x1c3		github.com/evcc-io/evcc/core/loadpoint_plan.go:118
#	0x1eca56f	github.com/evcc-io/evcc/core.(*Loadpoint).Update+0x70f			github.com/evcc-io/evcc/core/loadpoint.go:1845
#	0x1edfcff	github.com/evcc-io/evcc/core.(*Site).update+0x52f			github.com/evcc-io/evcc/core/site.go:901
#	0x1ee0d0b	github.com/evcc-io/evcc/core.(*Site).Run+0x22b				github.com/evcc-io/evcc/core/site.go:1024
#	0x20cf6fb	github.com/evcc-io/evcc/cmd.runRoot.func4+0x2b				github.com/evcc-io/evcc/cmd/root.go:312
```

andig, 2025-02-08:
Will take another look, than you!

andig, 2025-02-08:
@jeffborg would you be able to test the PR from source without merging? Unfortunately, since there is no tooling available, this has a high risk of introducing new bugs.