evcc-io/evcc #18604: Not fetching dynamic electricity prices

timpetersson81, 2025-02-05:
### Describe the bug

The last couple of days I have bin facing an issue where the electricity prices are not pulled. 

When I restart the homeaissistant device EVCC is installed on the prices are pulled. 

I use the custom part, and prices are here: https://linkcia.dk/elpriser.php and working.

I don't see anything in the logs related to this, and have no clue how to retry the pull request of the prices.

### Steps to reproduce

1. Wait for the price to update
2.
3.
...

### Configuration details

```yaml
network:
  port: 7070

log: debug

interval: 15s

meters:
  - name: my_grid
    type: template
    template: sungrow-hybrid
    usage: grid
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.1.63 # Hostname
    port: 502 # Port
    timeout: 10s # optional

  - name: my_pv
    type: template
    template: sungrow-hybrid
    usage: pv

    modbus: tcpip
    id: 1
    host: 192.168.1.63 # Hostname
    port: 502 # Port
    timeout: 10s # optional

  - name: my_battery
    type: template
    template: sungrow-hybrid
    usage: battery

    modbus: tcpip
    id: 1
    host: 192.168.1.63 # Hostname
    port: 502 # Port
    timeout: 10s # optional

chargers:
  - name: my_charger
    type: template
    template: pulsarplus-fw5

vehicles:
  - name: Xpeng
    type: template
[... 896 lines cut ...]
[lp-1  ] DEBUG 2025/02/05 04:29:31 detected active phases: 2p
[lp-1  ] DEBUG 2025/02/05 04:29:31 charge total import: 3202.158kWh
[lp-1  ] DEBUG 2025/02/05 04:29:31 charger status: C
[lp-1  ] DEBUG 2025/02/05 04:29:31 soc estimated: 92.96% (vehicle: 90.00%)
[lp-1  ] DEBUG 2025/02/05 04:29:31 vehicle soc: 93%
[lp-1  ] DEBUG 2025/02/05 04:29:31 vehicle range: 141km
[lp-1  ] DEBUG 2025/02/05 04:29:31 plan: charge 33m56s between 2025-02-05 05:00:00 +0100 CET until 2025-02-05 07:00:00 +0100 CET (power: 7360W, avg cost: 1.200)
[lp-1  ] DEBUG 2025/02/05 04:29:31 plan: avoid re-start within 1h0m0s, continuing for remaining 30m28s
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

andig, 2025-02-05:
> electricity prices are not pulled

Why do you think so? The log has no errors, maybe there are no prices? In a case like this, a trace log for the tariff is needed for the request that apparently doesn't retrieve data. Closing until log available.

closed 2025-02-05 by andig

timpetersson81, 2025-02-05:
> > electricity prices are not pulled
> 
> Why do you think so? The log has no errors, maybe there are no prices? In a case like this, a trace log for the tariff is needed for the request that apparently doesn't retrieve data. Closing until log available.

But the the prices are pull in when I restart home assistant then something seems not to be running, perhaps a cron or something? Can I fire it manually?

Prices become available in 25min, so I guess will return for some help then :)

I have the attached options for logs, what should I go with?

[image]

timpetersson81, 2025-02-05:
> > electricity prices are not pulled
> 
> Why do you think so? The log has no errors, maybe there are no prices? In a case like this, a trace log for the tariff is needed for the request that apparently doesn't retrieve data. Closing until log available.

No update in EVCC of price, they are updated in the API supplied. I restart the the hole home assistant device, and prices update as they should.

I have search logs for tariff, but nothing comes up.

Did the custom plugin for tariffs get removed? https://docs.evcc.io/en/docs/devices/tariffs#custom-plugin page is missing

andig, 2025-02-05:
See https://docs.evcc.io/en/docs/tariffs#custom-plugin

> No update in EVCC of price, they are updated in the API supplied. I restart the the hole home assistant device, and prices update as they should.

Tariffs should be updated every hour, starting with application start.