evcc-io/evcc #18610: Mindestladung bei Ankunft: Lädt nur 1-phasig bei Phaseneinstellung "automatisch"

An-Wi, 2025-02-05:
### Describe the bug

Erwartet hatte ich ein Laden mit 3 Phasen, um möglichst schnell die Mindestladung zu erreichen.

Die **Logs** habe ich selektiv angefügt (wichtigen Einträgen mit Stichwort "phase", den Ladestart sowie die manuellen Eingriffe).
Bei Bedarf liefere ich gern weitere Logs nach.

Ergänzend sei noch angemerkt, dass letzten Woche Freitag (31.1.) ein Firmware-Updates meiner Wallbox erfolgt ist.
Möglicherweise ist durch den Hersteller-Support zuletzt am 3.2. ein **Neustart der Wallbox** durchgeführt worden -> ggf. haben sich dadurch unerwartete Konstellationen ergeben.

### Steps to reproduce

1. Zuletzt geladen mit "Min+PV" mit Einstellung "3-phasig".
`[lp-1  ] DEBUG 2025/02/01 22:41:26 switched phases: 3p`
`[lp-1  ] INFO 2025/02/02 08:50:49 start charging ->`
`[lp-1  ] INFO 2025/02/02 16:37:30 stop charging <- `
2. (2 Tage später) Umstellung auf Phaseneinstellung "Automatisch" 10min bevor Auto angesteckt wurde.
`[lp-1  ] DEBUG 2025/02/04 19:33:28 set phases: 0`
3. Auto angesteckt
`[lp-1  ] INFO 2025/02/04 19:45:51 car connected`
4. =>evcc: startet Laden nur 1-phasig
`[lp-1  ] INFO 2025/02/04 19:46:11 start charging ->`
`[lp-1  ] DEBUG 2025/02/04 19:46:31 detected active phases: 1p`
5. Manuell auf "3-phasig" umgestellt
`[lp-1  ] DEBUG 2025/02/04 20:15:47 set phases: 3`
6. =>evcc: keine Phasenumschaltung! Läde weiter 1-phasig
`[lp-1  ] DEBUG 2025/02/04 20:15:47 detected active phases: 1p`
`[lp-1  ] DEBUG 2025/02/04 20:15:51 detected active phases: 1p`
7. Manuell auf "1-phasig" umgestellt
`[lp-1  ] DEBUG 2025/02/04 20:15:53 set phases: 1`
8. Anschließend direkt manuelle Phaseneinstellung auf "Automatisch"
`[lp-1  ] DEBUG 2025/02/04 20:15:55 set phases: 0`
9. =>evcc: Phasenwechsel und lädt nun 3-phasig
`[lp-1  ] DEBUG 2025/02/04 20:16:08 switched phases: 3p`

### Configuration details

```yaml
---
```

### Log details

```text
[site  ] DEBUG 2025/02/04 19:33:10 ----
[lp-1  ] DEBUG 2025/02/04 19:33:10 charge power: 0W
[lp-1  ] DEBUG 2025/02/04 19:33:10 charge currents: [0 0 0]A
[script] DEBUG 2025/02/04 19:33:10 /fupcService/pvCurrentPower.bat: 0
[site  ] DEBUG 2025/02/04 19:33:10 pv 1 power: 0W
[site  ] DEBUG 2025/02/04 19:33:11 grid power: 348W
[site  ] DEBUG 2025/02/04 19:33:11 grid currents: [0.692 0.852 0.679]A
[site  ] DEBUG 2025/02/04 19:33:11 site power: 248W
[lp-1  ] DEBUG 2025/02/04 19:33:11 charge total import: 9140.172kWh
[lp-1  ] DEBUG 2025/02/04 19:33:11 charger status: A
[lp-1  ] DEBUG 2025/02/04 19:33:11 next soc poll remaining time: 34m20s
[lp-1  ] DEBUG 2025/02/04 19:33:11 plan: charge 3h3m53s between 2025-02-05 04:26:07 +0100 CET until 2025-02-05 07:30:00 +0100 CET (power: 11040W, avg cost: 0.280)
[lp-1  ] DEBUG 2025/02/04 19:33:28 set phases: 0
[site  ] DEBUG 2025/02/04 19:33:28 ----
[lp-1  ] DEBUG 2025/02/04 19:33:28 charge power: 0W
[lp-1  ] DEBUG 2025/02/04 19:33:28 charge currents: [0 0 0]A
[script] DEBUG 2025/02/04 19:33:28 /fupcService/pvCurrentPower.bat: 0
[site  ] DEBUG 2025/02/04 19:33:28 pv 1 power: 0W
[site  ] DEBUG 2025/02/04 19:33:29 grid power: 378W
[site  ] DEBUG 2025/02/04 19:33:29 grid currents: [0.805 0.848 0.705]A
[site  ] DEBUG 2025/02/04 19:33:29 site power: 278W
[lp-1  ] DEBUG 2025/02/04 19:33:29 charge total import: 9140.172kWh
[lp-1  ] DEBUG 2025/02/04 19:33:29 charger status: A
[lp-1  ] DEBUG 2025/02/04 19:33:29 next soc poll remaining time: 34m1s
[lp-1  ] DEBUG 2025/02/04 19:33:29 plan: charge 3h3m53s between 2025-02-05 04:26:07 +0100 CET until 2025-02-05 07:30:00 +0100 CET (power: 11040W, avg cost: 0.280)
[site  ] DEBUG 2025/02/04 19:33:30 ----
[... 269 lines cut ...]
[lp-1  ] DEBUG 2025/02/04 20:16:51 charger status: C
[lp-1  ] DEBUG 2025/02/04 20:16:51 soc estimated: 14.61% (vehicle: 14.00%)
[lp-1  ] DEBUG 2025/02/04 20:16:51 vehicle soc: 15%
[lp-1  ] DEBUG 2025/02/04 20:16:51 vehicle range: 53km
[lp-1  ] DEBUG 2025/02/04 20:16:51 plan: charge 3h58m27s between 2025-02-05 03:31:33 +0100 CET until 2025-02-05 07:30:00 +0100 CET (power: 11040W, avg cost: 0.280)
[lp-1  ] DEBUG 2025/02/04 20:16:51 plan: continuing until end of slot
[lp-1  ] DEBUG 2025/02/04 20:16:51 forced charging at vehicle soc 15% (< 30% min soc)
[site  ] DEBUG 2025/02/04 20:17:10 ----
```

### What type of operating system or environment does evcc run on?

Windows

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

andig, 2025-02-05:
Missing config

An-Wi, 2025-02-05:
Config ergänzt in Beschreibung oben...
Sorry - dachte nicht, dass sie wichtig ist für dieses Problem.

andig, 2025-02-05:
Laut Log

    [lp-1  ] DEBUG 2025/02/04 19:45:51 forced charging at vehicle soc 13% (< 30% min soc)

passiert

```go
	// minimum or target charging
	case lp.minSocNotReached() || plannerActive:
		err = lp.fastCharging()
		lp.resetPhaseTimer()
		lp.elapsePVTimer() // let PV mode disable immediately afterwards
```

wobei der erste Aufruf auf 3p schaltet. Davon zeigt das Log aber nichts. Das müsste hier passieren:

```go
// scalePhasesIfAvailable scales if api.PhaseSwitcher is available
func (lp *Loadpoint) scalePhasesIfAvailable(phases int) error {
	if lp.configuredPhases != 0 {
		phases = lp.configuredPhases
	}

	if lp.hasPhaseSwitching() {
		return lp.scalePhases(phases)
	}

	return nil
}
```

@naltatis hältst Du es für möglich, dass mit den Phasen irgendwas prinzipiell kaputt ist?

andig, 2025-02-05:
@naltatis das sieht ähnlich aus wie https://github.com/evcc-io/evcc/issues/18606. Nur ist es hier v133- da war noch kein Config UI drin? Mit dieser Config ist der Charger ab Start immer auf 3p- was nicht sein sollte.

```yaml
network:
  port: 7070

log: debug

interval: 3s

javascript:
  - vm: shared
    script: |
      state = {
        residualpower: 500,
        pvpower: 0,
        batterypower: 0,
        batterySoc: 55,
        gridpower: -1000,
        loadpoints: [
          { enabled: false, vehicleSoc: 0, maxcurrent: 6, phases: 1, chargepower: 0 },
          { enabled: false, vehicleSoc: 22, maxcurrent: 0, phases: 3, chargepower: 0 }
        ]
      };
      function logState() {
        console.log("state:", JSON.stringify(state));
      }

meters:
  - name: grid
    type: custom
    power:
      source: js
      vm: shared
      script: |
        state.gridpower = state.loadpoints[0].chargepower + state.loadpoints[1].chargepower + state.residualpower - batterypower - pvpower;
        state.gridpower;
      in:
        - name: pvpower
          type: float
          config:
            source: js
            vm: shared
            script: |
              state.pvpower = 8000+500*Math.random();
              state.pvpower
        - name: batterypower
          type: float
          config:
            source: js
            vm: shared
            script: |
              state.batterypower = state.gridpower > 0 ? 1000 * Math.random() : 0;
              state.batterypower

  - name: pv
    type: custom
    power:
      source: js
      vm: shared
      script: 0

  - name: battery
    type: custom
    power:
      source: js
      vm: shared
      script: state.batterypower;
    soc:
      source: js
      vm: shared
      script: |
        if (state.batterypower < 0) state.batterySoc++; else state.batterySoc--;
        if (state.batterySoc < 10) state.batterySoc = 90;
        if (state.batterySoc > 90) state.batterySoc = 10;
        state.batterySoc;
    capacity: 13.4
    batterymode:
      source: js
      vm: shared
      script: |
        1

  - name: meter_charger_1
    type: custom
    power:
      source: js
      vm: shared
      script: state.loadpoints[0].chargepower;

chargers:
  - name: charger_1
    type: custom
    enable:
      source: js
      vm: shared
      script: |
        logState();
        var lp = state.loadpoints[0];
        lp.enabled = enable;
        enable;
      out:
        - name: enable
          type: bool
          config:
            source: js
            vm: shared
            script: |
              if (enable) lp.chargepower = lp.maxcurrent * 230 * lp.phases; else lp.chargepower = 0;
    enabled:
      source: js
      vm: shared
      script: |
        state.loadpoints[0].enabled;
    status:
      source: js
      vm: shared
      script: |
        if (state.loadpoints[0].enabled) "C"; else "B";
    maxcurrent:
      source: js
      vm: shared
      script: |
        logState();
        var lp = state.loadpoints[0];
        lp.maxcurrent = maxcurrent;
        if (lp.enabled) lp.chargepower = lp.maxcurrent * 230 * lp.phases; else lp.chargepower = 0;
    Phases1p3p:
      source: js
      vm: shared
      script: |
        logState();
        lp.phases = phases
    Tos: true

vehicles:
  - name: vehicle_1
    title: blauer e-Golf
    type: custom
    soc:
      source: js
      vm: shared
      script: 10
    capacity: 44

site:
  title: Zuhause
  meters:
    grid: grid
    pv: pv
    battery: battery

loadpoints:
  - title: Carport
    charger: charger_1
    mode: pv
    meter: meter_charger_1
    vehicle: vehicle_1
```

closed 2025-02-06 by naltatis