evcc-io/evcc #18622: FR: Force vehicle update (skip cache)

Schnabulation, 2025-02-06:
**Is your feature request related to a problem? Please describe.**
Die Vor-Klimatisierung ab Netzstrom funktioniert nur, wenn gerade dann die Auto-API neu angefragt und der Climater-Status neu abgefrat wird.

**Describe the solution you'd like**
Eine simple Lösung würde sein, dass z.B. via GUI oder API ein force-update getriggert werden könnte, dass den Cache umgeht und aktiv die API abfragt. Auch wäre denkbar, dass dies bei der (Neu-)Auswahl eines Fahrzeugs gemacht wird (Stichwort: API Befehl „Start vehicle detection“).

**Describe alternatives you've considered**
Ich dachte, dass vielleicht bei der Vehicle detection die API neu abgefragt wird, aber auch da läuft‘s nur in den Cache (was es ja auch soll).

andig, 2025-02-07:
> Ich dachte, dass vielleicht bei der Vehicle detection die API neu abgefragt wird, aber auch da läuft‘s nur in den Cache (was es ja auch soll).

Das passiert:

```go
	// request vehicle api refresh while waiting to identify
	select {
	case <-lp.vehicleDetectTicker.C:
		lp.log.DEBUG.Println("vehicle api refresh")
		util.ResetCached()
	default:
	}
```

Wir könnten ggf. das Wakeup API je Fahrzeug exposen (und dabei auch gleich die Caches leeren). Wie würde das dann genutzt werden?

Schnabulation, 2025-02-07:
In meinem Fall wäre die Idee, dass ich über Home Assistant eine Automation baue, die einerseits die Vorklimatisierung anstösst und andererseits evcc via API anweist, das Fahrzeug neu abzufragen. Das würde dann dazu führen, dass evcc sofort sieht, dass die Klima an ist und entsprechend die Wallbox freigibt - nicht erst nach Ablauf des Caches.

Ohne Home Assistant könnte dies vielleicht via Frontend-Button in evcc realisiert werden. Der User startet die Klima via Auto-App und klickt dann den Button in der GUI an.

skkumar25, 2025-02-08:
****indian army

Schnabulation, 2025-02-16:
Wurde das denn nun implementiert oder habt ihr auf einen Input von mir gewartet? Wieso wurde der FR geschlossen?

andig, 2025-02-17:
https://github.com/evcc-io/evcc/issues/18622#issuecomment-2642787130 ?

Schnabulation, 2025-02-17:
Wie es scheint verstehe ich deine Frage nicht.

In der aktuellen Situation ist es so, dass ich das Fahrzeug via App zum Vorklimatisieren anrege, das Fahrzeug pusht den neusten Stand dann an den Bluelink Server. evcc ruft diesen Status aber nur alle 15 Minuten ab (wenn im poll-mode: connected) - alles dazwischen läuft in den Cache.

Meine Idee wäre deshalb, dass man irgendwie als Enduser die Möglichkeit erhält, für die Fahrzeugabfrage einmalig den Cache zu umgehen respektive leeren, sodass die Abfrage aktiv zum Server geht. Beispielsweise durch die API oder vielleicht auch via Frontend Button.

andig, 2025-02-17:
Ok, also kein Wakeup, nur cache zurück setzen.

Schnabulation, 2025-02-17:
Wäre wohl das Einfachste, hät ich gesagt...

closed 2025-02-24 by andig