evcc-io/evcc #18626: "prevent grid usage" switch in EVCC

AloisKlingler, 2025-02-06:
I have 4 PV inverters, one of them with Battery.

there is no issue as long as enough sun is here.
as soon as there is not enough sun, I cannot use fast charge, because the inverter with the battery attached has insufficient power.

PV1: Fronius Gen 24 10.0 with 2x BYD HVM 22
AC-Power max: 10300W
Battery-only Power max: ~9300W (depends in case of fronius and BYD of the battery size and type how fast can be discharged / charged)

PV2: Fronius Symo 12.5
AC-Power max: 12600W

PV3: Fronius Symo 5.0
AC-Power max: 5050W

PV4: Hoymiles HMT-2250
AC-Power max: 2250W

I have a the smartmeter attached on the PV1 gen24, it is located between grid and house, it is configured and read in evcc

Using EVCC in "Fast" mode leads to grid consumption, if:
-) it is night (only ~9300W are available)
-) PV2, PV3 and PV4 produce less than ~3000W

It would be great to have an global option (which applies to Fast mode as well) to "prevent" grid usage.

suggestions:
-) make the max charge/discharge power configureable on the battery (so evcc knows if there is no PV power at all it must not set more power than this configuration)
-) have a switch to prevent grid usage (if no PV power is available, other equipment in house consumes energy which may lead to have less W available for changing

premultiply, 2025-02-06:
Battery power is not constant.
It depends on SoC, temperature, age etc.
So this will not be possible.

Did you try the battery boost mode?

jeffborg, 2025-02-07:
@AloisKlingler I think you can do this already in evcc via limiting via circuits and doing it externally.

```
# Using circuits with a dynamic limit - in home assistant I have a value to calculate the max power draw in any 1/2 hour block so it's dynamic. Values pushed into mqtt from home assistant but you can do via any plugin method
circuits:
- name: main
  meter: grid
  maxpower: 2000 # default value if mqtt topic not there
  GetMaxPower:
    source: mqtt # can be any plugin http etc...
    topic: amber/max_demand # your mqtt topic name

# OR VIA HEMS but the value isn't dynamically adjustable just ON/OFF
# Note I use the above one now.
 hems:
   type: relay
   maxPower: 2400
   limit: # boolean getter when true limit to 2400w
     # source: mqtt
     # topic: amber/demandActive
     # timeout: 24h
     source: js
     script: |
       // ausgrid EA116 demand active calculation
       var d = new Date()
       var DEMAND_MONTHS = [11,12,1,2,3,6,7,8]
       var isDemand = d.getHours() >= 15 && d.getHours() < 21 && DEMAND_MONTHS.indexOf(d.getMonth() + 1) !== -1
       console.log("HOUR: ", d.getHours(), "MONTH: ", d.getMonth() + 1, "ISDEMAND: ", isDemand)
       isDemand

```

premultiply, 2025-02-07:
> Battery Boost is not available in Fast mode.

Yes, intentionally not. Fast means fast regardless of anything else.

closed 2025-02-07 by premultiply