evcc-io/evcc #18628: Kostenberechnung bei der Nutzung von PV-Strom

StefanA-LU, 2025-02-06:
**Is your feature request related to a problem? Please describe.**
Meines Erachtens sind die Kosten für PV-Strom noch immer falsch berechnet. Berücksichtigt werden nur die "nicht gekauften Strommengen". Ggf. werden noch die durch fehlende Einspeisung fehlenden Erlöse berücksichtigt, aber das wäre an dieser Stelle auch noch zu prüfen.
Was aber definitiv fehlt, sind die Anschaffungs-/Abschreibungs-Berechnungen für eine PV-Anlage. Wer behauptet, dass der PV-Strom kostenlos ist, verschweigt dies. Angesichts der einmal mehr grundgesetzwidrigen Gesetzesänderung in 12/2022, wo rückwirkend zum 1.1.2022 alle Anlagen bis 30 KWp als "Spielerei" deklassiert wurden, weil man halt keine Ahnung hat, was die Vor-/Nachteile eines Gewerbebetriebs sind, haben sich je nach Erfolg der eingelegten Einsprüche (angehängt an BFH Az. III B 24/24 und Hauptsacheverfahren, das noch offen ist) schlicht die Bedingungen deutlich verschlechtert. Die Kalkulation vor der Investition wurden durch Ungleichbehandlung und Verstoß gegen Bestandschutzgrundlagen mit einem Schlag derart verschlechtert, dass es sich schlicht nicht mehr lohnt.

Warum? Ich habe es ausgerechnet. Bei wohlwollenden 20 Jahren Laufleistung fallen damit unter Beachtung von Anlagenkosten, Eigenstrom und Einspeisung 0,33 Euro/KWh an. Der Netzbezug kostet aktuell 0,38 Euro/KWh und wäre bei Bezug von 4500 KWh statt nur 1300 KWh bei 0,28 Euro/KWh.

Vor diesem rückwirkenden Gesetz, das bei vorheriger Kenntnis zu einer Entscheidung gegen die Anschaffung geführt hätte, hätte sich die Anlage durch Abschreibung bei nachgewiesener Gewinnabsicht (es wurde daher ja auch genehmigt!), nach 10 Jahren amortisiert. Legt man jetzt die obige Rechnung auf 10 Jahre aus, kostet die PV-KWh jetzt satte 0,75 Euro.

Soviel zu den guten Taten bei der Unterstützung der sogenannten Energiewende. Aber selbst ohne diesen neuerlichen Rohrkrepierer, der natürlich ab dem 23.03. NICHT korrigiert wird, wenn der BFH oder das BG / EUGH das nicht kassieren, müsste auch bei einem Gewerbebetrieb PV die reale Kostenstruktur abbildbar sein.

**Describe the solution you'd like**
Zur Berechnung der korrekten Kosten, sollten die Anschaffungs-/Betriebs-Kosten, sowie etwaige Abschreibungen mit erfasst werden und in die Berechnung einfließen. So wird eine ehrliche Bilanz daraus. Speziell in Kombination mit möglicher Einspeisung / erfolgter Einspeisung / Netzbezug.

**Describe alternatives you've considered**
Keine. Manuelle Exceltabellen aus den /bei mir E3DC) Statistiken und dem Zähler.

**Additional context**
-

VolkerK62, 2025-02-06:
dazu kommen evtl. noch Kreditzinsen und die USt für den Eigenverbrauch ....

Meine Meinung: 
Ich denke, das kann und sollte nicht Aufgabe von evcc sein. 
Es liegt an dir, das im Wert für `feedin` zu berücksichtigen.

naltatis, 2025-02-06:
Jo, wir haben nicht den Anspruch allgemeine Wirtschaftlichkeitsberechnungen für PV- und Speichersysteme zu implementieren. Anschaffungkosten und co. sind definitiv out-of-scope.

closed 2025-02-06 by naltatis

StefanA-LU, 2025-02-14:
Man muss ja nicht die Premiumlösung umsetzen. Aber zur Ehrlichkeit gehört eben, dass die Annahme, nur der "Einspeisepreis" die PV-Kosten sind, schlicht falsch ist. Dann macht ein einfaches Feld "PV-Kosten", wo sich das jeder selbst ausrechnen und eintragen kann, da kann er dann auch alle ggf. weiteren Parameter berücksichtigen und seinen Durchschnittswert eingeben. Das bildet es dann wenigstens sauber ab, nur die "Einspeisekosten" anzusetzen kann man sich auch ganz sparen, weil dann ist es Spielerei und hat mit der Realität nichts zu tun. Da ich im Widerspruchsverfahren bin, stelle ich eh Rechnungen als Gewerbebetrieb an den Netzbetreiber und mich als privat aus, zumal ich eh von der Kleinunternehmerregel §19 auf die Regelbesteuerung der Umsatzsteuer extra wegen der Anlage umgestiegen bin und das Thema auch noch mit dabei ist.