evcc-io/evcc #18629: SOC per MQTT wird nicht zuverlässig aktualisiert

ThomasWetzel, 2025-02-06:
### Describe the bug

Unser Kia Niro erhält seinen SOC über ein MQTT Topic. **(evcc/niro/soc)**
Der SOC wird auch erkannt und im Log festgehalten:

[sma   ] TRACE 2025/02/06 15:22:02 recv 10.2.201.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/02/06 15:22:03 recv 10.2.201.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/02/06 15:22:04 recv 10.2.201.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[mqtt  ] TRACE 2025/02/06 15:22:04 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/06 15:22:04 recv evcc/superb/soc: '31'
[mqtt  ] TRACE 2025/02/06 15:22:04 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/06 15:22:04 recv evcc/superb/range: '10'
[sma   ] TRACE 2025/02/06 15:22:05 recv 10.2.201.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/02/06 15:22:06 recv 10.2.201.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/02/06 15:22:07 recv 10.2.201.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/02/06 15:22:08 recv 10.2.201.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]

Allerdings werden die 36% nicht gespeichert:

[image]

Die 110 km Restreichweite werden allerdings empfangen und gespeichert.

Kann bitte jemand überprüfen, warum der SOC nicht zuverlässig gespeichert wird?

Ich kann gerne auch unterstützen.
Vielen Dank!

### Steps to reproduce

1. mir fällt kein Weg ein, das zu reproduzieren, außer das UPDATE-Statement in der Datenbank auszuwerten. Das wird aber leider nicht mitgetract.

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

log: trace
levels:
  cache: error

plant: e43e450bdc53f143da1713333cafc101b94e2d9c0caf32f6f7e72fb57ff0bcb2

interval: 30s

meters:
  - type: template
    template: sma-home-manager
    usage: grid
    host: 10.2.201.120
    name: sunny

  - name: solar-se-pv
    type: template
    template: sma-hybrid
    usage: pv
    modbus: tcpip
    id: 3
    host: 10.2.201.106
    port: 502

  - name: solar-se-battery
    type: template
    template: sma-hybrid
    usage: battery
    modbus: tcpip
    id: 3
[... 169 lines cut ...]
[sma   ] TRACE 2025/02/06 15:22:19 recv 10.2.201.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/02/06 15:22:20 recv 10.2.201.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/02/06 15:22:21 recv 10.2.201.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/02/06 15:22:22 recv 10.2.201.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/02/06 15:22:23 recv 10.2.201.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/02/06 15:22:24 recv 10.2.201.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/02/06 15:22:25 recv 10.2.201.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/02/06 15:22:26 recv 10.2.201.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.133.0

StefanSchoof, 2025-02-06:
Kann es sein, dass du den poll.mode entsprechend deiner Erwartung konfiguriert hast? https://docs.evcc.io/docs/reference/configuration/loadpoints#poll

ThomasWetzel, 2025-02-06:
Die Daten werden initial von einer HomeAssistant-Integration von Kia abgeholt und dann über eine zeitgesteuerte NodeRed-Automation jede Minute an das Topic aktualisiert.

Diese Daten kommen ja auch an. Das habe ich sowohl mit einem MQTT-Listener überprüft als auch wird der Wert von evcc erkannt, wie im Trace zu sehen ist.

Ein pull.mode kommt hier gar nicht imho vor. Die Reichweite, die 1:1 genauso konfiguriert ist, wird richtig gelesen und auch aktualisiert. Daher können wir den pull.mode - denke ich - ausschließen.

Das Verhalten ist leider auch nicht reproduzierbar, da durch einen Neustart die Werte wieder richtig aktualisiert werden.

Wäre es denkbar, die Datenbank-Kommandos zu tracen? Dann müssten wir, was in der Datenbank ankommt.

mfuchs1984, 2025-02-06:
Hast du eine längeres Logfile, das die Entwicklung zeigt?

ThomasWetzel, 2025-02-07:
Ich habe mal hier einen längeren MQTT-Auszug, der aber immer wieder das gleiche zeigt:

`[mqtt  ] TRACE 2025/02/07 09:00:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:00:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:00:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:00:09 recv evcc/superb/range: '18'
[mqtt  ] TRACE 2025/02/07 09:01:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:01:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:01:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:01:09 recv evcc/superb/range: '18'
[mqtt  ] TRACE 2025/02/07 09:02:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:02:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:02:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:02:09 recv evcc/superb/range: '18'
[mqtt  ] TRACE 2025/02/07 09:03:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:03:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:03:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:03:09 recv evcc/superb/range: '18'
[mqtt  ] TRACE 2025/02/07 09:04:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:04:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:04:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:04:09 recv evcc/superb/range: '18'
[mqtt  ] TRACE 2025/02/07 09:05:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:05:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:05:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:05:09 recv evcc/superb/range: '18'
[mqtt  ] TRACE 2025/02/07 09:06:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:06:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:06:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:06:09 recv evcc/superb/range: '18'
[mqtt  ] TRACE 2025/02/07 09:07:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:07:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:07:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:07:09 recv evcc/superb/range: '18'
[mqtt  ] TRACE 2025/02/07 09:08:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:08:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:08:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:08:09 recv evcc/superb/range: '18'
[mqtt  ] TRACE 2025/02/07 09:09:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:09:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:09:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:09:09 recv evcc/superb/range: '18'
[mqtt  ] TRACE 2025/02/07 09:10:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:10:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:10:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:10:09 recv evcc/superb/range: '18'
[mqtt  ] TRACE 2025/02/07 09:11:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:11:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:11:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:11:09 recv evcc/superb/range: '18'
[mqtt  ] TRACE 2025/02/07 09:12:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:12:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:12:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:12:09 recv evcc/superb/range: '18'
[mqtt  ] TRACE 2025/02/07 09:13:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:13:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:13:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:13:09 recv evcc/superb/range: '18'
[mqtt  ] TRACE 2025/02/07 09:14:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:14:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:14:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:14:09 recv evcc/superb/range: '18'
[mqtt  ] TRACE 2025/02/07 09:15:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:15:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:15:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:15:09 recv evcc/superb/range: '18'
[mqtt  ] TRACE 2025/02/07 09:16:09 recv evcc/niro/soc: '36'
[mqtt  ] TRACE 2025/02/07 09:16:09 recv evcc/superb/soc: '53'
[mqtt  ] TRACE 2025/02/07 09:16:09 recv evcc/niro/range: '110'
[mqtt  ] TRACE 2025/02/07 09:16:09 recv evcc/superb/range: '18'
[... 283 lines cut ...]
[mqtt ] TRACE 2025/02/07 09:38:16 delete: evcc/site/statistics/30d/solarPercentage
[mqtt ] TRACE 2025/02/07 09:38:16 delete: evcc/site/statistics/365d/solarPercentage
[mqtt ] TRACE 2025/02/07 09:38:16 delete: evcc/site/statistics/365d/chargedKWh
[mqtt ] TRACE 2025/02/07 09:38:16 delete: evcc/site/statistics/365d/avgPrice
[mqtt ] TRACE 2025/02/07 09:38:16 delete: evcc/site/statistics/365d/avgCo2
[mqtt ] TRACE 2025/02/07 09:38:16 delete: evcc/site/statistics/thisYear/solarPercentage
[mqtt ] TRACE 2025/02/07 09:38:16 delete: evcc/site/statistics/thisYear/chargedKWh
[mqtt ] TRACE 2025/02/07 09:38:16 delete: evcc/site/statistics/thisYear/avgPrice
[mqtt ] TRACE 2025/02/07 09:38:16 delete: evcc/site/statistics/thisYear/avgCo2
[mqtt ] TRACE 2025/02/07 09:38:16 delete: evcc/site/statistics/total/solarPercentage
[mqtt ] TRACE 2025/02/07 09:38:16 delete: evcc/site/statistics/total/chargedKWh
[mqtt ] TRACE 2025/02/07 09:38:16 delete: evcc/site/statistics/total/avgPrice
[mqtt ] TRACE 2025/02/07 09:38:16 delete: evcc/site/statistics/total/avgCo2
[mqtt ] TRACE 2025/02/07 09:39:09 recv evcc/niro/soc: '36'
[... cut ...]
[mqtt ] TRACE 2025/02/07 09:39:09 recv evcc/superb/range: '18'
[mqtt ] TRACE 2025/02/07 09:40:09 recv evcc/niro/soc: '36'
[mqtt ] TRACE 2025/02/07 09:40:09 recv evcc/superb/soc: '53'
[mqtt ] TRACE 2025/02/07 09:40:09 recv evcc/niro/range: '110'
[mqtt ] TRACE 2025/02/07 09:40:09 recv evcc/superb/range: '18'
[mqtt ] TRACE 2025/02/07 09:41:09 recv evcc/niro/soc: '36'
[mqtt ] TRACE 2025/02/07 09:41:09 recv evcc/superb/soc: '53'
[mqtt ] TRACE 2025/02/07 09:41:09 recv evcc/niro/range: '110'
[mqtt ] TRACE 2025/02/07 09:41:09 recv evcc/superb/range: '18'`

andig, 2025-02-07:
Die Logschnipsel helfen nicht da nirgendwo ersichtlich ist, welcher/ dass ein falscher vehicle soc verwendet würde. Ich mache zu bis es ein passendes Logfile gibt (loadpoint auf debug)

closed 2025-02-07 by andig

ThomasWetzel, 2025-02-07:
Ich denke, der Unterschied kommt daher, dass das EstimatedSoc angezeigt wird.

Das kommt aus der Zeile

soc estimated: 90.09% (vehicle: 70.00%)

Vielleicht liegt es ja daran, das die Schätzung herangezogen wird?

ThomasWetzel, 2025-02-07:
Irgendwie ist der Estimator der Meinung, dass die vorhandenen Werte keinen richtigen Sinn mehr ergeben und dann wird der SOC ermittelt, anstatt den vehicleSoc zu verwenden:

if s.estimate && s.virtualCapacity > 0 {
		socDelta := s.vehicleSoc - s.prevSoc
		energyDelta := max(chargedEnergy, 0) - s.prevChargedEnergy

		if socDelta != 0 || energyDelta < 0

wenn dies der Fall ist, dann wird der Soc berechnet:

} else {
			s.vehicleSoc = min(*fetchedSoc+energyDelta/s.energyPerSocStep, 100)
			s.log.DEBUG.Printf("soc estimated: %.2f%% (vehicle: %.2f%%)", s.vehicleSoc, *fetchedSoc)
		}

Der Wert kommt anscheinend richtig vom MQTT, aber entweder ist socDelta nicht genau 0 oder das energyDelta kleiner als 0.

Was könnte das für eine Ursache haben?

ThomasWetzel, 2025-02-07:
Ich glaube, den Grund für das Phänome gefunden zu haben:

Das Batterie-Level wird alle 30 Minuten abgefragt.
In dieser Zeit wird das Auto aber weiter fleißig geladen.

Nun kommt der Code an die Stelle, wo überprüft wird, ob das Auto die geladene Energie berücksichtigt.
Gibt es hier eine Differenz, wird auf den Fallback zurückgegriffen und die geschätzte Ladung angezeigt, anstatt auf den Vehicle-Wert zu vertrauen.

Das führt dazu, dass das Auto voller geschätzt wird, als es tatsächlich ist.

Wenn noch die chargedEnergy und die prevChargedEnergy getraced werden könnte, wären wir schlauer. :-)

Vielen Dank im übrigen für das tolle Projekt.

ThomasWetzel, 2025-02-07:
Noch eine Erkenntnis:

nach einiger Zeit ist socDiff > 10 und energyDiff > 0

und danach ist die Berechnung wieder intakt, wie man hier sieht:

hier geht der Ladestand schlagartig von 95% auf 77% (richtiger Wert) runter:

[lp-1 ] DEBUG 2025/02/07 14:03:17 pv disable timer remaining: 1m0s
[lp-1 ] DEBUG 2025/02/07 14:03:47 charge power: 2632W
[lp-1 ] DEBUG 2025/02/07 14:03:47 charge currents: [0.63 5.94 5.88]A
[lp-1 ] DEBUG 2025/02/07 14:04:17 charge power: 2629W
[lp-1 ] DEBUG 2025/02/07 14:04:17 charge currents: [0.63 5.92 5.87]A
[lp-1 ] DEBUG 2025/02/07 14:04:17 charge total import: 4399.992kWh
[lp-1 ] DEBUG 2025/02/07 14:04:17 charger status: C
[lp-1 ] DEBUG 2025/02/07 14:04:17 soc estimated: 95.20% (vehicle: 70.00%)
[lp-1 ] DEBUG 2025/02/07 14:04:17 vehicle soc: 95%
[lp-1 ] DEBUG 2025/02/07 14:04:17 vehicle range: 234km
[lp-1 ] DEBUG 2025/02/07 14:04:17 pv charge current: 0A = 6A + -9.61A (2209W @ 1p)
[lp-1 ] DEBUG 2025/02/07 14:04:17 projected site power 2209W >= 0W disable threshold
[lp-1 ] DEBUG 2025/02/07 14:04:17 pv disable in 0s
[lp-1 ] DEBUG 2025/02/07 14:04:17 pv disable timer remaining: 0s
[lp-1 ] DEBUG 2025/02/07 14:04:47 charge power: 2632W
[lp-1 ] DEBUG 2025/02/07 14:04:47 charge currents: [0.63 5.93 5.87]A
[lp-1 ] DEBUG 2025/02/07 14:05:17 charge power: 2632W
[lp-1 ] DEBUG 2025/02/07 14:05:17 charge currents: [0.63 5.93 5.86]A
[lp-1 ] DEBUG 2025/02/07 14:05:17 detected active phases: 2p
[lp-1 ] DEBUG 2025/02/07 14:05:17 charge total import: 4400.036kWh
[lp-1 ] DEBUG 2025/02/07 14:05:17 charger status: C
[lp-1 ] DEBUG 2025/02/07 14:05:17 soc gradient updated: soc: 77.0%, socDiff: 41.0%, energyDiff: 13302Wh, energyPerSocStep: 324.4Wh, virtualCapacity: 32444Wh
[lp-1 ] DEBUG 2025/02/07 14:05:17 vehicle soc: 77%
[lp-1 ] DEBUG 2025/02/07 14:05:17 vehicle range: 261km
[lp-1 ] WARN 2025/02/07 14:05:17 charger logic error: phases mismatch (got 3 measured, expected 1)
[lp-1 ] DEBUG 2025/02/07 14:05:17 !! pvScalePhases DOWN activePhases: 3, available power: 369W, scalable: true
[lp-1 ] DEBUG 2025/02/07 14:05:17 available power 369W < 4140W min 3p threshold
[lp-1 ] DEBUG 2025/02/07 14:05:17 start phase scale1p timer
[lp-1 ] DEBUG 2025/02/07 14:05:17 phase scale1p in 3m0s
[lp-1 ] DEBUG 2025/02/07 14:05:17 !! pvScalePhases UP maxPhases: 3, available power: 369W, scalable: false
[lp-1 ] DEBUG 2025/02/07 14:05:17 !! pvScalePhases EXIT
[lp-1 ] DEBUG 2025/02/07 14:05:17 pv charge current: 2.72A = 6A + -3.28A (2263W @ 3p)
[lp-1 ] DEBUG 2025/02/07 14:05:17 pv disable timer reset
[lp-1 ] DEBUG 2025/02/07 14:05:17 pv timer inactive
[lp-1 ] DEBUG 2025/02/07 14:05:47 charge power: 2631W
[lp-1 ] DEBUG 2025/02/07 14:05:47 charge currents: [0.63 5.94 5.88]A
[lp-1 ] DEBUG 2025/02/07 14:06:17 charge power: 2633W
[lp-1 ] DEBUG 2025/02/07 14:06:17 charge currents: [0.63 5.95 5.88]A
[lp-1 ] DEBUG 2025/02/07 14:06:17 detected active phases: 2p
[lp-1 ] DEBUG 2025/02/07 14:06:17 charge total import: 4400.081kWh
[lp-1 ] DEBUG 2025/02/07 14:06:17 charger status: C
[lp-1 ] DEBUG 2025/02/07 14:06:17 soc estimated: 77.14% (vehicle: 77.00%)
[lp-1 ] DEBUG 2025/02/07 14:06:17 vehicle soc: 77%
[lp-1 ] DEBUG 2025/02/07 14:06:17 vehicle range: 261km

ThomasWetzel, 2025-02-07:
Werde ich machen.

Der Estimator ist ja eine prima Sache, wenn er aber 20% zu viel Ladung berechnet wird, führt das im Endeffekt zu einem Aha-Effekt, wenn das Auto doch nicht geladen ist.

ThomasWetzel, 2025-02-07:
Dumme Frage:

Wo kann ich in meiner Konfiguration estimate: false setzen?

Ich habe es auf SOC-Level, Loadpoint und vehicle probiert.

Jeweils wurde mir gemeldet, dass die Konfiguration nicht gültig ist.

mfuchs1984, 2025-02-07:
Loadpoint -> so

https://docs.evcc.io/docs/reference/configuration/loadpoints#soc

ThomasWetzel, 2025-02-07:
Merci.

So wird die Konfiguration akzeptiert.