evcc-io/evcc #18653: Modbus RTU - Unterstützung für 'swapped' CRC

holgerscheller, 2025-02-07:
Ich habe versucht das relative neue Speichersysteme Venus E von Marstek über Modus in evcc zu integrieren. Bei der Recherche bin ich darauf gestossen, dass Marstek hier wohl bei der CRC Checksumme die Bytes vertauscht. 
Beim Modubus Protokoll, und in diesem Fall habe ich es mit Modus RTU über TCP/IP versucht, scheinen über die letzten Jahre einige Hersteller das bewusst oder unterbewusst gemacht zu haben. 
Meine Frage wäre ob man eine Option anbieten könnte, die einem es erlaubt über das Custom Template eine "getauschte" CRC Quersumme zu erlauben?

holgerscheller, 2025-02-07:
Hier die entsprechende Fehlermeldung beim Versuch ein Register auszulesen:
[hausbatterie-venus-e] TRACE 2025/02/07 09:43:39 modbus: send 01 03 7d 68 00 02 5d bb
[hausbatterie-venus-e] TRACE 2025/02/07 09:43:39 modbus: recv 01 03 04 00 0b 02 00 91 8a
Power: read failed: modbus: response crc '35473' does not match expected '37258'

andig, 2025-02-07:
> dass Marstek hier wohl bei der CRC Checksumme die Bytes vertauscht.

Das verletzt die Spezifikation. Ähnliche "wünsch dir was" Implmentierungen gibts auch bei Goodwe die eine falsche Modbus ID zurück geben. Gerne PR um das zu lösen- hat keine Prio fürs Kernteam.

premultiply, 2025-02-07:
Da wäre in erster Linie der Hersteller zu kontaktieren um das zu fixen.

closed 2025-02-07 by andig