evcc-io/evcc #18680: Tessie Custom Charger

bill340, 2025-02-08:
Is there still any possibility to implement a custom charger based on tessie?
https://github.com/evcc-io/evcc/pull/17274
works actually really well for me since some months now.
Only since some time it does not display any "charged" kWh anymore. It stays at 0. Don't know if it has anything to do with the tessie charger or something else.
Here is my evcc.yaml that I use sucessfully with the tessie charger:
...
chargers:
- name: tessiechargercher
#    # REPLACE VIN with your VIN
#    # REPLACE TOKEN with your token
#    # REPLACE {{ .lat }} and {{ .lon }} with your location's latitude and longitude
    type: custom
    power:
      source: http
      uri: https://api.tessie.com/XP7YGCEK8RBxxxxxx/state?use_cache=true
      headers:
        Authorization: Bearer xxxxxxxxxxxxxxxxx
      jq: |
        if ((.drive_state.latitude - xx.xxxx | abs) < 0.001) and
           ((.drive_state.longitude - xx.xxxx | abs) < 0.001) then
          .charge_state.charger_power * 1000
        else
          0
        end
      cache: 30s
    energy:
      source: http
      uri: https://api.tessie.com/XP7YGCEK8RBxxxxxx/state?use_cache=true
      headers:
        Authorization: Bearer xxxxxxxxxxxxxxxxx
      jq: |
        if ((.drive_state.latitude - xx.xxxx | abs) < 0.001) and
           ((.drive_state.longitude - xx.xxxx | abs) < 0.001) then
          .charge_state.charge_energy_added
        else
          0
        end
      cache: 30s
    status:
      source: http
      cache: 30s
      uri: https://api.tessie.com/XP7YGCEK8RBxxxxxx/state?use_cache=true
      headers:
        Authorization: Bearer xxxxxxxxxxx
      jq: |
        if ((.drive_state.latitude - xx.xxxx | abs) < 0.001) and
           ((.drive_state.longitude - xx.xxxx | abs) < 0.001) then
          if .charge_state.charging_state == "Charging" then "C"
          elif .charge_state.charging_state == "Complete" then "D"
          elif .charge_state.charge_port_door_open then "B"
          else "A"
          end
        else
          "A"
        end
    enabled:
      source: http
      uri: https://api.tessie.com/XP7YGCEK8RBxxxxxxx/state?use_cache=true
      headers:
        Authorization: Bearer xxxxxxxxxxxxxxxxxxx
      jq: |
        if ((.drive_state.latitude - xx.xxxx | abs) < 0.001) and
           ((.drive_state.longitude - xx.xxxx | abs) < 0.001) then
          .charge_state.charging_state == "Charging"
        else
          false
        end
      cache: 30s
    maxcurrent:
      source: http
      uri: [https://api.tessie.com/XP7YGCEK8RBxxxxxxxx/command/set_charging_amps?retry_duration=40&wait_for_completion=true&amps=${maxcurrent}](https://api.tessie.com/XP7YGCEK8RBxxxxxxxx/command/set_charging_amps?retry_duration=40&wait_for_completion=true&amps=$%7bmaxcurrent%7d)
      headers:
        Authorization: Bearer xxxxxxxxxxxxxxxxxxx
      method: POST
    enable:
      source: http
      uri: https://api.tessie.com/XP7YGCEK8xxxxxxxx/command/{{ if .enable }}start{{ else }}stop{{ end }}_charging?retry_duration=40&wait_for_completion=true
      headers:
        Authorization: Bearer xxxxxxxxxxxx
      method: POST

mucki12, 2025-02-08:
Sorry, no ideas from me. I just want to follow this thread because I'm currently considering every possible control option, as there are always problems with the final version of evcc and Tesla. Maybe that could be a solution.

andig, 2025-02-09:
Lets continue in https://github.com/evcc-io/evcc/pull/17274

closed 2025-02-09 by andig