evcc-io/evcc #18681: openWB Pro erkennt Vi

TobiasHuber1980, 2025-02-08:
### Describe the bug

Die Erkennung der Fahrzeuge funktioniert bei meiner openWB Pro nicht mehr. Es zeigt nur Gastfahrzeug mit den drehenden Pfeilen.
Evtl. seit https://github.com/evcc-io/evcc/pull/18137 

### Steps to reproduce

1. evcc starten
2. in diesem Fall ist der ID.3 angeschlossen; der ID.4 hängt an einer anderen Wallbox.
3.
...

### Configuration details

```yaml
alles im UI konfiguriert
```

### Log details

```text
[main ] INFO 2025/02/08 16:00:37 evcc 0.133.0 (f83da937)
[main ] INFO 2025/02/08 16:00:37 using config file: /etc/evcc.yaml
[db ] INFO 2025/02/08 16:00:37 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2025/02/08 16:00:38 listening at :7070
[lp-1 ] WARN 2025/02/08 16:00:43 deprecated: mode setting is ignored, please remove
[lp-1 ] WARN 2025/02/08 16:00:43 deprecated: soc setting is ignored, please remove
[lp-1 ] DEBUG 2025/02/08 16:00:43 set title: Garage
[lp-1 ] DEBUG 2025/02/08 16:00:43 set priority: 0
[lp-1 ] DEBUG 2025/02/08 16:00:43 set smart cost limit: <nil>
[lp-1 ] DEBUG 2025/02/08 16:00:43 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
[lp-1 ] DEBUG 2025/02/08 16:00:43 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-1 ] DEBUG 2025/02/08 16:00:43 set session energy limit: 0
[lp-1 ] DEBUG 2025/02/08 16:00:43 set session soc limit: 0
[lp-1 ] DEBUG 2025/02/08 16:00:43 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0x4000a8be88}
[lp-1 ] DEBUG 2025/02/08 16:00:43 set default mode: pv
[lp-1 ] DEBUG 2025/02/08 16:00:43 set phases: 0
[lp-1 ] DEBUG 2025/02/08 16:00:43 charge total import: 2014.238kWh
[site ] INFO 2025/02/08 16:00:44 site config:
[site ] INFO 2025/02/08 16:00:44 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2025/02/08 16:00:44 grid: power ✓ energy ✓ currents ✗
[site ] INFO 2025/02/08 16:00:44 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2025/02/08 16:00:44 battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site ] INFO 2025/02/08 16:00:44 vehicles:
[site ] INFO 2025/02/08 16:00:44 vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site ] INFO 2025/02/08 16:00:44 vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site ] INFO 2025/02/08 16:00:44 vehicle 3: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1 ] INFO 2025/02/08 16:00:44 loadpoint 1:
[lp-1 ] INFO 2025/02/08 16:00:44 mode: off
[lp-1 ] INFO 2025/02/08 16:00:44 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] INFO 2025/02/08 16:00:44 meters: charge ✓
[lp-1 ] INFO 2025/02/08 16:00:44 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2025/02/08 16:00:44 set buffer soc: 100
[site ] DEBUG 2025/02/08 16:00:44 set buffer start soc: 100
[site ] DEBUG 2025/02/08 16:00:44 set priority soc: 85
[site ] DEBUG 2025/02/08 16:00:44 set battery discharge control: false
[site ] DEBUG 2025/02/08 16:00:44 set residual power: 100
[lp-1 ] DEBUG 2025/02/08 16:00:44 phase timer inactive
[lp-1 ] DEBUG 2025/02/08 16:00:44 pv timer inactive
[site ] DEBUG 2025/02/08 16:00:44 ----
[lp-1 ] DEBUG 2025/02/08 16:00:44 charge power: 0W
[lp-1 ] DEBUG 2025/02/08 16:00:44 charge currents: [0 0 0]A
[site ] DEBUG 2025/02/08 16:00:44 grid power: -1257W
[site ] DEBUG 2025/02/08 16:00:44 battery 1 power: -4W
[site ] DEBUG 2025/02/08 16:00:44 battery 1 soc: 100%
[site ] DEBUG 2025/02/08 16:00:44 pv 1 power: 1823W
[site ] DEBUG 2025/02/08 16:00:44 site power: -1161W
[lp-1 ] DEBUG 2025/02/08 16:00:44 charge total import: 2014.238kWh
[lp-1 ] DEBUG 2025/02/08 16:00:44 charger status: B
[lp-1 ] DEBUG 2025/02/08 16:00:44 charge start energy: 2014.238kWh
[lp-1 ] INFO 2025/02/08 16:00:44 car connected
[lp-1 ] DEBUG 2025/02/08 16:00:44 vehicle api refresh
[lp-1 ] DEBUG 2025/02/08 16:00:44 pv timer elapse
[lp-1 ] DEBUG 2025/02/08 16:00:44 pv timer inactive
[lp-1 ] DEBUG 2025/02/08 16:00:44 charge total import: 2014.238kWh
[main ] DEBUG 2025/02/08 16:00:44 vehicle status: A (VW ID.5)
[main ] DEBUG 2025/02/08 16:00:45 vehicle status: B (VW ID.4)
[main ] DEBUG 2025/02/08 16:00:45 vehicle status: B (VW ID.3)
[main ] WARN 2025/02/08 16:00:45 vehicle status: >1 matches, giving up
[main ] DEBUG 2025/02/08 16:01:48 vehicle status: A (VW ID.5)
[main ] DEBUG 2025/02/08 16:01:48 vehicle status: B (VW ID.4)
[main ] DEBUG 2025/02/08 16:01:49 vehicle status: B (VW ID.3)
[main ] WARN 2025/02/08 16:01:49 vehicle status: >1 matches, giving up
[site ] DEBUG 2025/02/08 16:02:15 ----
[lp-1 ] DEBUG 2025/02/08 16:02:15 charge power: 0W
[lp-1 ] DEBUG 2025/02/08 16:02:15 charge currents: [0 0 0]A
[site ] DEBUG 2025/02/08 16:02:16 battery 1 power: -3W
[site ] DEBUG 2025/02/08 16:02:16 grid power: -1232W
[site ] DEBUG 2025/02/08 16:02:16 battery 1 soc: 100%
[site ] DEBUG 2025/02/08 16:02:16 pv 1 power: 1792W
[site ] DEBUG 2025/02/08 16:02:16 site power: -1135W
[lp-1 ] DEBUG 2025/02/08 16:02:16 charge total import: 2014.238kWh
[lp-1 ] DEBUG 2025/02/08 16:02:16 charger status: B
[main ] DEBUG 2025/02/08 16:02:16 vehicle status: A (VW ID.5)
[main ] DEBUG 2025/02/08 16:02:16 vehicle status: B (VW ID.4)
[main ] DEBUG 2025/02/08 16:02:16 vehicle status: B (VW ID.3)
[main ] WARN 2025/02/08 16:02:16 vehicle status: >1 matches, giving up
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0 (f83da937)

TobiasHuber1980, 2025-02-08:
ich glaube denkfehler meinerseits - ich mach hier erst mal zu

TobiasHuber1980, 2025-02-09:
ich mach hier noch mal auf, fahrzeugerkennung funktioniert bei mir nicht (mehr).

```
[main ] INFO 2025/02/09 13:12:58 evcc 0.133.0 (07fb01d2)
[main ] INFO 2025/02/09 13:12:58 using config file: /etc/evcc.yaml
[db ] INFO 2025/02/09 13:12:58 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2025/02/09 13:12:58 listening at :7070
[lp-1 ] DEBUG 2025/02/09 13:13:00 set title: Garage
[lp-1 ] DEBUG 2025/02/09 13:13:00 set priority: 0
[lp-1 ] DEBUG 2025/02/09 13:13:00 set smart cost limit: <nil>
[lp-1 ] DEBUG 2025/02/09 13:13:00 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
[lp-1 ] DEBUG 2025/02/09 13:13:00 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-1 ] DEBUG 2025/02/09 13:13:00 set session energy limit: 0
[lp-1 ] DEBUG 2025/02/09 13:13:00 set session soc limit: 0
[lp-1 ] DEBUG 2025/02/09 13:13:00 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0x4000d0a4a8}
[lp-1 ] DEBUG 2025/02/09 13:13:00 set default mode: pv
[lp-1 ] DEBUG 2025/02/09 13:13:00 set phases: 0
[lp-1 ] DEBUG 2025/02/09 13:13:01 charge total import: 2014.248kWh
[site ] INFO 2025/02/09 13:13:01 site config:
[site ] INFO 2025/02/09 13:13:01 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2025/02/09 13:13:01 grid: power ✓ energy ✓ currents ✗
[site ] INFO 2025/02/09 13:13:01 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2025/02/09 13:13:01 battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site ] INFO 2025/02/09 13:13:01 vehicles:
[site ] INFO 2025/02/09 13:13:01 vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site ] INFO 2025/02/09 13:13:01 vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site ] INFO 2025/02/09 13:13:01 vehicle 3: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1 ] INFO 2025/02/09 13:13:01 loadpoint 1:
[lp-1 ] INFO 2025/02/09 13:13:01 mode: pv
[lp-1 ] INFO 2025/02/09 13:13:01 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] INFO 2025/02/09 13:13:01 meters: charge ✓
[lp-1 ] INFO 2025/02/09 13:13:01 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2025/02/09 13:13:01 set buffer soc: 100
[site ] DEBUG 2025/02/09 13:13:01 set buffer start soc: 100
[site ] DEBUG 2025/02/09 13:13:01 set priority soc: 85
[site ] DEBUG 2025/02/09 13:13:01 set battery discharge control: false
[site ] DEBUG 2025/02/09 13:13:01 set residual power: 100
[lp-1 ] DEBUG 2025/02/09 13:13:01 phase timer inactive
[lp-1 ] DEBUG 2025/02/09 13:13:01 pv timer inactive
[lp-1 ] DEBUG 2025/02/09 13:13:01 max charge current: 6A
[site ] DEBUG 2025/02/09 13:13:01 ----
[lp-1 ] DEBUG 2025/02/09 13:13:01 charge power: 0W
[lp-1 ] DEBUG 2025/02/09 13:13:01 charge currents: [0 0 0]A
[site ] DEBUG 2025/02/09 13:13:01 battery 1 power: 344W
[site ] DEBUG 2025/02/09 13:13:01 grid power: -753W
[site ] DEBUG 2025/02/09 13:13:01 battery 1 soc: 100%
[site ] DEBUG 2025/02/09 13:13:01 pv 1 power: 3919W
[site ] DEBUG 2025/02/09 13:13:01 site power: -309W
[lp-1 ] DEBUG 2025/02/09 13:13:01 charge total import: 2014.248kWh
[lp-1 ] DEBUG 2025/02/09 13:13:01 charger status: C
[lp-1 ] DEBUG 2025/02/09 13:13:01 charge start energy: 2014.248kWh
[lp-1 ] INFO 2025/02/09 13:13:01 car connected
[lp-1 ] DEBUG 2025/02/09 13:13:01 vehicle api refresh
[lp-1 ] DEBUG 2025/02/09 13:13:01 pv timer elapse
[lp-1 ] DEBUG 2025/02/09 13:13:01 pv timer inactive
[lp-1 ] DEBUG 2025/02/09 13:13:01 charge total import: 2014.248kWh
[lp-1 ] DEBUG 2025/02/09 13:13:01 charge start energy: 2014.248kWh
[lp-1 ] INFO 2025/02/09 13:13:01 start charging ->
[lp-1 ] DEBUG 2025/02/09 13:13:01 wake-up timer: stop
**[main ] DEBUG 2025/02/09 13:13:02 vehicle status: A (VW ID.5)
[main ] DEBUG 2025/02/09 13:13:02 vehicle status: C (VW ID.4)
[main ] DEBUG 2025/02/09 13:13:04 vehicle status: B (VW ID.3)
[main ] WARN 2025/02/09 13:13:04 vehicle status: >1 matches, giving up**
[lp-1 ] DEBUG 2025/02/09 13:13:04 !! pvScalePhases DOWN activePhases: 3, available power: 309W, scalable: false
[lp-1 ] DEBUG 2025/02/09 13:13:04 !! pvScalePhases UP maxPhases: 3, available power: 309W, scalable: false
[lp-1 ] DEBUG 2025/02/09 13:13:04 !! pvScalePhases EXIT
[lp-1 ] DEBUG 2025/02/09 13:13:04 pv charge current: 2.45A = 2A + 0.448A (-309W @ 3p)
[lp-1 ] DEBUG 2025/02/09 13:13:04 pv disable timer reset
[lp-1 ] DEBUG 2025/02/09 13:13:04 pv timer inactive
[site ] DEBUG 2025/02/09 13:13:34 ----
[lp-1 ] DEBUG 2025/02/09 13:13:34 charge power: 1392W
[lp-1 ] DEBUG 2025/02/09 13:13:34 charge currents: [6.1 0 0]A
[site ] DEBUG 2025/02/09 13:13:34 battery 1 power: -23W
[site ] DEBUG 2025/02/09 13:13:34 grid power: -1873W
[site ] DEBUG 2025/02/09 13:13:34 pv 1 power: 4039W
[site ] DEBUG 2025/02/09 13:13:34 battery 1 soc: 100%
[site ] DEBUG 2025/02/09 13:13:35 site power: -1796W
[lp-1 ] DEBUG 2025/02/09 13:13:35 detected active phases: 1p
[lp-1 ] DEBUG 2025/02/09 13:13:35 charge total import: 2014.265kWh
[lp-1 ] DEBUG 2025/02/09 13:13:35 charger status: C
**[main ] DEBUG 2025/02/09 13:13:35 vehicle status: A (VW ID.5)
[main ] DEBUG 2025/02/09 13:13:35 vehicle status: C (VW ID.4)
[main ] DEBUG 2025/02/09 13:13:35 vehicle status: B (VW ID.3)
[main ] WARN 2025/02/09 13:13:35 vehicle status: >1 matches, giving up**
```

andig, 2025-02-09:
Das Log sagt Alles…

closed 2025-02-09 by andig