evcc-io/evcc #18684: "charged" kWh stuck at 0.

bill340, 2025-02-08:
### Describe the bug

When charging, the current power (KW) is shown correctly, but the "charged" kWh stays at 0.

### Steps to reproduce

Everytime I charge it is the same problem.

In the log:
"charge total import: 0.000kWh"
always stays at 0.000kWh

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 10s # control cycle interval

sponsortoken: xxx

chargers:
  - name: tessiechargercher
#    # REPLACE VIN with your VIN
#    # REPLACE TOKEN with your token
#    # REPLACE {{ .lat }} and {{ .lon }} with your location's latitude and longitude
    type: custom
    power:
      source: http
      uri: https://api.tessie.com/xxx/state?use_cache=true
      headers:
        Authorization: Bearer xxx
      jq: |
        if ((.drive_state.latitude - xx.6545 | abs) < 0.001) and
           ((.drive_state.longitude - xx.1651 | abs) < 0.001) then
          .charge_state.charger_power * 1000
        else
          0
        end
      cache: 30s
    energy:
      source: http
      uri: https://api.tessie.com/xxx/state?use_cache=true
      headers:
        Authorization: Bearer xxx
      jq: |
        if ((.drive_state.latitude - xx.6545 | abs) < 0.001) and
           ((.drive_state.longitude - xx.1651 | abs) < 0.001) then
          .charge_state.charge_energy_added
        else
          0
        end
      cache: 30s
    status:
      source: http
      cache: 30s
      uri: https://api.tessie.com/xx/state?use_cache=true
      headers:
        Authorization: Bearer xx
      jq: |
[... 153 lines cut ...]
[site ] DEBUG 2025/02/08 21:30:26 pv 1 power: -7W
[site ] DEBUG 2025/02/08 21:30:26 battery 1 power: 4260W
[site ] DEBUG 2025/02/08 21:30:26 grid power: -36W
[site ] DEBUG 2025/02/08 21:30:26 battery 1 soc: 26%
[site ] DEBUG 2025/02/08 21:30:26 site power: 4324W
[lp-1 ] DEBUG 2025/02/08 21:30:26 charge total import: 0.000kWh
[lp-1 ] DEBUG 2025/02/08 21:30:26 charger status: C
[lp-1 ] DEBUG 2025/02/08 21:30:27 soc estimated: 81.00% (vehicle: 81.00%)
[lp-1 ] DEBUG 2025/02/08 21:30:27 vehicle soc: 81%
[lp-1 ] DEBUG 2025/02/08 21:30:27 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/02/08 21:30:27 vehicle range: 414km
[lp-1 ] DEBUG 2025/02/08 21:30:27 pv charge current: min 5A > 0A (4324W @ 3p, battery: false)
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-02-09:
Probably because the value _is_ 0

closed 2025-02-09 by andig

bill340, 2025-02-09:
Ok, but where does this wrong value come from? I did not change anything and since some weeks it only shows 0.

andig, 2025-02-09:
You're logfile doesn't show that. Add a trace log.

bill340, 2025-02-09:
Here is the log. According Tessie GUI there was 0,2 kWh added.
Maybe a problem with the Tessie vehicle Integration?

[evcc-20250209-115104-trace.log](https://github.com/user-attachments/files/18724087/evcc-20250209-115104-trace.log)

[image]

andig, 2025-02-11:
Please check the log- does it contain he kWh in the Tessie response? Feel free to open PR for the Tessie template to fix the value.

bill340, 2025-02-11:
Can you please help me where I can open the PR? I cannot find the tessie template anywhere?

andig, 2025-02-16:
> Please check the log- does it contain he kWh in the Tessie response?

bill340, 2025-02-16:
No, in the log it always says
"charge_energy_added": 0,

But it was not like that some time ago...

djfanatix, 2025-02-16:
Nothing we can do, it looks like a Tessie issue. (you can see in the log it is 0)

You can check yourself on:

https://developer.tessie.com/reference/get-state

bill340, 2025-02-16:
Ok, but if I do the get-charges there, I get a value on
"energy_added": 0.85,
Is that not corresponding to the charged amount?

djfanatix, 2025-02-16:
I do not understand, but if you can see on the Tessie developer website a value, EVCC will also see that. (if you are at that location you specified)

In your logs it is 0.

bill340, 2025-02-16:
Yes, thats exactly the problem... In the log it is 0, but if I check on the website of the tessie api, there is a value.
Maybe the template does not read it correctly?
in the log it says:
"charge_energy_added": 0
But the value of the api is "energy_added".
Maybe something wrong here in the template?

bill340, 2025-02-16:
I think it is because you are using the query get-state.
But the value is only delivered if using get-charges.

djfanatix, 2025-02-16:
feel free to change it, it is just a template.
Try whatever suits you.

`    energy:
      source: http
      uri: https://api.tessie.com/xxx/charges
      headers:
        Authorization: Bearer xxx
      jq: |
        if ((.drive_state.latitude - xx.6545 | abs) < 0.001) and
           ((.drive_state.longitude - xx.1651 | abs) < 0.001) then
          .energy_added
        else
          0
        end
      cache: 30s`

bill340, 2025-02-16:
Sorry, I never did change anything in a template... Where can I find it / how can I do that? 
I am only familiar with the evcc.yaml...
Can you help?

djfanatix, 2025-02-16:
change it in your evcc.yaml

bill340, 2025-02-16:
Ok, I changed it, but now get the following errors in the log:

[lp-2 ] ERROR 2025/02/16 22:13:03 charge rater: charge total import: jq: query failed: cannot subtract: null and number (47.6545)
[lp-2 ] ERROR 2025/02/16 22:13:03 charge total import: jq: query failed: cannot subtract: null and number (47.6545)

bill340, 2025-02-16:
if I use only:
      jq: .energy_added

I get the following errors:
[lp-2 ] ERROR 2025/02/16 22:42:54 charge rater: charge total import: strconv.ParseFloat: parsing "<nil>": invalid syntax
[lp-2 ] ERROR 2025/02/16 22:42:54 charge total import: strconv.ParseFloat: parsing "<nil>": invalid syntax

djfanatix, 2025-02-17:
I still think it is a Tessie issue and the original config should work.
I have contacted the Tessie dev

bill340, 2025-02-17:
If I completly remove the "energy" part of the custom charger, it works.
I use the built in Tessie template for the cars and it seems to get the correct value from there...

andig, 2025-03-16:
@djfanatix has anything come up? Should we remove energy?

djfanatix, 2025-03-16:
But we are talking about a template, not included in EVCC? Do you want to implement it? (Does it matter?)

I don’t use the template, but I see that the energy_added is correct in the Tessie api.

bill340, 2025-03-16:
> [@djfanatix](https://github.com/djfanatix) has anything come up? Should we remove energy?

I would say "yes". I use it with the "energy" part completly removed without any issues.

andig, 2025-03-16:
Happy to take a PR.

djfanatix, 2025-03-17:
In a template it will not work as good. 

So I made it (with ai) in Go, but I have no clue how to test it.
Will take some time, otherwise it will be a bit dodgy template