evcc-io/evcc #18690: UI config: Grid meter widget (as defined in evcc.conf) disappeared

heinemannj, 2025-02-09:
### Describe the bug

After updating to the latest nightly build [evcc version 0.133.0 (f93ceffa)] the `grid meter` widget (as defined in evcc.conf) disappeared:

[image]

[image]

The creation of `grid meter` via UI is possible but after commenting out in `evcc.conf `, related `site` and `circuits` settings are failing.

### Steps to reproduce

see above

### Configuration details

```yaml
meters:
  #
  # Huawei SUN2000-6KTL-M1 (Hybrid inverter, High current)
  # - Huawei DTSU666-H Smart Power Sensor
  # - Huawei LUNA2000-10-S0 Battery 10kWh
  # - Huawei LUNA2000-5-S0 Battery 5kWh
  #
  - name: grid_1
    type: template
    template: huawei-sun2000
    id: 1
    host: 192.168.178.149
    port: 502
    usage: grid
    modbus: tcpip
    timeout: 15s
    storageunit: 1
  - name: pv_1
    type: template
    template: huawei-sun2000
    id: 1
    host: 192.168.178.149
    port: 502
    usage: pv
    modbus: tcpip
    timeout: 15s
    storageunit: 1
  - name: battery_1
    type: template
    template: huawei-sun2000
    id: 1
    host: 192.168.178.149
    port: 502
    usage: battery
    modbus: tcpip
    timeout: 15s
    storageunit: 1
    capacity: 10.0
  - name: battery_2
    type: template
    template: huawei-sun2000
    id: 1
    host: 192.168.178.149
    port: 502
    usage: battery
    modbus: tcpip
    timeout: 15s
    storageunit: 2
    capacity: 5.0

circuits:
  - name: grid
    title: Hauptstromkreis
    maxCurrent: 35        # 3 x 35A (L1 L2 L3)
    maxPower: 24150       # 24.15kW = 230V x 3 x 35A
    meter: grid_1
    parent:               # no parent, this is the root circuit
  - name: lpc
    title: SteuVE
    maxCurrent: 35        # 3 x 35A (L1 L2 L3)
    maxPower: 24150       # 24.15kW = 230V x 3 x 35A
    parent: grid
    # meter: em_6
  - name: pv_1
    title: Solar
    # maxCurrent: 20      # 3 x 20A (L1 L2 L3)
    maxPower: 6600        # 6.6kW
    parent: lpc
    meter: pv_1
  - name: battery_1
    title: Batterie_1
    # maxCurrent: 20      # 3 x 20A (L1 L2 L3)
    maxPower: 5000        # 5.0kW
    parent: lpc
    meter: battery_1
  - name: battery_2
    title: Batterie_2
    # maxCurrent: 20      # 3 x 20A (L1 L2 L3)
    maxPower: 2500        # 2.5kW
    parent: lpc
    meter: battery_2

site:
  title: Energy Management System (EMS)
  meters:
    grid: grid_1  # Hauptstromkreis           3 x 35A (L1 L2 L3) x 230V = 24150W , parent: root
    pv:
      - pv_1      # Inverter Input Power      3 x 20A (L1 L2 L3) x 230V = 13800W ,         lpc
    battery:
      - battery_1 # Battery Active Power      3 x 20A (L1 L2 L3) x 230V = 13800W ,         lpc
      - battery_2 # Battery Active Power      3 x 20A (L1 L2 L3) x 230V = 13800W ,         lpc
    aux:
      - em_1_a    # Haus_Waschen              1 x 16A (L1) x 230V       =  3680W ,         grid
      - em_1_b    # Haus_Trocknen             1 x 16A (L2) x 230V       =  3680W ,         grid
      - em_1_c    # Wärmepumpe (Kompressor)   1 x 16A (L3) x 230V       =  3680W ,         lpc
      - em_2_a    # Haus_Technik_1            1 x 16A (L1) x 230V       =  3680W ,         grid
      - em_2_b    # Wärmepumpe (Steuerung)    1 x 16A (L2) x 230V       =  3680W ,         grid
      - em_2_c    # Haus_Technik_2            1 x 16A (L3) x 230V       =  3680W ,         grid
      - em_3      # Wärmepumpe (Heizstab)     3 x 16A (L1 L2 L3) x 230V = 11040W ,         lpc
#      - em_4     # Haus_Unterverteilung      3 x 40A (L1 L2 L3) x 230V = 27600W ,         grid
```

### Log details

```text
see above
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.133.0 (f93ceffa)

VolkerK62, 2025-02-09:
>The creation of grid meter via UI is possible but after commenting out in evcc.conf , related site and circuits settings are failing.

You must delete in in the (yaml) site and you must rename the meter within the circuit configuration.
Check `evcc config` for the new name.

heinemannj, 2025-02-09:
> > The creation of grid meter via UI is possible but after commenting out in evcc.conf , related site and circuits settings are failing.
> 
> You must delete in in the (yaml) site and you must rename the meter within the circuit configuration. Check `evcc config` for the new name.

It works - Thanks for kicking me in the right direction!

```
  - name: grid
    title: Hauptstromkreis
    maxCurrent: 35        # 3 x 35A (L1 L2 L3)
    maxPower: 24150       # 24.15kW = 230V x 3 x 35A
    meter: db:10
    parent:               # no parent, this is the root circuit
```

heinemannj, 2025-02-09:
Above hack/handling is also working for `solar` and `battery meters`.

I have 3 x `Shelly Pro 3EM` meters and a couple of `custom` meters:
- 6 x `shelly-1pm`
- 1 x `shelly-pro-3em`
- 3 x `custom` with data source `mqtt`

Question:
Is there any hack available for their migrations as well?

closed 2025-02-09 by naltatis