evcc-io/evcc #18696: FritzDect lässt sich nicht mehr schalten.

bananajoe86, 2025-02-09:
### Describe the bug

Ich nutze eine FritzDect 200 zum schalten und messen eines Loadpoints. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:
- name: wallbox1234
  type: template
  template: fritzdect 
  uri: http://192.168.178.1
  user: xx
  password: xx
  ain: xx
  standbypower: 5
```

### Log details

```text
[lp-1 ] ERROR 2025/02/09 14:52:06 charge power: unexpected status: 403 (Forbidden)
[lp-1 ] ERROR 2025/02/09 14:52:17 charge power: unexpected status: 403 (Forbidden)
[lp-1 ] ERROR 2025/02/09 14:52:18 charge total import: unexpected status: 403 (Forbidden)
[lp-1 ] ERROR 2025/02/09 14:52:21 charger status: unexpected status: 403 (Forbidden)
[lp-1 ] ERROR 2025/02/09 14:52:26 charge power: unexpected status: 403 (Forbidden)
[lp-1 ] ERROR 2025/02/09 14:52:36 charge power: unexpected status: 403 (Forbidden)
[lp-1 ] ERROR 2025/02/09 14:52:47 charge power: unexpected status: 403 (Forbidden)
[lp-1 ] ERROR 2025/02/09 14:52:57 charge power: unexpected status: 403 (Forbidden)
[lp-1 ] ERROR 2025/02/09 14:53:09 charge power: unexpected status: 403 (Forbidden)
[lp-1 ] ERROR 2025/02/09 14:53:10 charge total import: unexpected status: 403 (Forbidden)
[lp-1 ] ERROR 2025/02/09 14:53:11 charger status: unexpected status: 403 (Forbidden)
[lp-1 ] ERROR 2025/02/09 14:53:16 charge power: unexpected status: 403 (Forbidden)
[lp-1 ] ERROR 2025/02/09 14:53:27 charge power: unexpected status: 403 (Forbidden)
[lp-1 ] ERROR 2025/02/09 14:53:37 charge power: unexpected status: 403 (Forbidden)
[lp-1 ] ERROR 2025/02/09 14:53:46 charge power: unexpected status: 403 (Forbidden)
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

andig, 2025-02-09:
Da sind wohl user/pw falsch.

closed 2025-02-09 by andig

bananajoe86, 2025-02-09:
Danke für deine Antwort.
User und Passwort mehrfach überprüft. Auch schon neue angelegt und jedes mal sichergestellt, dass der User die Smarthome Berechtigung hat. Immer der gleiche Error. Bei einer Falscheingabe würde auch der Error "charger status: invalid user or password" heißen.