evcc-io/evcc #18697: Deye SG01HP3 - missing second battery power

stormbreaka, 2025-02-09:
### Describe the bug

Hello together, I just recognized that there is a small issue if the second battery is used at Deye high voltage inverters with power rating >25kW
They have two battery DC connections. In the official Deye modbus protocol it is written to be misunderstood because battery 1 power is described with 'Battery output power' and battery 2 power is described with 'Battery2 Power'.

The second battery power is contained in register 595 with same decode as the first battery power. So for total battery power both registers need to be added (590 + 595).

### Steps to reproduce

1. SG01HP3 inverter with rated power >25kW has two battery connectors (each 50A)
2. use second battery connector and check battery power compared to display or modbus at Deye

### Configuration details

```yaml
# Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.254 # Hostname
    port: 502 # Port
    
  - name: my_battery
    type: template
    template: deye-hybrid-hp3
    usage: battery
```

### Log details

```text
[lp-1  ] DEBUG 2025/02/09 14:54:32 charge power: 1514W
[lp-1  ] DEBUG 2025/02/09 14:54:32 charge currents: [2.31 2.29 2.31]A
[site  ] DEBUG 2025/02/09 14:54:32 grid power: 190W
[site  ] DEBUG 2025/02/09 14:54:33 battery 1 power: -540W
[site  ] DEBUG 2025/02/09 14:54:33 pv 14 power: 176W
[site  ] DEBUG 2025/02/09 14:54:33 pv 3 power: 87W
[site  ] DEBUG 2025/02/09 14:54:33 pv 11 power: 51W
[site  ] DEBUG 2025/02/09 14:54:33 pv 4 power: 85W
[site  ] DEBUG 2025/02/09 14:54:33 pv 12 power: 86W
[site  ] DEBUG 2025/02/09 14:54:33 battery 1 soc: 35%
[site  ] DEBUG 2025/02/09 14:54:33 pv 1 power: 1160W
[site  ] DEBUG 2025/02/09 14:54:33 grid currents: [0.05 0.17 0.12]A
[site  ] DEBUG 2025/02/09 14:54:34 pv 13 power: 87W
[site  ] DEBUG 2025/02/09 14:54:34 pv 5 power: 85W
[site  ] DEBUG 2025/02/09 14:54:34 pv 2 power: 174W
[site  ] DEBUG 2025/02/09 14:54:34 pv 10 power: 86W
[site  ] DEBUG 2025/02/09 14:54:34 pv 6 power: 84W
[site  ] DEBUG 2025/02/09 14:54:35 pv 8 power: 86W
[site  ] DEBUG 2025/02/09 14:54:35 pv 7 power: 85W
[site  ] DEBUG 2025/02/09 14:54:35 pv 9 power: 175W
[site  ] DEBUG 2025/02/09 14:54:35 pv power: 2505W
[site  ] DEBUG 2025/02/09 14:54:35 site power: -350W
[lp-1  ] DEBUG 2025/02/09 14:54:35 detected active phases: 3p
[lp-1  ] DEBUG 2025/02/09 14:54:35 charge total import: 4844.261kWh
[lp-1  ] DEBUG 2025/02/09 14:54:35 charger status: C
[site  ] DEBUG 2025/02/09 14:55:02 ----
[lp-1  ] DEBUG 2025/02/09 14:55:02 charge power: 1381W
[lp-1  ] DEBUG 2025/02/09 14:55:02 charge currents: [2.12 2.12 2.13]A
[site  ] DEBUG 2025/02/09 14:55:02 grid power: 193W
[site  ] DEBUG 2025/02/09 14:55:03 battery 1 power: -530W
[site  ] DEBUG 2025/02/09 14:55:03 pv 14 power: 173W
[site  ] DEBUG 2025/02/09 14:55:03 pv 2 power: 172W
[site  ] DEBUG 2025/02/09 14:55:03 pv 7 power: 84W
[site  ] DEBUG 2025/02/09 14:55:03 pv 3 power: 86W
[site  ] DEBUG 2025/02/09 14:55:03 pv 8 power: 84W
[site  ] DEBUG 2025/02/09 14:55:03 grid currents: [0.05 0.17 0.12]A
[site  ] DEBUG 2025/02/09 14:55:03 battery 1 soc: 35%
[site  ] DEBUG 2025/02/09 14:55:03 pv 1 power: 1150W
[site  ] DEBUG 2025/02/09 14:55:04 pv 11 power: 50W
[site  ] DEBUG 2025/02/09 14:55:04 pv 10 power: 85W
[site  ] DEBUG 2025/02/09 14:55:04 pv 12 power: 84W
[site  ] DEBUG 2025/02/09 14:55:04 pv 5 power: 84W
[site  ] DEBUG 2025/02/09 14:55:04 pv 6 power: 83W
[site  ] DEBUG 2025/02/09 14:55:05 pv 13 power: 86W
[site  ] DEBUG 2025/02/09 14:55:05 pv 9 power: 172W
[site  ] DEBUG 2025/02/09 14:55:05 pv 4 power: 84W
[site  ] DEBUG 2025/02/09 14:55:05 pv power: 2477W
[site  ] DEBUG 2025/02/09 14:55:05 site power: -337W
[lp-1  ] DEBUG 2025/02/09 14:55:05 detected active phases: 3p
[lp-1  ] DEBUG 2025/02/09 14:55:05 charge total import: 4844.270kWh
[lp-1  ] DEBUG 2025/02/09 14:55:05 charger status: C
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-02-09:
Good catch- happy to take a PR. Should take soc into account, too!

stormbreaka, 2025-02-12:
Two or more SoC can only be taken into account if the the capacity is also entered. Otherwise a calculation of one general SoC is not possible.

andig, 2025-02-12:
To keep it simple we'd need to assume identical capacity or you'll need to build more complex logic into the template. Or assume soc2=soc1, then we can omit it.

PR #18754 (2025-02-12): Deye: add support for 2nd battery
Fixes #18697
closed 2025-02-12 by premultiply