evcc-io/evcc #18703: UI: Ladepunkte -> Standard-Modus wird nicht übernommen

TobiasHuber1980, 2025-02-09:
### Describe the bug

wenn ich im experimental ui den ladepunkt konfiguriere, also genauer ich ändere den standard-modus z.b. von pv auf min-pv, starte evcc neu, dann wird der neue modus nicht angewendet und somit startet das laden auch nicht (in diesem fall min-pv)

### Steps to reproduce

1. standard-mous ändern auf z.B. min-pv
2. evcc neu starten
3. ladung startet nicht 
...

### Configuration details

```yaml
alles im ui konfiguriert, wallbox openWB pro

auf gefallen ist mir:

[lp-1 ] DEBUG 2025/02/09 19:54:08 set default mode: minpv
[lp-1 ] INFO 2025/02/09 19:54:08 mode: pv
```

### Log details

```text
[main ] INFO 2025/02/09 19:54:04 evcc 0.133.0 (07fb01d2)
[main ] INFO 2025/02/09 19:54:04 using config file: /etc/evcc.yaml
[db ] INFO 2025/02/09 19:54:04 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2025/02/09 19:54:05 listening at :7070
[lp-1 ] DEBUG 2025/02/09 19:54:07 set title: Garage
[lp-1 ] DEBUG 2025/02/09 19:54:07 set priority: 0
[lp-1 ] DEBUG 2025/02/09 19:54:07 set smart cost limit: <nil>
[lp-1 ] DEBUG 2025/02/09 19:54:07 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
[lp-1 ] DEBUG 2025/02/09 19:54:07 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-1 ] DEBUG 2025/02/09 19:54:07 set session energy limit: 0
[lp-1 ] DEBUG 2025/02/09 19:54:07 set session soc limit: 100
[lp-1 ] DEBUG 2025/02/09 19:54:08 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0x4000a934f8}
[lp-1 ] DEBUG 2025/02/09 19:54:08 set default mode: minpv
[lp-1 ] DEBUG 2025/02/09 19:54:08 set phases: 0
[lp-1 ] DEBUG 2025/02/09 19:54:08 charge total import: 2025.427kWh
[site ] INFO 2025/02/09 19:54:08 site config:
[site ] INFO 2025/02/09 19:54:08 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2025/02/09 19:54:08 grid: power ✓ energy ✓ currents ✗
[site ] INFO 2025/02/09 19:54:08 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2025/02/09 19:54:08 battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site ] INFO 2025/02/09 19:54:08 vehicles:
[site ] INFO 2025/02/09 19:54:08 vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site ] INFO 2025/02/09 19:54:08 vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site ] INFO 2025/02/09 19:54:08 vehicle 3: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1 ] INFO 2025/02/09 19:54:08 loadpoint 1:
[lp-1 ] INFO 2025/02/09 19:54:08 mode: pv
[lp-1 ] INFO 2025/02/09 19:54:08 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] INFO 2025/02/09 19:54:08 meters: charge ✓
[lp-1 ] INFO 2025/02/09 19:54:08 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2025/02/09 19:54:08 set buffer soc: 100
[site ] DEBUG 2025/02/09 19:54:08 set buffer start soc: 100
[site ] DEBUG 2025/02/09 19:54:08 set priority soc: 85
[site ] DEBUG 2025/02/09 19:54:08 set battery discharge control: true
[site ] DEBUG 2025/02/09 19:54:08 set residual power: 100
[lp-1 ] DEBUG 2025/02/09 19:54:08 phase timer inactive
[lp-1 ] DEBUG 2025/02/09 19:54:08 pv timer inactive
[site ] DEBUG 2025/02/09 19:54:08 ----
[lp-1 ] DEBUG 2025/02/09 19:54:08 charge power: 0W
[lp-1 ] DEBUG 2025/02/09 19:54:08 charge currents: [0 0 0]A
[site ] DEBUG 2025/02/09 19:54:08 grid power: 1W
[site ] DEBUG 2025/02/09 19:54:08 battery 1 power: 442W
[site ] DEBUG 2025/02/09 19:54:08 pv 1 power: 0W
[site ] DEBUG 2025/02/09 19:54:09 battery 1 soc: 51%
[site ] DEBUG 2025/02/09 19:54:09 site power: 543W
[lp-1 ] DEBUG 2025/02/09 19:54:09 charge total import: 2025.427kWh
[lp-1 ] DEBUG 2025/02/09 19:54:09 charger status: B
[lp-1 ] DEBUG 2025/02/09 19:54:09 charge start energy: 2025.427kWh
[lp-1 ] INFO 2025/02/09 19:54:09 car connected
[lp-1 ] DEBUG 2025/02/09 19:54:09 vehicle api refresh
[lp-1 ] DEBUG 2025/02/09 19:54:09 pv timer elapse
[lp-1 ] DEBUG 2025/02/09 19:54:09 pv timer inactive
[lp-1 ] DEBUG 2025/02/09 19:54:09 charge total import: 2025.427kWh
[main ] DEBUG 2025/02/09 19:54:09 vehicle status: B (VW ID.5)
[main ] DEBUG 2025/02/09 19:54:10 vehicle status: A (VW ID.4)
[main ] DEBUG 2025/02/09 19:54:10 vehicle status: A (VW ID.3)
[lp-1 ] INFO 2025/02/09 19:54:10 vehicle updated: unknown -> VW ID.5
[lp-1 ] DEBUG 2025/02/09 19:54:10 vehicle status: B
[lp-1 ] DEBUG 2025/02/09 19:54:10 vehicle soc: 18%
[lp-1 ] DEBUG 2025/02/09 19:54:10 vehicle soc limit: 80%
[lp-1 ] DEBUG 2025/02/09 19:54:10 vehicle range: 68km
[lp-1 ] DEBUG 2025/02/09 19:54:10 !! pvScalePhases DOWN activePhases: 3, available power: -543W, scalable: true
[lp-1 ] DEBUG 2025/02/09 19:54:10 available power -543W < 4140W min 3p threshold
[lp-1 ] DEBUG 2025/02/09 19:54:10 phase scale1p in 0s
[lp-1 ] DEBUG 2025/02/09 19:54:10 switched phases: 1p
[lp-1 ] DEBUG 2025/02/09 19:54:10 phase timer inactive
[lp-1 ] DEBUG 2025/02/09 19:54:10 pv charge current: 0A = 0A + -2.36A (543W @ 1p)
[lp-1 ] DEBUG 2025/02/09 19:54:10 pv enable timer reset
[lp-1 ] DEBUG 2025/02/09 19:54:10 pv timer inactive
[site ] DEBUG 2025/02/09 19:54:40 ----
[lp-1 ] DEBUG 2025/02/09 19:54:40 charge power: 0W
[lp-1 ] DEBUG 2025/02/09 19:54:40 charge currents: [0 0 0]A
[site ] DEBUG 2025/02/09 19:54:40 grid power: 1W
[site ] DEBUG 2025/02/09 19:54:40 battery 1 power: 570W
[site ] DEBUG 2025/02/09 19:54:41 pv 1 power: 0W
[site ] DEBUG 2025/02/09 19:54:41 battery 1 soc: 51%
[site ] DEBUG 2025/02/09 19:54:41 site power: 671W
[lp-1 ] DEBUG 2025/02/09 19:54:41 vehicle odometer: 24259km
[lp-1 ] DEBUG 2025/02/09 19:54:41 charge total import: 2025.427kWh
[lp-1 ] DEBUG 2025/02/09 19:54:41 charger status: B
[lp-1 ] DEBUG 2025/02/09 19:54:41 pv charge current: 0A = 0A + -2.92A (671W @ 1p)
[site ] DEBUG 2025/02/09 19:55:10 ----
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0 (07fb01d2)

andig, 2025-02-16:
@naltatis any idea?

naltatis, 2025-02-16:
Ja. Hab das gesehen. Hab auch ne Idee. Initialisierung von yaml und db LP ist hier noch anders. Konnte es aber noch nicht ausprobieren.

closed 2025-02-22 by andig