evcc-io/evcc #18710: (einmaliger?) Fehler nach Update

VolkerK62, 2025-02-10:
### Describe the bug

Bis auf meine Custom Meter/Vehicle ist alles im UI konfiguriert
Als Charger habe ich eine Wallbox (ohne 1p/3p) und 3 Shelly.

Ich habe wegen Forecast Solar das aktuelle Nightly installiert.
Der Neustart wurde abgebrochen  
`[main  ] FATAL 2025/02/10 04:54:48 loadpoint [db:5] charger does not support phase switching`

Lösung war, in UI Konfiguration, den Loadpoint öffnen und (ohne Änderung) speichern.

Beim Neustart kam der gleiche Fehler für den nächsten Loadpoint.
Nachdem ich das dann 4x gemacht habe, funktioniert alles wieder.

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
Feb 10 04:54:10 raspberrypi evcc[15482]: [lp-4  ] DEBUG 2025/02/10 04:54:10 set session soc limit: 44
Feb 10 04:54:10 raspberrypi evcc[15482]: [lp-2  ] DEBUG 2025/02/10 04:54:10 charger status: A
Feb 10 04:54:10 raspberrypi evcc[15482]: [lp-2  ] DEBUG 2025/02/10 04:54:10 soc estimated: 99.00% (vehicle: 99.00%)
Feb 10 04:54:10 raspberrypi evcc[15482]: [lp-2  ] DEBUG 2025/02/10 04:54:10 vehicle soc: 99%
Feb 10 04:54:16 raspberrypi systemd[1]: Stopping evcc.service - evcc...
Feb 10 04:54:16 raspberrypi systemd[1]: evcc.service: Deactivated successfully.
Feb 10 04:54:16 raspberrypi systemd[1]: Stopped evcc.service - evcc.
Feb 10 04:54:16 raspberrypi systemd[1]: evcc.service: Consumed 44min 21.654s CPU time.
Feb 10 04:54:37 raspberrypi systemd[1]: Started evcc.service - evcc.
Feb 10 04:54:37 raspberrypi evcc[30375]: [main  ] INFO 2025/02/10 04:54:37 evcc 0.133.0 (5027ee6b)
Feb 10 04:54:37 raspberrypi evcc[30375]: [main  ] INFO 2025/02/10 04:54:37 using config file: /etc/evcc.yaml
Feb 10 04:54:37 raspberrypi evcc[30375]: [db    ] INFO 2025/02/10 04:54:37 using sqlite database: /var/lib/evcc/evcc.db
Feb 10 04:54:38 raspberrypi evcc[30375]: [mqtt  ] INFO 2025/02/10 04:54:38 connecting evcc-199711803 at tcp://localhost:1883
Feb 10 04:54:38 raspberrypi evcc[30375]: [mqtt  ] DEBUG 2025/02/10 04:54:38 tcp://localhost:1883 connected
Feb 10 04:54:38 raspberrypi evcc[30375]: [main  ] INFO 2025/02/10 04:54:38 listening at :7070
Feb 10 04:54:38 raspberrypi evcc[30375]: [proxy-5200] DEBUG 2025/02/10 04:54:38 modbus proxy for 192.168.178.26:502 listening at :5200
Feb 10 04:54:40 raspberrypi evcc[30375]: [circuit-WPSG] DEBUG 2025/02/10 04:54:40 validation of max power disabled
Feb 10 04:54:40 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:54:40 set title: Carport(L3)
Feb 10 04:54:40 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:54:40 set priority: 0
Feb 10 04:54:40 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:54:40 set smart cost limit: <nil>
Feb 10 04:54:40 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:54:40 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
Feb 10 04:54:47 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:54:47 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
Feb 10 04:54:47 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:54:47 set session energy limit: 0
Feb 10 04:54:47 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:54:47 set session soc limit: 0
Feb 10 04:54:47 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:54:47 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0x4000716658}
Feb 10 04:54:48 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:54:48 set default mode: off
Feb 10 04:54:48 raspberrypi evcc[30375]: [main  ] FATAL 2025/02/10 04:54:48 loadpoint [db:5] charger does not support phase switching
Feb 10 04:54:48 raspberrypi evcc[30375]: [main  ] FATAL 2025/02/10 04:54:48 will attempt restart in: 15m0s
Feb 10 04:57:07 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:57:07 set title: Carport(L3)
Feb 10 04:57:07 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:57:07 set priority: 0
Feb 10 04:57:07 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:57:07 set smart cost limit: <nil>
Feb 10 04:57:07 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:57:07 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
Feb 10 04:57:07 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:57:07 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
Feb 10 04:57:07 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:57:07 set session energy limit: 0
Feb 10 04:57:07 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:57:07 set session soc limit: 0
Feb 10 04:57:07 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:57:07 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0x4000e06158}
Feb 10 04:57:07 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:57:07 set default mode: off
Feb 10 04:57:07 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:57:07 set phases: 3
Feb 10 04:57:07 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:57:07 set min current: 6
Feb 10 04:57:07 raspberrypi evcc[30375]: [lp-1  ] DEBUG 2025/02/10 04:57:07 set max current: 16
Feb 10 04:57:13 raspberrypi evcc[30375]: [main  ] INFO 2025/02/10 04:57:13 evcc was stopped by user. OS should restart the service. Or restart manually.
Feb 10 04:57:13 raspberrypi systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Feb 10 04:57:13 raspberrypi systemd[1]: evcc.service: Failed with result 'exit-code'.
[... 451 lines cut ...]
Feb 10 04:59:50 raspberrypi evcc[30445]: [site  ] DEBUG 2025/02/10 04:59:50 grid power: 161W
Feb 10 04:59:50 raspberrypi evcc[30445]: [site  ] DEBUG 2025/02/10 04:59:50 grid powers: [44 54 63]W
Feb 10 04:59:50 raspberrypi evcc[30445]: [site  ] DEBUG 2025/02/10 04:59:50 grid currents: [0.761 0.684 0.62]A
Feb 10 04:59:50 raspberrypi evcc[30445]: [site  ] DEBUG 2025/02/10 04:59:50 site power: 169W
Feb 10 04:59:50 raspberrypi evcc[30445]: [lp-1  ] DEBUG 2025/02/10 04:59:50 charge voltages: [231 0 0]V
Feb 10 04:59:50 raspberrypi evcc[30445]: [lp-1  ] DEBUG 2025/02/10 04:59:50 detected connected phases: 1p
Feb 10 04:59:50 raspberrypi evcc[30445]: [lp-1  ] DEBUG 2025/02/10 04:59:50 charge total import: 2701.900kWh
Feb 10 04:59:50 raspberrypi evcc[30445]: [lp-1  ] DEBUG 2025/02/10 04:59:50 charger status: A
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0 (5027ee6b)

VolkerK62, 2025-02-10:
Was mir noch aufgefallen ist ...

Hier die LP-Config vor/nach dem Update.
Neu dazugekommen ist `phasesConfigured`
```
[main  ] INFO 2025/02/08 18:55:06 evcc 0.133.0 (f83da937)
db:5 type: {"charger":"db:4","circuit":"load","defaultMode":"off","limitSoc":0,"maxCurrent":20,"meter":"","mode":"off","phases":3,"soc":{"estimate":true,"poll":{"interval":3600000000000,"mode":"charging"}},"thresholds":{"disable":{"delay":180000000000,"threshold":0},"enable":{"delay":60000000000,"threshold":0}},"title":"Carport(L3)","vehicle":""}

[main  ] INFO 2025/02/10 05:57:21 evcc 0.133.0 (5027ee6b)
db:5 type: {"charger":"db:4","circuit":"load","defaultMode":"off","limitSoc":0,"maxCurrent":20,"meter":"","mode":"off","phases":3,"phasesConfigured":3,"soc":{"estimate":true,"poll":{"interval":3600000000000,"mode":"charging"}},"thresholds":{"disable":{"delay":180000000000,"threshold":0},"enable":{"delay":60000000000,"threshold":0}},"title":"Carport(L3)","vehicle":""}
```

naltatis, 2025-02-10:
Ja, wie du schon im Titel vermutest. Dieser Effekt sollte nur auftauchen, wenn du von einem Nightly der letzten Tage auf das aktuelle wechselst und LPs/Charger über das UI angelegt hattest. Wir musst dann noch was am Datenmodell ändern (`phases` > `phasesConfigured`) um rückwärtskompatible mit evcc.yaml Einstellungen zu bleiben.

Kurz: Ist in der Tat ein Breaking Change zwischen den letzten Nightlies 😁

closed 2025-02-10 by naltatis

VolkerK62, 2025-02-10:
Ok. Danke