evcc-io/evcc #18714: Schnell"-Laden nutzt PV-Leistung nicht (mehr in aktueller Version)

frankheit, 2025-02-10:
### Describe the bug

In Version 0.133.0 wird im Modus "Schnell-Laden" die komplette Ladeleistung/Hausverbrauch aus dem Netz bezogen (ca. 12 kw) obwohl die PV aktuell Leistung liefert (ca. 1,6 kw).
Mit der PV-Leistung wird weiterhin der Akku geladen, das ganze unabhängig von den %-Einstellungen des Hausakkus.

Erwartet hätte ich, dass mit der PV-Leistung geladen wird, die Batterie nicht mehr geladen wird und der Rest aus dem Netz bezogen wird (siehe Bilder) 

[image]

[image]

Da ich mir sicher war das dies nicht immer so war habe ich das Verhalten mir einer noch in der Schublade liegenden alten Version getestet.
Mit der Version 0.130.7 funktioniert alles wie erwartet, der Bedarf wird von PV und Netz bedient, Batterie wird nicht geladen (siehe Bilder) 

[image]

[image]

Um das Verhalten nachzustellen habe ich die SD-Karte mit der "alten" 0.130.7 geklont und die Kopie auf die aktuelle Version 0.133.0 geupdatet.
Es wurden keinerlei Konfigurationen geändert, der Fehler lässt sich innerhalb Minuten (ausstecken Raspberry, Karte wechseln, booten) reproduzieren.

### Steps to reproduce

1. wechseln der Versionen, Fehler lässt sich reproduzieren 
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 30s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.10.15  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.10.15  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery2
  minsoc: 10
- type: template
  template: kostal-ksem-inverter 
  id: 71  
  host: 192.168.10.15  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  name: grid3

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.10.238  
  name: wallbox5

vehicles:
- type: template
  template: tesla 
  title: Franks Model Y  
  accessToken: xxx
  refreshToken: xxx 
  vin: xxx 
  capacity: 65  
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  vehicle: ev4
  mode: pv

site:
  title: Zuhause
  meters:
    grid: grid3
    pv:
    - pv1
    battery:
    - battery2

tariffs:
  grid:
    # either
    type: tibber
    token: "xxx" # access token
```

### Log details

```text
-
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

frankheit, 2025-02-10:
hier noch die Logs:

[evcc-20250210-104733-debug.log](https://github.com/user-attachments/files/18731910/evcc-20250210-104733-debug.log)

andig, 2025-02-10:
Vmtl. funktioniert die Entladesperre nicht wie erwartet und sperrt komplett den AC Pfad?

frankheit, 2025-02-10:
> Vmtl. funktioniert die Entladesperre nicht wie erwartet und sperrt komplett den AC Pfad?

Wenn ich in den Einstellungen "Verhindere Entladung im Schnell-Modus und bei geplantem Laden" deaktiviere verhält sich alles wie erwartet (nur wird eben auch die Batterie leer gezogen).

premultiply, 2025-02-13:
> vielleicht ein Zusammenhang zur letzten Umstellung: limitsoc auf batterymode

Was / welchen PR meinst du damit konkret?

Gibt es dazu nicht die zwei unterschiedlichen Templates?

frankheit, 2025-02-13:
> > vielleicht ein Zusammenhang zur letzten Umstellung: limitsoc auf batterymode
> 
> Was / welchen PR meinst du damit konkret?
> 
> Gibt es dazu nicht die zwei unterschiedlichen Templates?

Ja, gibt es. Allerdings nutzen beide Versionen das  identische (alte, da ich einen Gen.1 Wechselrichter habe mit dem das Netzladen nicht funktioniert).

axel-w-gh, 2025-02-16:
Das tritt bei mir seit dem Update auch auf: Kostal-Plenticore-G3, KSEM und BYD HVM Batterie.

andig, 2025-02-16:
/cc @iseeberg79 ist das ein Fehler in #17793? Da wurde ja ein gen2 Tempalte eingeführt. Gleichzeitig wurde aber auch im alten Template von `limitsoc` auf `batterymode` umgestellt. War das beabsichtigt oder müsste dieser Teil nicht zurück gedreht werden da im alten Template ohnehin kein Netzladen möglich ist?

andig, 2025-02-16:
@frankheit @axel-w-gh könnt ihr bitte mal ein trace Log von 133 und 200 zur Verfügung stellen? @iseeberg79 hat gerade zu recht darauf hin gewiesen, dass die Implementierungen eigentlich identisch sind. Das sollte also NICHT zu besagtem Fehler bei Euch führen!

frankheit, 2025-02-16:
> [@frankheit](https://github.com/frankheit) [@axel-w-gh](https://github.com/axel-w-gh) könnt ihr bitte mal ein trace Log von 133 und 200 zur Verfügung stellen? [...]

Von der 133 und der 200?
Oder auch einen von der funktionierenden 0.130.7 ?

axel-w-gh, 2025-02-16:
Aktuell kann ich nur mit Logs von der 0.200.1 dienen. Bevor ich diesen Thread gesehen habe, war ich schon auf die 200 gegangen, um zu schauen, ob der Fehler dann noch auftritt. Das kann ich dann morgen sagen.

axel-w-gh, 2025-02-17:
Es sieht fast so aus, als wenn das Problem mit der 0.200.1 nicht auftritt.
Aber ich verstehe seit der 0.200 (und der DB-Konfiguration) allerdings nicht mehr so ganz, mit welcher Konfiguration evcc gerade arbeitet - konkret _kostal-plenticore-gen2_ oder _kostal-plenticore_. Das hatte ich zwischenzeitlich in der yml-Datei nämlich auch ausprobiert. D. h. es kann auch daran liegen, wenn sich in der 0.200 daran eigentlich nichts geändert hat.
Eins kann ich allerdings sicher sagen, dass das Problem mit _kostal-plenticore_ aufgetreten war.

frankheit, 2025-02-17:
Hallo zusammen,
ich hab nochmal versucht das alles inkl. der Trace-Logs zu dokumentieren.
Der Übersichtlichkeit nach mit Uhrzeiten damit es vielleicht besser nachvollziehbar ist. 
Alle Konfigurationen sind identisch (Copy & Paste).

Ca. 12:47 Uhr, 0.133.0 gebootet inkl. des dargestellten Problems.

[133 evcc-20250217-125008-trace.log](https://github.com/user-attachments/files/18825917/133.evcc-20250217-125008-trace.log)

Ca. 12:52 Uhr update auf die 0.200.1 weiterhin mit gleichem Problem

[200 evcc-20250217-125635-trace.log](https://github.com/user-attachments/files/18825925/200.evcc-20250217-125635-trace.log)

Gegen 12:58 Uhr Wechsel der SD-Karte auf 0.130.7 > Funktion wie erwartet

[130 evcc-20250217-130028-trace.log](https://github.com/user-attachments/files/18825950/130.evcc-20250217-130028-trace.log)

Vielen Dank.

andig, 2025-02-17:
Also 130.7 geht, 133 geht nicht? Problem beim Schnelladen und der damit verbundenen Batteriesperre? Die https://github.com/evcc-io/evcc/pull/17793 ist laut 

    git name-rev --tags --name-only b766c73

in 132.0 drin. Das würde also passen.

frankheit, 2025-02-17:
> Schwierig etwas zu sehen. Die Logs sind jeweils sehr kurz, bitte besser mehrere Minuten aufzeichnen (-5 bis +5), da der Wechselrichter durchaus träge reagieren könnte. Dazu würden mich übrigens die in der Diskussion genannten Register des WR zu den drei Zeitpunkten (vor, beim Start, während der Fzg.-Ladung) interessieren, insbesondere die, die zur Entladungssteuerung verwendet werden und welchen Betriebsmodus der Wechselrichter (wr website) hat.
> 
> **v0.130**: vor dem Start der Fahrzeugladung ist der WR im Entladungsmodus und entlädt mit ca. 500W - es ist eine erfolgreiche Umschaltung erkennbar, da die Entladung erfolgreich auf von >6500W auf 17W reduziert / gesperrt wird: [lp-1 ] DEBUG 2025/02/17 12:58:34 charge power: 0W [site ] DEBUG 2025/02/17 12:58:35 pv power: 1728W [site ] DEBUG 2025/02/17 12:58:35 battery soc: 94% [site ] DEBUG 2025/02/17 12:58:35 battery power: 288W [site ] DEBUG 2025/02/17 12:58:35 grid meter: 10W [lp-1 ] DEBUG 2025/02/17 12:58:38 pv timer reset [lp-1 ] DEBUG 2025/02/17 12:58:38 pv timer inactive **[lp-1 ] DEBUG 2025/02/17 12:58:39 charge power: 0W [site ] DEBUG 2025/02/17 12:58:39 pv power: 1726W [site ] DEBUG 2025/02/17 12:58:40 battery soc: 94% [site ] DEBUG 2025/02/17 12:58:40 battery power: 362W [site ] DEBUG 2025/02/17 12:58:40 grid meter: -4W [lp-1 ] DEBUG 2025/02/17 12:59:04 charge power: 510W [site ] DEBUG 2025/02/17 12:59:05 pv power: 1728W [site ] DEBUG 2025/02/17 12:59:05 battery soc: 94% [site ] DEBUG 2025/02/17 12:59:05 battery power: 502W [site ] DEBUG 2025/02/17 12:59:05 grid meter: 172W [lp-1 ] DEBUG 2025/02/17 12:59:34 charge power: 11182W [site ] DEBUG 2025/02/17 12:59:35 set battery mode: hold [site ] DEBUG 2025/02/17 12:59:35 pv power: 1754W [site ] DEBUG 2025/02/17 12:59:36 battery soc: 93% [site ] DEBUG 2025/02/17 12:59:36 battery power: 6566W [site ] DEBUG 2025/02/17 12:59:36 grid meter: 12093W [lp-1 ] DEBUG 2025/02/17 13:00:04 charge power: 11186W [site ] DEBUG 2025/02/17 13:00:05 pv power: 1745W [site ] DEBUG 2025/02/17 13:00:06 battery soc: 93% [site ] DEBUG 2025/02/17 13:00:06 battery power: 17W [site ] DEBUG 2025/02/17 13:00:06 grid meter: 12056W [lp-1 ] DEBUG 2025/02/17 13:00:16 charge power: 11189W** [site ] DEBUG 2025/02/17 13:00:16 set battery mode: normal [site ] DEBUG 2025/02/17 13:00:16 pv power: 1730W [site ] DEBUG 2025/02/17 13:00:17 battery soc: 93% [site ] DEBUG 2025/02/17 13:00:17 battery power: 17W [site ] DEBUG 2025/02/17 13:00:17 grid meter: 12013W
> 
> **v0.133:** die Batterie ist auch hier im Entladungsmodus (421W) vor dem Start der Fahrzeugladung. Dann startet die Fahrzeugladung und die Batterie entlädt sich kurz bis zur Umschaltung mit >6500W. Was dann passiert ist spannend, da der neue Wert der Entladung verschieden vom vorherigen ist. Ich verstehe nicht, was eine Ladung der Batterie von ca. 1700W ermöglicht, obwohl zuvor 6500W aus der Batterie entnommen wurden. Es wird per evcc ja keine Leistungsregulierung vorgenommen. [lp-1 ] DEBUG 2025/02/17 12:47:52 charge power: 0W [site ] DEBUG 2025/02/17 12:47:52 grid power: 27W [site ] DEBUG 2025/02/17 12:47:52 battery 1 power: -185W [site ] DEBUG 2025/02/17 12:47:53 battery 1 soc: 94% [site ] DEBUG 2025/02/17 12:47:53 pv 1 power: 3037W [site ] DEBUG 2025/02/17 12:47:53 battery has priority at soc 94% (< 100%) **[lp-1 ] DEBUG 2025/02/17 12:48:05 charge power: 0W [site ] DEBUG 2025/02/17 12:48:05 grid power: 51W [site ] DEBUG 2025/02/17 12:48:05 battery 1 power: 421W [site ] DEBUG 2025/02/17 12:48:05 battery 1 soc: 94% [site ] DEBUG 2025/02/17 12:48:05 pv 1 power: 2158W [lp-1 ] DEBUG 2025/02/17 12:48:35 charge power: 8233W [site ] DEBUG 2025/02/17 12:48:35 grid power: 2630W [site ] DEBUG 2025/02/17 12:48:35 battery 1 power: 6163W [site ] DEBUG 2025/02/17 12:48:35 battery 1 soc: 94% [site ] DEBUG 2025/02/17 12:48:36 pv 1 power: 1929W [lp-1 ] DEBUG 2025/02/17 12:49:05 charge power: 11180W [site ] DEBUG 2025/02/17 12:49:05 set battery mode: hold [site ] DEBUG 2025/02/17 12:49:05 grid power: 5998W [site ] DEBUG 2025/02/17 12:49:05 battery 1 power: 6569W [site ] DEBUG 2025/02/17 12:49:05 battery 1 soc: 94% [site ] DEBUG 2025/02/17 12:49:05 pv 1 power: 1873W [lp-1 ] DEBUG 2025/02/17 12:49:35 charge power: 11170W [site ] DEBUG 2025/02/17 12:49:35 grid power: 14108W [site ] DEBUG 2025/02/17 12:49:35 battery 1 power: -1735W [site ] DEBUG 2025/02/17 12:49:35 battery 1 soc: 94% [site ] DEBUG 2025/02/17 12:49:35 pv 1 power: 1815W [site ] DEBUG 2025/02/17 12:49:36 battery has priority at soc 94% (< 100%) [lp-1 ] DEBUG 2025/02/17 12:49:55 charge power: 11180W** [site ] DEBUG 2025/02/17 12:49:55 set battery mode: normal [site ] DEBUG 2025/02/17 12:49:55 grid power: 13788W [site ] DEBUG 2025/02/17 12:49:55 battery 1 power: -1697W [site ] DEBUG 2025/02/17 12:49:56 battery 1 soc: 94% [site ] DEBUG 2025/02/17 12:49:56 pv 1 power: 1780W [site ] DEBUG 2025/02/17 12:49:56 battery has priority at soc 94% (< 100%) [lp-1 ] DEBUG 2025/02/17 12:50:05 charge power: 0W [site ] DEBUG 2025/02/17 12:50:05 grid power: 5W [site ] DEBUG 2025/02/17 12:50:05 battery 1 power: 785W [site ] DEBUG 2025/02/17 12:50:05 battery 1 soc: 94% [site ] DEBUG 2025/02/17 12:50:05 pv 1 power: 1762W
> 
> **v0.200**: gleiches Verhalten mit 0.133; allerdings wird beim Start der Fzg.-Ladung die Batterie entladen mit ca. 300W, durch das Starten der Ladung erfolgt eine Entladung der Batterie i.H.v. >6500W; nach der Aktivierung der Batteriesperre wird auch hier etwa die PV Leistung in die Batterie gespeichert - ohne das ein Register daf [... cut ...]  soc 94% (< 100%) [lp-1 ] DEBUG 2025/02/17 12:56:22 charge power: 11177W [site ] DEBUG 2025/02/17 12:56:22 set battery mode: normal [site ] DEBUG 2025/02/17 12:56:22 grid power: 13349W [site ] DEBUG 2025/02/17 12:56:22 battery 1 power: -1737W [site ] DEBUG 2025/02/17 12:56:23 battery 1 soc: 94% [site ] DEBUG 2025/02/17 12:56:23 pv 1 power: 1819W
> 
> Ich sehe keinen erklärbaren Zusammenhang mit der gemachten Umstellung?

Die ca. 1700 Watt in der 0.133.0 ist ca. die Leistung der PV in dem Moment.

andig, 2025-02-17:
> Was dann passiert ist spannend

Es interessiert nur was das auf Modbus passiert- denn das ist die einzige Änderung die es gab.

frankheit, 2025-02-17:
> > Was dann passiert ist spannend
> 
> Es interessiert nur was das auf Modbus passiert- denn das ist die einzige Änderung die es gab.

Wäre es ein Problem auf das "funktionierende" Verhalten vor der Änderung für das Template "kostal-plenticore" zurück zu gehen?
Wenn ich es richtig verstanden habe waren die Änderungen ja nur für die Implementierung des Netzladens welches ja mir den Wechselrichtern der Gen.1 eh nicht funktionieren.
Fakt ist, dass vor der Änderung in meiner Kombination (und vermutlich bin ich da nicht der einzige, der Plenticore ist ja nicht so selten) alles perfekt funktioniert hat - jetzt nicht mehr :-(

andig, 2025-02-17:
Bingo! Der liegt in der gleichen Range:

    git range-diff 0.130.7...0.133.0 -- provider/watchdog.go

Das war leider eine Verschlimmbesserung. Der Code soll überall gleich aussehen aber die Konstanten müssen natürlich passen. Ist geändert! 

Ich mache hier mal tentatively zu- sollte morgen hoffentlich wieder gehen.

frankheit, 2025-02-18:
> [@frankheit](https://github.com/frankheit) schau mal bitte in den evcc Konfigurationen UND auf der Seite des Wechselrichters bzgl. watchdog und timeout der externen Steuerung.

In evcc ist er gar nicht konfiguriert (also Standart) und im Wechselrichter auf 60 Sekunden

frankheit, 2025-02-18:
Vielen Dank für eure Hilfe....
Ich teste dann sobald ich kann, wird aber vermutlich Freitag werden (bin tagsüber nicht zuhause, nachts scheint keine Sonne).

frankheit, 2025-02-21:
Hallo zusammen,
sorry, auch mit der aktuellen nightly keine Änderung des Verhaltens.
Siehe Bilder.

[image]

[image]

frankheit, 2025-02-21:
Direkt im Anschluss die 0.130.7 gebootet, hier alles wie es soll :-(

Trace-log ebenfalls anbei, Ladebeginn gegen 11:14 Uhr.

[image]

[image]

[evcc-20250221-111937-trace.log](https://github.com/user-attachments/files/18904866/evcc-20250221-111937-trace.log)

frankheit, 2025-02-21:
Die evcc-Konfiguration ist im ersten Post, ist seitdem unverändert und auf beiden Versionen identisch.

Kurz zur Erklärung: 
Eigentlich nutze ich evcc im Homeassistant.
Als ich den Fehler entdeckt habe habe ich, da ich mir sicher war das es letztes Jahr als evcc noch auf dem eigenen Raspi gelaufen ist funktioniert hat, den alten Raspi mit der noch darin enthaltenen 0.130.7 gebootet.

Als hier alles einwandfrei funktioniert hat habe ich diese alte SD-Karte geklont und eine Version geupdatet.

Ich habe hier also zwei identische Versionen, eine alte 0.130.7 und eine welche geupdated wurde auf aktueller Nightly.

(EVCC im HomeAssistant wurde natürlich deaktiviert).

Hier die Screenshots vom Wechselrichter mit den Einstellungen:

[image]

Die aktuellen Werte hier, allerdings scheint ja aktuell keine Sonne und ich lade auch nicht...
Möchtest du davon nochmal einen Screenshot beim Laden mit dem Problem?

[image]

frankheit, 2025-02-21:
Nachtrag: 
ich habe dazu noch gestern (weil ich mir nochmals sicher sein wollte das ich keinen Fehler in der Konfiguration habe) eine komplett frische Installation im HA angelegt welche rein über die UI konfiguriert wurde.
Leider selbes Ergebnis :-(

frankheit, 2025-02-21:
[image]

frankheit, 2025-02-21:
Aber jetzt mal ne andere (vielleicht doofe) Frage, aber vielleicht hat es ja doch was damit zu tun:

Das Template "kostal-plenticore" ist doch für die Wechselrichiter der Gen1 die Probleme mit dem Netzladen haben oder?
Ist quasi das alte Template ohne dieses Feature wenn ich das richtig verstanden habe.

Ist es dann trotzdem korrekt das ich das auswählen kann oder ist hier was schief?

[image]

[image]

frankheit, 2025-02-22:
> kurze Zusammenfassung: [@andig](https://github.com/andig) [@premultiply](https://github.com/premultiply)
> 
> * Korrektur des Watchdog richtig, aber hier nicht hilfreich -- der WR speist immer die PV Energie voll in die Batterie zusätzlich beim Ladevorgang (minsoc=100)
> * am Ende hab ich mit frankheit die Konfiguration so erstellt, dass die Definition des alten Templates bei ihm als custom-Battery aktiv ist und den Fehler damit lösen können
> * als letzten Versuch haben wir statt minsoc=100 mal einen minsoc=99 probiert in der neuen Version des Templates, ohne Ergebnis; ich brauche noch den Trace mit der Bestätigung, dass der Versuch technisch umgesetzt wurde...
> 
> Wir haben Traces erstellt und in der Diskussion [#18672](https://github.com/evcc-io/evcc/discussions/18672) abgelegt.
> 
> Ich schreib' im von dir erstellen PR weiter und denke, letztlich muss dieser zum revert der Template Einstellungen genutzt werden. Leider kann ich mir das noch immer nicht erklären - denn die Modbus-Nachrichten sind vergleichbar.
> 
> [@frankheit](https://github.com/frankheit) wir haben das Timeout des WR reduziert auf 30s - kann so bleiben; nachdem klar ist, wie's weiter geht solltest du dann letztlich von der durch uns gemachten Einstellung wieder weg auf das Standard-Template, aber erstmal hilft's dir ;)

Vielen Dank für deine Mühe und deine Zeit!

andig, 2025-02-22:
> Ich schreib' im von dir erstellen PR weiter und denke, letztlich muss dieser zum revert der Template Einstellungen genutzt werden. Leider kann ich mir das noch immer nicht erklären - denn die Modbus-Nachrichten sind vergleichbar.

Leider ergibt das tatsächlich gar keinen Sinn. Wenn auf Modbus das gleiche passiert dann muss es auch das gleiche Ergebnis haben- oder es gibt eine weitere Variable. Da ihr anscheinend lokal testen könnt (was großartig ist!) wäre es ja vllt. eine Option, mal eine 200er mit revertetem Commit auszuprobieren.

Ich mache hier mal zu bis es verbindliche Ergebnisse gibt. Vielen Dank @iseeberg79 für Deinen Einsatz!

closed 2025-02-22 by andig

andig, 2025-02-22:
> Einziges Rätsel, warum hier im Bild die 99 nicht gesendet wird:

Vergesst doch mal bitte diese nervigen Screenshots. Das kann man ganz einfach mit

    evcc meter

testen!

andig, 2025-02-22:
> Ausprobiert: richtige Modbus-Nachricht mit <99, aber ab 99 wird daraus 100: wir machen noch einen Test, wie sich der WR verhält.

Grmpf. Dazu hätte ich gerne config plus log. Ich sehe nicht, wo das her kommen sollte. Das muss durch den Debugger.

andig, 2025-02-23:
Ein trace log würde ja erstmal reichen. Und wieso sollte da überhaupt 99 kommen? Was hat das mit diesem Problem zu tun?

andig, 2025-02-23:
Du hast mich verloren. Ich hab keine Ahnung mehr worum es geht. Falls es noch Erkenntnisse geben sollte gerne teilen, sonst bleibts wie es ist…

frankheit, 2025-02-24:
> > Ein trace log würde ja erstmal reichen. Und wieso sollte da überhaupt 99 kommen? Was hat das mit diesem Problem zu tun?
> 
> Hab jetzt alles zusammen, und erkenne: die Umrechnung passt. Haken wir das ab unter 6 und 8 sind zu ähnlich - zu viele Logs gelesen:
> 
> 98 = 00 00 42 c4 99 = 00 00 42 c6 100 = 00 00 42 c8 101 = 00 00 42 ca
> 
> frankheit kann das nightly mit dem heutigen PR testen, bei Bedarf helf' ich beim Rückbau des Testaufbaus... PS: die Zeit hätte ich uns beiden gern erspart... sorry!

Von meiner Seite passt mit der aktuellen nightly von heute alles wieder und es wird sauber die PV-Leistung zum Laden genutzt!
Vielen Dank für eure Zeit und eure Nerven!