evcc-io/evcc #18721: Automatische Erkennung BMW i3 mit Edge

twixfix, 2025-02-10:
### Describe the bug

Das Problem mit der automatischen Erkennung beobachte ich nun schon länger bei unserem älteren i3, der nur noch über Edge, Daten übertragen kann. So wie ich das sehe, wird der Status teilweise erst nach einer gewissen Zeit beim BMW Server aktualisiert. Sehr selten wird das Auto automatisch erkannt. Meistens halt nicht. Und den logs nach schaut evcc "nur" 10 Minuten lang ob das Auto auf Status B springt? Holt sich aber erst alle 15 Minuten neue Daten vom Server und das auch nur wenn geladen wird, außer man stellt das auf dauerhaft um, was man ja lieber nicht machen sollte? Und ja auch mit welcomecharge wird es nicht besser.

Könnte man das verändern oder einstellen das evcc mindestens 15 Minuten schaut ob sich der Status verändert bzw. erst 2x oder 3x die Daten vom Server holt und dann nicht mehr überprüft?

### Steps to reproduce

Ladestart eines i3 mit Edge

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: i3

interval: 10s # control cycle interval

sponsortoken: i3

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- name: grid1
  type: template
  template: sungrow-hybrid
  usage: grid
  modbus: tcpip
  id: 1
  host: 192.168.178.96
  port: 502
- name: pv2
  type: template
  template: sungrow-hybrid
  usage: pv
  modbus: tcpip
  id: 1
  host: 192.168.178.96
  port: 502

chargers:
- type: template
  template: easee
  user: i3
  password: i3
  charger: i3
  name: easeehome
- type: template
  template: easee
  user: i3
  password: i3
  charger: i3
  name: easeechargelite

loadpoints:
- title: Carport
  charger: easeehome
  mode: pv
[... 351 lines cut ...]
[lp-1 ] TRACE 2025/01/30 17:27:26 slot from: 2025-01-31 02:16:06 +0100 CET to 2025-01-31 03:00:00 +0100 CET cost 0.360
[lp-1 ] TRACE 2025/01/30 17:27:26 slot from: 2025-01-31 03:00:00 +0100 CET to 2025-01-31 04:00:00 +0100 CET cost 0.359
[lp-1 ] DEBUG 2025/01/30 17:27:26 forced charging at vehicle soc 18% (< 20% min soc)
[easee ] TRACE 2025/01/30 17:27:27 ProductUpdate Easee: (2025-01-30 16:27:25 +0000 UTC) IN_CURRENT_T2 28.990999221801758
[easee ] TRACE 2025/01/30 17:27:32 ProductUpdate Easee: (2025-01-30 16:27:30 +0000 UTC) IN_CURRENT_T3 30.121000289916992
[easee ] TRACE 2025/01/30 17:27:32 ProductUpdate Easee: (2025-01-30 16:27:30 +0000 UTC) TOTAL_POWER 6.809000015258789
[easee ] TRACE 2025/01/30 17:27:32 ProductUpdate Easee: (2025-01-30 16:27:31 +0000 UTC) CIRCUIT_TOTAL_PHASE_CONDUCTOR_CURRENT_L1 30.121000289916992
[easee ] TRACE 2025/01/30 17:27:33 ProductUpdate Easee: (2025-01-30 16:27:31 +0000 UTC) IN_CURRENT_T2 30.26099967956543
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.133.0

closed 2025-02-10 by andig

twixfix, 2025-02-10:
Ich habe die config mit Hilfe der Dokumentation erstellt. Dort ist für welcomecharge kein Wert hinterlegt. Also habe ich es wie dort angegeben übernommen. Jetzt mit dem Wert "true", scheint es zu gehen. Danke!

mfuchs1984, 2025-02-10:
Es ist nicht bei allen BMWs notwendig, die neueren haben weder mit dem Zustand, noch mit dem späteren laden ohne welcomecharge Probleme, deshalb ist das ein optionaler Parameter.