evcc-io/evcc #18739: Wallbox: switch phases: modbus: exception '3' (illegal data value), function '6'

nst90, 2025-02-11:
### Describe the bug

Hallo zusammen,

Bekomme die Fehlermeldung „Wallbox: switch phases: modbus: exception '3' (illegal data value), function '6'“.
Benutze eine Mennekes Amtron Compact 2.0s 22kW. 
Der Fehler tritt seit neuestem auf, ich war auf Release 0.133.0+1739241145.
Habe dann wieder zurück downgegradet auf 0.133.0+1738574269, womit wieder alles funktioniert.

Die 22kW Variante der genannten Wallbox unterstützt übrigens kein Phasenswitch. Die 11 kW Variante schon. Steht im Kleingedruckten der technischen Daten der Wallbox :-(
Wird hier implizit vlt ein Phasenswitch ausgelöst? Habe gesehen, dass es zur Phasenauswahl zuletzt auch Änderungen gab (#18638 ).

VG

### Steps to reproduce

1. Start EVCC
2. Fehlermeldung tritt direkt auf

### Configuration details

```yaml
open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: warn
levels:
  cache: error
  bmw: warn
#  lp-1: trace
#  mennekes: trace

# unique installation id
plant: *****

sponsortoken: *****

interval: 10s # control cycle interval

meters:
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.1.193
  port: 502
  usage: grid
  modbus: tcpip
  timeout: 10s
  name: grid_sungrow
- type: template
  template: sma-inverter-modbus 
  id: 3  
  host: 192.168.1.48  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv_terrasse
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.1.193
  port: 502
  usage: battery
  modbus: tcpip
  timeout: 10s
  capacity: 5.12
  name: battery_sungrow
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.1.193
  port: 502
  usage: pv
  modbus: tcpip
  timeout: 10s
  name: pv_sungrow

vehicles:
- type: template
  template: bmw 
  title: *****
  icon: car  
  user: *****
  password: *****
  hcaptcha: *****
  vin: *****
  capacity: 80.7
  phases: 3
  cache: 10m  
  minCurrent: 6  
  maxCurrent: 16 
  priority: 1  
  region: EU  
  welcomecharge: false  
  name: vehicle_i4

chargers:
- name: mennekes
  type: template
  template: mennekes-compact
  id: 50
  device: /dev/ttyUSB0
  baudrate: 57600
  comset: "8N2"
  modbus: rs485serial

loadpoints:
  - title: Wallbox # display name for UI
#    charger: charger_dummy # Wallbox Referenz
    charger: mennekes
    vehicle: vehicle_i4 # Referenz auf Standardfahrzeug
    mode: pv # charge mode (off, now, minpv, pv)
    phases: 3
    soc:
      poll:
        mode: always
        interval: 15m
      estimate: true

site:
  title: *****
  meters:
    grid: grid_sungrow
    pv:
    - pv_sungrow
    - pv_terrasse
    battery:
    - battery_sungrow

influx:
  url: http://localhost:8086
  database: evcc # InfluxDB v2.x uses term `bucket` but for compatibility still named `database` here
  user: *****
  password: *****

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.2847 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0851 # [currency]/kWh

  co2:
    type: grünstromindex
    zip: *****

Version: `0.133.0 (b0c7750b)`
```

### Log details

```text
Feb 11 20:15:51 raspberrybonn2 evcc[122154]: [main  ] INFO 2025/02/11 20:15:51 evcc 0.133.0 (b9dc1454)
Feb 11 20:15:51 raspberrybonn2 evcc[122154]: [main  ] INFO 2025/02/11 20:15:51 using config file: /etc/evcc.yaml
Feb 11 20:15:53 raspberrybonn2 evcc[122154]: [lp-1  ] WARN 2025/02/11 20:15:53 poll mode '{always 15m0s}' may deplete your battery or lead to API misuse. USE AT >
Feb 11 20:15:56 raspberrybonn2 evcc[122154]: [lp-1  ] WARN 2025/02/11 20:15:56 ignoring deprecated phases: 3. please configure via UI
Feb 11 20:15:56 raspberrybonn2 evcc[122154]: [lp-1  ] WARN 2025/02/11 20:15:56 poll mode '{always 15m0s}' may deplete your battery or lead to API misuse. USE AT >
Feb 11 20:15:56 raspberrybonn2 evcc[122154]: [site  ] WARN 2025/02/11 20:15:56 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/refer>
Feb 11 20:15:56 raspberrybonn2 evcc[122154]: [lp-1  ] ERROR 2025/02/11 20:15:56 switch phases: modbus: exception '3' (illegal data value), function '6'
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0 (b9dc1454)

premultiply, 2025-02-11:
In der Tat geht die Implementierung derzeit davon aus, dass die Phasenumschaltung bei allen Varianten möglich ist.

Das lässt sich aber wahrscheinlich entsprechend anpassen.
Schaue ich mir an.

Die Frage warum dass erst jetzt neu auffällt bzw. auf die Füße fällt ist eine andere.

/cc @andig

premultiply, 2025-02-12:
Sieht zumindest so aus als ob es ginge.

PR #18757 (2025-02-16): Mennekes Compact: check 1p3p hw option
Fixes #18739

@nst90 Pls test
closed 2025-02-16 by premultiply