evcc-io/evcc #18753: HEMS device if not accepted

hybe2022, 2025-02-12:
### Describe the bug

I would like to integrate evcc into the SMA system. I used already SAE to get the go-e meter data into the ennexOS portal. Hence I would like to take over the vendorid and deviceid from SAE to keep that data and an let evcc now report the data to SHM2.

The deviceid of the charger in SAE has been defined with a value of '000000000001'. So I put this value into evcc.yaml. Unfortunately this value cannot be decoded and wehn starting evcc it fails with the following error message.

[lp-1 ] DEBUG 2025/02/12 17:28:05 charge total import: 4799.609kWh
[main ] FATAL 2025/02/12 17:28:05 failed configuring hems: device id: encoding/hex: odd length hex string
[main ] FATAL 2025/02/12 17:28:05 will attempt restart in: 15m0s

evcc.yaml:

hems:
  type: sma
  allowcontrol: false
  vendorid: XXXXXXXX
  deviceid: 000000000001

### Steps to reproduce

try to start evcc with the mentioned deviceid value under hems in evcc.yaml 

### Configuration details

```yaml
hems:
  type: sma
  allowcontrol: false
  vendorid: XXXXXXXX
  deviceid: 000000000001
```

### Log details

```text
[lp-1 ] DEBUG 2025/02/12 17:28:05 charge total import: 4799.609kWh
[main ] FATAL 2025/02/12 17:28:05 failed configuring hems: device id: encoding/hex: odd length hex string
[main ] FATAL 2025/02/12 17:28:05 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

premultiply, 2025-02-12:
This will not work the way you think it would work.
evcc constrains the values exactly to SEMP specification and always assigns the loadpoints in order as configured.
Your device id format does not conform to the expected format as specified.
See https://docs.evcc.io/docs/reference/configuration/hems#deviceid

hybe2022, 2025-02-12:
Why is it not compliant to your documentation or SEMP spec? Your doc mentions an ID of 12 symbols which is defined. Kepp in mind the XXXXXXXX 8 digits for the vedorid is only a placeholder. this is not the real value or problem if you closed the issue due to that...

000000000001 is the value that was used by SAE and with this id the loadpoint is integrated into ennexOS. Would just like to re-use and replace SAE with evcc

[image]

hybe2022, 2025-02-12:
Leider war der Kommentar nicht hilfreich. Entgegen der Dokumentation, die leider an einigen Stellen nicht eindeutig ist darf man nicht die Zeichenkette so eingeben, sondern muss sie als String in "" angeben.

So klappt es nun:

hems:
  type: sma
  allowcontrol: false
  deviceid: "000000000001"
  vendorid: "XXXXXXXX"

So steht's leider in der Doku:

hems:
  type: sma
  deviceid: BBBBBBBBBBBB
  vendorid: AAAAAAAA

premultiply, 2025-02-12:
Ok, da dieses Feature wohl kaum jemand braucht und nutzt ist das wohl nie aufgefallen.
Da geht es vor allem um den Transfer von einer Hardware auf eine andere.
evcc erwartet da Hexzahlen soweit ich mich erinnere.

Der integrierte Algoritmus zur automatischen ID-Erzeugung ist eigentlich ganz gut.
Bei mehr als einem Ladepunkt funktioniert das Vorgehen so ohnehin nicht mehr.

closed 2025-02-12 by premultiply

hybe2022, 2025-02-25:
Leider funktioniert die Konfiguration mit den Werten in Anführungszeichen als String in der UI nicht. 
Ich wollte es von der yaml in die UI Konfiguration schieben, aber musste es rückgängig machen, da evcc so nicht startet. Der PR müsste also wieder geöffnet werden und das in der UI Konfiguration gefixt werden...

[image]

hybe2022, 2025-02-25:
folgendes wird in der UI Konfiguration nicht akzeptiert

 type: sma
  allowcontrol: false
  deviceid: "000000000001"
  vendorid: "20099999"

hybe2022, 2025-02-25:
Ok, konnte es lösen. In der UI sieht es folgendermaßen aus. Man muss ' statt " benutzen

type: sma
allowcontrol: false
deviceid: '000000000001'
vendorid: '20099999'

hybe2022, 2025-02-25:
Übrigens scheint mit der neuen Version 0.200.3 die Konfiguration des HEMS über yaml kaputt zu sein. Ich hatte eben gesehen dass SEMP nicht gestartet wird, weil mein Ladevorgang von heute von der der SMA App und dem Portal auftauchte.
semp war somit auch nicht über die URL erreichbar.

Nach erfolgreichem Umzug der Konfiguration nun aus dem yaml in die UI klappt dies wieder.

hybe2022, 2025-02-26:
Nachdem nun alles wieder funktioniert und SEMP wieder startet habe ich die HEMS Konfiguration aus dem yaml entfernt. 

Nun zeigt er in der UI an es sei kein HEMS konfiguriert, obwohl es nun nur noch in der DB ist und auch wieder gestartet wird. 

[image]

Die Konfiguration ist dann sogar im Fenster sichtbar 

[image]