evcc-io/evcc #18759: DE API intgration for intelligent Octopus Go Germany

thecem, 2025-02-13:
INHALT FOLGT!

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

thecem, 2025-02-13:
Welche antowort bekommt ihr während der Ladung gurch octopus:

https://api.oeg-kraken/.energy/v1/graphql

**Variables:**
```
{"email": "account-email",
 "password": "Passwort",
  "accountNumber": "A-12345678",
}
```
**Headers:**
```
{ "Authorization": "token"
}

```
**Query:**
```
query Car($accountNumber: String = "accountNumber") {
  devices(accountNumber: $accountNumber) {
    integrationDeviceId
    name
    deviceType
    id
    status {
      current
      currentState
      isSuspended
    }
  }
}
```
**Antwort wenn nicht durch octopus geladen wird aber angesteckt ist:**

```
{
  "data": {
    "devices": [
      {
        "integrationDeviceId": null,
        "name": "Volkswagen ID.4",
        "deviceType": "ELECTRIC_VEHICLES",
        "id": "00000000-0002-4000-803c-0000000047e2",
        "status": {
          "current": "LIVE",
          "currentState": "SMART_CONTROL_CAPABLE",
          "isSuspended": false
        }
      }
    ]
  }
}
```
**"currentState": "SMART_CONTROL_CAPABLE"**

andig, 2025-02-13:
Sollen wir das in eine Discussion verwandeln? Es (mir) ist nicht klar, was das werden soll. Ein dynamischer Tarif? Etwas Anderes? Was?

thecem, 2025-02-13:
Hallo Andi, ich möchte das gerne hier offen lassen, da ich die jetzige octopus integration anpassen will, damit beide API´s in EVCC laufen.
Hintergrund sind nicht nur die Preis Zeiten sondern auch die Möglichkeit über den Steuerbefehl von Octopus die Ladung auslösen zu können, quasi als zusätzliches signal zum laden außerhalb von 0-5 Uhr.  Damit könnte auch Tagsüber die Ladung aller durch evcc gestreuten Geräte erfolgen (Batterie/WP/Vehicle etc.).

andig, 2025-02-13:
> die jetzige octopus integration

Das ist ein Tarif

> quasi als zusätzliches signal zum laden außerhalb von 0-5 Uhr

Das wäre ein externes HEMS. Haben wir z.B. bei SMA SEMP so umgesetzt. Da das core Team hier nichts tut/ das nicht im Backlog hat wäre es mir sehr lieb wenn wir eine Discussion draus machen könnten.

thecem, 2025-02-13:
Ok, mach eine Diskussion draus. Wichtig ist mir das diese „externe“ Steuerung genutzt werden kann aber so weit sind wir noch nicht.

phrozen77, 2025-02-14:
Da hätte ich in der Tat eine Frage dazu (Umstellung auf Octopus Intelligent Go innerhalb der nächsten Woche)...

Bei Intelligent Go ist es ja so, das man zwischen 0 und 5 Uhr einen (stark) verbilligten Ladetarif nutzen kann, zusätzlich man aber auch Octopus gestattet, das angesteckte Auto selbständig bis zum eingestellten Akkustand zu laden.

Da das bei vielen (allen?) Fahrzeugen über die jeweiligen APIs geht, darf evcc logischerweise weder blockieren noch "dazwischenpfuschen".

Und genau da tu ich mich grad ein bisschen schwer - muss ich evcc ausschalten dass das funktioniert? Könnte man evcc, sobald der tariff "octopus-api" oder "octopus-productcode" aktiviert ist, einen zusätzlichen Button "Offen für externe Steuerung" (neben PV, Schnell, ...) hinzufügen?

Was ich erreichen möchte, ist einerseits dass ich das schnell wieder umstellen kann und - vor allem - das evcc eine korrekte Sicht auf die geladene Menge, Kosten, etc. hat.

Ideen oder Meinungen dazu?

thecem, 2025-02-14:
**Der Sinn:**
- Nutzung wenn nicht genug PV vom Dach kommt
- Nutzung von 0-5 Uhr (hier nur ~14 €c/kWh)
- Nutzung bei Signal von Octopus (deshalb diese API) -> 14 €c/kWh auch außerhalb des 0-5 Uhr Zeitraum
- Laden weiterer Assets ( BESS/ Vehicle2 / WP / unameit) wenn das signal außerhalb des 0-5 Uhr Zeitraum kommt

0-5 Uhr -> geht heute schon
Ladesignal außerhalb des Zeitraums kollidiert mit der Laderegelung durch EVCC z.B. bei Mode: PV -> heute noch nicht möglich

Weiterhin übergibt die API Dispatch Zeiträume, die mit dem Go Tarif (z.B. 14 €c/kW) abgerechnet werden und in diesem Zeitraum des gesamte Melo Verbrauch zum günstigen Preis berechnet wird.

Ich habe das bereits in einem Node Red Flow zum testen und habe hierzu auch schon HA Sensoren.

@andig Du meintest das SMA hier schon ähnlich in evcc eingebunden ist, welche Datei ist das?

andig, 2025-02-14:
Das ist hems/semp und in der loadpoint.Update:

```go
		// Sunny Home Manager
		if lp.remoteControlled(loadpoint.RemoteSoftDisable) {
			remoteDisabled = loadpoint.RemoteSoftDisable
			targetCurrent = 0
		}
```

da brauchts dann wohl auch ein `RemoteEnable`. Sollte aber nicht weiter schwierig sein.

premultiply, 2025-02-14:
> Der Sinn:
>
> Nutzung wenn nicht genug PV vom Dach kommt
> Nutzung von 0-5 Uhr (hier nur ~14 €c/kWh)
> Nutzung bei Signal von Octopus (deshalb diese API) -> 14 €c/kWh auch außerhalb des 0-5 Uhr Zeitraum [...]

Vielleicht stehe ich ja auf dem Schlauch aber ich sehe da noch nicht wozu es da neben der reinen Preisinformation noch einweiteres "Signal".
Wenn man bei 14 Cent/kWh laden will kann man das doch einfach einstellen und gut ist.
Ob das jede Nacht und ab und an mal tagsüber so ist spielt ja keine Rolle.

andig, 2025-02-14:
> Nutzung bei Signal von Octopus (deshalb diese API) -> 14 €c/kWh auch außerhalb des 0-5 Uhr Zeitraum

Anscheinend ein Tarif mit Tarifdaten die dann doch nicht gelten. Oder sind die 14ct dann auch im FC? Dann reicht in der Tat der Tarif. Leider ist weiterhin die Frage offen, was das hier _wirklich_ werden soll bzw. muss ;)

premultiply, 2025-02-14:
3. könnte durch ein "Push"-Update des FC erfolgen?

Wie "ad-hoc" ist das denn überhaupt?

andig, 2025-02-14:
> 2 und 3 kann rund um die uhr passieren.

Die Frage ist doch, ob der Tarif dann entsprechend aktualisiert wird. Falls ja reicht der als Steuerungsinformation. Falls nein halt nicht. Das kann doch nicht so schwer zu beantworten sein?

thecem, 2025-02-14:
Der Forcast ist unabhängig vom adhoc signal

typxxi, 2025-02-15:
Wir kriegen jetzt auch Octopus Intelligent Go und zwar mit einer der 2 unterstützten Wallbox Hersteller 
* Wallbox (plusar plus etc.)
* myenergy (zappi)

So wie ich das bei Abschluss des Vertrag im November 2024 verstanden hatte, muss ich die Octopus App mit der Wallbox App verknüpfen und das klang damals so, dass man Octopus als OCPP Server eintragen müsste, so dass Octopus das Auto laden kann, wann sie möchten, wenn es verbunden ist.

Logisch, dass dann evcc Überschuss laden nicht mehr möglich war, weshalb ich in Absprache mit Octopus 2nd level Support die Idee entwickelte, eine 2. baugleiche Pulsar Plus Wallbox gebraucht zu kaufen und sozusagen parallel zur 1. Wallbox zu installieren, wobei hier immer nur 1 Auto an einer der beiden Wallboxen geladen werden kann: 
1. evcc gesteuertes Überschussladen per 1. Pulsar Plus
2. Octopus gesteuertes Laden per 2. Pulsar Plus

Wir treffen somit die Entscheidung beim Einstecken wie ab 18 Uhr an die 2. Wallbox zu gehen und tagsüber an 1. Wallbox.

Meine Frage ist nun (letzte Woche alles fertig installiert, Ocotpus Belieferung beginnt erst ab 1. April), ob sich dieses Thema auch für die positiv auswirken kann, die ein EV ohne APi nutzen ? 
Unser SoC kommt per MeatPi Dongle ins HomeAssistant und auch evcc.

Ich weiß aktuell nur, dass Octopus an einer neuen Lösung arbeiten wollte, aber habe keinen neuen Stand erhalten. 
Bisher hieß es immer mehr oder weniger, dass in der Wallbox der OCPP Server von Octopus einzutragen sei. 

Von daher wäre es schön zu wissen, ob unsere installierte Lösung damit hinfällig wäre oder evcc und Octopus künftig kompatibel würden. 
Meine Lösung mit 2x Pulsar Plus hat sich so verbreitet, dass die Gebrauchtpreise von 150€ bis 220€ auf das Doppelte angestiegen sind in den vergangengen 2 Monaten. 

P.S.: Ach ja, wir kriegen auch die §14a Lösung, wodurch bei uns 10,7 Cent Netzentgelt erstattet werden von 0 bis 6 Uhr, also jeden Tag von 0 bis 5 Uhr Strom für 4 Cent

thecem, 2025-02-16:
@andig ich denke dein Vorschlag ist der simpelste und am besten umzusetzen da dann auch die Batterie (Netzladen) Preisbezogen einfach geladen werden kann. Auch was die PV Prognose in Zukunft angeht wir das wohl am einfachsten sein.

Der IOG hat ja folgenden ablauf:

1. Vor Einbau iMSys Fester Preis 0-24
2. Nach Einbau IMSys 0-5 Go Rate / 5-24 Standard Rate plus den flexiblen zeitraum der Go-Rate (VollStunden Block/ Blöcke)
3. §14a Modul 3 in dieser Kombi ergibt sich zusätzlich eine überlagerung der M3 Zeiten und den NT / ST / HT Netzentgelt Arbeitspreisen.

Wenn nun der flexible Zeitraum von der oeg-kraken API genutzt wird um die zones zu dem jeweiligen zeitraum mit der Go-Rate zu überschreiben, sollte das doch ohne weiteres funktionieren?

  ```
grid:
    ## 1. octopus Intelligent Go & MME (GP 17,75€/m)
    type: fixed
    price: 0.2378 # EUR/kWh (default / OIG-Standard)
    ## 2. octopus Intelligent Go & IMSys (GP 17,75€/m)
    # type: fixed
    # price: 0.2827 # EUR/kWh (default / OIG-Standard)
    # zones:
    #   - days: Mo-So
    #     hours: 0-5
    #     price: 0.1507 # EUR/kWh (IOG-Go)
    ## 3. octopus Intelligent Go IMSys & EnWG $14a Modul 3 (GP 17,75€/m)
    #type: fixed
    #price: 0.2827 # EUR/kWh (default / OIG-Standard)
    # NT: 1.99
    # ST: 6.64
    # HT: 12.75
    #zones:
    #  - days: Mo-So
    #    hours: 0-5
    #    price: 0.1042 # EUR/kWh (IOG-Go-NT)
    #  - days: Mo-So
    #    hours: 5-6,11-15,23-0
    #    price: 0.2352 # EUR/kWh (IOG-ST & M3 NT)
    #  - days: Mo-So
    #    hours: 6-11,15-17,22-23
    #    price: 0.2817 # EUR/kWh (IOG-ST & M3 ST)
    #  - days: Mo-So
    #    hours: 17-22
    #    price: 0.3428 # EUR/kWh (IOG-ST & M3 HT)
```

andig, 2025-02-18:
Das hängt davon ab was "dynamisch" heissen soll. Das komische Ding passt aber in kein Raster. Z.B. ist gar nicht klar, wie lange dieser ad-hoc Preis eigentlich gelten soll? Ich denke du musst hier ein bisschen ausprobieren und coden und wenn du eine passende Lösung hast PR machen.

thecem, 2025-02-18:
das werde ich machen, allerdings sind ein paar infos nützlich: Ist ein type: 14a-Modul3 geplant? Damit das Modul 3 auch mit anderen Tarifen funktioniert (Tibber .etc da es ja hier zu eine Überlagerung kommt) oder soll das mit formula gemacht werden? Da ja Modul3 am Anschluss (DNO) hängt und nicht am (EVU), aber auf jeden tarif der eingetragen wird fixed, tibber, octopus, octopus germany, etc. wirkt.

thecem, 2025-02-18:
https://github.com/evcc-io/evcc/pull/18940

thecem, 2025-02-18:
https://github.com/evcc-io/evcc/pull/18947

closed 2025-03-02 by github-actions[bot]

thecem, 2025-03-02:
Side info, it is planed  create a ne PR. But for now I need the iOGo traif to test. Up to end of march there will be a working PR.