evcc-io/evcc #18763: Custom filtering for charging sessions

n0cturne, 2025-02-13:
**Is your feature request related to a problem? Please describe.**
I would like an overview of all charging sessions within a specified time period. For example, I want to see the total kWh loaded from February 1, 2024, to July 31, 2024.
Currently, the charging session overview only allows filtering by month, year, or total.
Unfortunately, there is also no option to export the entire year or all charging sessions.

**Describe the solution you'd like**
I would prefer the option to define a custom time span.

**Describe alternatives you've considered**
To filter externally via Excel, it would at least be helpful if exporting an entire year or total were possible.

VolkerK62, 2025-02-13:
>To filter externally via Excel, it would at least be helpful if exporting an entire year **or total** were possible.

choose "month" and at the end of the charging session list, you can download a CSV file with all sessions.

naltatis, 2025-02-14:
Implementing custom ranges for sessions would introduce quite a lot of complexity in the UI. We'd have to account for different groupings (month vs days, ...) as well.

Since your request with the "export" is already possible I'd close this here.

closed 2025-02-14 by naltatis