evcc-io/evcc #18772: Integrate discussion/issue with UI

andig, 2025-02-13:


andig, 2025-02-28:
Feel free to grab [#18874](https://github.com/evcc-io/evcc/pull/18874)

diddip21, 2025-04-25:
Hier ein kleiner Vorschlag von mir.
Die Hilfe um einen weiteren Button erweitern um damit direkt eine Erste-Hilfe Discussion zu öffnen (analog zu ./evcc discuss)
[image]
Hier öffnet sich ein weiteres Modal mit einer editierbaren Textbox zur Fehlerbeschreibung und zwei weitern Textboxen die jeweils die config beinhalten und einmal die letzten Log Einträge abholen (die Logeinträge lassen sich dann mit der checkbox IncludeLogs wahlweise mit übergeben oder eben nicht)
[image]

für das generieren der config bin ich leider noch etwas laienhaft unterwegs, bisher bediene ich mich der api folgendermaßen `api.get("/state");` das spiegelt aber natürlich nicht die config wieder. In der http.go hab ich den teil der config gefunden dort müsste man sich aber eine vollständige config (ähnlich dem config file) erst zusammen bauen. Ist da vielleicht schon etwas in der Pipeline ?

naltatis, 2025-04-25:
> das spiegelt aber natürlich nicht die config wieder

Der spannende Part ist eine redacted evcc.yaml via API zu bekommen. Schau dir mal `evcc discuss` an. Das macht das heute schon. Das muss aber ein Endpunkt mit Authentication (evtl. sogar erneute PW Eingabe) werden weil Redact nicht alles finden kann. Weiterer wichtiger (und neuer) Part sind die Config Daten aus der DB (Inhalt der Tabellen `configs` und `settings`) mit rein zu bekommen. Auch hier muss redacted werden und darf nur mit Auth ausgeliefert werden.

closed 2025-09-21 by andig