evcc-io/evcc #18777: Anderes Verhalten von Metern bei Lastmanagement seit 0.200.0

tankcom, 2025-02-13:
### Describe the bug

Seit 0.200.0 muss ein Meter das im Lastmanagement verwendet wird auch unter site definiert sein. Vorher hat es gereicht dass das meter unter meters definiert war.

Ich musste mein meter der den backup port von meinem sungrow überwacht unter site als aux meter hinzufügen um nach dem update starten zu können. Ist die Änderung beabsichtigt? Falls ja sollte das vielleicht in den Docs erwähnt werden. [https://docs.evcc.io/docs/features/loadmanagement](url)

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
meters:
  - name: grid1
    type: template
    template: sungrow-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: 192.168.2.6 # Hostname
    port: 5021 # Port
  - name: my_pv
    type: template
    template: sungrow-hybrid
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.2.6 # Hostname
    port: 5021 # Port
  - name: my_battery
    type: template
    template: sungrow-hybrid
    usage: battery
    modbus: tcpip
    id: 1
    host: 192.168.2.6 # Hostname
    port: 5021 # Port
    capacity: 9
  - name: openDTU
    type: template
    template: hoymiles-opendtu
    usage: pv
    host: 192.168.2.199 # IP-Adresse oder Hostname
  - name: backup_port
    type: custom
    power:
        source: mqtt
        topic: homeassistant/backup/total_power
    currents:
      - source: mqtt
        topic: homeassistant/backup/phase_a_current
      - source: mqtt
        topic: homeassistant/backup/phase_b_current
      - source: mqtt
        topic: homeassistant/backup/phase_c_current

site:
  - title: Mein Zuhause
    meters:
      grid: grid1
      aux: backup_port   # vor version 0.200.0 war diese Zeile nicht Notwendig
      pv:
        - my_pv
        - openDTU
      battery: my_battery
    residualPower: 10
    maxGridSupplyWhileBatteryCharging: 50

circuits:
  - name: main # eindeutiger Name, wird als Referenz für andere Circuits verwendet
    title: Hauptsicherung # Anzeige in der UI (später)
    maxCurrent: 30 # 63A (optional)
    meter: grid1
  - name: backup # eindeutiger Name, wird als Referenz für andere Circuits verwendet
    title: Backup-Port # Anzeige in der UI (später)
    maxCurrent: 16 # 63A (optional)
    maxPower: 11000 # 30kW (optional)
    parent: main
    meter: backup_port
```

### Log details

```text
-
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.0

andig, 2025-02-14:
Im Zweifel muss es ein ext meter sein, nicht aux. Mir wäre aber nicht klar, dass wir das geändert hätten.

tankcom, 2025-02-14:
Stimmt, aux meter haben ja einen einfluss auf die Überschussregelung. 

Wusste nicht dass es ext meter gibt, die sind leider kaum Dokumentiert, danke für den Tipp.

endurance1968, 2025-02-14:
Danke für den "Workaround?" so geht es zumindest auch bei mir.
Mir war weder noch bekannt, aber aux ist zumindest in der Doku drin.

andig, 2025-02-14:
@naltatis hier bleibt aber noch die Verhaltensänderung bzgl. non-site meters. Ich vermute das liegt am lazy init mit dem wir Leichen verhindern wollten. Entweder: lazy-init um circuits erweiteren oder breaking change dokumentieren. Abhängig davon, wie wir mit ext/aux meters umgehen wollen?

naltatis, 2025-02-14:
> Entweder: lazy-init um circuits erweiteren oder breaking change dokumentieren

Ich wäre für Letzteres. Gerade im Hinblick auf die Config UI, die wir für die Circuits bauen werden. Das Meter an einem Circuit kann ja durchaus auch noch eine weitere Funktion im System erfüllen (bspw. `grid`). Das ist bei den Ladepunkten (lazy init) anders.

Die Doku ist schon in diese Richtung angepasst: https://docs.evcc.io/docs/features/loadmanagement#messwerte
Können wir direkt aus den Releasenotes verlinken.

naltatis, 2025-02-14:
@ZaphodB72 Siehe Link im vorherigen Kommentar. Das Anlegen und Zuordnen von "ext"-Metern geht aktuell noch nicht über COnfig UI.

ZaphodB72, 2025-02-14:
Ja aber gerade das verstehe ich ja nicht. Ich habe bisher den Zähler meines Solax X3 als "grid" konfiguriert. Ich habe keinen anderen Zähler als Netzzähler. Was soll ich jetzt wo als "ext" konfigurieren? Das wird mir irgendwie in diesem kurzen Doku-Satz nicht klar...

tankcom, 2025-02-14:
Wenn bei dir was mit currents steht hat das nichts mit dem thema aus dem issue hier zu tun und du musst auch keinen ext meter erstellen.

Wahrscheinlich gibt dein Wechselrichter nicht die stromstärke je phase aus, da hat sich wohl was bei der einbindung geändert

andig, 2025-02-14:
Rel. Notes aktualisiert:

> All meters must be associated with the site. This is especially relevant for meters only used for load management.

@naltatis das bedeutet dann aber auch, dass so ein Meter falls es im LM angelegt würde automatisch auch der site zugewiesen werden müsste (und wieder entfernt). Passt das?

Damit können wir dann hier zu machen?

naltatis, 2025-02-14:
Ja passt.

closed 2025-02-14 by naltatis