evcc-io/evcc #18778: 0.200 fehler meter config wird beim start nicht erkannt

endurance1968, 2025-02-13:
### Describe the bug

Nach dem Upgrade von 0.133 erscheint der Fehler beim Start
Feb 13 22:16:28 lx-evcc evcc[24386]: [main  ] FATAL 2025/02/13 22:16:28 circuit cannot create circuit 'haus-verteiler-stromkreis': not found: haus-verteiler-ug-meter
Feb 13 22:16:28 lx-evcc evcc[24386]: [main  ] FATAL 2025/02/13 22:16:28 will attempt restart in: 15m0s

Ein zurückgehen auf die Version 133 scheint das Problem nicht zu lösen. Erst wenn man auch  /var/lib/evcc/evcc.db löscht fährt die version 133 wieder auch. Evtl. ein Migrationsproblem von yaml nach db?

### Steps to reproduce

1. configure meters
2. start with version 0.133 everything works
3. upgrade to 0.200 -> error as documented above
...

### Configuration details

```yaml
...
circuits:
  - name: main # unique name, used as reference, e.g. as parent in other circuits
    title: VNB-Netz
    maxcurrent: 32.5 # Fuse has 35A but to be on the safe side take 32.5A only
    maxpower: 22230 # min voltage seen so far 228V
    meter: victron-grid
    parent: # no parent, this is the root circuit
#  - name: lpc
#    title: Limitation of Power Consumption
#    parent: main
  - name: carport-ecke # unique name, used as reference, e.g. to associate loadpoints
    title: Carport-Ecke # used in the UI
    maxcurrent: 32
    maxpower: 21888 # min voltage seen so far 228V
    parent: main # parent to the main circuit
  - name: carport-treppe # unique name, used as reference, e.g. to associate loadpoints
    title: Carport-Treppe # used in the UI
    maxcurrent: 16
    maxpower: 10944 # min voltage seen so far 228V
    parent: main
  - name: haus-verteiler-stromkreis # unique name, used as reference, e.g. to associate loadpoints
    title: Haus-Verteiler # used in the UI
    maxcurrent: 32
    maxpower: 20000 #
    parent: main
    meter: haus-verteiler-ug-meter
  - name: waermepumpe-stromkreis # unique name, used as reference, e.g. to associate loadpoints
    title: Waermepumpe # used in the UI
    maxcurrent: 16
    maxpower: 11520 # typically we have ~240V therefor the power is pretty high
    parent: haus-verteiler-stromkreis
    meter: waermepumpen-meter
...
# mqtt meter
# https://github.com/evcc-io/evcc/discussions/8872
# https://docs.evcc.io/docs/reference/configuration/mqtt
# https://docs.evcc.io/docs/reference/plugins#mqtt-lesenschreiben
# Victron Energy
# https://docs.evcc.io/docs/devices/meters#energy
meters:
- type: template
  template: victron-energy
  usage: grid
  host: 192.168.1.71
  port: 502
  name: victron-grid
  meterid: 40
- type: template
  template: victron-energy
  usage: pv
  host: 192.168.1.71
  port: 502
  name: victron-pv
- type: template
  template: victron-energy
  usage: battery
  host: 192.168.1.71
  port: 502
  name: victron-battery
  minsoc: 15
  maxsoc: 85
- type: custom
  name: haus-verteiler-ug-meter
  power:
    source: mqtt
    topic: haus/verteiler/haus/leistung
    timeout: 30s # don't accept values older than timeout
  currents:
    - source: mqtt
      topic: haus/verteiler/haus/strom/l1
    - source: mqtt
      topic: haus/verteiler/haus/strom/l2
    - source: mqtt
      topic: haus/verteiler/haus/strom/l3
- type: custom
  name: waermepumpen-meter
  energy:
    source: mqtt
    topic: haus/wp/energie
    timeout: 45s
  power:
    source: mqtt
    topic: haus/wp/leistung
    timeout: 45s # don't accept values older than timeout
  currents:
    - source: mqtt
      topic: haus/wp/strom/l1
    - source: mqtt
      topic: haus/wp/strom/l2
    - source: mqtt
      topic: haus/wp/strom/l3
```

### Log details

```text
Feb 13 22:16:28 lx-evcc evcc[24386]: [main  ] FATAL 2025/02/13 22:16:28 circuit cannot create circuit 'haus-verteiler-stromkreis': not found: haus-verteiler-ug-meter
Feb 13 22:16:28 lx-evcc evcc[24386]: [main  ] FATAL 2025/02/13 22:16:28 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.0

andig, 2025-02-14:
Ist das https://github.com/evcc-io/evcc/issues/18777?

tankcom, 2025-02-14:
Poste mal deine site config dazu

endurance1968, 2025-02-14:
Moin
ich kann bestätigen mit dem aux in site wie unten funktioniert es. Ohne nicht..

```
site:
  title: Steinbach
  # https://docs.evcc.io/docs/reference/configuration/site#residualpower
  # Auf ca. 100W Netzbezug einpegeln
  residualPower: 100
  meters:
    grid: victron-grid
    aux: haus-verteiler-ug-meter
    pv:
    - victron-pv
    battery:
    - victron-battery
```

andig, 2025-02-14:
Aux ist trotzdem falsch- das verhagelt dir die Regelung…

closed 2025-02-14 by andig