evcc-io/evcc #18780: 401 Unauthorized: Invalid Tesla API User error even after refreshing tokens.

Stevansell, 2025-02-13:
### Describe the bug

Ever since the new update to tesla dev/ myteslamate, i have had nothing but issues with evcc talking to my car.
I have the issues of it not waking up the car, and now I have constant "Invalid Tesla API User" errors. I have tried over multiple days deleting the old tokens and generating new ones to try and resync it, but always the same result

### Steps to reproduce

1. Run EVCC
2. Try and change charge level - Solar to Max etc..
3. Get error
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 150s # control cycle interval

meters:
- name: grid
  type: template
  template: fronius-solarapi-v1
  usage: grid
  host: 192.168.1.213

- name: pv
  type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.1.213
  port: 502
  
chargers:
- type: template
  template: twc3 
  host: 192.168.1.195  
  name: wallbox

vehicles:
- name: ev
  type: template
  template: tesla
  title: 
  clientId: (client id)
  accessToken: (access token)
  refreshToken: (refresh token)
  vin: 
  capacity: 62
  commandProxy: https://api.myteslamate.com
  proxyToken: (proxy token)

loadpoints:
- title: Garage
  charger: wallbox
  vehicle: ev
  mode: pv

tariffs:
  currency: AUD # (default EUR)
  grid:
    # static grid price with price zones)
[... 957 lines cut ...]
[lp-1  ] DEBUG 2025/02/14 02:32:14 detected connected phases: 1p
[lp-1  ] DEBUG 2025/02/14 02:32:14 detected active phases: 1p
[lp-1  ] DEBUG 2025/02/14 02:32:14 charger status: C
[lp-1  ] DEBUG 2025/02/14 02:32:14 soc estimated: 82.37% (vehicle: 82.00%)
[lp-1  ] DEBUG 2025/02/14 02:32:14 vehicle soc: 82%
[lp-1  ] DEBUG 2025/02/14 02:32:14 vehicle soc limit: 100%
[lp-1  ] DEBUG 2025/02/14 02:32:14 vehicle range: 341km
[lp-1  ] ERROR 2025/02/14 02:32:16 max charge current 32A: 401 Unauthorized: Invalid Tesla API User
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

andig, 2025-02-14:
> have the issues of it not waking up the car

Thats fixed. What did you change for the other error to occur?

Stevansell, 2025-02-14:
> > have the issues of it not waking up the car
> 
> Thats fixed. What did you change for the other error to occur?

Since the update, the only changes I have made was changing the access and refresh token to come from myteslamate as well as adding the command proxy and proxy token and client id. 

When i do a full delete an recreation of the access and refresh token, the evcc control works for about an hour, but then it just starts with the 401 unauthorized: invalid tesla api user error again. My full config is above.

andig, 2025-02-14:
Please add a trace log including --log-headers so we can see where this error is even generated.

Stevansell, 2025-02-14:
> Please add a trace log including --log-headers so we can see where this error is even generated.

Okay, My Home assistant didn't register there was a version 0.200 available until i manually forced an update, so that may have been causing the wakeup issue still. As for the 401, I have realized that the Tesla Dev site was asking me to add a payment method just in case I went over the limit, not 100% but that may have been causing some issues. 
Thirdly I can see they added a 7th API Scope and the "Personal Information" was now unchecked. 

I have updated to 0.200, Added a payment method, and granted access to all API scopes. Refreshed the Access and Refresh tokens and rebooted EVCC.

So far things seem to be controlling the charge again. I will leave it overnight to see if it throws up errors again, and if they do I will run the trace and post it, if all is good in the morning I will close this ticket.

closed 2025-02-14 by andig