evcc-io/evcc #18782: go-e energy consumption wrong

hybe2022, 2025-02-13:
### Describe the bug

Unfortunately the energy consumption shown within evcc and hence also reported via SEMP to SMA home manager is almost 5% less than reported by go-e app and also with mqtt to home assistant.

[image]
[image]
[image]
[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:
- type: template
  template: go-e-v3
  host: 192.168.178.28
  name: wallbox1

loadpoints:
- title: Garage
  charger: wallbox1
  mode: pv
  circuit: garage
```

### Log details

```text
No logs from yesterday. Needs to be reproduced
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

StefanSchoof, 2025-02-14:
Siehe dazu https://github.com/evcc-io/evcc/issues/13726

hybe2022, 2025-02-14:
Du sagst bei deinem Problem, dass es nur bei zwei aufeinanderfolgenden Ladevorgängen passiert. Bei mir war es nur einer. 
Der Wert von der go-e wird hingegen korrekt übertragen mit diversen Nachkommastellen, wie man an dem mqtt Wert in HA sieht, die von der go-e bereitgestellt werden.

hybe2022, 2025-02-14:
Übrigens hat es der SAE stets exakt an den Home Manager übertragen und somit im SMA Portal dargestellt.
Im SAE habe ich statt Parameter dws (gelegene Energie) die Leistung über nrg[11] benutzt und den SAE die Energie berechnen lassen. Das hat exakt funktioniert. 
Eigentlich müsste es aber auch über des klappen, da man ja sieht, dass der Wert an sich stimmt und vom go-e korrekt übertragen wird, aber falsch dargestellt wird.

hybe2022, 2025-02-14:
Ich habe das Thema nun verstanden. Die go-e zeigt die Energie nach anstecken des Autos. Wenn am Vortag mit Überschuss und am nächsten Morgen das Auto über Planung oder so auf den gewünschten Wert gebracht wird ist das für die go-e ein Ladevorgang und somit wird die Energie erst nach abstecken von der WB genullt. Ist neben meiner Meinung auch richtig. 
evcc zählt das fertig laden am Morgen jedoch als neuen Ladevorgang und startet mit der Energie bei 0. Daher kommt die Differenz. 
Man sollte hier generell mit go-e gleichziehen. Macht meiner Meinung nach mehr Sinn. Man möchte ja wissen, was nach anstecken in das Auto ging. Betrifft dann denke ich alle Charger.

andig, 2025-02-14:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2025-02-14 by andig

hybe2022, 2025-02-14:
Das ist eher ein Design Thema. Da wird kein Log helfen, siehe letzten Kommentar von mir

mfuchs1984, 2025-02-14:
Ein logfile hilft, deine These zu belegen.

hybe2022, 2025-02-14:
Durchaus, aber es ist auch so deutlich ersichtlich und das Design und Code sind ja auch bekannt. 

Am Vortag wurden 0,643kWh nach abweichen anstecken geladen. Dann war der Überschuss vorbei. 

[image]

Am nächsten Morgen wurde dann zwangsgeladen um vor Abfahrt auf 50% zu kommen. 

[image]

Der go-e nimmt alles zwischen Anstecken und Abstecken und stellt dies auch im Parameter "Wh" mit 17.009Wh zur Verfügung.
evcc macht aus dem neuen Ladestart am nächsten Morgen einen neuen Ladevorgang und stellt dies somit mit 16,3kWh (übrigens falsch gerundet, ennexOS Portal richtigerweise 16,4kWh) dar. Das sind die 17009Wh minus der zum Start der neuen Session von 643Wh. 
Ist eigentlich eindeutig genug. 

Beim nächsten Mal kann ich gerne 17 Stunden Log zur Verfügung stellen, obwohl dies eigentlich nicht mehr nötig ist nach der Voranalyse

andig, 2025-02-14:
Screenshots nutzen uns überhaupt gar nichts. Die Wahrheit steht im Logfile. Da hilft auch keine Diskussion zu Designissues. Danke.