evcc-io/evcc #18790: Problem: Auto wird alle 15 Minuten abgefragt

VolkerK62, 2025-02-14:
### Describe the bug

Mit `0.133.0 (b428a06e)` ist alles in Ordung.
`0.200.0` lässt sich nicht installieren
Mit `0.200.0 (9bbac3e6)` findet sich alle 15 Minuten ein Trace Aufruf im Log.

Bis auf Custom Geräte ist alles im UI konfiguriert.

### Steps to reproduce

.

### Configuration details

```yaml
[main  ] INFO 2025/02/14 09:22:29 evcc 0.200.0 (9bbac3e6)
[main  ] INFO 2025/02/14 09:22:29 using config file: /etc/evcc.yaml
[db    ] INFO 2025/02/14 09:22:29 using sqlite database: /var/lib/evcc/evcc.db
[mqtt  ] INFO 2025/02/14 09:22:29 connecting evcc-613560790 at tcp://localhost:1883
[mqtt  ] DEBUG 2025/02/14 09:22:29 tcp://localhost:1883 connected
charger
---
db:4 type:template {"host":"localhost","port":5200,"template":"daheimladen-mb"}
db:6 type:template {"ain":"xxx","icon":"bike","integrateddevice":true,"password":"xxx","standbypower":10,"template":"fritzdect","uri":"xxx","user":"xxx"}
db:8 type:template {"channel":"0","heating":true,"host":"192.168.178.29","icon":"heatpump","integrateddevice":true,"standbypower":-1,"template":"shelly"}
db:10 type:template {"channel":"0","heating":true,"host":"192.168.178.21","icon":"heatpump","integrateddevice":true,"standbypower":10,"template":"shelly"}

meter
---
db:1 type:template {"host":"192.168.178.23","port":"502","template":"solarlog","usage":"pv"}
db:16 type:template {"host":"192.168.178.22","template":"shelly-pro-3em","usage":"grid"}

vehicle
---
db:2 type:template {"cache":"15m","capacity":73,"icon":"car","language":"de","mode":"pv","password":"xxx","template":"hyundai","title":"(T)Raumschiff","user":"xxx"}
db:17 type:template {"capacity":73,"icon":"bus","template":"offline","title":"ohne_Bluelink"}

loadpoint
---
db:5 type: {"charger":"db:4","circuit":"load","defaultMode":"off","limitSoc":0,"maxCurrent":20,"meter":"","mode":"off","phases":3,"phasesConfigured":3,"soc":{"estimate":true,"poll":{"interval":3600000000000,"mode":"charging"}},"thresholds":{"disable":{"delay":180000000000,"threshold":0},"enable":{"delay":60000000000,"threshold":0}},"title":"Carport(L3)","vehicle":""}
db:7 type: {"charger":"db:6","circuit":"main","defaultMode":"","maxCurrent":1,"meter":"","minCurrent":1,"mode":"off","phases":1,"phasesConfigured":1,"priority":1,"soc":{"estimate":true,"poll":{"interval":3600000000000,"mode":"charging"}},"thresholds":{"disable":{"delay":180000000000,"threshold":0},"enable":{"delay":60000000000,"threshold":0}},"title":"Garage","vehicle":"Vbike"}
db:9 type: {"charger":"db:8","circuit":"WPSG","maxCurrent":2,"meter":"","minCurrent":2,"phases":1,"phasesConfigured":1,"priority":3,"soc":{"estimate":false,"poll":{"interval":3600000000000,"mode":"charging"}},"thresholds":{"disable":{"delay":1200000000000,"threshold":0},"enable":{"delay":1200000000000,"threshold":0}},"title":"WP-SG+","vehicle":"WasserSG"}
db:11 type: {"charger":"db:10","circuit":"","defaultMode":"now","limitSoc":46,"maxCurrent":2,"meter":"","minCurrent":2,"mode":"now","phases":1,"phasesConfigured":1,"priority":5,"soc":{"estimate":false,"poll":{"interval":3600000000000,"mode":"charging"}},"thresholds":{"disable":{"delay":180000000000,"threshold":0},"enable":{"delay":60000000000,"threshold":0}},"title":"WP(L2)-Info-","vehicle":"WasserWP"}
```

### Log details

```text
Feb 14 08:57:00 raspberrypi evcc[73069]: [hyundai] TRACE 2025/02/14 08:57:00 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/718cba00-f613-468b-9b32-4befc32058da/status/latest
Feb 14 08:57:00 raspberrypi evcc[73069]: [hyundai] TRACE 2025/02/14 08:57:00 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":49.310089,"lon":8.539175,"alt":0,"type":0},"head":326,"speed":{"value":0,"unit":0},"accuracy":{"hdop":0,"pdop":0},"time":"20250212110539"},"vehicleStatus":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"02H","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":false,"batteryStatus":75,"batteryPlugin":0,"remainTime2":{"etc1":{"value":7,"unit":1},"etc2":{"value":480,"unit":1},"etc3":{"value":135,"unit":1},"atc":{"value":0,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{"value":276,"unit":1},"totalAvailableRange":{"value":276,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[4],"time":{"time":"1140","timeSection":0}},"reservChargeSet":true,"reservFatcSet":{"defrost":false,"airTemp":{"value":"0AH","unit":0,"hvacTempType":1},"airCtrl":0,"heating1":0}}},"offpeakPowerInfo":{"offPeakPowerTime1":{"starttime":{"time":"0100","timeSection":0},"endtime":{"time":"1100","timeSection":1}},"offPeakPowerFlag":1},"reserveChargeInfo2":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"00H","unit":0,"hvacTempType":0},"airCtrl":0,"heating1":0}}},"reservFlag":0,"ect":{"start":{"day":0,"time":{"time":"1200","timeSection":0}},"end":{"day":0,"time":{"time":"1200","timeSection":0}}},"targetSOClist":[{"targetSOClevel":80,"plugType":0},{"targetSOClevel":100,"plugType":1}]},"chargePortDoorOpenStatus":2,"batteryPreconditioning":false,"batterySoh":0,"batteryPower":{"batteryFstChrgPower":0,"batteryStndChrgPower":0,"batteryDischrgPower":0}},"ign3":true,"hoodOpen":false,"transCond":false,"steerWheelHeat":0,"sideBackWindowHeat":0,"tirePressureLamp":{"tirePressureLampAll":0,"tirePressureLampFL":0,"tirePressureLampFR":0,"tirePressureLampRL":0,"tirePressureLampRR":0},"battery":{"batSoc":81,"batState":0,"sjbDeliveryMode":0,"batSignalReferenceValue":{"batWarning":65},"powerAutoCutMode":2},"lampWireStatus":{"stopLamp":{"leftLamp":false,"rightLamp":false},"headLamp":{"headLampStatus":false,"leftLowLamp":false,"rightLowLamp":false,"leftHighLamp":false,"rightHighLamp":false,"leftBifuncLamp":false,"rightBifuncLamp":false},"turnSignalLamp":{"leftFrontLamp":false,"rightFrontLamp":false,"leftRearLamp":false,"rightRearLamp":false}},"smartKeyBatteryWarning":false,"washerFluidStatus":false,"breakOilStatus":false,"sleepModeCheck":false,"time":"20250214075416","remoteWaitingTimeAlert":{"remoteControlAvailable":1,"remoteControlWaitingTime":168,"elapsedTime":"45:19:11"},"systemCutOffAlert":0,"tailLampStatus":0,"hazardStatus":0},"odometer":{"value":27722.6,"unit":1}}},"msgId":"454495b0-eaa9-11ef-b07e-a8cdb9c6241c"}
Feb 14 08:57:00 raspberrypi evcc[73069]: [hyundai] TRACE 2025/02/14 08:57:00 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/718cba00-f613-468b-9b32-4befc32058da/status
Feb 14 08:57:18 raspberrypi evcc[73069]: [hyundai] TRACE 2025/02/14 08:57:18 {"retCode":"S","resCode":"0000","resMsg":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"02H","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":false,"batteryStatus":75,"batteryPlugin":0,"remainTime2":{"etc1":{"value":7,"unit":1},"etc2":{"value":480,"unit":1},"etc3":{"value":135,"unit":1},"atc":{"value":135,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{"value":276,"unit":1},"totalAvailableRange":{"value":276,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[4],"time":{"time":"1140","timeSection":0}},"reservChargeSet":true,"reservFatcSet":{"defrost":false,"airTemp":{"value":"0AH","unit":0,"hvacTempType":1},"airCtrl":0,"heating1":0}}},"offpeakPowerInfo":{"offPeakPowerTime1":{"starttime":{"time":"0100","timeSection":0},"endtime":{"time":"1100","timeSection":1}},"offPeakPowerFlag":1},"reserveChargeInfo2":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"00H","unit":0,"hvacTempType":0},"airCtrl":0,"heating1":0}}},"reservFlag":0,"ect":{"start":{"day":0,"time":{"time":"1200","timeSection":0}},"end":{"day":0,"time":{"time":"1200","timeSection":0}}},"targetSOClist":[{"targetSOClevel":80,"dte":{"rangeByFuel":{"evModeRange":{"value":293,"unit":1},"totalAvailableRange":{"value":293,"unit":1}},"type":2},"plugType":0},{"targetSOClevel":100,"dte":{"rangeByFuel":{"evModeRange":{"value":375,"unit":1},"totalAvailableRange":{"value":375,"unit":1}},"type":2},"plugType":1}]},"chargePortDoorOpenStatus":2,"batteryPreconditioning":false,"batterySoh":0,"batteryPower":{"batteryFst [... cut ...] ardStatus":0},"odometer":{"value":27722.6,"unit":1}}},"msgId":"7916cfa0-eaab-11ef-97fe-06daa1655595"}
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.0 (9bbac3e6)

VolkerK62, 2025-02-14:
Ergänzung:
Das Auto ist nicht angesteckt.

andig, 2025-02-14:
Was sagt

    evcc config loadpoint

(bitte cleanen!)

mfuchs1984, 2025-02-14:
Passiert auch mit angestecktem Fahrzeug, loadpoint interval steht auf 20 Minuten, die API wird aber alle 15:30 Minuten abgefragt.

mfuchs1984, 2025-02-14:
https://github.com/evcc-io/evcc/pull/18405/commits/ddd2d496ce4436f05912e0559f7924830c98bab2

Der? Cache ist 15 Minuten.

andig, 2025-02-14:
> 0.200.0 lässt sich nicht installieren

@VolkerK62 was heisst das?

andig, 2025-02-14:
@naltatis haben wir irgendein problem mit soc.poll.interval?

naltatis, 2025-02-14:
@VolkerK62 Siehst anderen Poll-bezogene Logmeldungen beim Start?

VolkerK62, 2025-02-14:
@andig  
```
volker@raspberrypi:~ $ sudo apt install evcc=0.200.0
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Paket evcc ist nicht verfügbar, wird aber von einem anderen Paket
referenziert. Das kann heißen, dass das Paket fehlt, dass es abgelöst
wurde oder nur aus einer anderen Quelle verfügbar ist.

E: Version »0.200.0« für »evcc« konnte nicht gefunden werden.
```
@naltatis   muss ich mal schauen. Ich habe immer nur auf "hyundai" gefiltert.

andig, 2025-02-14:
```
Unable to find image 'debian:latest' locally
latest: Pulling from library/debian
106abeaee908: Download complete 
Digest: sha256:4abf773f2a570e6873259c4e3ba16de6c6268fb571fd46ec80be7c67822823b3
Status: Downloaded newer image for debian:latest
root@5e736e5c2f48:/# sudo apt install -y debian-keyring debian-archive-keyring apt-transport-https curl
bash: sudo: command not found
root@5e736e5c2f48:/# apt install -y debian-keyring debian-archive-keyring apt-transport-https curl
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'apt' instead of 'apt-transport-https'
E: Unable to locate package debian-keyring
E: Unable to locate package curl
root@5e736e5c2f48:/# apt update
Get:1 http://deb.debian.org/debian bookworm InRelease [151 kB]
Get:2 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:3 http://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Get:4 http://deb.debian.org/debian bookworm/main arm64 Packages [8693 kB]
Get:5 http://deb.debian.org/debian bookworm-updates/main arm64 Packages [13.3 kB]
Get:6 http://deb.debian.org/debian-security bookworm-security/main arm64 Packages [241 kB]
Fetched 9201 kB in 6s (1553 kB/s)                       
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
root@5e736e5c2f48:/# apt install -y debian-keyring debian-archive-keyring apt-transport-https curl
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
debian-archive-keyring is already the newest version (2023.3+deb12u1).
debian-archive-keyring set to manually installed.
The following additional packages will be installed:
  ca-certificates dirmngr gnupg gnupg-l10n gnupg-utils gpg gpg-agent gpg-wks-client gpg-wks-server gpgconf gpgsm krb5-locales libassuan0 libbrotli1
  libcurl4 libgpm2 libgssapi-krb5-2 libk5crypto3 libkeyutils1 libkrb5-3 libkrb5support0 libksba8 libldap-2.5-0 libldap-common libncursesw6 libnghttp2-14
  libnpth0 libpsl5 libreadline8 librtmp1 libsasl2-2 libsasl2-modules libsasl2-modules-db libsqlite3-0 libssh2-1 libssl3 openssl pinentry-curses
  publicsuffix readline-common
Suggested packages:
  dbus-user-session libpam-systemd pinentry-gnome3 tor parcimonie xloadimage scdaemon gpm krb5-doc krb5-user libsasl2-modules-gssapi-mit
  | libsasl2-modules-gssapi-heimdal libsasl2-modules-ldap libsasl2-modules-otp libsasl2-modules-sql pinentry-doc readline-doc
The following NEW packages will be installed:
  apt-transport-https ca-certificates curl debian-keyring dirmngr gnupg gnupg-l10n gnupg-utils gpg gpg-agent gpg-wks-client gpg-wks-server gpgconf gpgsm
  krb5-locales libassuan0 libbrotli1 libcurl4 libgpm2 libgssapi-krb5-2 libk5crypto3 libkeyutils1 libkrb5-3 libkrb5support0 libksba8 libldap-2.5-0
  libldap-common libncursesw6 libnghttp2-14 libnpth0 libpsl5 libreadline8 librtmp1 libsasl2-2 libsasl2-modules libsasl2-modules-db libsqlite3-0 libssh2-1
  libssl3 openssl pinentry-curses publicsuffix readline-common
0 upgraded, 43 newly installed, 0 to remove and 1 not upgraded.
Need to get 45.1 MB of archives.
After this operation, 68.1 MB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main arm64 readline-common all 8.2-1.3 [69.0 kB]
Get:2 http://deb.debian.org/debian bookworm/main arm64 libssl3 arm64 3.0.15-1~deb12u1 [1811 kB]
Get:3 http://deb.debian.org/debian bookworm/main arm64 openssl arm64 3.0.15-1~deb12u1 [1392 kB]
Get:4 http://deb.debian.org/debian bookworm/main arm64 ca-certificates all 20230311 [153 kB]                                                              
Get:5 http://deb.debian.org/debian bookworm/main arm64 krb5-locales all 1.20.1-2+deb12u2 [62.8 kB]                                                        
Get:6 http://deb.debian.org/debian bookworm/main arm64 apt-transport-https all 2.6.1 [25.2 kB]                                                            
Get:7 http://deb.debian.org/debian bookworm/main arm64 libbrotli1 arm64 1.0.9-2+b6 [269 kB]                                                               
Get:8 http://deb.debian.org/debian bookworm/main arm64 libkrb5support0 arm64 1.20.1-2+deb12u2 [31.8 kB]                                                   
Get:9 http://deb.debian.org/debian bookworm/main arm64 libk5crypto3 arm64 1.20.1-2+deb12u2 [79.4 kB]                                                      
Get:10 http://deb.debian.org/debian bookworm/main arm64 libkeyutils1 arm64 1.6.3-2 [8844 B]                                                               
Get:11 http://deb.debian.org/debian bookworm/main arm64 libkrb5-3 arm64 1.20.1-2+deb12u2 [313 kB]                                                         
Get:12 http://deb.debian.org/debian bookworm/main arm64 libgssapi-krb5-2 arm64 1.20.1-2+deb12u2 [124 kB]                                                  
Get:13 http://deb.debian.org/debian bookworm/main arm64 libsasl2-modules-db arm64 2.1.28+dfsg-10 [20.8 kB]                                                
Get:14 http://deb.debian.org/debian bookworm/main arm64 libsasl2-2 arm64 2.1.28+dfsg-10 [58.0 kB]                                                         
Get:15 http://deb.debian.org/debian bookworm/main arm64 libldap-2.5-0 arm64 2.5.13+dfsg-5 [171 kB]                                                        
[... cut ...]
After this operation, 89.3 MB of additional disk space will be used.
Get:1 https://dl.evcc.io/public/evcc/stable/deb/debian bookworm/main arm64 evcc arm64 0.200.0 [21.6 MB]
Fetched 21.6 MB in 5s (4055 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package evcc.
(Reading database ... 7056 files and directories currently installed.)
Preparing to unpack .../evcc_0.200.0_arm64.deb ...
Unpacking evcc (0.200.0) ...
Setting up evcc (0.200.0) ...
root@5e736e5c2f48:/# 
```

geht!

VolkerK62, 2025-02-14:
@naltatis  mir ist nichts aufgefallen. Die LP 2 bis 4 sind Shelly bzw. FritzDect.

[log.txt](https://github.com/user-attachments/files/18798099/log.txt)

VolkerK62, 2025-02-14:
was mir auffällt.
Beim Start wird das Auto zwar initialisiert, aber kein Status abgefragt.
Danach bleibt das Log bzgl "hyundai" auch leer.
Erst wenn ich im UI die Konfiguration öffen, wird vom Auto der Status abgefragt und danach dann alle 15 Minuten.

andig, 2025-02-14:
Mach doch das Auto im UI mal bitte zu

VolkerK62, 2025-02-14:
das ist zu. 
Ich geh einmal in die Konfiguration, warte kurz bis alles (loadpoints und vehicles) angezeigt wird, und mach dann die Konfiguration wieder zu.

VolkerK62, 2025-02-14:
@naltatis  ich denke, ich habe die Ursache gefunden.
Ich habe verschiedene Sachen ausprobiert.

Kurz ausgeholt:
Nach einem Neustart wird das Auto initialisiert. Danach ist das Log ruhig.
Am Loadpoint das (abgesteckte) Auto ausgewählt. Dann wird der Status vom Auto geholt. Danach ist das Log ruhig.
Das Konfigurationsfenster geöffnet. Kurz gewartet, bis alles angezeigt wurde und wieder zurück auf die normale Webansicht. Dadurch wurde der Status vom Auto geholt. Und danach dann alle 15 Minuten.
Ich habe dann die Ansicht aktualisiert (F5). Dann hört der Spuk auf.

Es scheint so zu sein, als würde das Verlassen der Konfigurationsansicht nicht registriert.
Ich vermute, wenn man diese Ansicht geöffnet hält, wird regelmäßig der Status aktualisiert.

naltatis, 2025-02-14:
Ok. Gute Beobachtung. Damit kann ich arbeiten. Ich schau mal dass ich das reproduziert bekomme.

closed 2025-02-15 by naltatis