evcc-io/evcc #18797: Max. Ladestrom je nach Phasen

ZaphodB72, 2025-02-14:
Ich lebe in einem Wohnpark mit eingeschränktem Stromnetz, weswegen ich nicht mehr als 9kW aus dem Netz nehmen darf.
Ich möchte außerdem mit PV-Überschuss laden.
Dementsprechend möchte ich bei dreiphasigem laden maximal mit 12A laden, bei einphasigem laden darf aber ruhig mit 16A geladen werden.

Ich habe gelesen, dass das demnächst optimal über ein Landmanagement geregelt werden könnte, beim Versuch, dieses zu aktivieren, konnte EVCC aber aufgrund einer fehlenden Konfiguration des Netzzählers nicht mehr starten und eine Änderung war dementsprechend im UI auch nicht mehr möglich, also musste ich komplett deinstallieren und neu installieren...

StefanSchoof, 2025-02-14:
Siehe https://github.com/evcc-io/evcc/issues/14661

VolkerK62, 2025-02-14:
Ein ähnlicher Fall, 22kW Wallbox, aber nur 11kW genehmigt.

```
circuits: # Loadmanagement
- name: main
  title: Hausanschluss
  maxCurrent: 35
  maxpower: 24000
  meter: grid
 
- name: load
  title: Carport
  maxCurrent: 21  # wegen Schieflast@1p
  maxpower: 11500 # wegen Genehmigung
```

ZaphodB72, 2025-02-14:
can't create circuit "main", cannot find "grid"

unter "meters" habe ich den Solax mit "name= grid" und "Usage=grid" konfiguriert

andig, 2025-02-14:
Siehe https://github.com/evcc-io/evcc/releases/tag/0.200.0

closed 2025-02-14 by andig

ZaphodB72, 2025-02-14:
ok, hab grid unter site hinzugefügt. Neue Fehlermeldung:

circuit cannot create circuit 'main': meter does not support phase currents

andig, 2025-02-14:
Was hat an der Stelle HA mit evcc zu tun? In dem Issue hier geht es auch nicht um Solax.

andig, 2025-02-14:
> hier geht es auch nicht um Solax.

tankcom, 2025-02-14:
Du kannst maxPower nutzen aber nicht maxcurrent.

Oder du berechnest in home assistant die ströme falls leistung und Spannung je phase verfügbar sind und gibst sie per mqtt rüber.

ZaphodB72, 2025-02-14:
Ok, hab in der evcc.yaml "maxcurrent" rausgenommen, den circuit "load" gelöscht, da ich ja nur den einen Eingangszähler habe, die Wallbox wieder unter "chargers" eingefügt und unter "loadpoints" Wallbox und Circuit verknüpft. In der UI hab ich dann die dort eingefügte Wallbox gelöscht, die aus der Yaml steht ja schon drin. Unter Lastmanagment sehe ich jetzt korrekterweise den Hausanschluss "main" mit der Leistung 0kW/9kW. 
Danke, scheint jetzt so zu funktionieren!
Hier nochmal die Einträge in der yaml:

meters:
  - type: template
    template: solax
    id: 1
    host: 192.168.178.90
    port: 502
    usage: grid
    modbus: tcpip
    name: grid1
circuits:
- name: main
  title: Hausanschluss
  maxpower: 10000
  meter: grid1
chargers:
  - type: template
    template: go-e-v3
    host: 192.168.178.196
    name: wallbox5
loadpoints:
  - title: Go-E
    circuit: main
    charger: wallbox5
site:
  title: Home
  meters:
    grid: grid1

ZaphodB72, 2025-02-14:
Wenn mir jetzt noch einer erklärt, wie ich den Code hier korrekt einfügen kann...🫣