evcc-io/evcc #18807: Version 0.200 ignoring deprecated guardduration: 15m0s. please configure via UI

erikgieseler, 2025-02-14:
### Describe the bug

Hallo,
seit Version 0.200 erscheint im Log die Meldung `ignoring deprecated guardduration: 15m0s. please configure via UI`
`Please configure via UI` kann ich leider nicht machen da sich dafür keine Option in der UI findet.
Ist diese Meldung so gewollt?

### Steps to reproduce

-

### Configuration details

```yaml
-
```

### Log details

```text
[lp-3  ] INFO 2025/02/14 15:38:05 loadpoint 3:
[lp-3  ] INFO 2025/02/14 15:38:05   mode:        off
[lp-3  ] INFO 2025/02/14 15:38:05   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-3  ] INFO 2025/02/14 15:38:05   meters:      charge ✓
[lp-3  ] INFO 2025/02/14 15:38:05     charge:    power ✓ energy ✗ currents ✗
[lp-3  ] WARN 2025/02/14 15:38:05 ignoring deprecated guardduration: 15m0s. please configure via UI
[lp-3  ] DEBUG 2025/02/14 15:38:05 phase timer inactive
[lp-3  ] DEBUG 2025/02/14 15:38:05 pv timer inactive
[lp-3  ] INFO 2025/02/14 15:38:05 vehicle updated: unknown -> Wärmepumpenanforderung
[lp-3  ] WARN 2025/02/14 15:38:05 poll mode '{always 1m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-3  ] DEBUG 2025/02/14 15:38:06 charge power: 2698W
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200

VolkerK62, 2025-02-14:
`guardduration` gibt es schon lange nicht mehr.
Der Hinweis auf UI ist falsch.

erikgieseler, 2025-02-14:
Kann man das jetzt gar nicht mehr steuern?

closed 2025-02-14 by erikgieseler