evcc-io/evcc #18811: Bitte PR #7088 rückgängig machen

mucki12, 2025-02-14:
Habe das Problem zu meinem Diskussionsbeitrag gefunden: https://github.com/evcc-io/evcc/pull/7088

Hintergrund:
PV Erträge werden von evcc immer als positiver Wert erwartet.

Scheinbar haben einige bei einem Shelly EM die Wandlerzange (aus meiner und aus evcc Sicht) falsch herum montiert und der Ertrag wir dann fälschlicherweise als negativer Wert übermittelt.

Mit dem PR https://github.com/evcc-io/evcc/pull/7088 wurde es dann umgesetzt, dass ALLE Werte als positiver Wert gezählt werden und dabei vollständig ignoriert, dass der Shelly EM tatsächlich in beide Richtungen messen kann.

In meinem Fall ist die Erzeugung bereits ein positiver Wert (da die Wandlerzange für die evcc Nutzung korrekt installiert wurde). Hört die Erzeugung auf wird allerdings ein wenig Strom vom Erzeuger selber verbraucht. Dieser wird vom Shelly EM auch korrekter Weise als negativer Wert erfasst. Durch https://github.com/evcc-io/evcc/pull/7088 von evcc allerdings fälschlicherweise ebenfalls als Ertrag gewertet.

Würde mich sehr darüber freuen wenn daher https://github.com/evcc-io/evcc/pull/7088 wieder rückgängig gemacht wird und der Shelly damit behandelt wird was er halt ist: ein Zweirichtungszähler.

Vielen Dank.

### Discussed in https://github.com/evcc-io/evcc/discussions/18792

<div type='discussions-op-text'>

<sup>Originally posted by **mucki12** February 14, 2025</sup>
Hallo,

im Zuge der ständigen Erweiterungen der Konfiguration über die UI (vielen Dank dafür) habe ich eine Verständnisfrage zu den Möglichkeiten unter "PV & Batterie".

Derzeit habe ich u. a. einen Leistungserzeuger mit einem Shelly EM eingebunden. Wenn der Erzeuger wirklich gerade Strom erzeugt ist der Wert positiv und falls gerade keine Erzeugung stattfindet (leicht) negativ.

Um dieses aufzufangen sieht meine Einbindung derzeit so aus:
```
  - name: pt
    type: custom
    power:
      source: http
      uri: http://192.168.2.XXX/emeter/0
      jq: if .power < 0 then 0 else .power end
```

Das klappt wie gewünscht, aber mit dem neuen Release 0.200.0 beschäftige ich mich verstärkt mit den Einstellungen direkt über das UI. Hier kann der Shelly EM ja ebenfalls direkt eingebunden werden. Dabei wird dann das Template shelly-1pm.yaml genutzt, welches ja auch als meter: pv genutzt werden kann.

Allerdings werden durch dieses Template keine negativen Werte abgefangen. evcc meter zeigt den Wert zwar noch als negativ an, in der Berechnung wird dieser Wert dann allerdings als positive Erzeugung gewertet.

Ich hatte das in der Vergangenheit so verstanden, dass bei den Templates für "richtige" PV Erzeuger oftmals im Template ebenfalls negative Werte abgefangen werden (kann mich aber auch irren).

Daher meine wahrscheinlich unbedarfte Frage:
Müsste das bei Templates für meters wie Shelly, AVM Fritz etc. pp nicht ebenso gemacht werden? Natürlich nur, wenn diese als usage: pv genutzt werden (bei grid und charge wäre das natürlich kontraproduktiv :-) )

Vielen Dank!</div>

andig, 2025-02-14:
Das hat jetzt 2 Jahre lang problemlos funktioniert. Es ist nicht erkennbar, warum hier für einen Einzelfall ein revert notwndig sein sollte?

mucki12, 2025-02-14:
Meiner Meinung nach war das damals eine Verkettung mehrerer Umstände:

- evcc Nutzer, die nur in eine Richtung messen wollten, haben sich (warum auch immer) mit dem Shelly EM einen Zweirichtungszähler angeschafft
- bei diesem wurde dann die Wandlerzange so angeschlossen, dass zu evcc ein negativer Wert übermittelt wurde
- als "Fix" dieser beiden Fehler wurde das so umgebaut, dass alle Werte (also positive und negative) immer als positiv gewertet werden und somit der Sinn eines Zweirichtungszählers geplättet wurde.

Bei allen, die jetzt einen Shelly EM für PV einsetzen passiert damit folgendes:

- der (geringe) Standby Verbrauch der PV-Anlage wird vom Shelly EM korrekterweise als negativer Wert gemessen und dann von evcc als PV Ertrag gewertet (der evcc Nutzer lebt halt damit und hat immer einen PV Ertrag)
- diejenigen die nicht damit leben wollten, haben den Shelly EM daraufhin als custom meter eingebaut und die negativen Werte weggebügelt.

Warum man damals überhaupt einen Fix benötigte für diejenigen, die die Wandlerzange partout für evcc nicht drehen wollten, erschließt sich mir immer noch nicht.

Wenn man den Shelly EM dafür nutzt wofür er vorgesehen ist (Zweirichtungszähler), kann er so niemals über die UI von evcc Nutzern eingebunden werden, die ihn auch wie vorgesehen nutzen.

Und nein, das hat nicht zwei Jahre problemlos funktioniert. Da bin ich schon vor über einem Jahr drüber gestolpert und konnte mir schon damals das fehlerhafte Messverhalten nicht erklären: https://github.com/evcc-io/evcc/discussions/12104

mucki12, 2025-02-14:
Habe den Shelly EM kurz über die UI eingebunden:

[image]

[image]

mucki12, 2025-02-15:
Ich habe noch einmal eine Nacht drüber geschlafen und wollte nur noch einmal anmerken, dass es mir auf keinen Fall darum geht diejenigen mit einer anderen Richtung der Wandlerzange ihre Umsetzung kaputt zu machen.

Wäre halt nur schön, wenn wie beschrieben bei einem Zweirichtungszähler nicht alle Werte (also negativ und positiv) als positiver PV-Ertrag gewertet werden. Zur Not drehe ich meine Wandlerzange auch falsch herum :-)

Spaß beiseite:
Bin nicht so fit im Coding - wie wird das denn bei dem Shelly Pro3M im monophase Betrieb gehandhabt. Da werden doch wahrscheinlich negative Werte auch nicht als positiver PV Ertrag gewertet, oder?

thierolm, 2025-02-15:
@mucki12 lass mal mit dem drehen. Ich hatte noch keine Zeit mir das genau anzuschauen. Da ich selbst keine 3EM habe, habe ich die Anpassungen für den 3 EM immer auf Basis des Inputs der Nutzer hier im Forum gemacht. Kann mich dunkel an den Einbau der abs Funktion erinnern... 

Wir können einen Parameter ergänzen, der dazu führt, dass das "original" Verhalten des Shellies bezüglich Messwerte ermöglicht.

Lass uns das aber zusammen gut testen. Ich melde mich asap, wenn ich Zeit gefunden habe mir das nochmal anzuschauen.

thierolm, 2025-02-15:
Mein Vorschlag wäre, den Issue in eine Discussion umzuwandeln.

mucki12, 2025-02-15:
Sehr gerne, stehe natürlich auch für jeden Test zur Verfügung.

thierolm, 2025-02-15:
In der Zwischenzeit kannst du vielleicht ein paar Original JSON Responses der relevanten API Endpoints mit korrekt angeschlossener Mess-Zange teilen. :-)

Die JSONS selbst werden in evcc im folgenden Code getestet: 
https://github.com/evcc-io/evcc/blob/master/meter/shelly/types_test.go