evcc-io/evcc #18812: Wärmepumpe Lambda/Zewotherm EU-L Serie

chriz77, 2025-02-14:
Liebe Entwickler,

könnt ihr euch bitte mal die Wärmepumpen von Lambda  (AT) / Zewotherm (DE) ansehen? Es wäre mega cool, wenn ihr da was einbauen könntet!!! :-D Die WP sind über Modbus TCP steuerbar, was aktuell bei mir recht umständlich über Loxone läuft. Ich habe die EU08L in Verbindung mit einem Fronius Gen24 Plus und einem Fronius Smartmeter. Wenn ihr mehr Infos braucht, oder einen Betatester, dann könnt ihr mich gerne kontaktieren.

Modellbezeichnung der gängisten Modelle:
EU08L
EU10L
EU13L

Modbus Doku:
https://lambda-wp.at/wp-content/uploads/2025/02/Modbus-Protokoll-und-Beschreibung-1.pdf

Viele Grüße

Chris

andig, 2025-02-15:
Bitte ausprobieren ob die Register das tun was du erwartest. Vorher machts keinen Sinn da Arbeit rein zu stecken.

hausi24, 2025-02-15:
Eine einbindung der Lambda würde ich ebenfalls sehr begrüßen!
Meine wird im Frühjahr installiert.

Grüße
Hausi

jannisif, 2025-02-15:
Wir haben hier auch eine Lambda, die Register kann man auf jeden Fall auslesen. Man muss halt nur mit den Werten gucken, dass man diese Einheiten entsprechend anpasst. Steuern/Modi setzen habe ich noch nicht direkt getestet.
Ich weiß nicht, wie das im Allgemeinen gehandhabt wird, aber wenn ich das richtig in der Modbus-Beschreibung sehe, basiert das Lambda-System auf Modulen, sodass individuelle Installationen verschiedene Register haben können... korrigiert mich da gerne!
Ich habe mich zuletzt mal mit den Templates und der modbus-Doku auseinandergesetzt, noch nicht ganz verstanden, aber mit Unterstützung kriegen wir das hoffentlich hin.
Ich habe aktuell nicht viel Zeit, weswegen ich mich erst im März damit auseinandersetzen könnte. Lasst uns diesen Issue/FR doch zu einer Diskussion umwandeln, dann machen wir das dort in Ruhe!
@chriz77 Wie sieht die Loxone-Integration aus, wenn ich fragen darf? Vielleicht können wir davon einiges abgucken :)

thecem, 2025-02-15:
eine idee wäre schon mal den grid wert per modbus dem em der lambda zu übergeben:

```
data:
  address: 102
  hub: EU08L
  value:
    - "{{states('sensor.ksem_ac_meter_watts') | int | bitwise_and(65535)}}"
action: modbus.write_register
```

der muss aktiv in die lambda geschrieben werden. Hierdurch kann die lambda bei Einspeisung in den erhöten betrieb gehen. 
Dies könnte dann auch geziehlt genutzt werden , quasi als (SG Kontakt).
Weitere Steuermöglichkeiten sehe ich im Moment nicht und sind auch nicht im aktuellen [Modbus Manual](https://lambda-wp.at/wp-content/uploads/2025/02/Modbus-Protokoll-und-Beschreibung-1.pdf) hinterlegt.

Leistugsvorgabe geht immer muss aber ein HK oder FBH Ventil öffnung anstossen sonst sinnlos. DHW würde noch gehen. Mal sehen wann Lambda das EnegieManger Update raus bringt.

Johprit, 2025-02-16:
Die Einbindung von Lambda WP wäre super.
Meine - EU13L - kommt auch bald.

andig, 2025-02-16:
> address: 102

Davon sehe ich nichts in der Doku (oder ich verstehe sie nicht). Was erwartest du konkret bei "inaktiv" und bei "pv verfügbar" sowie "dimmen"? Gerne jeweils Register und Wert. Mach gerne einen Vorschlag als PR, Stiebel dient als Vorlage.

premultiply, 2025-02-16:
So wie ich das Manual verstanden habe (nur kurz überflogen) kann man der WP per Modbus ein eigenes Gird-Meter anbinden welches ihr dann den Überschuss am Netzübergang liefert. Daraus rechnet sie sich selbst aus was sie damit anfangen kann und will.

Das ist ein ähnliches Konzept z.B. zu den mypv-Heizstäben und erfordert keinerlei externe Steuerung.

chriz77, 2025-02-18:
Sorry,  jetzt hat die Antwort ein wenig länger gedauert, aber die letzten Tage war nichts mit PV-Strom und ich wollte euch ein paar Screenshots machen. Ich gebe über Loxone den PV-Ertrag meiner Anlage an die Lambda. Man könnte natürlich genau so gut einen Fake Wert vorgeben, wenn der Strompreis günstig ist.

0102 Actual Input Power-General E-Manager
[image]

[image]

[image]

[image]

[image]

[image]

andig, 2025-02-18:
> Das ist ein ähnliches Konzept z.B. zu den mypv-Heizstäben und erfordert keinerlei externe Steuerung.

@premultiply beim myPV kann man mit evcc den Zähler einsparen- das ginge hier sicher auch. Ob das zu implementieren lohnt? Don't know...

jannisif, 2025-02-18:
Ja genau, man kann sowohl einen Zähler von Lambda einbinden (verhältnismäßig teuer + Einbau) als auch einen externen über modbus tcpip einbinden, welche sich ja hier gut anbieten würde. Von daher auf die Frage von andig: klares JA!

Randnotiz: Was mir soeben vor die Nase gekommen ist: Die WP ist zwar SG-Ready, dies anscheinend aber nur für PV-Überschuss mithilfe eines potenzialfreien Kontakt. Jetzt habe ich die EVU-Sperre (eigentlich für Netzbetreiber) gesehen, dies ist ein extra 24 V Eingang am E-Manager, also für uns nicht wirklich nutzbar. Heißt also, rein theoretisch wäre es sogar möglich, zumindest zu drosseln.
Wir werden uns an Lambda wenden, ob da in der Zukunft noch eine Steuerung über Modbus möglich wäre(geringe Chance vermutlich).

Anyway:
Ich möchte nun im ersten Schritt nur die Leistung der WP in evcc angezeigt bekommen. Im Zweiten könnte man dann den Grid-Wert an diese übermitteln. Danach möchte ich erstmal nur testen und später dann ein Template erstellen.
Ich brauche also für die WP in der evcc.yaml ein Meter, einen Charger und einen Loadpoint, welches sich dann gegenseitig zugewiesen wird:

```
loadpoints:
  - title: Lambda Wärmepumpe
     charger: lambda_control
     meter: lambda_strom
     integrateddevice: true # Fahrzeugauswahl deaktivieren

meters:
 - type: custom
    name: lambda_strom
    power:
      source: modbus
      uri: xxx:502
      id: 1
      register:
        address: 104 # aktuelle Aufnahmeleistung der WP
        type: input
        decode: int16
     scale: 1000

chargers:
  - name: lambda_control # erstmal nur Werte sehen
     type: custom
     uri: xxx:502 # ModBus address
# wird hier noch was benötigt?

```

Wichtig: hier soll jetzt erstmal nur die Ansicht der WP realisiert werden, noch nicht die Steuerung!

Abseits der sonstigen Config: was fehlt bzw. ist offensichtlich fehlerhaft? Ich habe noch nie etwas individuell in evcc eingebunden, so verzeiht mir meine fehlende Kenntnis. Ich möchte auch lernen, und das lässt sich am besten an der eigenen Umgebung  :D

andig, 2025-02-18:
Du brauchst egtl. keinen Charger sondern einfach ein aux meter. PR welcome. Ggf können wir auch die Grid Leistung als Modbus Server verfügbar machen, dann kann die WP sich die selbst abholen.

premultiply, 2025-02-18:
Oder sogar nur ein ext-Meter.
Egal wie: Das kannst du dir ganz einfach als Custom-Meter zusammenschrauben - wenn du der internen Rechnerei deiner WP ausreichend Glauben schenkst.
In der Regel taugen diese internen "Messwerte" nicht so viel und es ist eher ratsam einfach einen externern Zähler in die Anschlussleitung hinzuzufügen und nur diesen abzufragen.
Vielleicht ist Lambda aber auch eine große Ausnahme 🤷🏻‍♂️
Probier es aus, miss es nach.

jannisif, 2025-02-19:
Ich habe es jetzt über ein ext-Meter gemacht, dies erschien mir sinnvoller.
Wie schon gesagt, habe ich die Register-Werte der WP mit mbpoll erfolgreich lesen können: 
`mbpoll -m tcp -r 104 10.0.0.96 `
```
mbpoll 1.0-0 - ModBus(R) Master Simulator
Protocol configuration: ModBus TCP
Slave configuration...: address = [1]
                        start reference = 104, count = 1
Communication.........: 10.0.0.96, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1... Ctrl-C to stop)
[104]: 	1716
-- Polling slave 1... Ctrl-C to stop)
[104]: 	1716
-- Polling slave 1... Ctrl-C to stop)
[104]: 	1715
-- Polling slave 1... Ctrl-C to stop)
[104]: 	1714
-- Polling slave 1... Ctrl-C to stop)
[104]: 	1713
^C--- 10.0.0.96 poll statistics ---
5 frames transmitted, 5 received, 0 errors, 0.0% frame loss

everything was closed.
```
Nun habe ich das versucht, in evcc.yaml anzuwenden:
```
- name: lambda_power
  type: custom
  power:
    source: modbus
    uri: 10.0.0.96:502
    id: 1 
    register:
      address: 104 # aktuelle Aufnahmeleistung der WP
      type: holding
      decode: int16

```
Es erscheinen keine Fehler, das ist schonmal gut. In den Logs sehe ich aber nun folgendes:

```
[lambda_power] TRACE 2025/02/19 00:15:53 modbus: send 00 19 00 00 00 06 01 03 00 68 00 01
[lambda_power] TRACE 2025/02/19 00:15:53 modbus: recv 00 19 00 00 00 05 01 03 02 00 00
[lambda_power] TRACE 2025/02/19 00:16:08 modbus: send 00 1a 00 00 00 06 01 03 00 68 00 01
[lambda_power] TRACE 2025/02/19 00:16:08 modbus: recv 00 1a 00 00 00 05 01 03 02 00 00
[lambda_power] TRACE 2025/02/19 00:16:23 modbus: send 00 1b 00 00 00 06 01 03 00 68 00 01
[lambda_power] TRACE 2025/02/19 00:16:23 modbus: recv 00 1b 00 00 00 05 01 03 02 00 00

```
`evcc meter --log trace` ergibt nichts überraschendes: (bei Bedarf kann ich auch nochmal alles posten)
```
lambda_power
------------
[lambda_power] TRACE 2025/02/19 00:27:12 modbus: send 00 01 00 00 00 06 01 03 00 68 00 01
[lambda_power] TRACE 2025/02/19 00:27:12 modbus: recv 00 01 00 00 00 05 01 03 02 00 00
Power: 0W
```
Resultat: Es kommt was an, aber die Werte werden nicht richtig bzw. als 0 decodiert. Ich habe auch `uint16` versucht und den Wechsel auf `input` ebenfalls.

Übersehe ich irgendwas?
Woran kann es liegen, dass mbpoll die Werte decodieren kann, aber evcc nicht?
Habt ihr noch Ideen? Danke für die großartige Unterstützung aller bisher!

premultiply, 2025-02-19:
```yaml
- name: lambda_power
  type: custom
  power:
    source: modbus
    uri: 10.0.0.96:502
    id: 1 
    register:
      address: 103 # Actual power consumption
      type: holding
      decode: int16
```

andreas-bulling, 2025-02-19:
Ich sehe ehrlich gesagt noch nicht, was eine Integration genau bringen soll. Die Lambda macht Erhöhung bei Überschuss wunderbar alleine - man muss nur den aktuellen Grid Meter Wert regelmäßig in das entsprechende Register schreiben. Tut bei uns hervorragend mit einer HomeAssistant Automation. 

Welchen Vorteil bietet eine EVCC Integration darüber hinaus?

jannisif, 2025-02-19:
Ich verstehe dich da komplett, auch ich nutze Home Assistant, aber eben als Teil eines "mehrschichtigen" Systems.
Das wäre eine super Neuigkeit, die direkte Unterstützung von dyn. Stromtarifen durch Lambda wäre natürlich umso besser, sodass die WP selber nach den optimalen Zeitfenstern plant. Hoffen wir mal, dass das auch wirklich  kommt!  Das Sperren sehe ich auch kritisch, aber damit hätte man halt Preisspitzen umgehen können. Das ist natürlich keine saubere Lösung.

Aber jetzt weiter im Text: dann wäre es jetzt erstmal wichtig, wie man die Griddaten per Modbus an die WP bekommt, also wie wir Register 102 mit Werten beschreiben können? Wie würde das denn durch evcc gehandhabt werden? Also woher bekommt man den Grid Wert und wie kann diesen automatisch in das Register 102 schreiben?
Ich kann selbst vermutlich nicht als Testobjekt dienen, da wir schon ein Gridmeter mitinstalliert bekommen haben, und an der Einstellung kann/möchte ich erstmal nichts ändern.

Ich mache mich im März an das Lambda-Template, wenn bis dahin keine Mini-LPs in Aussicht sind. Dann habe ich mehr Zeit und Ruhe, und kann testen :) Und vielleicht bekomme ich (oder ihr) es noch hin, dass man noch mehr Werte sehen kann.

Für alle, die es eben jetzt schon haben wollen, finden ja oben die Anleitung/Config! 

Vielen Dank für alle eure Bemühungen bis hierhin und sorry für den Spam in den letzten Tagen :D

andig, 2025-02-20:
> Ich sehe ehrlich gesagt noch nicht, was eine Integration genau bringen soll. Die Lambda macht Erhöhung bei Überschuss wunderbar alleine - man muss nur den aktuellen Grid Meter Wert regelmäßig in das entsprechende Register schreiben. Tut bei uns hervorragend mit einer HomeAssistant Automation.

Ich glaube der Punkt hier ist, dass es "jemand" tun muss. Das kann auch evcc sein. Siehe auch https://github.com/evcc-io/evcc/issues/19002. Das scheint kein Einzelfall zu sein. Man könnte also einen weiteren einfachen "leistungs-kontrollierten" Charger draus machen bei dem sich optional noch Temperatur und Leistung konfigurieren ließen. Das könnte sogar charmanter sein als ext/aux meter die beiden den Kern verfehlen weilen die Leistung zum Teil (aber nur zum Teil) steuerbar ist.

/cc @premultiply

hausi24, 2025-02-22:
Interessant wäre die Einbindung auch für die § 14a Regelung, bzw. Leistungsbeschränkung der WP durch EVCC als Energiemanagement System.
Auch die Priorisierung von PV Überschuss über das Lastmanagement von EVCC ergibt für mich Sinn und wäre ein tolles Feature.

andig, 2025-02-22:
Schaut Euch mal bitte an, ob https://github.com/evcc-io/evcc/pull/19008 funktionieren kann. Da wird einfach der WP die verfügbare Leistung mitgeteilt. Temperatursensor fehlt noch (welche/n bräuchte es dafür? gerne PR direkt ergänzen).

Kann das jemand testen?

SolarPowerEV, 2025-02-28:
> Wie doof, meine Sturheit... 103 ist korrekt, warum auch immer mbpoll dort den Gridmeter findet. Ok, nächstes Problem: die Werte stimmen und werden auch im Log korrekt angezeigt, nur im Dashboard von evcc nicht.. was ja mein eigentliches Ziel war. Die Ansage mit dem ext/aux-Meter hat mich etwas irritiert und man findet in der Doku kaum etwas dazu. Die ext/aux-Werte werden aktuell (wie auch immer) nur für die interne Berechnung verwendet.
> 
> [#6984](https://github.com/evcc-io/evcc/discussions/6984) und [Diskussion 7235](https://github.com/evcc-io/evcc/discussions/7235#discussioncomment-12246021) zeigen ein wenig das Dilemma, in dem ich mich gerade befinde. Also ich stehe jetzt zwischen den beiden Möglichkeiten:
> 
>     1. Warten wir auf die Mini-LPs oder Anzeige der ext-meters, und machen es dann vernünftig, sodass auch eine minimale Steuerung(ergo der Boost) möglich wäre **oder**
> 
>     2. Integriere ich die WP in einem extra WP-LP, was aber nicht wirklich schön ist (bei Switch zum Mini-LP wird das dann auch wieder "Theater" geben mit der Datenbank/Statistik)

Ich stehe gerade vor demselben Problem. Ich fänds super, wenn evcc im GUI mir den aktuellen Verbrauch meiner Lambda direkt anzeigen würde.

Bei der ersten Konfig als ext-meter kann ich mir den Wert zwar im CLI ausgeben lassen, die Konfig scheint also zu stimmen und schmeißt auch keinen Fehler. Aber im GUI ist der Verbrauch bzw. dieses ext-meter (trotz Eintrags unter site:) beim besten Willen nirgends zu sehen. Auch nicht in der Auflistung aller Geräte im Bereich "Home / Konfiguration".

Bei der zweiten Variante als WP-Loadpoint wird der Wert angezeigt, allerdings sehr groß als vollwertiger Loadpoint (von denen habe ich allerdings schon drei, was die Nutzung auf dem Handydisplay nicht gerade einfacher macht) und dann mit etwas "schrägen" Angaben wie _Verbunden_ und einem dicken grünen Balken ohne Funktion:

[image]

andig, 2025-03-01:
@SolarPowerEV Du willst der WP einfach nur den aktuellen Überschuss "verkaufen", richtig? Dann ist https://github.com/evcc-io/evcc/pull/19008/files#diff-839ae08ecd3f8c911f56336c1c17582a559be1931575329a83baf8737fb4b529 die Lösung. Falls nein bitte separate Diskussion- die Umsetzung hier ist erfolgt.

JanPeer82, 2025-03-01:
Hallo zusammen, bei mir funktioniert die einbingung super Vielen Dank dafür.
@chriz77 Moin, wie bist du an das Level 2 passwort für die Wärmepumpe gekommen. Ich besitze diese bereits seit über einem Jahr und mein Installateur sowie Zewotherm waren bisher nicht in der lage mir diese zu übergeben. Ich muss wegen jeder kleinen änderung den Zewotherm support anschreiben. Ich würde gerne die "Leistungsanforderung Einschaltverzögerung" anpassen.

SolarPowerEV, 2025-03-02:
> [@SolarPowerEV](https://github.com/SolarPowerEV) Du willst der WP einfach nur den aktuellen Überschuss "verkaufen", richtig? Dann ist [...]

Ich möchte im ersten Schritt erst einmal nur den Stromverbrauch meiner Lambda in evcc separat ausgewiesen bekommen. Da die Informationen in diesem Thread etwas unklar sind, hatte ich gehofft, von @jannisif oder jemand anderem Kompetenten eine klare Antwort zu erhalten, ob ich ggf. etwas übersehe und man ein ext-meter vielleicht doch noch irgendwie angezeigt bekommen kann oder ein vollständiger Loadpoint wirklich die einzige Option ist.

Die mit #19008 ermöglichte Steuerung der Lambda probiere ich gerne dann im nächsten Schritt aus. Ich bin mir allerdings nicht sicher, ob das "setmaxpower" in #19008 am Ende wirklich so richtig funktionieren wird, denn das Register 103 ist nach der [mir vorliegenden Dokumentation von Lambda](https://lambda-wp.at/wp-content/uploads/2025/02/Modbus-Beschreibung-und-Protokoll.pdf) als "RO" (= read-only) markiert und zur Übergabe des PV-Überschuss ist wohl Register 102 vorgesehen (das ist auch markiert als "RW") . Aber vielleicht missverstehe ich hier nur etwas grundlegend. 

[image]

SolarPowerEV, 2025-03-02:
@JanPeer82 Ich kann Dir helfen, sogar mit Level 3 access. Kontaktier mich bitte direkt entweder hier oder in Slack.

andreas-bulling, 2025-03-02:
@SolarPowerEV 

> Ich bin mir allerdings nicht sicher, ob das "setmaxpower" in [#19008](https://github.com/evcc-io/evcc/pull/19008) am Ende wirklich funktionieren wird, denn das Register 103 ist nach der [mir vorliegenden Dokumentation von Lambda](https://lambda-wp.at/wp-content/uploads/2025/02/Modbus-Beschreibung-und-Protokoll.pdf) als "RO" (= read-only) markiert und zur Übergabe des PV-Überschuss ist wohl Register 102 vorgesehen (das ist auch markiert als "RW") . Aber vielleicht missverstehe ich hier auch nur etwas grundlegend?

Du hast Recht. Das wird mit Register 103 nicht funktionieren. Überschuss an 102 übergeben und fertig. Lambda regelt den Rest.

Unter anderem deswegen meinte ich weiter oben, dass ich nicht verstehe, was das Ziel sein soll. Für die Überschussregelung braucht man m.E. keine zusätzliche externe Steuerung.

andreas-bulling, 2025-03-02:
Hallo zusammen,

ist für manche von euch vielleicht auch interessant: Ich habe heute morgen die Integration der LAMBDA in Venus OS (Victron Ökosystem) fertiggestellt. Damit lassen sich die aktuell seitens Victron unterstützten Werte (aktuelle Leistung, Energieverbrauch gesamt) in der Remote Console anzeigen.

Das VRM Portal zeigt momentan leider noch keine Wärmepumpen an und auch die Temperaturwerte werden noch nirgendwo angezeigt. Das hat Victron noch nicht umgesetzt.

Den Code findet ihr hier:
https://github.com/andreas-bulling/dbus-lambda

andig, 2025-03-02:
> Du hast Recht. Das wird mit Register 103 nicht funktionieren. Überschuss an 102 übergeben und fertig. Lambda regelt den Rest.

Ändere ich in 30s. Könnt ihr bitte nochmal die Doku Teilen? Der Lambda Link ist leider tot.

andreas-bulling, 2025-03-02:
Höh?! Der Link tut bei mir: https://lambda-wp.at/wp-content/uploads/2025/02/Modbus-Beschreibung-und-Protokoll.pdf

andig, 2025-03-02:
Yo, jetzt wieder. Eben 404.

chriz77, 2025-03-02:
@andig @andreas-bulling Vielen lieben Dank für euren Einsatz!!! Ich bin gerade dabei auf Victron umzustellen und werde das dann auch mal testen.

SolarPowerEV, 2025-03-02:
> Du hast Recht. Das wird mit Register 103 nicht funktionieren. Überschuss an 102 übergeben und fertig. Lambda regelt den Rest.
>
> Unter anderem deswegen meinte ich weiter oben, dass ich nicht verstehe, [...]

Da stimme ich jetzt nicht ganz zu. M.E. hat @andig schon recht, wenn er sagt, dass jemand die Kontrolle übernehmen und die Daten regelmäßig der Lambda mitteilen muss. Warum sollte evcc nicht diese Rolle übernehmen? 
Das ist doch die Kernkompetenz von evcc, den PV-Überschuss zu ermitteln und dann zu verteilen und mir wäre es auch lieber, dass evcc das zentral regelt, als dass ich das dann über einen extra Modbus Stromzähler von Zewotherm machen muss.

Die Frage, die ich noch nicht abschließend klären konnte, ist allerdings, ob man im Menü "Energiemanagement" der Lambda die Betriebsart auf "Manuell" oder auf "Automatik" stellen muss:

[image]

Also bedeutet "Automatik" bei der Lambda, dass diese sich die Daten dann selbständig von einem externen Modbus-Stromzähler aktiv per Modbus abholt und alles weitere dann selbständig regelt?

Oder muss "Automatik" auch eingestellt werden, wenn man über das Register 102 der Lambda den Überschuss regelmäßig mitteilt (wenn ja, in welchen Intervallen?) und die Lambda entscheidet daraufhin alles weitere selbständig?

Oder bedeutet das Schreiben des Registers 102 bereits, dass man die Lambda quasi fernsteuert und deswegen der Modus "Manuell" der richtige wäre?

chriz77, 2025-03-02:
Guck doch mal bei meinem Screenshots weiter oben. Das ist doch genau die Einstellung die du brauchst. Die Betriebsart muss auf Automatik stehen, dann berechet die Lambda den Psoll Wert automatisch. Ich verstehe aber nicht wirklich, was die Leistungsanforderung Min/Max Werte bedeuten. Zumindst läuft es bei mir jetzt ganz gut. Die WP nutzt brav den PV-Überschuss. Allerdings bekommt sie den aktuell noch über Loxone.

andreas-bulling, 2025-03-02:
@SolarPowerEV Das war dann missverständlich. Mit "zusätzlicher externer Steuerung" meinte ich das aktive Aktivieren des Überschuss-Modus, das externe Setzen von irgendwelchen Leistungsparametern etc. 

Dass nicht jeder ein HomeAssistant am Laufen hat, um den Überschuss in Register 102 zu schreiben, habe ich oben ja schon zugestimmt. Aber darüber hinaus braucht man m.E. nichts.

Bei uns funktioniert die Überschussregelung der Lambda hervorragend.

SolarPowerEV, 2025-03-03:
> Guck doch mal bei meinem Screenshots weiter oben. Das ist doch genau die Einstellung die du brauchst. Die Betriebsart muss auf Automatik stehen, dann berechet die Lambda den Psoll Wert automatisch. [...]

Ja, den Screenshot hatte ich gesehen. Ich hatte aber gedacht, der diente mehr der generellen Darstellung, was die Lambda grundsätzlich könnte und dient nicht als Doku, dass alle Settings genau so eingerichtet werden müssen. 

Ich bin mir bzgl. der Einstellung "Automatik" immer noch nicht ganz sicher. Wenn "Automatik" nicht nur für den Fall eingestellt werden muss, dass die Lambda sich die Daten selbständig von dem per Modbus angeschlossenen Stromzähler von Zewotherm (bzw. genau genommen Carlo Gavazzi) holt und verwaltet, sondern auch für den Fall, dass man der Lambda die Werte ins Register 102 schreibt: Wofür genau dient dann noch der Modus "Manuell"? 
Abfragen lassen sich die Register per Modbus TCP jedenfalls auch alle im Setting Manuell.

Um Deine Frage noch zu beantworten, was die Settings im Energie Management-Untermenü der Lambda bedeuten:

**Anlagengrundlast (Default 0.30 kW)**
-     Definiert die Mindestleistung, die immer für andere Verbraucher im Haus reserviert bleibt.
-     Die Wärmepumpe beginnt erst mit der Nutzung von PV-Überschuss, wenn mehr als 0.30 kW frei sind.

**Leistungsanforderung Minimum (Default 0.5 kW)**
-     Dieser Wert regelt, dass die Wärmepumpe erst bei mindestens 0.5 kW verfügbarem PV-Überschuss aktiviert wird.
-     Falls der Überschuss unter 0.5 kW liegt, bleibt die Wärmepumpe ausgeschaltet.

**Leistungsanforderung Maximum (Default 4.0 kW)**
-     Dieser Wert definiert die maximale Leistung, die die Wärmepumpe aus dem PV-Überschuss ziehen darf.
-     Mit diesem Default-Wert wird sie maximal 4.0 kW aus der PV-Anlage nutzen.
-     Falls mehr Überschuss verfügbar wäre (gemeldet wird), wird dieser nicht weiter genutzt.

Wobei ich mich jetzt frage, ob die 0.30 kW und die 0.5 kW kumulativ sind und die Lambda dann mit diesen Werten faktisch erst bei 0.8 kW Überschuss anfängt, loszulegen. Ich vermute aber mal, dass dem so sein müsste.

thecem, 2025-03-03:
**Manuell:** Es wird auf einen fest vorgegeben Wert geregelt.
**Auto:** Die Leistung wird auf den Überschuss und Bedarf zwischen min und max geregelt.

SolarPowerEV, 2025-03-03:
@thecem Danke für die Info! Weißt Du zufällig auch noch, über welches Register der Lambda dann der fest vorgegebene Wert übergeben werden müsste?
Register 102 kann es ja nicht sein, denn das bekommt nur den gerade verfügbaren Überschuss mitgeteilt, d.h. das dürfte nur für die Automatik funktionieren.

Sind das dann die Register auf den Seiten 12 und 13 der Modbus-Doku? Also mit Beschreibungen wie diesen für den Buffer:
[image]

oder diesen für den Heizkreis?
[image]

thecem, 2025-03-03:
Wie schon gesagt, ist Lambda mit dem Modbus am Anfang überfordert gewesen, heute würden sie es anders machen.

Der Wert für Manuell kann nur per Oberfläche eingegeben werden, es gibt kein Register um das zu schreiben.

Wenn ein Wert für einen bestimmten Zeitraum nicht übergeben wird, wird das Programm mit dem jeweiligen timeout (teilw. Konfigurierbar) fertig gefahren und sobald der Register Time Out (1min) greift ein Fehler geworfen.

Ich hatte vorgeschlagen alle (ALLE)  Werte der Steuerung per Modbus ab Register 10000 zu übergeben und jeweils ein zusätzliches Lese Register für die R/W Register anzulegen. Somit wäre dann auch alles möglich, aktuell können bestimmte Register keine Rückgabe liefern. Egal, wird sich vielleicht ändern.

SolarPowerEV, 2025-03-03:
Vielen Dank für die Infos!
Vor allem das mit dem timeout ist sehr hilfreich zu wissen. 

Ich habe gerade mal versucht, über die Lambda-Oberfläche eine Überschuss-Regelung (User hat Level 3 Rechte) zu provozieren. 

Im Automatik-Setting kann ich nur einen Wert bei "Aktuelle Messleistung" eintragen. Der wird aber munter ignoriert, egal was ich da eingebe. Am Ende erscheint immer wieder einfach nur "0.00 kW"

Im Setting "Manuell" werden die bei "Aktuelle Messleistung" eingetragenen Werte ebenfalls ignoriert. Dafür wird aber noch zusätzlich das Feld "Psoll" (im Untermenü dann als "Regelwert Aufnahmeleistung" bezeichnet) editierbar. Nachdem ich dort einen höheren Wert (3 kW - die Pumpe lief vorher auf 2,1 kW) eingetragen hatte, wurde das Icon "Solarüberschuss" angezeigt und die Pumpe ist nun dabei, das Brauchwasser aufhzuheizen. Obwohl lustigerweise dort kein Solarüberschuss-Icon angezeigt wird, dafür aber daneben beim Pufferspeicher:
[image]

Und komischerweise fährt die Wärmepumpe gerade auch deutlich höher, als mit Psoll vorgegeben:

[image]

Ich nehme an, dass wenn die PV-Automatik einmal getriggert ist, die Lambda einfach ihr Programm durchfährt (im Rahmen der von mir oben beschriebenen Min/Max-Parameter) und den tatsächlichen weiteren PV-Überschuss dann einfach ignoriert. Was aus Verdichter-Sicht ja vermutlich auch Sinn macht. Das bedeutet allerdings, dass man hier aus Nutzer- bzw. Steuerungssicht damit rechnen muss, dass die Pumpe sich auch mal 1 bis 2 kW mehr nimmt, als man ihr als verfügbar mitgeteilt hat.

chriz77, 2025-03-03:
@SolarPowerEV Wenn du manuell den Psoll Wert verändest, dann spult die Lambda ihr PV Programm ab, egal wie hoch der Überschuss ist. Den Psoll Wert musst du wieder auf 0 setzten, sonst wird sie immer wieder mit dem PV-Programm anfangen. Der manuelle Wert macht für mich keinen Sinn, da man diesen leider nicht über Modbus ändern kann. Wie geschrieben regelt die WP das selber, wenn sie PV-Überschuss über das Register 102 bekommt. Übrigens reichen User Rechte 2, um den Psoll Wert manuell zu setzen.

Wenn die WP das Brauchwasser überheitzt, dann sollte das PV Symbol eigentlich auch beim Brauchwasser zu sehen sein.

Wie thecem richtig geschrieben hat, sind die Steuerungsmöglichkeiten über Modbus leider sehr rudimentär. Hoffentlich kommt da in Zukunft noch mehr. Allerdings frage ich mich, wie man dann an das FW-Update kommt? Vermutlich muss sich dann jemamand von Lambda oder Zewotherm aufschalten und das aufspielen.

Die Leistungsanforderungswerte hätte ich auch so interpretiert wie du. Allerdings war ich mir nicht sicher, was mit der maximalen Leistungsanforderung gemeint ist. Meine WP hat noch nie mehr als 1,5 kW gezogen, was auch locker zu reichen scheint. Bei uns war es auf jeden Fall immer kuschlig warm :-)

SolarPowerEV, 2025-03-03:
> [@SolarPowerEV](https://github.com/SolarPowerEV) Wenn du manuell den Psoll Wert verändest, dann spult die Lambda ihr PV Programm ab, egal wie hoch der Überschuss ist. Den Psoll Wert musst du wieder auf 0 setzten, sonst wird sie immer wieder mit dem PV-Programm anfangen. 

Ja, das habe ich heute auch gemerkt, als ich ein bisschen damit rumgespielt habe. Und in der Tat zum Beenden des PV-Modus unbedingt den Wert auf "0" setzen und nicht etwa die Betriebsart auf "Aus"! 
Weil da hat die Lambda zwar auch den PV-Überschussmodus beendet, aber anschließend direkt gemerkt, dass Pufferspeicher und Heizkreis mehr Wärme hatten, als nötig und ist direkt mal in den Modus "keine Anforderung" gegangen, sprich hat sich für eine Weile ganz abgeschaltet. 🤪
Das war zwar logisch und folgerichtig, aber natürlich kontraproduktiv, weil in Abschaltzeit (die auch eine Sperrzeit enthält) die ganze Überladung erstmal wieder abgerufen wurde. 🙈

> Wenn die WP das Brauchwasser überheitzt, dann sollte das PV Symbol eigentlich auch beim Brauchwasser zu sehen sein.

Das war auch meine Erwartungshaltung, aber der Screenshot oben ist da ja eindeutig. Die Symbole beim Heizkreis zeigen klar eine Anforderung für das Brauchwasser an, aber ausgerechnet beim Brauchwasser wird kein Icon angezeigt. 

> Die Leistungsanforderungswerte hätte ich auch so interpretiert wie du. Allerdings war ich mir nicht sicher, was mit der maximalen Leistungsanforderung gemeint ist. Meine WP hat noch nie mehr als 1,5 [...]

Wenn man keinen ungedämmten Altbau hat, wie wir, dann reicht das definitiv. Aber es hat einen Grund, warum wir eine EU20L eingebaut bekommen haben. Und die kann auch mal locker 4 oder 5 kW ziehen, wenn es darum geht, Brauchwasser zu erwärmen oder gegen Minusgrade draußen anzuheizen. Aber ist in jedem Fall ein tolles Gerät. COPs von über 4 bei Null Grad finde ich schon echt beachtlich (aktuell fährt sie bei 10 Grad mit einem COP von 5,11)

Bzgl. der Leistungsanforderung ist mir noch eingefallen, dass ich vielleicht auch einen Denkfehler hatte. 
Wenn die Lambda bereits mit z.B. 1,5 kW im Regelbetrieb läuft und ich ihr dann den von evcc angezeigten Überschuss von z.B. 3,5 kW melde, dann ist Überschuss ja schon nach dem vorhandenen Verbrauch i.H.v. 1,5 kW berechnet. D.h. in der Tat stünden der Lambda in dem Szenario ja auch 1,5 kW + 3,5 kW Überschuss = 5 kW zur Verfügung. 
Und dann würde es Sinn machen, dass die Lambda sich einen Wert von mehr als 3,5 kW genehmigt hat.

andig, 2025-03-03:
> D.h. in der Tat stünden der Lambda in dem Szenario ja auch 1,5 kW + 3,5 kW Überschuss = 5 kW zur Verfügung.

Ich kann der ganzen Diskussion seit mindestens 30 Posts nicht mehr folgen und schlage Euch vor in eine Discussion zu wechseln. Die Implementierung ist fertig und die Diskussion scheint sich um alles andere zu drehen. Danke.

SolarPowerEV, 2025-03-03:
D.h. zum Auslesen des Verbrauchs wird jetzt das Register 103 verwendet und zum Übergeben des Überschusses das Register 102? 

Gibt es irgendwo ein yaml-Template für die Lambda, mit welchem ich die Implementierung mit der nächsten Version (0.200.6) dann einmal austesten kann? Denn meine yaml enthält nur einen Loadpoint mit Verweis auf das Register 103 und mir ist nicht klar, wie ich das erweitern muss, damit die Steuerung aktiviert wird und ob das so (also Loadpoint) überhaupt richtig ist.

In jedem Fall schonmal vielen Dank an alle Beteiligten für die Implementierung der Lambda in evcc! Highly appreciated! 🙏

andig, 2025-03-03:
> Gibt es irgendwo ein yaml-Template für die Lambda

Ungefähr (nein, genau!) seitdem dieses Issue geschlossen wurde...

SolarPowerEV, 2025-03-03:
Deinem Wunsch entsprechend habe ich meine Fragen jetzt in diesen Diskussions-Thread verlagert:
https://github.com/evcc-io/evcc/discussions/19384

SolarPowerEV, 2025-03-08:
Ich weiß nicht, ob dieser Thread jetzt der richtig ist, aber hat schon jemand die Lambda erfolgreich mit dem aktuellen Release 0.200.8 ansteuern können? Der Fix für Register 102 statt 103 zur Ansteuerung sollte darin ja eigentlich enthalten sein.

Bei mir kam leider in der Lambda kurz nach dem Umschalten des Loadpoints "Lambda Wärmepumpe" eine dicke rote Fehlermeldung: "Alarm: E-Manager ungültiger Wert". 

[image]

Irgendwie scheint der Lambda der Wert nicht zu schmecken, den evcc in das Register 102 schreibt. 

Klappt das bei Euch? Ich wollte erstmal hören, ob ich der einzige mit dem Fehler bin und es bei anderen läuft oder ob ich ein Issue aufmachen soll mit Logfile etc. pp.

Rocky-1-RB, 2025-05-01:
@TRON4R 
um für meine neue Lambda-WP EU13 den PV-Überschuss nutzen zu können, benötige ich ja auch die Rechte für Level 2 access (auch Level 3? möglich).
Du hast dazu Infos? - Danke!

TRON4R, 2025-05-02:
Du kannst Dir die nötigen Einstellungen in der Lambda auch vom Support (die reagiere auf eine E-Mail binnen 24 Stunden, ansonsten hilft auch anrufen) freischalten lassen. Ansonsten schreib mich gerne in Slack mal direkt an.

G3m1b2, 2025-05-19:
@TRON4R Mich würde der Level 2 und 3 Zugang ebenfalls interessieren, checke aber nicht wie ich dich in Slack kontaktieren kann. Danke für Infos und deine Hilfe.

TRON4R, 2025-05-25:
Such in Slack nach meinem Namen. Andere haben mich darüber auch gefunden. Es ist schade, dass GitHub keine vernünftige Messaging-Funktion bietet. Und meine E-Mail-Adresse will ich hier auf keinen Fall breittreten, sorry. Ich bekomme jetzt schon mehr als genug SPAM.

dev-kalleno, 2025-07-03:
Hey TRON4R, ich habe auch Interesse an erhöhten Rechten in meiner Lambda, momentan kann ich ja nicht mal meine Zirkulationspumpe selbst steuern 😅. Ich habe auch bereits bei Slack Kontakt gesucht, aber vielleicht habe ich den falschen angeschrieben?

Paradiesvogel1989, 2025-09-16:
Hi, @TRON4R ich hätte auch interesse an den Passwörtern, ich kann dich bei Slack allerdings unter dem Namen TRON4R nicht finden.

TRON4R, 2025-10-06:
Sorry, aber bitte erfragt die Zugangsdaten für höhere Level als Level 1 (der Anwender-Level hat kein Passwort) bei Eurem Heizungsmonteur oder bei Lambda/Zewotherm oder lasst Euch die nötigen Einstellungen von deren Support anpassen. Die reagieren super-fix auf E-Mails und Anrufe.