evcc-io/evcc #18815: 401 Unauthorized: Invalid API User error

Stevansell, 2025-02-15:
### Describe the bug

[evcc-20250215-110510-trace.log](https://github.com/user-attachments/files/18806795/evcc-20250215-110510-trace.log)

Once again, after about 12 hours of running, All functionality is blocked for a 401 error. The same as my fault raised yesterday.

### Steps to reproduce

1. Run EVCC
2. Leave it run on 'Solar"
3. After around 12 hours it generates errors and stops working
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 30s # control cycle interval

meters:
- name: grid
  type: template
  template: fronius-solarapi-v1
  usage: grid
  host: 192.168.1.213

- name: pv
  type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.1.213
  port: 502
  
chargers:
- type: template
  template: twc3 
  host: 192.168.1.195  
  name: wallbox

vehicles:
- name: ev
  type: template
  template: tesla
  title: 
  clientId: token
  accessToken: token
  refreshToken: token
  vin: 
  capacity: 62
  commandProxy: https://api.myteslamate.com
  proxyToken: token

loadpoints:
- title: Garage
  charger: wallbox
  vehicle: ev
  mode: pv

tariffs:
  currency: AUD # (default EUR)
  grid:
    # static grid price with price zones)
    type: fixed
    price: 0.2915 # EUR/kWh
    zones:
      - days: Mo-So
        hours: 15-21
        price: 0.5126
      - days: Mo-So
        hours: 11-14
        price: 0.0001
      - days: Mo-So
        hours: 0-6
        price: 0.08
    
    
  feedin:
    type: fixed
    price: 0.055

site:
  title: Home
  meters:
    grid: grid
    pv: pv

sponsortoken: token
```

### Log details

```text
[site  ] DEBUG 2025/02/15 11:04:40 ----
[twc3  ] TRACE 2025/02/15 11:04:40 GET http://192.168.1.195/api/1/vitals
[twc3  ] TRACE 2025/02/15 11:04:40 {"contactor_closed":false,"vehicle_connected":true,"session_s":50610,"grid_v":242.8,"grid_hz":49.843,"vehicle_current_a":0.5,"currentA_a":0.5,"currentB_a":0.5,"currentC_a":0.3,"currentN_a":0.7,"voltageA_v":2.0,"voltageB_v":5.5,"voltageC_v":0.0,"relay_coil_v":11.8,"pcba_temp_c":20.7,"handle_temp_c":17.0,"mcu_temp_c":26.8,"uptime_s":633457,"input_thermopile_uv":-187,"prox_v":0.0,"pilot_high_v":8.6,"pilot_low_v":-11.6,"session_energy_wh":21717.699,"config_status":5,"evse_state":4,"current_alerts":[],"evse_not_ready_reasons":[1]}
[lp-1  ] DEBUG 2025/02/15 11:04:40 charge power: 0W
[lp-1  ] DEBUG 2025/02/15 11:04:40 charge currents: [0.5 0.5 0.3]A
[sunspec] TRACE 2025/02/15 11:04:40 modbus: send 03 e7 00 00 00 06 01 03 9d 3f 00 08
[grid  ] TRACE 2025/02/15 11:04:40 GET http://192.168.1.213/solar_api/v1/GetPowerFlowRealtimeData.fcgi
[sunspec] TRACE 2025/02/15 11:04:40 modbus: recv 03 e7 00 00 00 13 01 03 10 ff fc ff fe ff fe ff fe ff ff ff ff 00 02 ff ff
[sunspec] TRACE 2025/02/15 11:04:40 modbus: send 03 e8 00 00 00 06 01 03 9d 47 00 14
[sunspec] TRACE 2025/02/15 11:04:40 modbus: recv 03 e8 00 00 00 2b 01 03 28 00 01 4d 50 50 54 20 31 00 00 00 00 00 00 00 00 00 00 47 a3 3e a5 72 e1 17 72 5c f6 2f 42 96 97 80 00 ff ff ff ff ff ff
[sunspec] TRACE 2025/02/15 11:04:40 model 160 block 1 point DCW: 294.09000000000003
[sunspec] TRACE 2025/02/15 11:04:40 modbus: send 03 e9 00 00 00 06 01 03 9d 3f 00 08
[sunspec] TRACE 2025/02/15 11:04:40 modbus: recv 03 e9 00 00 00 13 01 03 10 ff fc ff fe ff fe ff fe ff ff ff ff 00 02 ff ff
[sunspec] TRACE 2025/02/15 11:04:40 modbus: send 03 ea 00 00 00 06 01 03 9d 5b 00 14
[sunspec] TRACE 2025/02/15 11:04:40 modbus: recv 03 ea 00 00 00 2b 01 03 28 00 02 4d 50 50 54 20 32 00 00 00 00 00 00 00 00 00 00 3d 27 7c 6a c2 c6 30 1b 7d aa 2f 42 96 97 80 00 ff ff ff ff ff ff
[sunspec] TRACE 2025/02/15 11:04:40 model 160 block 2 point DCW: 498.62
[site  ] DEBUG 2025/02/15 11:04:40 pv 1 power: 793W
[grid  ] TRACE 2025/02/15 11:04:40 {
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 1,
               "E_Day" : null,
               "E_Total" : 11468355.939722221,
               "E_Year" : null,
               "P" : 735.3187255859375
            }
         },
         "SecondaryMeters" : {},
         "Site" : {
            "BackupMode" : false,
            "E_Day" : null,
            "E_Total" : 11468355.939722221,
            "E_Year" : null,
            "Meter_Location" : "grid",
            "Mode" : "meter",
            "P_Akku" : null,
            "P_Grid" : -200.47,
            "P_Load" : -534.84872558593747,
            "P_PV" : 790.78250122070312,
            "rel_Autonomy" : 100.0,
            "rel_SelfConsumption" : 72.736992405537364
         },
         "Smartloads" : {
            "Ohmpilots" : {}
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2025-02-15T00:04:40+00:00"
   }
}
[site  ] DEBUG 2025/02/15 11:04:40 grid power: -200W
[site  ] DEBUG 2025/02/15 11:04:40 site power: -200W
[lp-1  ] DEBUG 2025/02/15 11:04:40 charge voltages: [2 5.5 0]V
[lp-1  ] DEBUG 2025/02/15 11:04:40 charger status: B
[lp-1  ] DEBUG 2025/02/15 11:04:40 smart cost active: 0.00
[tesla ] TRACE 2025/02/15 11:04:40 POST https://api.myteslamate.com/api/1/vehicles/LRWYHCFS8NC579070/command/set_charging_amps
[tesla ] TRACE 2025/02/15 11:04:41 {"charging_amps": 32}
--
{"error":"Invalid Tesla API User"}
[lp-1  ] ERROR 2025/02/15 11:04:41 max charge current 32A: 401 Unauthorized: Invalid Tesla API User
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.0

andig, 2025-02-15:
Du hast sicher gestellt, dass mta das Token nicht refreshed?

Stevansell, 2025-02-15:
> Have you made sure that mta does not refresh the token?

How do i get it to not refresh the token? every time i hit the delete tokens from myteslamate button it refreshes everything anyway. But i dont see any other buttons to stop refresh token refreshing... it does look like only the refresh token is changing, the access token is the same

andig, 2025-02-15:
You MUST remove the token vom mtm refresh.

andig, 2025-02-15:
You MUST NOT share the token between apps.

jlestel, 2025-02-18:
> Once again, after about 12 hours of running, All functionality is blocked for a 401 error. The same as my fault raised yesterday.

Hi @Stevansell Tesla tokens seem to have been invalidated, MyTeslamate can no longer initiate orders.

To fix it, you only need to log in once here: https://app.myteslamate.com/fleet
Let me know when this is done, so I can track the behavior specifically on your user.

Stevansell, 2025-02-19:
> > Once again, after about 12 hours of running, All functionality is blocked for a 401 error. The same as my fault raised yesterday.
> 
> Hi [@Stevansell](https://github.com/Stevansell) Tesla tokens seem to have been invalidated, MyTeslamate can no longer initiate commands.
> 
> To fix it, you only need to log in once here: https://app.myteslamate.com/fleet Let me know when this is done, so I can track the behavior specifically on your user.

Hey @jlestel Thanks for getting back to me. I logged back into the myteslamate fleet api page, and it looks like EVCC has started working again. 
I am not sure if I am correct here, but I think this is user error on my part as it is a little confusing.

When I put the refresh token into my EVCC YAML, it works for a couple of hours and then I get the 401 error. When I log back into the myteslamate page it displays different refresh and access tokens, so I assume that its all been changed. So i got in a loop of updating those tokens every time I tried this. 
I also tried deleting those tokens after andy told me i must remove the token, and a big sign saying "Important! Only one app can refresh tokens. After pasting them into your app, click the button below to remove the tokens from MyTeslamate. This ensures that your app is the only one using them. If you want to use MyTeslamate proxies or view your vehicles here, you will be able to log in again.", and when i did that it immediately threw up the 401 error.

So I am hoping with the simple act of logging back in and it no longer throwing up 401 errors, it may be working now. I will give it 12 hours and let you know tomorrow morning if it fixed the issue.

Stevansell, 2025-02-21:
@jlestel @andig After leaving it for over 24 hours it looks like it is remaining stable. No issues have shown up.
Thanks for the assistance

closed 2025-02-21 by andig