evcc-io/evcc #18816: Wärmepumpe Kermi (Bösch) x-change dynamic pro AW E - Modbus TCP

czam70, 2025-02-15:
An das Entwicklerteam,

Ich besitze eine **Kermi x-change dynamic pro 10 AW E**.
Die Zahl sagt nur was über die Leistung aus, die gibt es auch als "6 AW E" oder "16 AW E". Das Modbus-Interface ist natürlich identisch.
Die Bösch Wärmepumpen, die in Österreich verkauft werden, sollten damit übrigens baugleich sein. 

Über Umwege (auf https://library.loxone.com/detail/fajslfdsadsa-1567/overview) habe ich das angehängte Modbus Dokument von Kermi gefunden.

Auf jeden Fall ist das eine vollständige Doku der Kermi Modbus Register.
Ich vermute, dass man die Register 300 bis 303 unter der Rubrik "PV Modulation Wärmepumpe" (Seite 2 im Dokument) nutzen könnte um die Wärmepumpe mit PV-Überschuss zu steuern.

Ich werde es mir aber nochmal von Kermi bestätigen lassen, dass das Dokument aktuell ist (aber warum sollte es das nicht sein) und welche Register genutzt werden sollten.

[Kermi Modbus TCP_RTU_Quick Guide_DE.pdf](https://github.com/user-attachments/files/18809267/Kermi.Modbus.TCP_RTU_Quick.Guide_DE.pdf)

andig, 2025-02-15:
Bitte ausprobieren ob die Register das tun was du erwartest. Vorher machts keinen Sinn da Arbeit rein zu stecken.

czam70, 2025-02-15:
Danke für die Rückmeldung, aber das kann ich leider nicht, dazu reicht meine Expertise nicht aus.
Aber ich frage mal bei Kermi nach. Wenn die das bestätigen sollte es doch passen, oder?

andig, 2025-02-15:
Bitte ausprobieren!

andig, 2025-02-16:
V.a. müsstest Du mal festlegen, was Du eigentlich von evcc erwartest. Welches Wunder soll passieren, wenn PV verfügbar ist? Einfach nur auf "comfort" schalten und anschließend wieder auf "eco"? Das scheint sehr einfach möglich zu sein. Was das dann aber tatsächlich tut kannst du nur an Deiner WP ausprobieren.

czam70, 2025-02-16:
Ich erwarte kein Wunder sondern lediglich, dass die Kermi WP per Modbus TCP integriert wird und das EVCC als "übergeordnetes System (wie Kermi es damit ja explizit ermöglicht) bei zu definierenden Bedingungen (sprich Menge des PV-Überschuss) die richtigen Register setzt (und ggf. vorher liest).

Und von der WP erwarte ich dann, dass sie damit korrekt umgeht, also v.a. nicht taktet. Das probiere ich dann auch gern aus und melde das Ergebnis für die Community und andere Nutzer zurück. Aktuell komme ich aber nicht weiter, darum https://github.com/evcc-io/evcc/discussions/18821

Eine Anfrage an Kermi, welche Register zu setzten wäre habe ich geschickt und warte jetzt mal ab was da kommt. War schon früher aus anderen Gründen mit denen in Kontakt, die waren eigtl. ganz auskunftsfreudig. Mal schauen was da kommt.

andig, 2025-02-16:
> Und von der WP erwarte ich dann, dass sie damit korrekt umgeht, also v.a. nicht taktet.

Womit umgeht? Die Frage war, was überhaupt passieren soll. Comfort Modus einschalten? Etwas Anderes? Was?

czam70, 2025-02-22:
Zwischenstand: bin mit Kermi in Kontakt bzgl. Steuerung über Modbus TCP. Werde das hier ergänzen, sobald meine Fragen geklärt sind.

andig, 2025-02-22:
Probier gerne mal den PR aus- der sollte bei Überschuss auf Comfort schalten.

czam70, 2025-02-23:
Hallo,

Inzwischen habe ich https://github.com/evcc-io/evcc/discussions/18821 noch ergänzt mit der Info, die ich bekommen habe. Ich bin aber noch weiter mit Kermi dran...

Zur Nutzung von  "Komfort": bei Kermi (und vermutlich allen Heizungen und Wärmepumpen) kann man mit den div. Einstellungen eine Parallelverschiebung der Heizkurve veranlassen, z. Bsp. bei mir so: für Komfort +2 K, Eco -2 K und Normal ist natürlich 0 K.
Das ist aber nicht ideal um PV-Überschuss am besten auszunutzen. Da will man ja den Pufferspeicher gezielt überheizen, entweder für Trinkwasser (macht immer Sinn) oder eben auch für das Heizwasser. Und Modbus Register gibt es dafür ja...

premultiply, 2025-02-23:
Wir können nicht für jede Wärmepumpe am Markt alle erdenklichen Hydraulikschemata hinterlegen.

Die Einbindung muss sich aus Sicht von evcc an dieser Stelle auf zwei Dinge beschränken:
- Boost an/aus
- Dimmung an/aus

Was die jeweilige WP und die indivduell installierte Hydraulik daraus macht muss das WP-System jeweils selbst entscheiden und dazu intern hinterlegt haben.

Im Zweifelsfall geht es auch einfach nicht oder bringt in der Praxis keinen Mehrwert.

czam70, 2025-02-24:
Es muss kein Hydraulikschema hinterlegt (oder verstanden) werden - nur welche Register man wie lesen oder schreiben müsste um PV Überschuss zu nutzen. Ist bei den Batterien ja auch so.

Aber klar, das verstehe ich bzgl. Anpassungsaufwand. Ich teile in https://github.com/evcc-io/evcc/discussions/18821 einfach meine Erkenntnisse, was ich mit Kermi in Erfahrung bringen kann. Vielleicht nützt es jemanden was, auch außerhalb von EVCC, der evtl. diese Info hier findet.
Wenn es dann halt nicht realisiert werden kann ist das halt so.

andig, 2025-02-24:
Mengels Testergebnis geschlossen. Schade :(

czam70, 2025-02-25:
Läuft noch... Musste erst noch Abklärungen mit Kermi machen und habe ja auch nicht jeden Tag Zeit dafür. Würd noch etwas dauern aber ich bleibe dran.

czam70, 2025-02-25:
Update - ERFOLG - ES KLAPPT ALLES WIE ERWARTET! Siehe https://github.com/evcc-io/evcc/discussions/18821.

Alles was EVCC bei einer Integration also tun müsste, ist regelmäßig das Register 301 der Modbus Geräte ID 40 mit dem aktuellen "PV-Überschuss in W x 10" zu aktualisieren (es simuliert also einen s0 Zähler) - den Rest macht die WP nach den Grundeistellungen!
Das Intervall (in Minuten) sollte man evtl. einstellen können.

andig, 2025-02-25:
Der entsprechende WP Typ „heating“ ist bereits vorhanden.

diddip21, 2025-03-22:
Hallo, versuche auch eine x-change dynamic von Kermin einzubinden, den Modbus habe ich mir über den support bereits freigeschaltet. Kann mir villeicht bitte jemand kurz eine Anleitung verlinken wie ich diese Implementation in evcc testen kann?
besten Dank!

andig, 2025-03-22:
Das Template aus dem verlinkten PR müsste getestet werden. Das geht ganz gut mit Hilfe von https://github.com/evcc-io/evcc/issues/19489.

diddip21, 2025-03-22:
Habe mir das kermi.yaml als raw auf den server geladen (unter /tmp), den evcc service gestoppt und mit 
evcc --template-type charger --template /tmp/kermi.yaml --database /var/lib/evcc/evcc.db 
gestartet, doch leider finde ich anschließend in der UI den Charger nicht zur Auswahl.

diddip21, 2025-03-23:
Danke, werde ich heute Abend neu versuchen.

Zur ID 40 steht in der kermi TCP Beschreibung aus dem ersten Post. Auf der ersten Seite unten rechts.

Geräteadresse Wärmepumpe=40

andig, 2025-03-23:
Schön wären ja noch Temperaturquellen- also WW oder Pufferspeicher?

czam70, 2025-03-23:
> Schön wären ja noch Temperaturquellen- also WW oder Pufferspeicher?

Ja die gibt es, habe ich aber nicht getestet weil mir das nicht so wichtig ist. Aber evtl. wollen die anderen Kermi Besitzer das ausprobieren?

Die passenden Register gemäss Kermi müssten diese sein:

Mit ID 51:
100 R °C 0x03 Isttemperatur TWE (read only)
101 R °C 0x03 Solltemperatur TWE (read only)

Und mit ID 50:
151 R °C 0x03 Isttemperatur Heizkreis (read only)
152 R 0/85 °C 0x03 Solltemperatur Heizkreis (read only)

andig, 2025-03-23:
Danke- eingebaut und gemerged.

closed 2025-03-23 by andig

diddip21, 2025-03-25:
Entschuldigt meine späte Rückmeldung, ich habe mir dem Kermi Support kommuniziert und dabei erfahren, dass die hier beschreibene Modbus TCP verbindung **nur mit einem PRO** gerät funktioniert.
Für meinen Anwendungsfall mit der Kermi x-change dynamic braucht es noch eine separate "verkabelung".
Vielleicht ist das noch einhiflreicher Hinweis, welcher im Template hinterlegt werden könnte.

andig, 2025-03-25:
Gerne PR 👍🏻