evcc-io/evcc #18820: Load settings ignored in latest release

felberch, 2025-02-15:
### Describe the bug

I have updated to v0.200.0
My setting is: pv

But with no power from pv the evcc keeps loading. 

### Steps to reproduce

1. Set charge mode to minpv
2. Set charge mode to pv
3. Timeout reaches 
4. even with not enough surplus charge does not stop
...

### Configuration details

```yaml
network:
# setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: xxx

# log settings
log: info
levels:
  site: trace
  lp-1: trace
#  lp-2: trace
#  lp-3: trace
  cache: error
  db: error
  ocpp: trace
  charger: trace
  # modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for documentation see https://docs.evcc.io/docs/devices/chargers
   - name: my_grid
     type: custom
     power:
       source: http
       uri: http://192.168.0.195:8675/aggregates
       jq: .site.instant_power
       timeout: 10s
     currents: # optional currents (A)
     - source: http # L1 plugin type
       uri: http://192.168.0.195:8675/aggregates #tedapi/status
[... 404 lines cut ...]
[ocpp  ] TRACE 2025/02/15 10:37:57 recv SN521853: [3,"1443574938",
{
"status": "Accepted"
}]
[ocpp  ] TRACE 2025/02/15 10:37:57 recv SN521853: [2,"ea699374-abe1-467d-8973-86592ecab4d4","MeterValues",{"meterValue": [{"timestamp": "2025-02-15T09:37:56Z","sampledValue": [{"value": "1089","format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh"},{"value": "2812","format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W"},{"value": "6","format": "Raw","location": "Outlet","context": "Trigger","measurand": "Current.Offered","unit": "A"},{"value": "4156","format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Offered","unit": "W"},{"value": "6","format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A"},{"value": "225.000000","format": "Raw","location": "Outlet","context": "Trigger","phase": "L1-N","measurand": "Voltage","unit": "V"} ]} ],"connectorId": 1,"transactionId": 1739609517}]
[ocpp  ] TRACE 2025/02/15 10:37:57 send SN521853: [3,"ea699374-abe1-467d-8973-86592ecab4d4",{}]
[ocpp  ] TRACE 2025/02/15 10:37:59 recv SN521853: [2,"124d342f-a7f1-4e87-a8fd-bd8adcaef4fa","MeterValues",{"meterValue": [{"timestamp": "2025-02-15T09:37:58Z","sampledValue": [{"value": "1090","format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh"},{"value": "2812","format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W"},{"value": "6","format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A"},{"value": "4156","format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W"},{"value": "6","format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A"},{"value": "225.000000","format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1-N","measurand": "Voltage","unit": "V"} ]} ],"connectorId": 1,"transactionId": 1739609517}]
[ocpp  ] TRACE 2025/02/15 10:37:59 send SN521853: [3,"124d342f-a7f1-4e87-a8fd-bd8adcaef4fa",{}]
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

V0.200.0

StefanSchoof, 2025-02-15:
In deiner log ist die Zeit noch nicht abgelaufen 

```
[lp-1  ] DEBUG 2025/02/15 10:37:44 pv disable timer remaining: 1m49s
```

Würde dann nach der Zeit abgeschaltet?

felberch, 2025-02-15:
Sry das log wird so schnell riesig:

Ich hoffe ich habe jetzt die richtige Sektion erwischt. Nein leider nach Ablauf des timers lädt er fröhlich weiter.
Ich konnte den Ladevorgang nur stoppen indem ich den Ladevorgang des Auto abgeschaltet habe. 

`site  ] DEBUG 2025/02/15 10:20:14 ----
[lp-1  ] DEBUG 2025/02/15 10:20:14 charge power: 2858W
[lp-1  ] DEBUG 2025/02/15 10:20:14 charge currents: [6 0 0]A
[site  ] DEBUG 2025/02/15 10:20:14 grid power: 4553W
[site  ] DEBUG 2025/02/15 10:20:14 grid currents: [6.65 5.87 7.81]A
[site  ] DEBUG 2025/02/15 10:20:14 battery 1 power: 1W
[site  ] DEBUG 2025/02/15 10:20:14 pv 1 power: 490W
[site  ] DEBUG 2025/02/15 10:20:14 battery 1 soc: 9%
[site  ] DEBUG 2025/02/15 10:20:14 site power: 4554W
[lp-1  ] DEBUG 2025/02/15 10:20:14 charge voltages: [226 0 0]V
[lp-1  ] DEBUG 2025/02/15 10:20:14 detected connected phases: 1p
[lp-1  ] DEBUG 2025/02/15 10:20:14 detected active phases: 1p
[lp-1  ] DEBUG 2025/02/15 10:20:14 charge total import: 0.090kWh
[lp-1  ] DEBUG 2025/02/15 10:20:14 charger status: C
[lp-1  ] DEBUG 2025/02/15 10:20:14 soc estimated: 33.35% (vehicle: 33.00%)
[lp-1  ] DEBUG 2025/02/15 10:20:14 vehicle soc: 33%
[lp-1  ] DEBUG 2025/02/15 10:20:14 vehicle soc limit: 50%
[lp-1  ] DEBUG 2025/02/15 10:20:14 vehicle range: 178km
[lp-1  ] DEBUG 2025/02/15 10:20:14 pv charge current: 0A = 6A + -19.8A (4554W @ 1p)
[lp-1  ] DEBUG 2025/02/15 10:20:14 projected site power 4554W >= 0W disable threshold
[lp-1  ] DEBUG 2025/02/15 10:20:14 pv disable in 0s
[lp-1  ] DEBUG 2025/02/15 10:20:14 pv disable timer remaining: 0s
[ocpp  ] TRACE 2025/02/15 10:20:22 recv SN521853: [2,"4044657d-a356-4b8c-8b67-8b92048d4c94","Heartbeat",{}]
[ocpp  ] TRACE 2025/02/15 10:20:22 send SN521853: [3,"4044657d-a356-4b8c-8b67-8b92048d4c94",{"currentTime":"2025-02-15T09:20:22Z"}]
[ocpp  ] TRACE 2025/02/15 10:20:23 send SN521853: [2,"1002585979","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/02/15 10:20:23 recv SN521853: [3,"1002585979",
{
"status": "Accepted"
}]
[ocpp  ] TRACE 2025/02/15 10:20:23 recv SN521853: [2,"3a98db49-8897-426c-bf51-4267652d2b0d","MeterValues",{"meterValue": [{"timestamp": "2025-02-15T09:20:22Z","sampledValue": [{"value": "99","format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh"},{"value": "2870","format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W"},{"value": "6","format": "Raw","location": "Outlet","context": "Trigger","measurand": "Current.Offered","unit": "A"},{"value": "4156","format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Offered","unit": "W"},{"value": "6","format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A"},{"value": "226.000000","format": "Raw","location": "Outlet","context": "Trigger","phase": "L1-N","measurand": "Voltage","unit": "V"} ]} ],"connectorId": 1,"transactionId": 1739609517}]
[ocpp  ] TRACE 2025/02/15 10:20:23 send SN521853: [3,"3a98db49-8897-426c-bf51-4267652d2b0d",{}]
[ocpp  ] TRACE 2025/02/15 10:20:24 recv SN521853: [2,"ad5a0123-89d5-4d88-88e6-75ab78a32adf","MeterValues",{"meterValue": [{"timestamp": "2025-02-15T09:20:22Z","sampledValue": [{"value": "99","format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh"},{"value": "2870","format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W"},{"value": "6","format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A"},{"value": "4156","format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W"},{"value": "6","format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A"},{"value": "226.000000","format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1-N","measurand": "Voltage","unit": "V"} ]} ],"connectorId": 1,"transactionId": 1739609517}]
[ocpp  ] TRACE 2025/02/15 10:20:24 send SN521853: [3,"ad5a0123-89d5-4d88-88e6-75ab78a32adf",{}]
[ocpp  ] TRACE 2025/02/15 10:20:33 send SN521853: [2,"3572711679","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/02/15 10:20:33 recv SN521853: [3,"3572711679",
{
"status": "Accepted"
}]
[ocpp  ] TRACE 2025/02/15 10:20:33 recv SN521853: [2,"afc924c7-f5f8-4b75-900b-1382ac42ebc7","MeterValues",{"meterValue": [{"timestamp": "2025-02-15T09:20:31Z","sampledValue": [{"value": "107","format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh"},{"value": "2876","format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W"},{"value": "6","format": "Raw","location": "Outlet","context": "Trigger","measurand": "Current.Offered","unit": "A"},{"value": "4156","format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Offered","unit": "W"},{"value": "6","format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A"},{"value": "226.000000","format": "Raw","location": "Outlet","context": "Trigger","phase": "L1-N","measurand": "Voltage","unit": "V"} ]} ],"connectorId": 1,"transactionId": 1739609517}]
[... cut ...]
[lp-1  ] DEBUG 2025/02/15 10:21:14 vehicle soc limit: 50%
[lp-1  ] DEBUG 2025/02/15 10:21:14 vehicle range: 179km
[lp-1  ] DEBUG 2025/02/15 10:21:14 pv charge current: 0A = 0A + -20.7A (4757W @ 1p)
[lp-1  ] DEBUG 2025/02/15 10:21:14 pv enable timer reset
[lp-1  ] DEBUG 2025/02/15 10:21:14 pv timer inactive
[ocpp  ] TRACE 2025/02/15 10:21:17 send SN521853: [2,"788188209","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/02/15 10:21:17 recv SN521853: [3,"788188209",
{ `

mfuchs1984, 2025-02-16:
Deine Wallbox befolgt den Befehl zum Abschalten nicht. Er wird von ihr zwar akzeptiert, sie lädt aber munter weiter. Welche Firmware hast du genau auf der Pulsar?

felberch, 2025-02-16:
Hab ein Ticket offen bei Wallbox - da das ding nicht richtig funktioniert. (Die WB liefert inkorrekte werte der Phasen; bei 5.5.11 werden nur die Daten 1 Phase (V und A) zurück gesendet. Mit Firmware 6.4.14 immer zeigt er 2 Phasen an; Obwohl alles 3 Phasig angeschlossen wurde …)

Aktuell 5.5.11 (Werkseinstellung)

mfuchs1984, 2025-02-16:
>Mit Firmware 6.4.14 immer zeigt er 2 Phasen an

Passier laut Berichten hier leider auch mit 6.7.17.

felberch, 2025-02-16:
Ich hatte auch 6.7.17 installiert. Die Version funktioniert aber auch nicht stabil. 
Verstehe nicht warum diese Probleme nicht konstant für alle auftreten… .
Sehr spannend hingegen ist nach meinen Problemen mit 6.7.17 hab ich die WB auf Werks Einstellungen zurückgesetzt. Danach auf 6.4.14 aktualisiert die 6.7.17 wird danach nicht mehr angeboten.

Ich bin schon ernsthaft am überlegen mir eine andere Wallbox zu kaufen… . 
Wallbox scheint ja ABL gekauft zu haben, ob es damit besser wird ist ab zu warten.

andig, 2025-02-16:
> 10:37:44 pv disable timer remaining: 1m49s

Der Teil wo das dann endlich passiert (3min ist schon sehr lang) fehlt im Log.

closed 2025-02-16 by andig

mfuchs1984, 2025-02-16:
>Danach auf 6.4.14 aktualisiert die 6.7.17 wird danach nicht mehr angeboten.

Wurde aufgrund von Problemen wieder zurückgezogen. Insgesamt scheint das die grottigste Wallbox überhaupt zu sein