evcc-io/evcc #18827: Adding a heat pump via UI meter shelly3em 401

Cheppetto, 2025-02-15:
### Describe the bug

When trying to add a load point (shelly integration) for a heat pump and then trying to add a shelly 3em as energy meter for it this always ends in 401s. Username and PW are correct 100% 

With the same user password combination I could add the 3em as loadpoint itself (shelly integration) - Which does not work properly as it only reads the first phase. 

### Steps to reproduce

1. Create load point shelly for WP
2. Add meter shelly3em to loadpoint
3. See authentication fail with 401 all the time
...

### Configuration details

```yaml
Not available as I am trying to switch to UI config only
```

### Log details

```text
Not available
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.0

andig, 2025-02-16:
Could you check the browsers developer console which requests are executed? Does the charger request look correct?

Cheppetto, 2025-02-16:
The request looks fine, besides authentication does not seem to be done - 

`template: "shelly-3em", type: "template", host: "xxx.xxx.xxx.xxx", user: "user",…}

host: "xxx.xxx.xxx.xxx"
password: "password"
template: "shelly-3em"
type: "template"
usage: "charge"
user: "user"`

Payload sieht ok aus Frontend --> Backend

andig, 2025-02-16:
Und wie sieht der backend request aus? Falls du die Headers braucht mit —log-headers mit loggen.

Cheppetto, 2025-02-16:
Hmm, Ich gehe davon aus das es nur dne einen gibt, das ist der verfiy für das shelly device, response ist ein 401er not authorized
Sonst müsstest du mir auf die Sprünge helfen

Das oben ist die payload vom request - obfuscated

Cheppetto, 2025-02-16:
Da ich ohne positives validate nicht speichern kann, gibt es dazu fürchte ich nichts

andig, 2025-02-16:
Bitte nicht raten. Hast du ins Log geschaut? Das UI schickt dir eher keinen 401.

Cheppetto, 2025-02-16:
dann würde ich ich rate dazu schreiben, aber hier hast du das log, wie gesagt es schweigt sich komplett aus.

[evcc-20250216-150144-debug.log](https://github.com/user-attachments/files/18814913/evcc-20250216-150144-debug.log)

steps:

- Evcc vorher neu gestartet 
- neuen shelly loadpoint angelegt (wieder einmal)
- zusätzlichen meter shelly 3em hinzufügen versucht
- fehler beim validate
- log gezogen

Falls du etwas anderes brauchst, bitte beschreiben wie/wo/wann/was, danke
Bin jetzt aber sicher 2-3h nicht erreichbar, also gerne reinschreiben, aber auf keine unmittelbare Antwort warten, danke

andig, 2025-02-16:
Ah, es braucht ein trace log für den Shelly- sonst sind die Requests nicht sichtbar.

Cheppetto, 2025-02-16:
So, with the 8th try I was able to create the load point.
Attaching the log, still broken AF

[evcc-20250216-210015-trace.log.txt](https://github.com/user-attachments/files/18816168/evcc-20250216-210015-trace.log.txt)

Shelly log shows no traces of the meter only load point logs the 401s

Cheppetto, 2025-02-17:
The password is being sent with URL encoded quotes in the request, please fix @naltatis @andig

andig, 2025-02-17:
Wo kommen die Quotes her? Url Encoding ist ja erstmal richtig.

Cheppetto, 2025-02-17:
> Wo kommen die Quotes her? Url Encoding ist ja erstmal richtig.

Ich hoffe doch dass ihr das heraus findet.
Leider müsste ich raten und das soll ich ja nicht.

Aber eines weiß ich 100% evcc wo auch immer dichtet sie dazu, weil ich habe sie nicht konfiguriert.

andig, 2025-02-18:
Es braucht bitte mal dein bzw ein ähnliches pw das auch dazu führt als reproducer.

Cheppetto, 2025-02-18:
want happen, either close the bug, or maybe try it once? everything needed - or much more is in the ticket.
I was trying to test that feature - looks like I will have to think about using evcc at all.

andig, 2025-02-18:
We need a reproducer.

Cheppetto, 2025-02-18:
Ok it is a bit more tricky then I thought, try an exclamation mark as first character, should work consistently. 

Exclamation marks are tricky I know, neither the less I never encountered the problem before. - it is also not an issue for the first step only if you add a meter

andig, 2025-02-27:
In dem Fall würde ich erwarten:

    http://<user>:%21...@ip-of-shelly/whatever

funktioniert der Request wenn du ihn von Hand machst?

andig, 2025-02-27:
I've found the root cause, but not yet the fix :O

andig, 2025-02-27:
I'm afraid, for time being, I can only recommend not to use trailing !. Sorry for not having a better fix.

closed 2025-02-27 by andig

Cheppetto, 2025-02-27:
> Exclamation marks are tricky I know, neither the less I never encountered the problem before. - it is also not an issue for the first step only if you add a meter

Yeah, I know ;) Exclamation marks are a bit tricky ;)