evcc-io/evcc #18828: Switchable loadpoints do not show up in ui

Cheppetto, 2025-02-15:
### Describe the bug

I created a switchable load point for a heat pump with the UI, when wanting to edit it it can't be found, tough present in the DB.
Ad wegen creating a new one it gets the next ID

Load points of all kinds should be shown/edit/deletable via UI

### Steps to reproduce

1.create integrated device load point (shelly)
2. Try to find it in the UI 
3. Verify it still exists in DB by creating another one and seeing the index go up
...

### Configuration details

```yaml
Ui config only - evcc yaml just had one line so that evcc starts
```

### Log details

```text
Nothing to be found in ui
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.0

naltatis, 2025-02-15:
Can you please elaborate what you're trying to do exactly? Not sure what you mean with "switchable loadpoint". Please provide a clearer step by step instruction of what you're trying to do.

Cheppetto, 2025-02-15:
Of course, sorry.
I just add images as they can say more then words:

New charger (type selection) 

[image]

Add type shelly (as in documentation) 

[image]

Store

andig, 2025-02-16:
> I just add images as they can say more then words:

We prefer clear and concise steps to reproduce ;)

Cheppetto, 2025-02-16:
There is no more I can share

Cheppetto, 2025-02-18:
Issue did not occur anymore

closed 2025-02-18 by Cheppetto