evcc-io/evcc #18836: title from loadpoint not used in 0.200

lgusewski, 2025-02-15:
### Describe the bug

Ich habe seit dem 01.02.2025 eine neue Datenbank mit Version 0.133 und yaml-Konfiguration benutzt. Mein Ladepunkt hat einen Titel.

`loadpoints:
  - title: Stellplatz
    charger: wallbox
    circuit: carport`

Bis zum letzten Ladevorgang mit der 0.133 am 13.02.2025 wurde im Ladeprotokoll in der Spalte "Ladepunkt" auch der Titel "Stellplatz" eingetragen. Danach bin ich auf die Version 0.200 gegangen und habe heute unter der neuen Version einen Ladevorgang gestartet. Danach stand im Ladeprotokoll allerdings nicht "Stellplatz" sondern "Ladepunkt" im Protokoll.

### Steps to reproduce

Siehe oben, Konfiguration nur über yaml und Ladungen sollte zwischen Version 0.133 oder älter und Version 0.200 an den Ladepunkten unterschiedlich sein.

### Configuration details

```yaml
Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

meters:
  - name: shm
    type: template
    template: sma-home-manager
    usage: grid
    host: 192.168.x.x
  - name: pv1
    type: template
    template: sma-hybrid
    usage: pv
    modbus: tcpip
    id: 3
    host: 192.168.x.x
    port: 502
  - name: battery
    type: template
    template: sma-hybrid
    usage: battery
    capacity: 19.3
    modbus: tcpip
    id: 3
    host: 192.168.x.x
    port: 502
  - name: pv2
    type: template
    template: sma-inverter-modbus
    usage: pv
    modbus: tcpip
    id: 3
    host: 192.168.x.x
    port: 502

circuits:
  - name: main
    title: Hauptstromkreis
    maxCurrent: 63 # 63 A
    maxPower: 30000 # 30 kW
    meter: shm
  - name: carport
    title: Stellplatz
    maxCurrent: 25 # 25 A
    maxPower: 11000 # 11 kW
    parent: main

hems:
  type: sma
  vendorid: 28081973
  deviceid: *****
  allowcontrol: false

chargers:
  - name: wallbox
    type: template
    template: keba-modbus
    modbus: tcpip
    id: 255
    host: 192.168.x.x
    port: 502
    welcomecharge: true

vehicles:
  - name: my_company_car
    type: template
    template: vw
    user: *****
    password: *****
    vin: *****
    title: ID.7 Tourer Pro
    capacity: 77 # kWh
    timeout: 10s
    identifiers:
      - xxxxxxxx

site:
  - title: Mein Zuhause
    meters:
      grid: shm
      pv:
        - pv1
        - pv2
      battery:
        - battery
    residualPower: 100

loadpoints:
  - title: Stellplatz
    charger: wallbox
    circuit: carport

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.2871
  feedin:
    type: fixed
    price: 0.0748
  co2:
    type: grünstromindex
    zip: 58730

mqtt:
  broker: x.x.x.x:1883
  topic: evcc
  user: *****
  password: *****

sponsortoken: *****

Version: `0.200.0`
```

### Log details

```text
Kein Log für diesen Fehler vorhanden.
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.0

julred, 2025-02-15:
Same here. Not possible to adjust name of load point or car using historical tabular view

andig, 2025-02-16:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

lgusewski, 2025-02-16:
Ich hatte zu diesem Zeitpunkt den evcc-Dienst bereits geschlossen, später die anonymisierte Konfiguration erstellen lassen. In dem Log der UI stand dazu nichts sinnvolles mehr.

Die [Diskussion](https://github.com/evcc-io/evcc/discussions/18837) ist ja offensichtlich bekannt, daher finde ich ein es befremdliche diesen Issue einfach ohne Behebung des Fehlers zu schliessen. Andere könnten hier, wenn es denn so wichtig ist, bei einem offenen Issue auch ihre Logs anhängen. Daher bitte ich um erneutes Öffnen dieses Issues.

Ich kann programmiertechnisch diesem Projekt nichts beitragen, habe jedoch früher einmal in anderen Sprachen programmiert und würde aus dieser Erfahrung sagen, es liegt bei der Einarbeitung der UI eine fehlerhafte genutzte Variable vor, die auf die Zeichenkette des Ladepunkt-Titels verweisen sollte, jedoch auf die Spaltenüberschrift zeigt.

andig, 2025-02-16:
>Die https://github.com/evcc-io/evcc/discussions/18837 ist ja offensichtlich bekannt, daher finde ich ein es befremdlich diesen Issue einfach ohne Behebung des Fehlers zu schliessen. 

@lgusewski bitte erspare uns die Diskussion. Im Issuetemplate steht glasklar drin was es braucht. Das ist Zeitverschwendung.

@naltatis du willst https://github.com/evcc-io/evcc/issues/18850

closed 2025-02-16 by andig