evcc-io/evcc #18838: SUNGROW Batterie Anzeige falsch ("In" Anzeige anstelle "Out")

SmartVolker, 2025-02-15:
### Describe the bug

In der Version 0.200.0 habe ich folgende Beobachtung gemacht, dass die Anzeige der Batterie (laden/entladen) nicht stimmt. Die Batterie habe ich auf Laden bei Tarif eingestellt, in der Anzeige von evcc wurde der Status falsch angezeigt als entladen (Batterie versorgt das Gebäude) angezeigt. Es wird in evcc "Batterie entladen" angezeigt anstelle "Batterie laden".

Im Screenshot kann man sehen zum einen, wie in der Sungrow App die Batterie geladen wird mit 5,4kW (7,8kW gesamt) und in der Tibber App wie der Gesamtnetzbezug ist. 

### Steps to reproduce

[image]
[image]
[image]
[image]
[image]

1. Netzbezug der Sungrow Batterie
2. Anzeige steht auf "
3.
...

### Configuration details

```yaml
Sungrow Hybrid Wechselricher mit Sungrow Batterie
Tibber Pulse als Grid Meter
```

### Log details

```text
Können bei Bedarf zur Verfügung gestellt werden
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.0

VolkerK62, 2025-02-15:
das entwickelt sich zur never-ending-story
https://github.com/evcc-io/evcc/pull/18473

premultiply, 2025-02-15:
Auch hier:
Ich brauche die Ausgabe von `evcc meter --log trace,db:error` dazu.
Riesige Screenshots nützen überhaupt nichts.

SmartVolker, 2025-02-16:
[evcc-20250216-112648-trace.log](https://github.com/user-attachments/files/18814393/evcc-20250216-112648-trace.log)

Hier nun die Trace logs. Einmal mit Batterieladung und ohne.

Was wir aufviel, das ich unter Hausbatterie Menu, nicht das Netzladen UI gesehen habe. 2-3 mal probiert, dann war es wieder zu sehen.

premultiply, 2025-02-16:
Ich brauche die Ausgabe von `evcc meter --log trace,db:error` jeweils in den beiden definierten Zuständen
1. Batterie laden und
2. Batterie entladen.

premultiply, 2025-02-16:
Standby:
```
[db:3  ] TRACE 2025/02/16 11:17:33 modbus: send 00 03 00 00 00 06 01 04 32 c8 00 01
[db:3  ] TRACE 2025/02/16 11:17:33 modbus: recv 00 03 00 00 00 05 01 04 02 00 00
[db:3  ] TRACE 2025/02/16 11:17:33 modbus: send 00 04 00 00 00 06 01 04 32 dc 00 01
[db:3  ] TRACE 2025/02/16 11:17:33 modbus: recv 00 04 00 00 00 05 01 04 02 00 00
[db:3  ] TRACE 2025/02/16 11:17:33 modbus: send 00 05 00 00 00 06 01 04 32 dd 00 01
[db:3  ] TRACE 2025/02/16 11:17:33 modbus: recv 00 05 00 00 00 05 01 04 02 00 00
[site  ] DEBUG 2025/02/16 11:17:33 battery 1 power: 0W
```

Laden:
```
[db:3  ] TRACE 2025/02/16 11:20:16 modbus: send 00 23 00 00 00 06 01 04 32 c8 00 01
[db:3  ] TRACE 2025/02/16 11:20:16 modbus: recv 00 23 00 00 00 05 01 04 02 00 00
[db:3  ] TRACE 2025/02/16 11:20:16 modbus: send 00 24 00 00 00 06 01 04 32 dc 00 01
[db:3  ] TRACE 2025/02/16 11:20:16 modbus: recv 00 24 00 00 00 05 01 04 02 01 03
[db:3  ] TRACE 2025/02/16 11:20:16 modbus: send 00 25 00 00 00 06 01 04 32 dd 00 01
[db:3  ] TRACE 2025/02/16 11:20:16 modbus: recv 00 25 00 00 00 05 01 04 02 14 a2
[site  ] DEBUG 2025/02/16 11:20:16 battery 1 power: 5282W
```

@SmartVolker fehlt nur noch Entladen. Kannst du das bitte noch nachliefern?
Eines der Register müsste beim entladen (hoffentlich) negativ werden.
Sonst hätten wir einen ganz neuen Fall und ein Problem.

SmartVolker, 2025-02-17:
Ja, das schicke ich dir dir Traces zum Entladen. Ich habe die neue Version 0.200.1 installiert:

Aktuell scheint die Sonne und die Darstellung ist nicht richtig:
Anzeige Sungrow:

[image]

Anzeige evcc:

[image]

SmartVolker, 2025-02-17:
Hier nun die TRACE zum "Entladen"
[site ] DEBUG 2025/02/17 18:00:05 ----
[site ] DEBUG 2025/02/17 18:00:05 grid power: 75W
[site ] DEBUG 2025/02/17 18:00:05 grid currents: [0 0 0]A
[db:2 ] TRACE 2025/02/17 18:00:05 modbus: send 10 82 00 00 00 06 01 04 13 98 00 02
[db:2 ] TRACE 2025/02/17 18:00:05 modbus: recv 10 82 00 00 00 07 01 04 04 00 00 00 00
[site ] DEBUG 2025/02/17 18:00:05 pv 1 power: 0W
[db:2 ] TRACE 2025/02/17 18:00:05 modbus: send 10 83 00 00 00 06 01 04 32 ca 00 02
[db:2 ] TRACE 2025/02/17 18:00:05 modbus: recv 10 83 00 00 00 07 01 04 04 03 92 00 00
[site ] DEBUG 2025/02/17 18:00:05 battery 1 power: 330W
[site ] DEBUG 2025/02/17 18:00:05 battery 1 soc: 57%
[site ] DEBUG 2025/02/17 18:00:05 site power: 405W
[site ] DEBUG 2025/02/17 18:00:45 ----
[site ] DEBUG 2025/02/17 18:00:45 grid power: 70W
[site ] DEBUG 2025/02/17 18:00:45 grid currents: [0 0 0]A
[db:2 ] TRACE 2025/02/17 18:00:45 modbus: send 10 89 00 00 00 06 01 04 13 98 00 02
[db:2 ] TRACE 2025/02/17 18:00:45 modbus: recv 10 89 00 00 00 07 01 04 04 00 00 00 00
[site ] DEBUG 2025/02/17 18:00:45 pv 1 power: 0W
[db:2 ] TRACE 2025/02/17 18:00:45 modbus: send 10 8a 00 00 00 06 01 04 32 ca 00 02
[db:2 ] TRACE 2025/02/17 18:00:45 modbus: recv 10 8a 00 00 00 07 01 04 04 03 92 00 00
[site ] DEBUG 2025/02/17 18:00:45 battery 1 power: 330W
[site ] DEBUG 2025/02/17 18:00:45 battery 1 soc: 57%
[site ] DEBUG 2025/02/17 18:00:45 site power: 400W
[site ] DEBUG 2025/02/17 18:01:25 ----
[site ] DEBUG 2025/02/17 18:01:25 grid power: 103W
[site ] DEBUG 2025/02/17 18:01:25 grid currents: [0 0 0]A
[db:2 ] TRACE 2025/02/17 18:01:25 modbus: send 10 90 00 00 00 06 01 04 13 98 00 02
[db:2 ] TRACE 2025/02/17 18:01:25 modbus: recv 10 90 00 00 00 07 01 04 04 00 00 00 00
[site ] DEBUG 2025/02/17 18:01:25 pv 1 power: 0W
[db:2 ] TRACE 2025/02/17 18:01:25 modbus: send 10 91 00 00 00 06 01 04 32 ca 00 02
[db:2 ] TRACE 2025/02/17 18:01:25 modbus: recv 10 91 00 00 00 07 01 04 04 03 92 00 00
[site ] DEBUG 2025/02/17 18:01:25 battery 1 power: 187W
[site ] DEBUG 2025/02/17 18:01:25 battery 1 soc: 57%
[site ] DEBUG 2025/02/17 18:01:25 site power: 290W
[site ] DEBUG 2025/02/17 18:02:05 ----
[site ] DEBUG 2025/02/17 18:02:05 grid power: 97W
[site ] DEBUG 2025/02/17 18:02:05 grid currents: [0 0 0]A
[db:2 ] TRACE 2025/02/17 18:02:05 modbus: send 10 97 00 00 00 06 01 04 13 98 00 02
[db:2 ] TRACE 2025/02/17 18:02:05 modbus: recv 10 97 00 00 00 07 01 04 04 00 00 00 00
[site ] DEBUG 2025/02/17 18:02:05 pv 1 power: 0W
[db:2 ] TRACE 2025/02/17 18:02:05 modbus: send 10 98 00 00 00 06 01 04 32 ca 00 02
[db:2 ] TRACE 2025/02/17 18:02:05 modbus: recv 10 98 00 00 00 07 01 04 04 03 92 00 00
[site ] DEBUG 2025/02/17 18:02:05 battery 1 power: 258W
[site ] DEBUG 2025/02/17 18:02:05 battery 1 soc: 57%
[site ] DEBUG 2025/02/17 18:02:05 site power: 355W
[site ] DEBUG 2025/02/17 18:02:45 ----
[site ] DEBUG 2025/02/17 18:02:45 grid power: 100W
[site ] DEBUG 2025/02/17 18:02:45 grid currents: [0 0 0]A
[db:2 ] TRACE 2025/02/17 18:02:45 modbus: send 10 9e 00 00 00 06 01 04 13 98 00 02
[db:2 ] TRACE 2025/02/17 18:02:45 modbus: recv 10 9e 00 00 00 07 01 04 04 00 00 00 00
[site ] DEBUG 2025/02/17 18:02:45 pv 1 power: 0W
[db:2 ] TRACE 2025/02/17 18:02:45 modbus: send 10 9f 00 00 00 06 01 04 32 ca 00 02
[db:2 ] TRACE 2025/02/17 18:02:45 modbus: recv 10 9f 00 00 00 07 01 04 04 03 92 00 00
[site ] DEBUG 2025/02/17 18:02:45 battery 1 power: 251W
[site ] DEBUG 2025/02/17 18:02:45 battery 1 soc: 57%
[site ] DEBUG 2025/02/17 18:02:45 site power: 351W 

[image]

premultiply, 2025-02-17:
> Hier nun die TRACE zum "Entladen"

Leider fehlt das entscheidene Gerät db:3 und dessen relevante Modbus-Kommunikation 😃

@SmartVolker Guck nochmal bitte. Zu viel bzw. das Falsche rausgefiltert?

SmartVolker, 2025-02-18:
@premultiply OK mach ich. Habs leider übersehen. Sind dies die richtigen ?

[image]

SmartVolker, 2025-02-18:
Komisch, ich sehen den Kommentar von glumanda nicht mehr von gestern. Dies war der Eintrag:

glumanda left a comment [(evcc-io/evcc#18838)](https://deref-web.de/mail/client/PbNH1JTku6c/dereferrer/?redirectUrl=https%3A%2F%2Fgithub.com%2Fevcc-io%2Fevcc%2Fissues%2F18838%23issuecomment-2663885265)

Hallo zusammen, das könnte mit einer Änderung durch Sungrow in der Firmware v95.03 zusammenhängen. Ich hatte das auch in meiner Home Assistant config anpassen müssen. Sungrow hat für das Register 13022 (Adresse 13021) battery power das Format von uint16 auf int16 (somit nun vorzeichenbehaftet) umgestellt, wobei ein negativer Wert das Laden der Batterie anzeigt und ein positiver ein Entladen.
Vor dieser Firmware Änderung habe ich immer den powerflow state (Register 13001, Adresse 13000) ausgewertet und den vorzeichenlosen Wert mit -1 multipliziert, wenn entladen wurde. Wenn in evcc dieselbe Logik implementiert ist, wird jetzt für die Batterieleistung immer ein negativer Wert ermittelt und dadruch vmtl. immer als Entladung interpretiert und dargestellt.
Die Lösung wäre den Wert von battery power einfach immer mit -1 zu multiplizieren.

premultiply, 2025-02-18:
Genau das tun wir eigentlich im Template.
Bei dir scheint aber noch irgendwas anders zu sein oder irgendwas ist doch noch im Template falsch. 🤷🏻‍♂️

Es hilft nix, ich brauche die Logdaten von `db:3` und `site`.

SmartVolker, 2025-02-18:
Hier nun vom "Entladen":

[site ] DEBUG 2025/02/18 17:40:05 pv 1 power: 0W
[db:2 ] TRACE 2025/02/18 17:40:05 modbus: send 4a f2 00 00 00 06 01 04 32 ca 00 02
[db:2 ] TRACE 2025/02/18 17:40:05 modbus: recv 4a f2 00 00 00 07 01 04 04 04 03 00 00
[db:3 ] TRACE 2025/02/18 17:40:05 modbus: send 4a f3 00 00 00 06 01 04 32 c8 00 01
[db:3 ] TRACE 2025/02/18 17:40:05 modbus: recv 4a f3 00 00 00 05 01 04 02 00 00
[db:3 ] TRACE 2025/02/18 17:40:05 modbus: send 4a f4 00 00 00 06 01 04 32 dc 00 01
[db:3 ] TRACE 2025/02/18 17:40:05 modbus: recv 4a f4 00 00 00 05 01 04 02 00 0d
[db:3 ] TRACE 2025/02/18 17:40:05 modbus: send 4a f5 00 00 00 06 01 04 32 dd 00 01
[db:3 ] TRACE 2025/02/18 17:40:05 modbus: recv 4a f5 00 00 00 05 01 04 02 00 b8
[site ] DEBUG 2025/02/18 17:40:05 battery 1 power: 184W
[db:3 ] TRACE 2025/02/18 17:40:05 modbus: send 4a f6 00 00 00 06 01 04 32 e2 00 02
[db:3 ] TRACE 2025/02/18 17:40:05 modbus: recv 4a f6 00 00 00 07 01 04 04 04 0e 00 00
[db:3 ] TRACE 2025/02/18 17:40:05 modbus: send 4a f7 00 00 00 06 01 04 32 de 00 01
[db:3 ] TRACE 2025/02/18 17:40:05 modbus: recv 4a f7 00 00 00 05 01 04 02 02 b6
[site ] DEBUG 2025/02/18 17:40:05 battery 1 soc: 69%
[site ] DEBUG 2025/02/18 17:40:05 site power: 217W
[site ] DEBUG 2025/02/18 17:40:45 ----
[site ] DEBUG 2025/02/18 17:40:45 grid power: 26W
[site ] DEBUG 2025/02/18 17:40:45 grid currents: [0 0 0]A
[db:2 ] TRACE 2025/02/18 17:40:45 modbus: send 4a f8 00 00 00 06 01 04 13 98 00 02
[db:2 ] TRACE 2025/02/18 17:40:45 modbus: recv 4a f8 00 00 00 07 01 04 04 00 00 00 00
[site ] DEBUG 2025/02/18 17:40:45 pv 1 power: 0W
[db:2 ] TRACE 2025/02/18 17:40:45 modbus: send 4a f9 00 00 00 06 01 04 32 ca 00 02
[db:2 ] TRACE 2025/02/18 17:40:45 modbus: recv 4a f9 00 00 00 07 01 04 04 04 03 00 00
[db:3 ] TRACE 2025/02/18 17:40:45 modbus: send 4a fa 00 00 00 06 01 04 32 c8 00 01
[db:3 ] TRACE 2025/02/18 17:40:45 modbus: recv 4a fa 00 00 00 05 01 04 02 00 00
[db:3 ] TRACE 2025/02/18 17:40:45 modbus: send 4a fb 00 00 00 06 01 04 32 dc 00 01
[db:3 ] TRACE 2025/02/18 17:40:45 modbus: recv 4a fb 00 00 00 05 01 04 02 00 0d
[db:3 ] TRACE 2025/02/18 17:40:45 modbus: send 4a fc 00 00 00 06 01 04 32 dd 00 01
[db:3 ] TRACE 2025/02/18 17:40:45 modbus: recv 4a fc 00 00 00 05 01 04 02 00 bc
[site ] DEBUG 2025/02/18 17:40:45 battery 1 power: 188W
[db:3 ] TRACE 2025/02/18 17:40:45 modbus: send 4a fd 00 00 00 06 01 04 32 e2 00 02
[db:3 ] TRACE 2025/02/18 17:40:45 modbus: recv 4a fd 00 00 00 07 01 04 04 04 0e 00 00
[db:3 ] TRACE 2025/02/18 17:40:45 modbus: send 4a fe 00 00 00 06 01 04 32 de 00 01
[db:3 ] TRACE 2025/02/18 17:40:45 modbus: recv 4a fe 00 00 00 05 01 04 02 02 b6
[site ] DEBUG 2025/02/18 17:40:45 battery 1 soc: 69%
[site ] DEBUG 2025/02/18 17:40:45 site power: 214W
[site ] DEBUG 2025/02/18 17:41:25 ----
[site ] DEBUG 2025/02/18 17:41:25 grid power: 30W
[site ] DEBUG 2025/02/18 17:41:25 grid currents: [0 0 0]A
[db:2 ] TRACE 2025/02/18 17:41:25 modbus: send 4a ff 00 00 00 06 01 04 13 98 00 02
[db:2 ] TRACE 2025/02/18 17:41:25 modbus: recv 4a ff 00 00 00 07 01 04 04 00 00 00 00
[site ] DEBUG 2025/02/18 17:41:25 pv 1 power: 0W
[db:2 ] TRACE 2025/02/18 17:41:25 modbus: send 4b 00 00 00 00 06 01 04 32 ca 00 02
[db:2 ] TRACE 2025/02/18 17:41:25 modbus: recv 4b 00 00 00 00 07 01 04 04 04 03 00 00
[db:3 ] TRACE 2025/02/18 17:41:25 modbus: send 4b 01 00 00 00 06 01 04 32 c8 00 01
[db:3 ] TRACE 2025/02/18 17:41:25 modbus: recv 4b 01 00 00 00 05 01 04 02 00 00
[db:3 ] TRACE 2025/02/18 17:41:25 modbus: send 4b 02 00 00 00 06 01 04 32 dc 00 01
[db:3 ] TRACE 2025/02/18 17:41:25 modbus: recv 4b 02 00 00 00 05 01 04 02 00 0d
[db:3 ] TRACE 2025/02/18 17:41:25 modbus: send 4b 03 00 00 00 06 01 04 32 dd 00 01
[db:3 ] TRACE 2025/02/18 17:41:25 modbus: recv 4b 03 00 00 00 05 01 04 02 00 b7
[site ] DEBUG 2025/02/18 17:41:25 battery 1 power: 183W
[db:3 ] TRACE 2025/02/18 17:41:25 modbus: send 4b 04 00 00 00 06 01 04 32 e2 00 02
[db:3 ] TRACE 2025/02/18 17:41:25 modbus: recv 4b 04 00 00 00 07 01 04 04 04 0e 00 00
[db:3 ] TRACE 2025/02/18 17:41:25 modbus: send 4b 05 00 00 00 06 01 04 32 de 00 01
[db:3 ] TRACE 2025/02/18 17:41:25 modbus: recv 4b 05 00 00 00 05 01 04 02 02 b6
[site ] DEBUG 2025/02/18 17:41:25 battery 1 soc: 69%
[site ] DEBUG 2025/02/18 17:41:25 site power: 213W

glumanda, 2025-02-18:
Dann belebe ich mal meinen gestrigen Eintrag wieder:

Hallo zusammen, das könnte mit einer Änderung durch Sungrow in der Firmware v95.03 zusammenhängen. Ich hatte das auch in meiner Home Assistant config anpassen müssen. Sungrow hat für das Register 13022 (Adresse 13021) battery power das Format von uint16 auf int16 (somit nun vorzeichenbehaftet) umgestellt, wobei ein negativer Wert das Laden der Batterie anzeigt und ein positiver ein Entladen.
Vor dieser Firmware Änderung habe ich immer den powerflow state (Register 13001, Adresse 13000) ausgewertet und den vorzeichenlosen Wert mit -1 multipliziert, wenn entladen wurde. Wenn in evcc dieselbe Logik implementiert ist, wird jetzt für die Batterieleistung immer ein negativer Wert ermittelt und dadruch vmtl. immer als Entladung interpretiert und dargestellt.
Die Lösung wäre den Wert von battery power einfach immer mit -1 zu multiplizieren.

ALLERDINGS: Ich hatte noch ein selber gebautes Docker Image (musste die mqtt-Topics für den tasmota Zählersensor ändern, hatte ich bereits in tasmota verändert und hätte zu viel Umbauarbeiten im Bestand benötigt). Nach gestrigem Blick in das sungrow-hybrid Template hatte ich gesehen, dass es funktionieren müsste. Und ja tat es heute (war viel Sonne bei uns) auch. Für mich somit alles fein.

Gestern hatte ich meinen Beitrag einfach gelöscht .... ging ganz schnell und rückstandsfrei.

evcc läuft bei mir noch im "Leerlauf": in Kürze kommt da noch eine Wallbox und ein Auto dazu. Bin schon jetzt sehr beeindruckt. Danke für Eure Arbeit!

premultiply, 2025-02-18:
@SmartVolker Danke, aber leider ganz schlecht:
In deinem Fall gibt es bislang absolut keinen Indikator ob die Batterie lädt oder entlädt.

Da kann ich leider nichts tun.

Ich kann nur nochmal etwas verzweifelt zu Allgemeinplätzen greifen:
- WR neustarten oder nochmal Firmware auf neusten Stand bringen
- Vom hinteren LAN-Anschluss auf Dongle wecheln (oder umgekehrt)
- Nochmal neuste evcc nightly probieren

SmartVolker, 2025-02-19:
Ich habe bereits den neusten Stand der Firmware, gemäss der Einstelungen. Die Anlage wurde vor 3 Wochen installiert. Der Hybrid Wechselrichter ist per WLAN angebunden. Wie kann ich die Firmware Version des Hybrid WR auslesen? Ich habe evcc Version 0.200.1 installiert. 

Sind die Werte immer positiv? Dann wäre der Hinweis von glumanda relevant zusätzlich den powerflow state auszuwerten,  powerflow state (Register 13001, Adresse 13000)

zachelnet, 2025-02-19:
Hi @SmartVolker 
entsprechende Register wird Ausgewertet, siehe https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/sungrow-hybrid.yaml#L121

Hast du ein SHxxRT oder ein der neuen SHxxT Wechselrichter, ist bei dir WiNet-S oder WiNet-S2 verbaut.

Grüße
Martin

SmartVolker, 2025-02-19:
@zachelnet Martin, ich habe einSH6.0RT-V112 verbaut. Win WiNet-S ist verbaut.

zachelnet, 2025-02-19:
@SmartVolker Danke für die Information. Bezüglich Firmware prüfen meines wissen kannst du es ansehen mit ein Installationsaccount. Alternative, wenn noch die Nachricht vorhanden ist bezüglich Updates von deinem Installateur, im Nachrichten-Center. Ansonsten kann ich nur dasselbe sagen wie @premultiply.

closed 2025-02-20 by SmartVolker

premultiply, 2025-02-20:
Puh 😅